Koenkk/zigbee2mqtt #17984: Debounce not working for exact duplicate messages in the same second.

nateww, 2023-06-10:
### What happened?

I was monitoring the zigbee2mqtt logs to debug an unrelated configuration issue, and noticed one of my devices was sending a LOT of duplicate messages, which is taking up a lot of bandwidth.  FYI, the device/site being monitored is 100 miles away connected over a cellular network to my home network over a wireguard VPN connection, so bandwidth IS an issue.

After setting debounce to get rid of duplicate messages, no change is being made to the published messages.  Messages below are AFTER changing the settings and restarting zigbee2MQTT
```
Jun 10 09:23:31 argon-pi npm[2189]: Zigbee2MQTT:info  2023-06-10 09:23:31: MQTT publish: topic 'zigbee2mqttseeley/seeley-camera-plug', payload '{"last_seen":"2023-06-10T09:23:31-06:00","linkquality":69,"state":"ON"}'
Jun 10 09:23:31 argon-pi npm[2189]: Zigbee2MQTT:info  2023-06-10 09:23:31: MQTT publish: topic 'zigbee2mqttseeley/seeley-camera-plug', payload '{"last_seen":"2023-06-10T09:23:31-06:00","linkquality":69,"state":"ON"}'
Jun 10 09:23:31 argon-pi npm[2189]: Zigbee2MQTT:info  2023-06-10 09:23:31: MQTT publish: topic 'zigbee2mqttseeley/seeley-camera-plug', payload '{"last_seen":"2023-06-10T09:23:31-06:00","linkquality":69,"state":"ON"}'
Jun 10 09:23:31 argon-pi npm[2189]: Zigbee2MQTT:info  2023-06-10 09:23:31: MQTT publish: topic 'zigbee2mqttseeley/seeley-camera-plug', payload '{"last_seen":"2023-06-10T09:23:31-06:00","linkquality":69,"state":"ON"}'
Jun 10 09:23:31 argon-pi npm[2189]: Zigbee2MQTT:info  2023-06-10 09:23:31: MQTT publish: topic 'zigbee2mqttseeley/seeley-camera-plug', payload '{"last_seen":"2023-06-10T09:23:31-06:00","linkquality":69,"state":"ON"}'
Jun 10 09:23:31 argon-pi npm[2189]: Zigbee2MQTT:info  2023-06-10 09:23:31: MQTT publish: topic 'zigbee2mqttseeley/seeley-camera-plug', payload '{"last_seen":"2023-06-10T09:23:31-06:00","linkquality":69,"state":"ON"}'
Jun 10 09:23:33 argon-pi npm[2189]: Zigbee2MQTT:info  2023-06-10 09:23:33: MQTT publish: topic 'zigbee2mqttseeley/seeley-camera-plug', payload '{"last_seen":"2023-06-10T09:23:31-06:00","linkquality":69,"state":"ON"}'
```

And, just a few moments later
```
Jun 10 09:28:31 argon-pi npm[2189]: Zigbee2MQTT:info  2023-06-10 09:28:31: MQTT publish: topic 'zigbee2mqttseeley/seeley-camera-plug', payload '{"last_seen":"2023-06-10T09:28:31-06:00","linkquality":69,"state":"ON"}'
Jun 10 09:28:31 argon-pi npm[2189]: Zigbee2MQTT:info  2023-06-10 09:28:31: MQTT publish: topic 'zigbee2mqttseeley/seeley-camera-plug', payload '{"last_seen":"2023-06-10T09:28:31-06:00","linkquality":69,"state":"ON"}'
Jun 10 09:28:31 argon-pi npm[2189]: Zigbee2MQTT:info  2023-06-10 09:28:31: MQTT publish: topic 'zigbee2mqttseeley/seeley-camera-plug', payload '{"last_seen":"2023-06-10T09:28:31-06:00","linkquality":69,"state":"ON"}'
Jun 10 09:28:31 argon-pi npm[2189]: Zigbee2MQTT:info  2023-06-10 09:28:31: MQTT publish: topic 'zigbee2mqttseeley/seeley-camera-plug', payload '{"last_seen":"2023-06-10T09:28:31-06:00","linkquality":69,"state":"ON"}'
Jun 10 09:28:31 argon-pi npm[2189]: Zigbee2MQTT:info  2023-06-10 09:28:31: MQTT publish: topic 'zigbee2mqttseeley/seeley-camera-plug', payload '{"last_seen":"2023-06-10T09:28:31-06:00","linkquality":69,"state":"ON"}'
Jun 10 09:28:31 argon-pi npm[2189]: Zigbee2MQTT:info  2023-06-10 09:28:31: MQTT publish: topic 'zigbee2mqttseeley/seeley-camera-plug', payload '{"last_seen":"2023-06-10T09:28:31-06:00","linkquality":69,"state":"ON"}'
Jun 10 09:28:31 argon-pi npm[2189]: Zigbee2MQTT:info  2023-06-10 09:28:31: MQTT publish: topic 'zigbee2mqttseeley/seeley-camera-plug', payload '{"last_seen":"2023-06-10T09:28:31-06:00","linkquality":69,"state":"ON"}'
Jun 10 09:28:31 argon-pi npm[2189]: Zigbee2MQTT:info  2023-06-10 09:28:31: MQTT publish: topic 'zigbee2mqttseeley/seeley-camera-plug', payload '{"last_seen":"2023-06-10T09:28:31-06:00","linkquality":69,"state":"ON"}'
Jun 10 09:28:31 argon-pi npm[2189]: Zigbee2MQTT:info  2023-06-10 09:28:31: MQTT publish: topic 'zigbee2mqttseeley/seeley-camera-plug', payload '{"last_seen":"2023-06-10T09:28:31-06:00","linkquality":69,"state":"ON"}'
Jun 10 09:28:31 argon-pi npm[2189]: Zigbee2MQTT:info  2023-06-10 09:28:31: MQTT publish: topic 'zigbee2mqttseeley/seeley-camera-plug', payload '{"last_seen":"2023-06-10T09:28:31-06:00","linkquality":69,"state":"ON"}'
Jun 10 09:28:31 argon-pi npm[2189]: Zigbee2MQTT:info  2023-06-10 09:28:31: MQTT publish: topic 'zigbee2mqttseeley/seeley-camera-plug', payload '{"last_seen":"2023-06-10T09:28:31-06:00","linkquality":69,"state":"ON"}'
Jun 10 09:28:31 argon-pi npm[2189]: Zigbee2MQTT:info  2023-06-10 09:28:31: MQTT publish: topic 'zigbee2mqttseeley/seeley-camera-plug', payload '{"last_seen":"2023-06-10T09:28:31-06:00","linkquality":69,"state":"ON"}'
Jun 10 09:28:31 argon-pi npm[2189]: Zigbee2MQTT:info  2023-06-10 09:28:31: MQTT publish: topic 'zigbee2mqttseeley/seeley-camera-plug', payload '{"last_seen":"2023-06-10T09:28:31-06:00","linkquality":69,"state":"ON"}'
Jun 10 09:28:33 argon-pi npm[2189]: Zigbee2MQTT:info  2023-06-10 09:28:33: MQTT publish: topic 'zigbee2mqttseeley/seeley-camera-plug', payload '{"last_seen":"2023-06-10T09:28:31-06:00","linkquality":69,"state":"ON"}'
```

devices.yaml
```
'0x00124b00ffffffff':
  friendly_name: seeley-camera-plug
  debounce: 2
```
(Device id modified for security.)

Device is a Sonoff S31 Lite zb (firmware version 20191113)
[image]

Version of Zigbee2MQTT is 1.30.2 (haven't upgraded as there 1.31 is experiencing a lot of updates, and this is a production setup).

### What did you expect to happen?

Not send duplicate messages for the device.

Note, I have debounce on a Tuya motion detection device and that seems to be working fine.
[image]
```
'0x804b50ffffffffff':
  friendly_name: seeley-mmwave
  debounce: 15
```

### How to reproduce it (minimal and precise)

Use a Sonoff zigbee plug and connect it to Zigbee2MQTT.

### Zigbee2MQTT version

1.30.2

### Adapter firmware version

20210708

[... 71 lines cut ...]
info  2023-06-10 09:53:31: MQTT publish: topic 'zigbee2mqttseeley/seeley-camera-plug', payload '{"last_seen":"2023-06-10T09:53:31-06:00","linkquality":72,"state":"ON"}'
info  2023-06-10 09:53:31: MQTT publish: topic 'zigbee2mqttseeley/seeley-camera-plug', payload '{"last_seen":"2023-06-10T09:53:31-06:00","linkquality":72,"state":"ON"}'
info  2023-06-10 09:53:31: MQTT publish: topic 'zigbee2mqttseeley/seeley-camera-plug', payload '{"last_seen":"2023-06-10T09:53:31-06:00","linkquality":72,"state":"ON"}'
debug 2023-06-10 09:53:31: Received Zigbee message from 'seeley-camera-plug', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
debug 2023-06-10 09:53:31: Received Zigbee message from 'seeley-camera-plug', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
debug 2023-06-10 09:53:31: Received Zigbee message from 'seeley-camera-plug', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
debug 2023-06-10 09:53:31: Received Zigbee message from 'seeley-camera-plug', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
debug 2023-06-10 09:53:31: Received Zigbee message from 'seeley-camera-plug', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
debug 2023-06-10 09:53:31: Received Zigbee message from 'seeley-camera-plug', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
debug 2023-06-10 09:53:31: Received Zigbee message from 'seeley-camera-plug', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
debug 2023-06-10 09:53:31: Received Zigbee message from 'seeley-camera-plug', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
debug 2023-06-10 09:53:31: Received Zigbee message from 'seeley-camera-plug', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
debug 2023-06-10 09:53:31: Received Zigbee message from 'seeley-camera-plug', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
info  2023-06-10 09:53:31: MQTT publish: topic 'zigbee2mqttseeley/seeley-camera-plug', payload '{"last_seen":"2023-06-10T09:53:31-06:00","linkquality":72,"state":"ON"}'
[... cut ...]
info  2023-06-10 09:53:31: MQTT publish: topic 'zigbee2mqttseeley/seeley-camera-plug', payload '{"last_seen":"2023-06-10T09:53:31-06:00","linkquality":72,"state":"ON"}'
info  2023-06-10 09:53:31: MQTT publish: topic 'zigbee2mqttseeley/seeley-camera-plug', payload '{"last_seen":"2023-06-10T09:53:31-06:00","linkquality":72,"state":"ON"}'
info  2023-06-10 09:53:33: MQTT publish: topic 'zigbee2mqttseeley/seeley-camera-plug', payload '{"last_seen":"2023-06-10T09:53:31-06:00","linkquality":72,"state":"ON"}'
```

nateww, 2023-07-13:
This issue has not been addressed

nateww, 2023-08-14:
Comment to indicate the issue still exists.

nateww, 2023-09-14:
Not stale

nateww, 2023-10-17:
Still not stale

nateww, 2023-10-25:
This issue still exists in V1.33.1 (stock version now as I've switched to using a docker container).

Fecos, 2024-01-14:
I have the same problem with a MOES UFO-R11. The "debounce" does not work.

rsamd5, 2024-01-18:
Same issue with _TZE200_bcusnqt8

config
```
  '0xXXX':
    friendly_name: energy_input
    debounce: 5
    debounce_ignore:
      - energy
      - produced_energy
```

```
info  2024-01-18 13:16:49: MQTT publish: topic 'zigbee2mqtt/energy_input', payload '{"current":0.58,"energy":817.06,"last_seen":"2024-01-18T13:16:49+01:00","linkquality":140,"power":75,"produced_energy":0,"voltage":235.9}'
info  2024-01-18 13:16:49: MQTT publish: topic 'zigbee2mqtt/energy_input', payload '{"current":0.58,"energy":817.06,"last_seen":"2024-01-18T13:16:49+01:00","linkquality":140,"power":75,"produced_energy":0,"voltage":235.9}'
info  2024-01-18 13:16:50: MQTT publish: topic 'zigbee2mqtt/energy_input', payload '{"current":0.58,"energy":817.06,"last_seen":"2024-01-18T13:16:49+01:00","linkquality":140,"power":75,"produced_energy":0,"voltage":235.9}'
```

Without debounce there are many more msgs for this device
```
info  2024-01-18 11:49:53: MQTT publish: topic 'zigbee2mqtt/energy_input', payload '{"current":0.9,"energy":816.42,"last_seen":"2024-01-18 11:49:52+01:00","linkquality":120,"power":186,"produced_energy":0,"voltage":236.5}'
info  2024-01-18 11:49:53: MQTT publish: topic 'zigbee2mqtt/energy_input', payload '{"current":0.9,"energy":816.42,"last_seen":"2024-01-18 11:49:52+01:00","linkquality":120,"power":186,"produced_energy":0,"voltage":236.5}'
info  2024-01-18 11:49:54: MQTT publish: topic 'zigbee2mqtt/energy_input', payload '{"current":0.9,"energy":816.42,"last_seen":"2024-01-18 11:49:52+01:00","linkquality":120,"power":186,"produced_energy":0,"voltage":236.5}'
info  2024-01-18 11:49:54: MQTT publish: topic 'zigbee2mqtt/energy_input', payload '{"current":0.9,"energy":816.42,"last_seen":"2024-01-18 11:49:52+01:00","linkquality":120,"power":186,"produced_energy":0,"voltage":236.5}'
info  2024-01-18 11:49:54: MQTT publish: topic 'zigbee2mqtt/energy_input', payload '{"current":0.9,"energy":816.42,"last_seen":"2024-01-18 11:49:52+01:00","linkquality":120,"power":186,"produced_energy":0,"voltage":236.5}'
info  2024-01-18 11:49:54: MQTT publish: topic 'zigbee2mqtt/energy_input', payload '{"current":0.9,"energy":816.42,"last_seen":"2024-01-18 11:49:52+01:00","linkquality":120,"power":186,"produced_energy":0,"voltage":236.5}'
info  2024-01-18 11:49:54: MQTT publish: topic 'zigbee2mqtt/energy_input', payload '{"current":0.9,"energy":816.86,"last_seen":"2024-01-18 11:49:52+01:00","linkquality":120,"power":186,"produced_energy":0,"voltage":236.5}'
info  2024-01-18 11:49:54: MQTT publish: topic 'zigbee2mqtt/energy_input', payload '{"current":0.9,"energy":816.86,"last_seen":"2024-01-18 11:49:52+01:00","linkquality":120,"power":186,"produced_energy":0,"voltage":236.5}'
info  2024-01-18 11:49:54: MQTT publish: topic 'zigbee2mqtt/energy_input', payload '{"current":0.9,"energy":816.86,"last_seen":"2024-01-18 11:49:52+01:00","linkquality":120,"power":186,"produced_energy":0,"voltage":236.5}'
info  2024-01-18 11:49:54: MQTT publish: topic 'zigbee2mqtt/energy_input', payload '{"current":0.9,"energy":816.86,"last_seen":"2024-01-18 11:49:52+01:00","linkquality":120,"power":186,"produced_energy":0,"voltage":236.5}'
info  2024-01-18 11:49:54: MQTT publish: topic 'zigbee2mqtt/energy_input', payload '{"current":0.9,"energy":816.86,"last_seen":"2024-01-18 11:49:52+01:00","linkquality":120,"power":186,"produced_energy":0,"voltage":236.5}'
info  2024-01-18 11:49:54: MQTT publish: topic 'zigbee2mqtt/energy_input', payload '{"current":0.9,"energy":816.86,"last_seen":"2024-01-18 11:49:52+01:00","linkquality":200,"power":186,"produced_energy":0,"voltage":236.5}'
info  2024-01-18 11:49:54: MQTT publish: topic 'zigbee2mqtt/energy_input', payload '{"current":0.9,"energy":816.86,"last_seen":"2024-01-18 11:49:52+01:00","linkquality":200,"power":186,"produced_energy":0,"voltage":236.5}'
info  2024-01-18 11:49:54: MQTT publish: topic 'zigbee2mqtt/energy_input', payload '{"current":0.9,"energy":816.86,"last_seen":"2024-01-18 11:49:52+01:00","linkquality":200,"power":186,"produced_energy":0,"voltage":236.5}'
info  2024-01-18 11:49:54: MQTT publish: topic 'zigbee2mqtt/energy_input', payload '{"current":0.9,"energy":816.86,"last_seen":"2024-01-18 11:49:52+01:00","linkquality":200,"power":186,"produced_energy":0,"voltage":236.5}'
info  2024-01-18 11:49:54: MQTT publish: topic 'zigbee2mqtt/energy_input', payload '{"current":0.9,"energy":816.86,"last_seen":"2024-01-18 11:49:52+01:00","linkquality":200,"power":186,"produced_energy":0,"voltage":236.5}'
info  2024-01-18 11:49:54: MQTT publish: topic 'zigbee2mqtt/energy_input', payload '{"current":0.9,"energy":816.86,"last_seen":"2024-01-18 11:49:52+01:00","linkquality":200,"power":186,"produced_energy":0,"voltage":236.5}'
info  2024-01-18 11:49:54: MQTT publish: topic 'zigbee2mqtt/energy_input', payload '{"current":0.9,"energy":816.86,"last_seen":"2024-01-18 11:49:52+01:00","linkquality":200,"power":186,"produced_energy":0,"voltage":236.5}'
info  2024-01-18 11:49:54: MQTT publish: topic 'zigbee2mqtt/energy_input', payload '{"current":0.9,"energy":816.86,"last_seen":"2024-01-18 11:49:53+01:00","linkquality":200,"power":186,"produced_energy":0,"voltage":236.5}'
info  2024-01-18 11:49:54: MQTT publish: topic 'zigbee2mqtt/energy_input', payload '{"current":0.9,"energy":816.86,"last_seen":"2024-01-18 11:49:53+01:00","linkquality":200,"power":186,"produced_energy":0,"voltage":236.5}'
[... cut ...]
info  2024-01-18 11:49:57: MQTT publish: topic 'zigbee2mqtt/energy_input', payload '{"current":0.9,"energy":816.86,"last_seen":"2024-01-18 11:49:57+01:00","linkquality":152,"power":186,"produced_energy":0,"voltage":236.5}'
info  2024-01-18 11:49:58: MQTT publish: topic 'zigbee2mqtt/energy_input', payload '{"current":0.9,"energy":816.86,"last_seen":"2024-01-18 11:49:58+01:00","linkquality":152,"power":186,"produced_energy":0,"voltage":236.5}'
info  2024-01-18 11:49:58: MQTT publish: topic 'zigbee2mqtt/energy_input', payload '{"current":0.9,"energy":816.86,"last_seen":"2024-01-18 11:49:58+01:00","linkquality":152,"power":186,"produced_energy":0,"voltage":236.5}'
```

hackepeterOli, 2024-01-20:
Same here with a TS0601 _TZE204_aoclfnxz
zigbee2mqtt Version 1.35.1

Koenkk, 2024-01-21:
Could you provide the debug log of this + your `configuration.yaml`?

nateww, 2024-01-22:
Which kind of debug log, standard, or including zigbee-herdsman?

https://www.zigbee2mqtt.io/guide/usage/debug.html

hackepeterOli, 2024-01-22:
Here is one debug log. The Spam-device is ThermostatBuero.
With debounce = 60 or 10 i get no more mqtt messages (that should actually be the goal, to get mqtt messages every 60s) 

Here is a log with debounce = 5

`Debug 2024-01-22 17:33:27Received Zigbee message from 'ThermostatBuero', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":36}],"seq":2182}' from endpoint 1 with groupID 0
Debug 2024-01-22 17:33:27Received Zigbee message from 'ThermostatBuero', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":36}],"seq":2438}' from endpoint 1 with groupID 0
Debug 2024-01-22 17:33:27Received Zigbee message from 'ThermostatBuero', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":36}],"seq":1670}' from endpoint 1 with groupID 0
Debug 2024-01-22 17:33:27Received Zigbee message from 'ThermostatBuero', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":36}],"seq":1926}' from endpoint 1 with groupID 0
Debug 2024-01-22 17:33:27Received Zigbee message from 'ThermostatBuero', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":36}],"seq":1926}' from endpoint 1 with groupID 0
Debug 2024-01-22 17:33:28Received Zigbee message from 'ThermostatBuero', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":36}],"seq":2182}' from endpoint 1 with groupID 0
Debug 2024-01-22 17:33:28Received Zigbee message from 'ThermostatBuero', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":36}],"seq":2438}' from endpoint 1 with groupID 0
Debug 2024-01-22 17:33:28Received Zigbee message from 'ThermostatBuero', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":36}],"seq":2694}' from endpoint 1 with groupID 0
Debug 2024-01-22 17:33:28Received Zigbee message from 'ThermostatBuero', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":36}],"seq":2182}' from endpoint 1 with groupID 0
Debug 2024-01-22 17:33:28Received Zigbee message from 'ThermostatBuero', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":36}],"seq":2438}' from endpoint 1 with groupID 0
Debug 2024-01-22 17:33:28Received Zigbee message from 'ThermostatBuero', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":36}],"seq":2694}' from endpoint 1 with groupID 0
Debug 2024-01-22 17:33:28Received Zigbee message from 'ThermostatBuero', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":36}],"seq":2438}' from endpoint 1 with groupID 0
Debug 2024-01-22 17:33:28Received Zigbee message from 'ThermostatBuero', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":36}],"seq":2694}' from endpoint 1 with groupID 0
Debug 2024-01-22 17:33:28Received Zigbee message from 'ThermostatBuero', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":36}],"seq":2950}' from endpoint 1 with groupID 0
Debug 2024-01-22 17:33:28Received Zigbee message from 'ThermostatBuero', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":36}],"seq":1670}' from endpoint 1 with groupID 0
Debug 2024-01-22 17:33:28Received Zigbee message from 'ThermostatBuero', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":36}],"seq":1926}' from endpoint 1 with groupID 0
Debug 2024-01-22 17:33:28Received Zigbee message from 'ThermostatBuero', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":36}],"seq":1926}' from endpoint 1 with groupID 0
Debug 2024-01-22 17:33:28Received Zigbee message from 'ThermostatBuero', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":36}],"seq":2182}' from endpoint 1 with groupID 0
Debug 2024-01-22 17:33:28Received Zigbee message from 'ThermostatBuero', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":36}],"seq":2438}' from endpoint 1 with groupID 0
Debug 2024-01-22 17:33:28Received Zigbee message from 'ThermostatBuero', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":36}],"seq":1926}' from endpoint 1 with groupID 0
Debug 2024-01-22 17:33:28Received Zigbee message from 'ThermostatBuero', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":36}],"seq":2182}' from endpoint 1 with groupID 0
[... cut ...]
Debug 2024-01-22 17:33:52Received Zigbee message from 'ThermostatBuero', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":36}],"seq":4486}' from endpoint 1 with groupID 0
Debug 2024-01-22 17:33:53Received Zigbee message from 'ThermostatBuero', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":36}],"seq":4486}' from endpoint 1 with groupID 0
Debug `

uncle-fed, 2024-01-22:
I have a seemingly similar issue with the device `TS0121` / `_TZ3000_g5xawfcq` [(BlitzWolf BW-SHP1](https://www.zigbee2mqtt.io/devices/TS0121_plug.html#tuya-ts0121_plug)).

Here is the general config:

```
homeassistant: false
permit_join: false
frontend:
  port: 8883
mqtt:
  base_topic: z2m
  server: mqtt://127.0.0.1
serial:
  port: /dev/ttyUSB0
advanced:
  pan_id: ...
  ext_pan_id: [ ... ]
  network_key: [ ... ]
  channel: 25
  legacy_api: false
  legacy_availability_payload: false
  cache_state: false
  cache_state_persistent: false
  cache_state_send_on_startup: false
  last_seen: ISO_8601
  log_level: debug
  homeassistant_legacy_entity_attributes: false
availability:
  active:
    timeout: 5
  passive:
    timeout: 160
device_options:
  retain: false
  legacy: false
devices:
    '0x842e14fffe05afdb':
    friendly_name: socket1
    debounce: 1
```

Here is the debug log that shows 3 messages being published instantaneously, even though `debounce: 1`  is set for this device.

Zigbee2MQTT:debug 2024-01-22 20:20:24: Received MQTT message on 'z2m/socket1/set' with data '{"state":"OFF"}'
Zigbee2MQTT:debug 2024-01-22 20:20:24: Publishing 'set' 'state' to 'home/socket/bedroom1/xmas'
  zigbee-herdsman:controller:endpoint Command 0x842e14fffe05afdb/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) +466ms
  zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x842e14fffe05afdb:6745/1 (0,0,1) +461ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":6745,"destendpoint":1,"srcendpoint":1,"clusterid":6,"transid":88,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[1,26,0]}} +461ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,13,36,1,89,26,1,1,6,0,88,0,30,3,1,26,0,51] +461ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100] +457ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0} +461ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,88,158] +2ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,88,158] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,88] - 158 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":88} +461ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,196,89,26,0,193] +17ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,196,89,26,0,193] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 196 - [89,26,0] - 193 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":6745,"relaycount":0,"relaylist":[]} +18ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,27,68,129,0,0,6,0,89,26,1,1,0,98,0,77,249,235,0,0,7,24,79,10,0,0,16,0,89,26,29,178] +7ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,27,68,129,0,0,6,0,89,26,1,1,0,98,0,77,249,235,0,0,7,24,79,10,0,0,16,0,89,26,29,178] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 27 - 2 - 4 - 129 - [0,0,6,0,89,26,1,1,0,98,0,77,249,235,0,0,7,24,79,10,0,0,16,0,89,26,29] - 178 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":6745,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":98,"securityuse":0,"timestamp":15464781,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[24,79,10,0,0,16,0]}} +8ms
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":79,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":0,"dataType":16,"attrData":0}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":6745,"endpoint":1,"linkquality":98,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}' +511ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +5ms
[... 11 lines cut ...]
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 196 - [118,41,0] - 221 +1ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":10614,"relaycount":0,"relaylist":[]} +842ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,35,68,129,0,0,0,0,118,41,1,1,0,65,0,46,223,236,0,0,15,8,114,10,1,0,32,68,226,255,32,31,228,255,32,0,118,41,29,164] +48ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,35,68,129,0,0,0,0,118,41,1,1,0,65,0,46,223,236,0,0,15,8,114,10,1,0,32,68,226,255,32,31,228,255,32,0,118,41,29,164] +1ms
[... cut ...]
  zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0xa4c138cc6261e318:31210/1 (0,0,1) +702ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":31210,"destendpoint":1,"srcendpoint":1,"clusterid":61184,"transid":90,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[16,189,11,2,0]}} +701ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,15,36,1,234,121,1,1,0,239,90,0,30,5,16,189,11,2,0,179] +702ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +7ms
```

elenril2, 2024-02-24:
Debounce does not work for the device BHT-002-GCLZB either. The [documentation ](https://www.zigbee2mqtt.io/devices/BHT-002-GCLZB.html#stop-message-flooding)advises to use this function, but the device still sends several dozen messages per second.

nateww, 2024-03-20:
Ping to see if this could get looked at...

YogoGit, 2024-04-03:
Ping o' rama to keep this alive.

leah-potato, 2024-06-28:
Also seeing the same with a [Tuya ZS06](https://www.zigbee2mqtt.io/devices/ZS06.html#tuya-zs06) ir blaster.

ivanfmartinez, 2024-06-28:
debounce does not work for this devices, as specified in debouce documentation : https://www.zigbee2mqtt.io/guide/configuration/devices-groups.html#common-device-options

"
Debounces messages of this device. When setting e.g. debounce: 1 and a message from a device is received, Zigbee2MQTT will not immediately publish this message. But it will combine it with other messages that are received max 1 seconds apart from each other. So there needs to be "N second of silence" from the device before the combined message is sent out.
"

Debounce works for devices that have some silence, and many devices are sending lots of messages that can be just ignored before sending to mqtt.

Here I have level sensor and ambient sensor with this behaviour that send same measurement with less than one second. 

- https://github.com/Koenkk/zigbee2mqtt/issues/21015

- https://www.zigbee2mqtt.io/devices/TS0601_air_quality_sensor.html#tuya-ts0601_air_quality_sensor

- https://www.zigbee2mqtt.io/devices/TS0601_smart_air_house_keeper.html#tuya-ts0601_smart_air_house_keeper

@Koenkk any other suggestion that can be applied to at least reduce number of mqtt messages for this devices ?

They need something to combine and use last value, but send each message at least X seconds, in my devices one each 30 or 60 seconds will be fine because ambient values and water level does not change very fast.

anoblet, 2024-06-28:
Perhaps a `throttle` method similar to lodash(https://lodash.com/docs/#throttle) could be useful?

https://medium.com/@bs903944/debounce-and-throttling-what-they-are-and-when-to-use-them-eadd272fe0be

ivanfmartinez, 2024-08-04:
I have made some tests and have a "dirty" proof of concept implementation that solves the problem for my devices.

```js
--- receive.js.ori      2024-08-04 14:39:39.437046808 -0300
+++ receive.js  2024-08-04 16:04:46.670479142 -0300
@@ -139,6 +139,32 @@
                 const now = Date.now();
                 if (this.elapsed[data.device.ieeeAddr]) {
                     payload.elapsed = now - this.elapsed[data.device.ieeeAddr];
+
+                    // very simple and dirty anti-spamming https://github.com/Koenkk/zigbee2mqtt/issues/17984
+                    //    as a proof of concept maybe Koenkk can find a better solution as the debounce does not help for my SPAMMER devices
+                    //       ambient sensor and water level that sometimes send mupliple messages on same second
+                    //    this will not help on zigbee network, but at least help on mqtt and homeassistant recorder and history
+                    //    this will not work for devices that have actions and specific events that are important
+                    //    it solves the SPAMMING on sensor devices that does not change values too fast and messages can be ignored
+                    // I dont know all the side effects of this code, but here is the ones that I found already
+                    //   - on web ui, the last-seen is only updated after a non ignored message
+                    //   - web ui are more responsive than before 
+                    //   - my homeassistant does not have a lot of data from this devices that are not need                    
+                    //   - my homeassistant became more responsive
+                    //   - the CPU load are sensible lower
+                    // using "SPAMMER" in description is an easy way to test without changing options on yaml
+                    if (data.device.options.min_elapsed || (data.device.options.description && data.device.options.description.includes("SPAMMER"))) {
+                        let min_elapsed = 30000;
+                        if (data.device.options.min_elapsed) {
+                            min_elapsed = data.device.options.min_elapsed;
+                        } 
+                        
+                        if (payload.elapsed < min_elapsed) {
+                            logger_1.default.debug(`Ignoring message from SPAMMER - ${data.device.ieeeAddr} -  ${data.device.options.friendly_name} - elapsed=${payload.elapsed} - min_elapsed=${min_elapsed}`);
+                            return;
+                        }
+                    }
+                    
                 }
                 this.elapsed[data.device.ieeeAddr] = now;
             }

```

jesse-kaufman, 2024-08-16:
If the web UI is more responsive with your dirty fix in place, I wonder if this is also part of the underlying cause of another issue I have where the settings reset before I can hit save: https://github.com/Koenkk/zigbee2mqtt/issues/23320

To me, it would make sense if the UI is being updated with every message .. *especially* on a network with 5 "spammy" devices (all Tuya mmWave sensors) .. I also would wonder if this is what's causing me to have to click buttons multiple times in the web UI to get a response 

I have two Zigbee networks, each having at least one spammy device .. the only differences are the coordinator firmware (just haven't upgraded the one, yet) and the number/types of devices .. here's my setup:

**Primary:**
- SMLIGHT SLZB-06 (non-M), firmware version: 20240315
- Z2M version: 1.39.1 commit: e132316
- 128 total devices
- 5 spammy devices (all Tuya mmWave sensors)
- Experience:
  - nearly impossible to use the web UI to change settings
  - as soon as I type the change, it's reset to the previous value (before I can hit save)
  - buttons often take numerous clicks to respond

**Secondary:**
- SMLIGHT SLZB-06 (non-M), firmware: 20240710
- Z2M version: 1.39.1 commit: e132316
- 22 total devices
- 1 spammy device (Tuya ultrasonic water level sensor)
- Experience
  - settings usually can be saved
  - settings that require more clicks/typing (filter, debounce ignore, cache ignore, etc.) are difficult but not always impossible to get saved
  - buttons are usually responsive and only occasionally require multiple clicks

I'm planning on updating the firmware in the next couple days, so that will take that variable out .. after I do that, I might toss the one spammy device from my secondary network onto the primary network and see if that changes anything on the responsiveness of the web UI

pcrednet, 2024-08-31:
> He realizado algunas pruebas y tengo una implementación de prueba de concepto "sucia" que resuelve el problema de mis dispositivos.
> 
> ```js
> --- receive.js.ori      2024-08-04 14:39:39.437046808 -0300
> +++ receive.js  2024-08-04 16:04:46.670479142 -0300
> @@ -139,6 +139,32 @@
>                  const now = Date.now();
>                  if (this.elapsed[data.device.ieeeAddr]) {
>                      payload.elapsed = now - this.elapsed[data.device.ieeeAddr];
> +
> +                    // very simple and dirty anti-spamming https://github.com/Koenkk/zigbee2mqtt/issues/17984
> +                    //    as a proof of concept maybe Koenkk can find a better solution as the debounce does not help for my SPAMMER devices
> +                    //       ambient sensor and water level that sometimes send mupliple messages on same second
> +                    //    this will not help on zigbee network, but at least help on mqtt and homeassistant recorder and history
> +                    //    this will not work for devices that have actions and specific events that are important
> +                    //    it solves the SPAMMING on sensor devices that does not change values too fast and messages can be ignored
> +                    // I dont know all the side effects of this code, but here is the ones that I found already
> +                    //   - on web ui, the last-seen is only updated after a non ignored message
> +                    //   - web ui are more responsive than before 
> +                    //   - my homeassistant does not have a lot of data from this devices that are not need                    
> +                    //   - my homeassistant became more responsive
> +                    //   - the CPU load are sensible lower
> +                    // using "SPAMMER" in description is an easy way to test without changing options on yaml
> +                    if (data.device.options.min_elapsed || (data.device.options.description && data.device.options.description.includes("SPAMMER"))) {
> +                        let min_elapsed = 30000;
> +                        if (data.device.options.min_elapsed) {
> +                            min_elapsed = data.device.options.min_elapsed;
> +                        } 
> +                        
> +                        if (payload.elapsed < min_elapsed) {
> +                            logger_1.default.debug(`Ignoring message from SPAMMER - ${data.device.ieeeAddr} -  ${data.device.options.friendly_name} - elapsed=${payload.elapsed} - min_elapsed=${min_elapsed}`);
> +                            return;
> +                        }
> +                    }
> +                    
>                  }
>                  this.elapsed[data.device.ieeeAddr] = now;
>              }
> ```

How do I implement this solution? On HassOs is possible?, or only on z2m in docker?
Thx in advance.

ivanfmartinez, 2024-08-31:
> How do I implement this solution? On HassOs is possible?, or only on z2m in docker?
> Thx in advance.

I don´t have knowledge about z2m on hassOS.
I use docker, and have override the container file using volume definition.
I have copied the original file, made the changes and override it using :docker-compose.override.yaml do dont make changes in my standard definition file.

this the  docker-compose.override.yaml to change this file : 

```
services:
  zigbee2mqtt_3:
    volumes:
      - ./data/tests/receive.js:/app/dist/extension/receive.js:ro

```

pcrednet, 2024-09-01:
the original file is receive.ts?

I have downloaded zigbee2mqtt 1.39.1 and there I see the file /lib/extension/receive.ts.
I renonmbrado it to receive.js and I have made the changes, when I start in docker, I get this error:

```python
Using '/app/data' as data directory
Starting Zigbee2MQTT without watchdog.
/app/dist/extension/receive.js:1
import bind from 'bind-decorator';
^^^^^^
SyntaxError: Cannot use import statement outside a module
    at internalCompileFunction (node:internal/vm:76:18)
    at wrapSafe (node:internal/modules/cjs/loader:1283:20)
    at Module._compile (node:internal/modules/cjs/loader:1328:27)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1422:10)
    at Module.load (node:internal/modules/cjs/loader:1203:32)
    at Function.Module._load (node:internal/modules/cjs/loader:1019:12)
    at Module.require (node:internal/modules/cjs/loader:1231:19)
    at require (node:internal/modules/helpers:177:18)
    at Object.<anonymous> (/app/lib/controller.ts:26:1)
    at Module._compile (node:internal/modules/cjs/loader:1364:14)
``` 

this is my stack ( on portainer... ):

```python
services:
  zigbee2mqtt:
    container_name: z2m2
    restart: always
    image: koenkk/zigbee2mqtt
    volumes:
      - /docker/z2m2/data:/app/data
      - /run/udev:/run/udev:ro
      - /docker/z2m2/receive.js:/app/dist/extension/receive.js:ro
    ports:
      - 84:8080
    environment:
      - TZ=Europe/Madrid

``` 
Could you explain it for newbies, thanks.

ivanfmartinez, 2024-09-01:
> I have downloaded zigbee2mqtt 1.39.1 and there I see the file /lib/extension/receive.ts.

The **ts** is the source for the **js** file. 

If you have only the **ts** file you must change the **ts** file, the **js** should be changed when it was already generated.

The changes are the same, but on different lines, just search for the lines of code that calculate the elapsed attribute.

pcrednet, 2024-09-01:
thank you for your help.
I have retrieved the js file from another istance of z2m that I have running and modified what you indicate.
The container starts, tomorrow I will test how it works with the multiple spammer devices I have.

Thank you very much.

YogoGit, 2025-01-05:
This hasn't been resolved, and still exists with the latest release 1.* release (1.42.0)

DaeMonSxy, 2025-02-07:
most likely involved my link also: 
https://github.com/Koenkk/zigbee2mqtt/discussions/26244

YogoGit, 2025-04-09:
Still not done

nateww, 2025-08-10:
Still not stale, just ignored.

closed 2025-10-20 by github-actions[bot]