Koenkk/zigbee2mqtt #17989: Tuya ZTH01 (_TZE200_yjjdcqsq) does not report data except LQI + LED shines red

dayt47, 2023-06-10:
### What happened?

My newly arrived Tuya Temperature and Humidity sensor ZTH01 (_TZE200_yjjdcqsq) does not report any data except the signal LQI although it states on the [supported device page](https://www.zigbee2mqtt.io/devices/ZTH01.html) to be supported. The sensor has 2 AAA batteries. It pairs without problems. Strange thing is that the red LED of the sensor is shining permanently. Tried a couple times to forcely remove and repair, but it did not work.

### What did you expect to happen?

Expected to get temperature and humidity data from the sensor

### Zigbee2MQTT version

1.31.2

### Adapter firmware version

20221226

### Adapter

CC2652P

dayt47, 2023-06-10:
Such a strange thing, now I get data reported. I really don't know what the problem was. Maybe I have to wait 30min or so? But Ii pushed the button several times, does this not push the data to the coordinator?

dayt47, 2023-06-12:
It doesn't work again...

sopelt, 2023-06-12:
@dayt47 did the sensor work at some point? I actually see the same behavior on ZHA/zigpy as well ...

dayt47, 2023-06-12:
Yes but I only got one data point and it is since than stuck at that specific temperature. P.s. The red LED is shining permanently again...

closed 2023-06-13 by dayt47