Koenkk/zigbee2mqtt #17995: [New device support]: _TZE200_5toc8efa

VoltickVL, 2023-06-11:
### Link

https://aliexpress.ru/item/1005004644629279.html?sku_id=12000030167735642

### Database entry

{"id":9,"type":"Router","ieeeAddr":"0x003c84fffef71d75","nwkAddr":28830,"manufId":4098,"manufName":"_TZE200_5toc8efa","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0005","65506":31,"65508":0,"65534":0,"modelId":"TS0601","manufacturerName":"_TZE200_5toc8efa","stackVersion":0,"dateCode":"","zclVersion":3,"appVersion":72,"powerSource":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-708457359},"lastSeen":1686477913285,"defaultSendRequestWhen":"immediate"}

### Comments

I tried to foillow the supporting new device page but got stuck and have warning:

`zigbee-herdsman-converters:Moes BHT-002: Unrecognized DP #102 with data {"dp":102,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,240]}}`

### External converter

```shell
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const fz = {...require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
const tz = {...require('zigbee-herdsman-converters/converters/toZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').toZigbee};
const ota = require('zigbee-herdsman-converters/lib/ota');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const dataPoints = {
    valve_state: 102,
};

const fzLocal = {
    valve_state: {
        cluster: 'manuSpecificTuya',
        type: ['commandDataReport'],
        convert: (model, msg, publish, options, meta) => {
            const result = {};
            for (const dpValue of msg.data.dpValues) {
                const dp = dpValue.dp; // First we get the data point ID
                const value = tuya.getDataValue(dpValue); // This function will take care of converting the data to proper JS type
                switch (dp) {
                case dataPoints.valve_state: {
                    result.valve_state = value;
                    break;
                }
                default: {
                    meta.logger.debug(`>>> UNKNOWN DP #${dp} with data "${JSON.stringify(dpValue)}"`);
                }
                }
            }
            return result;
        },
    },
};
const tzLocal = {
    valve_state: {
        key: ['valve_state'],
        convertSet: async (entity, key, value, meta) => {
            await tuya.sendDataPointValue(entity, dataPoints.valve_state, value);
        },
    },
};

const exposesLocal = {
    hour: (name) => exposes.numeric(name, ea.STATE_SET).withUnit('h').withValueMin(0).withValueMax(23),
    minute: (name) => exposes.numeric(name, ea.STATE_SET).withUnit('m').withValueMin(0).withValueMax(59),
    program_temperature: (name) => exposes.numeric(name, ea.STATE_SET).withUnit('°C')
        .withValueMin(5).withValueMax(35).withValueStep(0.5),
};
[... 185 lines cut ...]
                .withFeature(exposesLocal.hour('sunday_p2_hour'))
                .withFeature(exposesLocal.minute('sunday_p2_minute'))
                .withFeature(exposesLocal.program_temperature('sunday_p2_temperature'))
                .withFeature(exposesLocal.hour('sunday_p3_hour'))
                .withFeature(exposesLocal.minute('sunday_p3_minute'))
                .withFeature(exposesLocal.program_temperature('sunday_p3_temperature'))
                .withFeature(exposesLocal.hour('sunday_p4_hour'))
                .withFeature(exposesLocal.minute('sunday_p4_minute'))
                .withFeature(exposesLocal.program_temperature('sunday_p4_temperature')),
        ],
    meta: {
        tuyaDatapoints: [
            [101, 'schedule_monday', tuya.valueConverter.thermostatScheduleDayMultiDP],
            [102, 'schedule_tuesday', tuya.valueConverter.thermostatScheduleDayMultiDP],
            [103, 'schedule_wednesday', tuya.valueConverter.thermostatScheduleDayMultiDP],
            [104, 'schedule_thursday', tuya.valueConverter.thermostatScheduleDayMultiDP],
            [105, 'schedule_friday', tuya.valueConverter.thermostatScheduleDayMultiDP],
            [106, 'schedule_saturday', tuya.valueConverter.thermostatScheduleDayMultiDP],
            [107, 'schedule_sunday', tuya.valueConverter.thermostatScheduleDayMultiDP],
        ],
    },
	icon: 'data:image/jpeg;base64,/9j/4QRzRXhpZgAATU0AKgAAAAgADAEAAAMAAAABAoAAAAEBAAMAAAABAoAAAAECAAMAAAADAAAAngEGAAMAAAABAAIAAAESAAMAAAABAAEAAAEVAAMAAAABAAMAAAEaAAUAAAABAAAApAEbAAUAAAABAAAArAEoAAMAAAABAAMAAAExAAIAAAAdAAAAtAEyAAIAAAAUAAAA0YdpAAQAAAABAAAA6AAAASAACAAIAAgALcbAAAAnEAAtxsAAACcQQWRvYmUgUGhvdG9zaG9wIENDIChXaW5kb3dzKQAyMDIzOjAxOjIwIDExOjEyOjQ5AAAAAAAEkAAABwAAAAQwMjIxoAEAAwAAAAH//wAAoAIABAAAAAEAAAD6oAMABAAAAAEAAAD6AAAAAAAAAAYBAwADAAAAAQAGAAABGgAFAAAAAQAAAW4BGwAFAAAAAQAAAXYBKAADAAAAAQACAAACAQAEAAAAAQAAAX4CAgAEAAAAAQAAAu0AAAAAAAAASAAAAAEAAABIAAAAAf/Y/+0ADEFkb2JlX0NNAAL/7gAOQWRvYmUAZIAAAAAB/9sAhAAMCAgICQgMCQkMEQsKCxEVDwwMDxUYExMVExMYEQwMDAwMDBEMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMAQ0LCw0ODRAODhAUDg4OFBQODg4OFBEMDAwMDBERDAwMDAwMEQwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAwMDAz/wAARCAAYABgDASIAAhEBAxEB/90ABAAC/8QBPwAAAQUBAQEBAQEAAAAAAAAAAwABAgQFBgcICQoLAQABBQEBAQEBAQAAAAAAAAABAAIDBAUGBwgJCgsQAAEEAQMCBAIFBwYIBQMMMwEAAhEDBCESMQVBUWETInGBMgYUkaGxQiMkFVLBYjM0coLRQwclklPw4fFjczUWorKDJkSTVGRFwqN0NhfSVeJl8rOEw9N14/NGJ5SkhbSVxNTk9KW1xdXl9VZmdoaWprbG1ub2N0dXZ3eHl6e3x9fn9xEAAgIBAgQEAwQFBgcHBgU1AQACEQMhMRIEQVFhcSITBTKBkRShsUIjwVLR8DMkYuFygpJDUxVjczTxJQYWorKDByY1wtJEk1SjF2RFVTZ0ZeLys4TD03Xj80aUpIW0lcTU5PSltcXV5fVWZnaGlqa2xtbm9ic3R1dnd4eXp7fH/9oADAMBAAIRAxEAPwDvrer14+Hk5+dZ6GPivt9RzQ5wbXVY6lrtjA+x/wBH81ZTf8Z31IDQ09Ul3j6GR/6QQvrP/wCI7rf9XK/8/uXjBeA3Y6GgnV5/zeP5KPZD9E9M6pidUxac/Au9fEyNxqs2ubuDS+t3stayxv6Rn7qSw/8AFsR/zN6R/Ut/8+5CSX8FP//Q7ptLDTfi5OO66uy2/fW+lz63MfY97Wua9pZY1zUJvRugAy/o1BH/AIRZ/wCkl86pI610R9r9N4FVFQZRi4/2aioHbWKvSY0Hd7a2ANZ9J35qS+ZEktVaP//Z/+0MTlBob3Rvc2hvcCAzLjAAOEJJTQQEAAAAAAAPHAFaAAMbJUccAgAAAgAAADhCSU0EJQAAAAAAEM3P+n2ox74JBXB2rq8Fw044QklNBDoAAAAAAQ8AAAAQAAAAAQAAAAAAC3ByaW50T3V0cHV0AAAABQAAAABQc3RTYm9vbAEAAAAASW50ZWVudW0AAAAASW50ZQAAAABJbWcgAAAAD3ByaW50U2l4dGVlbkJpdGJvb2wAAAAAC3ByaW50ZXJOYW1lVEVYVAAAAA0AQwBhAG4AbwBuACAATQBGADMAMAAxADAAAAAAAA9wcmludFByb29mU2V0dXBPYmpjAAAAFQQfBDAEQAQwBDwENQRCBEAESwAgBEYEMgQ1BEIEPgQ/BEAEPgQxBEsAAAAAAApwcm9vZlNldHVwAAAAAQAAAABCbHRuZW51bQAAAAxidWlsdGluUHJvb2YAAAAJcHJvb2ZDTVlLADhCSU0EOwAAAAACLQAAABAAAAABAAAAAAAScHJpbnRPdXRwdXRPcHRpb25zAAAAFwAAAABDcHRuYm9vbAAAAAAAQ2xicmJvb2wAAAAAAFJnc01ib29sAAAAAABDcm5DYm9vbAAAAAAAQ250Q2Jvb2wAAAAAAExibHNib29sAAAAAABOZ3R2Ym9vbAAAAAAARW1sRGJvb2wAAAAAAEludHJib29sAAAAAABCY2tnT2JqYwAAAAEAAAAAAABSR0JDAAAAAwAAAABSZCAgZG91YkBv4AAAAAAAAAAAAEdybiBkb3ViQG/gAAAAAAAAAAAAQmwgIGRvdWJAb+AAAAAAAAAAAABCcmRUVW50RiNSbHQAAAAAAAAAAAAAAABCbGQgVW50RiNSbHQAAAAAAAAAAAAAAABSc2x0VW50RiNSbHRA6soAAAAAAAAAAAp2ZWN0b3JEYXRhYm9vbAEAAAAAUGdQc2VudW0AAAAAUGdQcwAAAABQZ1BDAAAAAExlZnRVbnRGI1JsdAAAAAAAAAAAAAAAAFRvcCBVbnRGI1JsdAAAAAAAAAAAAAAAAFNjbCBVbnRGI1ByY0BZAAAAAAAAAAAAEGNyb3BXaGVuUHJpbnRpbmdib29sAAAAAA5jcm9wUmVjdEJvdHRvbWxvbmcAAAAAAAAADGNyb3BSZWN0TGVmdGxvbmcAAAAAAAAADWNyb3BSZWN0UmlnaHRsb25nAAAAAAAAAAtjcm9wUmVjdFRvcGxvbmcAAAAAADhCSU0D7QAAAAAAEAL6AAAAAgABAvoAAAACAAE4QklNBCYAAAAAAA4AAAAAAAAAAAAAP4AAADhCSU0EDQAAAAAABAAAAB44QklNBBkAAAAAAAQAAAAeOEJJTQPzAAAAAAAJAAAAAAAAAAABADhCSU0nEAAAAAAACgABAAAAAAAAAAE4QklNA/UAAAAAAEgAL2ZmAAEAbGZmAAYAAAAAAAEAL2ZmAAEAoZmaAAYAAAAAAAEAMgAAAAEAWgAAAAYAAAAAAAEANQAAAAEALQAAAAYAAAAAAAE4QklNA/gAAAAAAHAAAP////////////////////////////8D6AAAAAD/////////////////////////////A+gAAAAA/////////////////////////////wPoAAAAAP////////////////////////////8D6AAAOEJJTQQIAAAAAAAQAAAAAQAAAkAAAAJAAAAAADhCSU0EHgAAAAAABAAAAAA4QklNBBoAAAAAA2UAAAAGAAAAAAAAAAAAAAD6AAAA+gAAABgAQgBTAEUARQBEAC0AWgBpAGcAYgBlAGUALgBqAHAAZwBfADYANAAwAHgANgA0ADAAAAABAAAAAAAAAAAAAAAAAAAAAAAAAAEAAAAAAAAAAAAAAPoAAAD6AAAAAAAAAAAAAAAAAAAAAAEAAAAAAAAAAAAAAAAAAAAAAAAAEAAAAAEAAAAAAABudWxsAAAAAgAAAAZib3VuZHNPYmpjAAAAAQAAAAAAAFJjdDEAAAAEAAAAAFRvcCBsb25nAAAAAAAAAABMZWZ0bG9uZwAAAAAAAAAAQnRvbWxvbmcAAAD6AAAAAFJnaHRsb25nAAAA+gAAAAZzbGljZXNWbExzAAAAAU9ia [... cut ...] Gv+/D1rxm/hHXv9ln6e/wCeXi/6mf8ASHv30Y/34et+O38I67Hxn6dDajtWFv8ABpCQf8DZRx701pU18THWvHb+Edd/7LT09qDDatOpHFlbj/eUN/evovWQ9b8dvMD9nXbfGrp5v1bUpj/X1WJP1vcKD78LLOZDTrRmYilBTpV7R6d652NXvlttbYoKHKNH4RX6WkqY4SCpjhdyRCGVjcqATfn26lrGtCaluqF2OPLoTvahRpAHVOve9jgK8evde9+691737r3Xvfuvdf/W3+Pfuvde9+691737r3Xvfuvde9+691737r3Xvfuvde9+691737r3Xvfuvde9+691737r3Xvfuvde9+691737r3Xvfuvde9+691737r3Xvfuvde9+691737r3Xvfuvde9+691737r3Xvfuvde9+691737r3Xvfuvde9+691737r3Xvfuvde9+691//9k='
};
module.exports = definition;
```

VoltickVL, 2023-06-29:
Plz tell me, when you add new device?

closed 2023-08-06 by github-actions[bot]

Malaysk, 2023-10-02:
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const fz = {...require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
const tz = {...require('zigbee-herdsman-converters/converters/toZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').toZigbee};
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const exposesLocal = {
hour: (name) => exposes.numeric(name, ea.STATE_SET).withUnit('h').withValueMin(0).withValueMax(23),
minute: (name) => exposes.numeric(name, ea.STATE_SET).withUnit('m').withValueMin(0).withValueMax(59),
program_temperature: (name) => exposes.numeric(name, ea.STATE_SET).withUnit('°C')
.withValueMin(5).withValueMax(35).withValueStep(0.5),
};

const definition = {
zigbeeModel: ['TS0601'],
model: 'BHT-006-GCLZB',
fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_5toc8efa'}],
vendor: 'BSED',
description: 'BSED thermostat',
fromZigbee: [fz.legacy.moes_thermostat],
toZigbee: [tz.legacy.moes_thermostat_child_lock, tz.legacy.moes_thermostat_current_heating_setpoint, tz.legacy.moes_thermostat_mode,
tz.legacy.moes_thermostat_standby, tz.legacy.moes_thermostat_sensor, tz.legacy.moes_thermostat_calibration,
tz.legacy.moes_thermostat_deadzone_temperature, tz.legacy.moes_thermostat_max_temperature_limit, tz.legacy.moes_thermostat_min_temperature_limit],
exposes: [e.child_lock(), e.deadzone_temperature(), e.max_temperature_limit(), e.min_temperature_limit(),
exposes.climate().withSetpoint('current_heating_setpoint', 5, 35, 1, ea.STATE_SET)
.withLocalTemperature(ea.STATE).withLocalTemperatureCalibration(-30, 30, 0.1, ea.STATE_SET)
.withSystemMode(['off', 'heat'], ea.STATE_SET).withRunningState(['idle', 'heat', 'cool'], ea.STATE)
.withPreset(['hold', 'program']),
e.temperature_sensor_select(['IN', 'AL', 'OU']),
exposes.composite('program', 'program', ea.STATE_SET).withDescription('Time of day and setpoint to use when in program mode')
.withFeature(exposesLocal.hour('weekdays_p1_hour'))
.withFeature(exposesLocal.minute('weekdays_p1_minute'))
.withFeature(exposesLocal.program_temperature('weekdays_p1_temperature'))
.withFeature(exposesLocal.hour('weekdays_p2_hour'))
.withFeature(exposesLocal.minute('weekdays_p2_minute'))
.withFeature(exposesLocal.program_temperature('weekdays_p2_temperature'))
.withFeature(exposesLocal.hour('weekdays_p3_hour'))
.withFeature(exposesLocal.minute('weekdays_p3_minute'))
.withFeature(exposesLocal.program_temperature('weekdays_p3_temperature'))
.withFeature(exposesLocal.hour('weekdays_p4_hour'))
.withFeature(exposesLocal.minute('weekdays_p4_minute'))
.withFeature(exposesLocal.program_temperature('weekdays_p4_temperature'))
.withFeature(exposesLocal.hour('saturday_p1_hour'))
.withFeature(exposesLocal.minute('saturday_p1_minute'))
.withFeature(exposesLocal.program_temperature('saturday_p1_temperature'))
.withFeature(exposesLocal.hour('saturday_p2_hour'))
.withFeature(exposesLocal.minute('saturday_p2_minute'))
.withFeature(exposesLocal.program_temperature('saturday_p2_temperature'))
.withFeature(exposesLocal.hour('saturday_p3_hour'))
.withFeature(exposesLocal.minute('saturday_p3_minute'))
.withFeature(exposesLocal.program_temperature('saturday_p3_temperature'))
.withFeature(exposesLocal.hour('saturday_p4_hour'))
.withFeature(exposesLocal.minute('saturday_p4_minute'))
.withFeature(exposesLocal.program_temperature('saturday_p4_temperature'))
.withFeature(exposesLocal.hour('sunday_p1_hour'))
.withFeature(exposesLocal.minute('sunday_p1_minute'))
.withFeature(exposesLocal.program_temperature('sunday_p1_temperature'))
.withFeature(exposesLocal.hour('sunday_p2_hour'))
.withFeature(exposesLocal.minute('sunday_p2_minute'))
.withFeature(exposesLocal.program_temperature('sunday_p2_temperature'))
.withFeature(exposesLocal.hour('sunday_p3_hour'))
.withFeature(exposesLocal.minute('sunday_p3_minute'))
.withFeature(exposesLocal.program_temperature('sunday_p3_temperature'))
.withFeature(exposesLocal.hour('sunday_p4_hour'))
.withFeature(exposesLocal.minute('sunday_p4_minute'))
.withFeature(exposesLocal.program_temperature('sunday_p4_temperature')),
],
onEvent: tuya.onEventSetLocalTime,
};

module.exports = definition;