Koenkk/zigbee2mqtt #17997: TS0601_soil is not shown temperature

smarteshome2020, 2023-06-11:
### What happened?

Hi,

my new device is not shown the temperature value.
Is this a bug of zigbee2mqtt or is the device broken?

{
    "battery": 50,
    "battery_state": "medium",
    "last_seen": "2023-06-11T15:23:54+02:00",
    "linkquality": 58,
    "soil_moisture": 94,
    "temperature": 0,
    "temperature_unit": "celsius"
}

### Zigbee2MQTT version

1.31.2-dev commit: 0dd89967

### Adapter firmware version

 20221226

### Adapter

zStack3x0

### Debug log

no debug existing.

alialaur, 2023-07-14:
Mine also stopped updating temperature after zigbee2mqtt update. At least it was working when I bought the device. 
Now it's stuck at 33 C.

Error 2023-07-14 10:40:24Exception while calling fromZigbee converter: 'temperature_precision' is not a number, got string ()}
Info 2023-07-14 10:40:24MQTT publish: topic 'zigbee2mqtt_garden/Soil sensor', payload '{"battery":80,"battery_state":"high","last_seen":"2023-07-14T10:40:24+03:00","linkquality":127,"soil_moisture":97,"temperature":33,"temperature_unit":"celsius"}'

closed 2023-08-22 by github-actions[bot]

herbalizer404, 2024-03-13:
Same issue for me