Koenkk/zigbee2mqtt #18002: Tuya JM-TRH-ZGB-V1 clock error

z-x-f, 2023-06-12:
### What happened?

Since the update of (https://github.com/Koenkk/zigbee2mqtt/commit/ddcab827ad086949b87d5eaff8d261d8f2d529bf), my tuya jm-trh-zgb-v1 has a clock error. Please help me to find out where the problem is.

### Zigbee2MQTT version

1.31.2-dev commit: 0dd89967

### Adapter firmware version

20230410

### Adapter

zigstar

b2un0, 2023-06-12:
i have the same issue with this two devices:
[M-TRH-ZGB-V1](https://www.zigbee2mqtt.io/devices/JM-TRH-ZGB-V1.html#tuya-jm-trh-zgb-v1)
 [SZ-T04](https://www.zigbee2mqtt.io/devices/SZ-T04.html#nous-sz-t04)

Koenkk, 2023-06-12:
@z-x-f does it still work in z2m 1.31.0?

z-x-f, 2023-06-12:
[1.31.1](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.31.1) commit: [3572c607](https://github.com/Koenkk/zigbee2mqtt/commit/3572c607), have the same problem.

z-x-f, 2023-06-12:
https://github.com/Koenkk/zigbee2mqtt/commit/21f51258a9749d2ccba212997c3ab2f419d62142 works normally, but https://github.com/Koenkk/zigbee2mqtt/commit/7df6b2bab54616ce6573ca8633bec4b0a4b8fbd6 is abnormal again.

Koenkk, 2023-06-13:
@z-x-f can you add a log statement here: https://github.com/Koenkk/zigbee-herdsman-converters/blob/1d4e3cc732397fe1a2cd7ba7075f888d404ba3dd/src/lib/tuya.ts#L106 and a `console.log(error)` here: https://github.com/Koenkk/zigbee-herdsman-converters/blob/1d4e3cc732397fe1a2cd7ba7075f888d404ba3dd/src/lib/tuya.ts#L122 to see where it goes wrong?

z-x-f, 2023-06-16:
```js
DataConfirmError: Command 0xa4c1383a1e6417c1/1 manuSpecificTuya.mcuSyncTime({"payloadSize":8,"payload":[100,null,140,236,100,null,140,108]}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))
    at ZStackAdapter.sendZclFrameToEndpointInternal (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:415:23)
    at Queue.executeNext (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32) {
  code: 205 
```

Koenkk, 2023-06-17:
It is, this is the time sync command. It seems for some reason the device is not reachable, did you already try to re-pair it?

b2un0, 2023-06-17:
I tried to repair the affected devices, it does not help.

z-x-f, 2023-06-19:
When the time synchronization is wrong, I found that there is null in the payload, I'm not sure why it is like this?

This is the normal log when the time is synchronized:
```js
  Zigbee2MQTT:debug 2023-06-19 21:06:00: Received Zigbee message from '0xa4c1386584b6bf43', type 'commandMcuSyncTime', cluster 'manuSpecificTuya', data '{"payloadSize":117}' from endpoint 1 with groupID 0
  zigbee-herdsman:controller:endpoint Command 0xa4c1386584b6bf43/1 manuSpecificTuya.mcuSyncTime({"payloadSize":8,"payload":[100,144,82,184,100,144,195,56]}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) +131ms
  zigbee-herdsman:controller:endpoint DefaultResponse 0xa4c1386584b6bf43/1 61184(36, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) +0ms
```
This is the log when the time synchronization is not normal:
```js
Zigbee2MQTT:debug 2023-06-19 21:01:09: Received Zigbee message from '0xa4c1386584b6bf43', type 'commandMcuSyncTime', cluster 'manuSpecificTuya', data '{"payloadSize":75}' from endpoint 1 with groupID 0
  zigbee-herdsman:controller:endpoint Command 0xa4c1386584b6bf43/1 manuSpecificTuya.mcuSyncTime({"payloadSize":8,"payload":[100,null,144,149,100,null,144,21]}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) +129ms
  zigbee-herdsman:controller:endpoint DefaultResponse 0xa4c1386584b6bf43/1 61184(36, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) +0ms
```

z-x-f, 2023-06-19:
I modified `convertDecimalValueTo4ByteHexArray` as follows, which temporarily solved the problem. Is this modification feasible?

```js
function convertDecimalValueTo4ByteHexArray(value) {
    const hexValue = Number(value).toString(16).padStart(8, '0');
    const chunk1 = parseInt(hexValue.substring(0, 2), 16);
    const chunk2 = parseInt(hexValue.substring(2, 4), 16);
    const chunk3 = parseInt(hexValue.substring(4, 6), 16);
    const chunk4 = parseInt(hexValue.substring(6), 16);
    return [chunk1, chunk2, chunk3, chunk4];
}
```

```js
=============
utcTime: 1687191707
localTime: 1687220507
=============
{
  payloadSize: 8,
  payload: [
    100, 144, 128,
    155, 100, 144,
    241,  27
  ]
}
```

Koenkk, 2023-06-19:
Yes, but I don't know what the diff with existing code is. Does the following also work?

```js
function convertDecimalValueTo4ByteHexArray(value) {
    const hexValue = Number(value).toString(16).padStart(8, '0');
    const chunk1 = hexValue.substring(0, 2);
    const chunk2 = hexValue.substring(2, 4);
    const chunk3 = hexValue.substring(4, 6);
    const chunk4 = hexValue.substring(6);
    return [chunk1, chunk2, chunk3, chunk4].map((hexVal) => parseInt(hexVal, 16));
}
```

z-x-f, 2023-06-20:
This is also working fine.
```js
function convertDecimalValueTo4ByteHexArray(value) {
    const hexValue = Number(value).toString(16).padStart(8, '0');
    const chunk1 = hexValue.substring(0, 2);
    const chunk2 = hexValue.substring(2, 4);
    const chunk3 = hexValue.substring(4, 6);
    const chunk4 = hexValue.substring(6);
    return [chunk1, chunk2, chunk3, chunk4].map((hexVal) => parseInt(hexVal, 16));
}

=============
utcTime: 1687228201
localTime: 1687257001
=============
{
  payloadSize: 8,
  payload: [
    100, 145,  15,  41,
    100, 145, 127, 169
  ]
}
```
By the way, test the original again, there is still an error.
```js
=============
utcTime: 1687228553
localTime: 1687257353
=============
{
  payloadSize: 8,
  payload: [
    100, NaN, 145, 137,
    100, NaN, 145,   9
  ]
}
```

Koenkk, 2023-06-20:
Thanks, pushed the fix!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-06-20 by Koenkk

b2un0, 2023-06-21:
> i have the same issue with this two devices: [M-TRH-ZGB-V1](https://www.zigbee2mqtt.io/devices/JM-TRH-ZGB-V1.html#tuya-jm-trh-zgb-v1) [SZ-T04](https://www.zigbee2mqtt.io/devices/SZ-T04.html#nous-sz-t04)

can confirm, both devices works again.  🎉 
thanks!