Koenkk/zigbee2mqtt #18005: [New device support]: Namron 4512749-N Thermostat outlet socket

danielfinans, 2023-06-12:
### Link

https://homey.app/en-us/app/net.filllip-namron/Namron/

### Database entry

{"id":68,"type":"Router","ieeeAddr":"0x7fff0001000002ba","nwkAddr":50832,"manufId":4644,"manufName":"NAMRON AS","powerSource":"Mains (single phase)","modelId":"4512749-N","epList":[1,2,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[0,3,4,5,6,9,1026,1794,2820,2821,4096,64716],"outClusterList":[25,4096],"clusters":{"genBasic":{"attributes":{"appVersion":0,"stackVersion":0}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":769,"inClusterList":[0,4,5,513],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"appVersion":0,"stackVersion":0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":102,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":0,"hwVersion":1,"dateCode":"NULL","swBuildId":"6.9.1.0","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1686598764528,"defaultSendRequestWhen":"immediate"}

### Comments

The Namron 4512749 Thermostat outlet socket is supported by Z2M. 
When pairing, this plug shows up as 4512749**-N**.

Is it possible to get support for the 4512749-N? 

The only place I could actually find the "N" version whas from the linked Homey-site. 

This is the one i ordered: https://www.elektroimportoren.no/namron-zigbee-termostatplugg-16a/4512749/Product.html 
And this is the attached user manual: https://www.elektroimportoren.no/docs/lib/4512749-Brukerveiledning-5.pdf 
The physical one in the box contains same numbering (70110005) but is not an exact match.

danielfinans, 2023-06-13:
First take. 
Tried creating 4512749-N.js in same folder as Z2M configuration.yaml. 
Used the converter example for a switch, along with info from the existing converter 4512749 from [here](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/namron.ts)

Added 
`external_converters:
  - 4512749-N.js`
to configuration.yaml

The 4512749-N.js looks like this: 
`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['4512749-N'],
    model: '4512749-N',
    vendor: 'Namron',
    fromZigbee: [fz.metering, fz.electrical_measurement, fz.on_off, fz.temperature],
    toZigbee: [tz.on_off, tz.power_on_behavior],
    exposes: [e.temperature(), e.power(), e.current(), e.voltage(), e.switch(), e.power_on_behavior()],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement', 'msTemperatureMeasurement']);
        await endpoint.read('haElectricalMeasurement', ['acVoltageMultiplier', 'acVoltageDivisor']);
        await endpoint.read('haElectricalMeasurement', ['acPowerMultiplier', 'acPowerDivisor']);
        await endpoint.read('haElectricalMeasurement', ['acCurrentMultiplier', 'acCurrentDivisor']);
        await reporting.onOff(endpoint);
        await reporting.temperature(endpoint);
        await reporting.rmsVoltage(endpoint);
        await reporting.rmsCurrent(endpoint);
        await reporting.activePower(endpoint);
    },
};

module.exports = definition;`

And gives the following errors: 
`[08:54:05] INFO: Starting Zigbee2MQTT...
/app/node_modules/zigbee-herdsman-converters/index.js:48
        assert.notStrictEqual(undefined, definition[field], `Converter field ${field} is undefined`);
               ^
AssertionError [ERR_ASSERTION]: Converter field description is undefined
    at validateDefinition (/app/node_modules/zigbee-herdsman-converters/index.js:48:16)
    at Object.addDefinition [as addDeviceDefinition] (/app/node_modules/zigbee-herdsman-converters/index.js:73:5)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:15:17)
    at new Controller (/app/lib/controller.ts:84:58)
    at start (/app/index.js:106:18)`

Second take: 
Changed the naming of the converter to 4512749N.js. Changed this in config.yaml as well. 
Same error.

danielfinans, 2023-06-18:
Device stopped reporting states. Unpaired and paired again. 

No luck, will have to dig deeper. [image]

[image]

danielfinans, 2023-06-19:
Fixed for now. Creating Pull. Closing issue.

closed 2023-06-19 by danielfinans

ok-67, 2023-10-13:
It seems like this only is supported as an on/off plug with temperature, voltage and power measurements.
In ZHA it is supported as a thermostat (however the thermostat modus must be sett using ZHA-toolkit to set the vendor proprietary attributt 0x1000 to 1 for thermostat operation {or 0 for on/off operation}).