Koenkk/zigbee2mqtt #18007: LifeControl MCLH-04

ChapayTv, 2023-06-13:
### What happened?

Battery level always shows 50%

### Zigbee2MQTT version

1.31.2-1

### Adapter firmware version

20221031195840-if00

### Adapter

usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2

### Debug log

[mqtt-833b8daecdfa6cbb4c4dbdc0b0067543-Датчик окна на кухне-0be5853d64b527ad9b667ed9d148e3b1.json.txt](https://github.com/Koenkk/zigbee2mqtt/files/11730781/mqtt-833b8daecdfa6cbb4c4dbdc0b0067543-.-0be5853d64b527ad9b667ed9d148e3b1.json.txt)

ChapayTv, 2023-06-13:
> ### What happened?
> Battery level always shows 50%
>
> ### What did you expect to happen?
> _No response_
>
> ### How to reproduce it (minimal and precise)
> _No response_
>
> ### Zigbee2MQTT version
> 1.31.2-1 [...]

Koenkk, 2023-06-13:
Fixed!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-06-13 by Koenkk

ChapayTv, 2023-06-13:
> Fixed!
> 
> Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Thanks, I'll be waiting for the general release.