Koenkk/zigbee2mqtt #18011: [Feature request]: Zigbee Fingerbot Additional Options

Gr8z, 2023-06-13:
### Is your feature request related to a problem? Please describe

The zigbee fingerbot only has limited options such as:

Switch on/off and power_on_behaviour

### Describe the solution you'd like

add support for more options such as:

- Mode (Click/Switch)
- Touch Control (On/Off)
- Down Movement (0-100%)
- Up Movement (0-100%)
- etc

### Describe alternatives you've considered

N/A

### Additional context

N/A

Gr8z, 2023-06-13:
Maybe this helps?
```
debug 2023-06-13 20:56:13: No converter available for 'TS0001_1' with cluster 'manuSpecificTuya' and type 'commandActiveStatusReport' and data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":101},{"data":{"data":[0,0,0,85],"type":"Buffer"},"datatype":2,"dp":102}],"seq":10752}'
debug 2023-06-13 20:56:13: No converter available for 'TS0001_1' with cluster 'manuSpecificTuya' and type 'commandActiveStatusReport' and data '{"dpValues":[{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":103},{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":104}],"seq":11008}'
debug 2023-06-13 20:56:14: No converter available for 'TS0001_1' with cluster 'manuSpecificTuya' and type 'commandActiveStatusReport' and data '{"dpValues":[{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":106},{"data":{"data":[0,0,0,100],"type":"Buffer"},"datatype":2,"dp":105}],"seq":11264}'
debug 2023-06-13 20:56:14: No converter available for 'TS0001_1' with cluster 'manuSpecificTuya' and type 'commandActiveStatusReport' and data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":108},{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":107}],"seq":11520}'
debug 2023-06-13 20:56:18: No converter available for 'TS0001_1' with cluster 'manuSpecificTuya' and type 'commandActiveStatusReport' and data '{"dpValues":[{"data":{"data":[255,255,255,0],"type":"Buffer"},"datatype":0,"dp":109}],"seq":11776}'
debug 2023-06-13 20:56:18: No converter available for 'TS0001_1' with cluster 'manuSpecificTuya' and type 'commandActiveStatusReport' and data '{"dpValues":[{"data":{"data":[0,0,0,237],"type":"Buffer"},"datatype":2,"dp":111}],"seq":12032}'
```

Gr8z, 2023-06-14:
I did some Zigbee sniffing and compiled this list as well:

code | Name | type | values
-- | -- | -- | --
101 | Mode | enum | Click/Switch/Program
107 | Touch Control | bool | on/off
104 | Reverse | enum | up_on/up_off
102 | Down Movement | number | 50-100
106 | Up Movement | number | 0-50
103 | Sustain Time | value | 0s-10s
1 | Switch | enum | On/Off
105 | Battery | number | 0-100

Gr8z, 2023-06-14:
Tried this code by following the docs but couldn't get anywhere :(

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee')
const tz = require('zigbee-herdsman-converters/converters/toZigbee')
const exposes = require('zigbee-herdsman-converters/lib/exposes')
const reporting = require('zigbee-herdsman-converters/lib/reporting')
const extend = require('zigbee-herdsman-converters/lib/extend')
const e = exposes.presets
const ea = exposes.access
const tuya = require('zigbee-herdsman-converters/lib/tuya')

const definition = {
  // Since a lot of TuYa devices use the same modelID, but use different datapoints
  // it's necessary to provide a fingerprint instead of a zigbeeModel
  fingerprint: [
    {
      // The model ID from: Device with modelID 'TS0601' is not supported
      // You may need to add \u0000 at the end of the name in some cases
      modelID: 'TS0001_1',
      // The manufacturer name from: Device with modelID 'TS0601' is not supported.
      manufacturerName: '_TZ3210_dse8ogfy',
    },
  ],
  zigbeeModel: ['TS0001'],
  model: 'TS0001_1',
  vendor: 'Adaprox',
  description: 'White Zigbee FingerBot plus',
  fromZigbee: [tuya.fz.datapoints],
  toZigbee: [tuya.tz.datapoints],
  configure: tuya.configureMagicPacket,
  exposes: [
    // Here you should put all functionality that your device exposes
    e.battery(),
    e.enum('101', ea.STATE_SET, ['Click', 'Switch', 'Program']),
    e.boolean('107', ea.STATE_SET),
    e.enum('104', ea.STATE_SET, ['up_on', 'up_off']),
    e.percent('102', ea.STATE_SET),
    e.percent('106', ea.STATE_SET),
    e.value('103', ea.STATE_SET, {min: 0, max: 10, unit: 's'}),
    e.enum('1', ea.STATE_SET, ['On', 'Off']),
  ],
  meta: {
    // All datapoints go in here
    tuyaDatapoints: [
      [101, 'mode', tuya.converters.enum({0: 'Click', 1: 'Switch', 2: 'Program'})],
      [107, 'touch_control', tuya.converters.boolean],
      [104, 'reverse', tuya.converters.enum({0: 'up_on', 1: 'up_off'})],
      [102, 'down_movement', tuya.converters.percent],
      [106, 'up_movement', tuya.converters.percent],
      [103, 'sustain_time', tuya.converters.value({min: 0, max: 10, unit: 's'})],
      [1, 'switch', tuya.converters.enum({1: 'On', 0: 'Off'})],
      [105, 'battery', tuya.converters.value({min: 0, max: 100})],
    ],
  },
}

module.exports = definition
```

Koenkk, 2023-06-14:
Instead of e.g. `e.enum('101', ea.STATE_SET, ['Click', 'Switch', 'Program']),` use `e.enum('mode', ea.STATE_SET, ['Click', 'Switch', 'Program']),`

Koenkk, 2023-06-15:
This should be fixed after https://github.com/Koenkk/zigbee-herdsman-converters/pull/5883 is merged.

Gr8z, 2023-06-16:
Awesome, thank you!

closed 2023-06-16 by Gr8z

didi789, 2023-06-20:
https://github.com/Koenkk/zigbee-herdsman-converters/pull/5883/commits merged, now what?

kirovilya, 2023-06-20:
@didi789 update to dev/edge version https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html

kmanan, 2024-04-25:
> I did some Zigbee sniffing and compiled this list as well:
>
> code	Name	type	values
> 101	Mode	enum	Click/Switch/Program
> 107	Touch Control	bool	on/off
> 104	Reverse	enum	up_on/up_off
> 102	Down Movement [...]

What does up_on and up_off mean in this case, sorry maybe it's obvious to everyone but for reverse up_on and up_off don't make sense to me.

leec-666, 2026-09-01:
@TheToto thanks for this.  After getting help from ChatGPT to make the program I need for my case, it correctly determined that the second and third numbers are for repetitions and final position.  Don't know about the first value.

This program
* pushes to 100% and holds for 2 seconds
* retracts to 50%
* pushes to 100%
* retracts to 50%
* pushes to 100%
* retracts to 0% and ends with no extra repetitions

```
  payload: |-
    {
      "program": [
        0, 1, 0, 5,        # 0 - don't know, 1: one repetition, 0: final position 0% (raise arm back to top of device)
        100, 0, 2,
        50, 0, 0,
        100, 0, 0,
        50, 0, 0,
        100, 0, 0
      ]
    }
```