Koenkk/zigbee2mqtt #18012: [Wrong device]:

rufabf, 2023-06-13:
### Link

https://www.zigbee2mqtt.io/devices/TS0121_plug.html#tuya-ts0121-plug

### Model

TS0121  _TZ3000_8nkb7mof

### Description

Plug

### Vendor

Blitzwolf

### Picture (link)

https://www.zigbee2mqtt.io/images/devices/TS0121_plug.jpg

### Database entry

{"id":8,"type":"Router","ieeeAddr":"0x00124b0022d4c386","nwkAddr":15322,"manufId":0,"manufName":"eWeLink","powerSource":"Mains (single phase)","modelId":"SA-003-Zigbee","epList":[1],"endpoints":{"1":{"profId":49246,"epId":1,"devId":16,"inClusterList":[0,3,4,5,6],"outClusterList":[0],"clusters":{"genBasic":{"attributes":{"modelId":"SA-003-Zigbee","manufacturerName":"eWeLink"}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x847127fffec9a54c","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":5,"hwVersion":1,"dateCode":"20201207","swBuildId":"1.0.5","zclVersion":1,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1686676031525,"defaultSendRequestWhen":"immediate"}

Koenkk, 2023-06-13:
I think you posted the wrong database.db entry, I don't see `TS0121` or `_TZ3000_8nkb7mof`

rufabf, 2023-06-13:
This is the

> I think you posted the wrong database.db entry, I don't see `TS0121` or `_TZ3000_8nkb7mof`

This is the model that zigbee2mqtt show

Koenkk, 2023-06-14:
Yes, but the database.db entry you posted doesn't match that (`"manufName":"eWeLink","powerSource":"Mains (single phase)","modelId":"SA-003-Zigbee"`). Please post your complete data/database.db file.

rufabf, 2023-06-14:
> Yes, but the database.db entry you posted doesn't match that (`"manufName":"eWeLink","powerSource":"Mains (single phase)","modelId":"SA-003-Zigbee"`). Please post your complete data/database.db file.

Thats the model that z2m think it is, i have 2 same plugs one is right and the other it shows like ewelink 

`{"id":1,"type":"Coordinator","ieeeAddr":"0x847127fffec9a54c","nwkAddr":0,"manufId":0,"epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":48879,"inClusterList":[0,3,6,10,25,26,768],"outClusterList":[0,3,4,5,6,8,32,768,1024,1026,1029,1030,1280,2817,2819,2820,1794,4096,64513,64514],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":true,"meta":{},"lastSeen":1686763426494,"defaultSendRequestWhen":"immediate"}
{"id":2,"type":"EndDevice","ieeeAddr":"0x00124b0022eae12c","nwkAddr":46680,"manufId":0,"manufName":"eWeLink","powerSource":"Battery","modelId":"WB01","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":0,"inClusterList":[0,3,1],"outClusterList":[6,3],"clusters":{"genBasic":{"attributes":{"modelId":"WB01","manufacturerName":"eWeLink","powerSource":3,"zclVersion":0,"appVersion":5,"hwVersion":0,"dateCode":"20201026"}},"genPowerCfg":{"attributes":{"batteryVoltage":30,"batteryPercentageRemaining":200}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x847127fffec9a54c","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x847127fffec9a54c","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":32,"minRepIntval":3600,"maxRepIntval":7200,"repChange":0},{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":7200,"repChange":0}],"meta":{}}},"appVersion":5,"hwVersion":0,"dateCode":"20201026","zclVersion":0,"interviewCompleted":true,"meta":{"configured":956609490},"lastSeen":1686761099923,"defaultSendRequestWhen":"immediate"}
{"id":3,"type":"Router","ieeeAddr":"0x842e14fffe13bd57","nwkAddr":24614,"manufId":4098,"manufName":"_TZ3000_odygigth","powerSource":"Mains (single phase)","modelId":"TS0505A","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0505A","manufacturerName":"_TZ3000_odygigth","powerSource":1,"zclVersion":3,"appVersion":80,"stackVersion":0,"hwVersion":1,"dateCode":""}},"lightingColorCtrl":{"attributes":{"colorCapabilities":29}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":80,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1686756922125,"defaultSendRequestWhen":"immediate"}
{"id":4,"type":"EndDevice","ieeeAddr":"0x04cf8cdf3c7ca897","nwkAddr":14135,"manufId":4718,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sen_ill.mgl01","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":262,"inClusterList":[0,1024,3,1],"outClusterList":[3],"clusters":{"genBasic":{"attributes":{"modelId":"lumi.sen_ill.mgl01","manufacturerName":"LUMI","powerSource":3,"zclVersion":3,"appVersion":21,"stackVersion":2,"hwVersion":1,"dateCode":"20191118","swBuildId":"2019\u0000www."}},"msIlluminanceMeasurement":{"attributes":{"measuredValue":26929}},"genPowerCfg":{"attributes":{"batteryVoltage":29}}},"binds":[{"cluster":1024,"type":"endpoint","deviceIeeeAddress":"0x847127fffec9a54c","endpointID":1}],"configuredReportings":[{"cluster":1024,"attrId":0,"minRepIntval":15,"maxRepIntval":3600,"repChange":500}],"meta":{}}},"appVersion":21,"stackVersion":2,"hwVersion":1,"dateCode":"20191118","swBuildId":"2019\u0000www.","zclVersion":3,"interviewCompleted":true,"meta":{"configured":855622161},"lastSeen":1686763834101,"defaultSendRequestWhen":"immediate"}
{"id":5,"type":"EndDevice","ieeeAddr":"0xa4c138886c1f741a","nwkAddr":9234,"manufId":4417,"manufName":"_TZ3000_oxslv1c9","powerSource":"Battery","modelId":"TS0203","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[1,3,1280,0],"outClusterList":[3,4,5,6,8,4096,25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0203","manufacturerName":"_TZ3000_oxslv1c9","powerSource":3,"zclVersion":3,"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":""}},"ssIasZone":{"attributes":{"iasCieAddr":"0x847127fffec9a54c","zoneState":1}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":166,"batteryVoltage":27}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x847127fffec9a54c","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0},{"cluster":1,"attrId":32,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0}],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-305470159},"lastSeen":1686762926240,"defaultSendRequestWhen":"immediate"}
{"id":6,"type":"EndDevice","ieeeAddr":"0xa4c13850c9837670","nwkAddr":35247,"manufId":4417,"manufName":"_TZ3000_mcxw5ehu","powerSource":"Battery","modelId":"TS0202","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[1,1280,3,0],"outClusterList":[4096,6,25,10],"clusters":{"genBasic":{"attributes":{"modelId [... cut ...] inds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x847127fffec9a54c","endpointID":1},{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x847127fffec9a54c","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0}],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-429758583},"lastSeen":1686759922191,"defaultSendRequestWhen":"immediate"}`

rufabf, 2023-06-14:
> I see, if `SA-003-Zigbee` is used as the modelID we cannot distinguish it (since both manufactur name and modelID are the same).

Then how can i modify the ID to use the correct model? 
XH-002P [](https://zigbee.blakadder.com/Tuya_XH-002P.html) 
[image]

With `SA-003-Zigbee` only can on and off there is no power monitor

Koenkk, 2023-06-15:
> This is the
> 
> 
> 
> > I think you posted the wrong database.db entry, I don't see `TS0121` or `_TZ3000_8nkb7mof`
> 
> 
> 
> This is the model that zigbee2mqtt show 

Where does it show this, can you make a screenshot?

rufabf, 2023-06-15:
[image]The number 3 should be like 2, both where buys at the same time but z2m treated like different

Koenkk, 2023-06-15:
I think you got 2 different devies but with the same housing. Looking at the inClusterList from the OP, this device does not support electrical measurements.

rufabf, 2023-06-16:
> I think you got 2 different devies but with the same housing. Looking at the inClusterList from the OP, this device does not support electrical measurements.

Ahh ok, thanks anyway

closed 2023-06-16 by rufabf