Koenkk/zigbee2mqtt #18013: Ikea E2123 buttons not working after firmware device update

howesey2k, 2023-06-13:
### What happened?

After updating to the latest firmware, none of the buttons work.

### What did you expect to happen?

After re adding the remote several times, all I can see is the exposed battery and LQI level. None of the actions are triggered. It was working before I updated the firmware.

### Zigbee2MQTT version

1.31.2

### Adapter firmware version

Unknown

### Adapter

Home Assistant SkyConnect

SqyD, 2023-06-14:
Just updated an new, previously unused E2123 straight to 1.0.35. All buttons works fine except for the 1dot and 2dot onces. Other button events arrive just fine. Legacy configuration option doesn't change things.

howesey2k, 2023-06-14:
Strange. Mine is new too. Is there any way you know of to do a clean firmware reset? Thanks for your help in advance.

SqyD, 2023-06-14:
Not sure if it's possible with the Ikea devices, but this shows how to specify a firmware version for a device. The doc does mention that not all vendors allow downgrades.

SqyD, 2023-06-14:
Managed to fix the missing dot actions by removing and repairing again. This was covered in https://github.com/Koenkk/zigbee2mqtt/issues/17142

kumme74, 2023-06-15:
I have here two new E2123 one with original 1.0.012 and one I updated to 1.0.35.

1.0.012 works well, but with the 1.0.35 there is a problem with the periodic volume_up_hold and volume_down_hold function.

When the volume up/down button is pressed for a longer time periodic messages arrive as expected. When the button is released, there are still messages arriving even seconds after the button was released.

Even if another button e.g. dot button is pressed in the meantime, some spurious volume_x_hold messages are received after the dot press messages.

This does not happen with 1.0.0.012

I attached a log explaining the behaviour.
Any idea what is happening there?
[E2123 Problem.txt](https://github.com/Koenkk/zigbee2mqtt/files/11755132/E2123.Problem.txt)

closed 2023-07-29 by github-actions[bot]