Koenkk/zigbee2mqtt #18016: After upgrade to 1.31.2 the add-on failed to start up

cdr93, 2023-06-13:
### Link

https://fr.aliexpress.com/item/4001237395363.html?spm=a2g0o.order_list.order_list_main.175.21ef5e5bkWDnxK&gatewayAdapt=glo2fra

### Database entry

{"id":11,"type":"Router","ieeeAddr":"0x84fd27fffed9e235","nwkAddr":58476,"manufId":4098,"manufName":"_TZE200_axgvo9jh","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0006","65506":31,"65508":0,"65534":0,"modelId":"TS0601","manufacturerName":"_TZE200_axgvo9jh","powerSource":1,"zclVersion":3,"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{}},"genOnOffSwitchCfg":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-708457359},"lastSeen":1686678917550,"defaultSendRequestWhen":"immediate"}

### Comments

Hello, before the 1.31.2 update, this hardware worked with a converter. Since this update, I have to remove the converter otherwise Zigbee2MQTT does not start.
Thanks in advance for your help in resolving this issue.

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const definition = {
    // Since a lot of Tuya devices use the same modelID, but use different data points
    // it's usually necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE200_axgvo9jh'
        },
    ],
 	model: 'ZSTY-SM-1DMZG-US-W',
    vendor: 'Somgoms',
    description: 'Curtain switch',
    fromZigbee: [fz.tuya_cover, fz.ignore_basic_report],
    toZigbee: [tz.tuya_cover_control, tz.tuya_cover_options],
	exposes: [e.cover_position().setAccess('position', ea.STATE_SET)],
};

module.exports = definition;
```

cdr93, 2023-06-13:
I add in case the debug at the start of Zigbee2MQTT:
`> zigbee2mqtt@1.31.2 start
> node index.js
/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/index.js:79
if (converter.options) {
^
TypeError: Cannot read properties of undefined (reading 'options')
at Object.addDefinition [as addDeviceDefinition] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/index.js:79:23)
at new ExternalConverters (/opt/zigbee2mqtt/lib/extension/externalConverters.ts:15:17)
at new Controller (/opt/zigbee2mqtt/lib/controller.ts:84:58)
at start (/opt/zigbee2mqtt/index.js:106:18)`

Thanking you in advance for any help you can give me,
Cordially,

Koenkk, 2023-06-14:
- on top of the file add `const legacy = require('zigbee-herdsman-converters/lib/legacy');`
- change `fz.tuya_cover` to `legacy.fz.tuya_cover`
- change `tz.tuya_cover_control` to `legacy.tz.tuya_cover_control`
- change `tz.tuya_cover_options` to `legacy.tz.tuya_cover_options`

Koenkk, 2023-06-14:
I've also added the device for out-of-the-box support.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-06-14 by Koenkk