Koenkk/zigbee2mqtt #18023: IKEA E2002 OTA

KC-inDomus, 2023-06-14:
### What happened?

ZigBee2MQTT misunderstand the newest OTA firmware

See screenshot: https://pasteboard.co/uDOFBGAbDkTP.png

### What did you expect to happen?

To let me update the firmware with latest versione 1.0.0.24

### How to reproduce it (minimal and precise)

-

### Zigbee2MQTT version

latest

### Adapter firmware version

20200708

### Adapter

Sonoff ZBDongle-P (20230507)

Koenkk, 2023-06-14:
What do you mean with misunderstood? This info comes from the device, it seems the changed the numbering scheme.

KC-inDomus, 2023-06-14:
> What do you mean with misunderstood? This info comes from the device, it seems the changed the numbering scheme.

Installed version date: 07/07/2020
new available firmware date: 05/30/2023

but zigbee2mqtt says that the newer is... older than the installed one.

Koenkk, 2023-06-14:
Can you provide the debug log when checking for an update?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

KC-inDomus, 2023-06-14:
```
Debug 2023-06-14 20:32:29Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/check' with data '{"id":"ikea_switch_1","transaction":"cp5ya-1"}'
Info 2023-06-14 20:32:29Checking if update available for 'ikea_switch_1'
Info 2023-06-14 20:32:29MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Checking if update available for 'ikea_switch_1'","meta":{"device":"ikea_switch_1","status":"checking_if_available"},"type":"ota_update"}'
Debug 2023-06-14 20:32:29Check if update available for '0x04cd15fffe844299' (Remote Control N2)
Debug 2023-06-14 20:32:29Using endpoint '1'
Debug 2023-06-14 20:32:30Received Zigbee message from '0xa4c138c6273ef6db', type 'attributeReport', cluster 'genBasic', data '{"65506":54,"65508":0,"appVersion":192}' from endpoint 1 with groupID 0
```

Koenkk, 2023-06-14:
The logging stops too early, please wait until the update check is done.

KC-inDomus, 2023-06-14:
you're right, sorry.

```
Debug 2023-06-14 20:42:15Received Zigbee message from '0xa4c138c6273ef6db', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":35}' from endpoint 1 with groupID 0
Info 2023-06-14 20:42:15MQTT publish: topic 'zigbee2mqtt/0xa4c138c6273ef6db', payload '{"child_lock":"UNLOCK","current":0.04,"device":{"applicationVersion":192,"dateCode":"","friendlyName":"0xa4c138c6273ef6db","hardwareVersion":1,"ieeeAddr":"0xa4c138c6273ef6db","manufacturerID":4417,"manufacturerName":"_TZ3000_w0qqde0g","model":"TS011F_plug_1","networkAddress":43009,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"energy":0,"indicator_mode":"off/on","linkquality":145,"power":34,"power_outage_memory":"off","state":"ON","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":false,"voltage":241}'
Debug 2023-06-14 20:42:15Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/check' with data '{"id":"ikea_switch_1","transaction":"obl5z-3"}'
Info 2023-06-14 20:42:15Checking if update available for 'ikea_switch_1'
Info 2023-06-14 20:42:15MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Checking if update available for 'ikea_switch_1'","meta":{"device":"ikea_switch_1","status":"checking_if_available"},"type":"ota_update"}'
Debug 2023-06-14 20:42:15Check if update available for '0x04cd15fffe844299' (Remote Control N2)
Debug 2023-06-14 20:42:15Using endpoint '1'
Debug 2023-06-14 20:42:17Received Zigbee message from 'ikea_switch_1', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":65572,"imageType":4555,"manufacturerCode":4476}' from endpoint 1 with groupID 0
Debug 2023-06-14 20:42:17Got OTA request '{"fieldControl":1,"manufacturerCode":4476,"imageType":4555,"fileVersion":65572}'
Debug 2023-06-14 20:42:17Received Zigbee message from '0xa4c138c6273ef6db', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID 0
Info 2023-06-14 20:42:17MQTT publish: topic 'zigbee2mqtt/0xa4c138c6273ef6db', payload '{"child_lock":"UNLOCK","current":0.04,"device":{"applicationVersion":192,"dateCode":"","friendlyName":"0xa4c138c6273ef6db","hardwareVersion":1,"ieeeAddr":"0xa4c138c6273ef6db","manufacturerID":4417,"manufacturerName":"_TZ3000_w0qqde0g","model":"TS011F_plug_1","networkAddress":43009,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"energy":0,"indicator_mode":"off/on","linkquality":149,"power":0,"power_outage_memory":"off","state":"ON","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":false,"voltage":241}'
Debug 2023-06-14 20:42:18Is new image available for '0x04cd15fffe844299', current '{"fieldControl":1,"manufacturerCode":4476,"imageType":4555,"fileVersion":65572}', latest meta '{"fileVersion":581,"url":"http://fw.ota.homesmart.ikea.net/global/GW1.0/01.20.065/bin/10078031-zingo_kt_styrbar_remote-2.4.5.ota.ota.signed"}'
Debug 2023-06-14 20:42:18Update available for '0x04cd15fffe844299': NO
Warning 2023-06-14 20:42:18**Firmware on '0x04cd15fffe844299' is newer than latest firmware online.**
Info 2023-06-14 20:42:18No update available for 'ikea_switch_1'
```

Koenkk, 2023-06-17:
According to the zigbee spec, z2m is correct, current = `"fileVersion":65572`, latest is `"fileVersion":581`. What you can do is force the OTA update, see https://www.zigbee2mqtt.io/guide/usage/ota_updates.html#local-ota-index-and-firmware-files on how to do that.

KC-inDomus, 2023-06-17:
surely that's an anomaly - not z2m side, I mean. ok, I'll try

rossowl, 2023-07-03:
How to change the address for IKEA tradfri OTA upgrade?

zigbee_ota_override_index_location: index.json not working. OTA is still querying the IKEA server.

my local index.json
[{
    "url": "10078031-zingo_kt_styrbar_remote-2.4.5.ota.ota.signed",
    "force": true
}]

Log:
2023-07-03 10:50:28New image available for '0x003c84fffe32bd5d', current '{"fieldControl":1, "manufacturerCode":4476, "imageType":4555, "fileVersion":65572}', latest meta '{"fileVersion":581, "url": "http://fw.ota.homesmart.ikea.net/global/GW1.0/01.20.065/bin/10078031-zingo_kt_styrbar_remote-2.4.5.ota.ota.signed"}'
debug 2023-07-03 10:50:28Update available for '0x003c84fffe32bd5d': NO
warning 2023-07-03 10:50:28Firmware for '0x003c84fffe32bd5d' is newer than the latest firmware online.
 
zigbee-herdsman-converters/src/lib/ota-tradfri.ts
const productionURL = 'http://fw.ota.homesmart.ikea.net/feed/version_info.json';

And there is:
"fw_binary_url": "http://fw.ota.homesmart.ikea.net/global/GW1.0/01.20.065/bin/10078031-zingo_kt_styrbar_remote-2.4.5.ota.ota.signed"

jamesmelville, 2023-07-06:
Looks like the override index only works for devices which use [generic zigbee OTA configuration](https://github.com/Koenkk/zigbee2mqtt/issues/12554) 

I made almost the same configuration as you - downloaded the ota file and placed in config dir, configured the JSON file and put filename in settings.
I found this line of this file on my filesystem
https://github.com/Koenkk/zigbee-herdsman-converters/blob/06d2ac8c08e783d9914c3b3c92626d4a5960f005/src/devices/ikea.js#L754
(for me it was in `/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices/ikea.js`)
changed the line from `ota: ota.tradfri,` to `ota: ota.zigbeeOTA,`

Things I also did but may not have needed to 🤷 :
- renamed the file to `10078031-zingo_kt_styrbar_remote-2.4.5.ota` and updated index.json
- restarted zigbee2mqtt after changing `ikea.js`

I then successfully applied the latest firmware to my E2002 using the OTA menu in the webui.

So I see 2 issues here:
1. I think a fix/enhancement that resolves 12554 would be ideal, but I guess the docs https://www.zigbee2mqtt.io/guide/usage/ota_updates.html#update-to-latest-firmware could also be updated to note that custom firmware updates only work for certain devices
2. I think I've understood that the root of this issue was that Ikea published some inconsistent version numbers. I'm not sure if zigbee2mqtt should work around that.

TheManii, 2023-07-10:
I would like to reiterate the problem:
The Z2M UI does not allow overriding what ota files to apply, it's purely "go/no go" if it considers it a valid update (json has an entry for a version newer than what the device itself reports)

The underlying Z2M code allows you to manually override the json entries and/or the files hosted, if the file is uploaded ahead of time and if the device lands in the main zigbeeOTA namespace

The issue with this:

- Files must be available ahead of time, either via http (reachable by Z2M), or uploaded into Z2M's container
- Install options ("force": true) must be declared ahead of time (if this is what the user wishes to do)
- The existing json files must be overriden, but only the ota.zigbeeota namespace can handle overrides as of this time

Options:

1. Add a "install anyway?" option to the existing ota dialog
2. Add in a "upload file" dialog, as shown below is the one from ZwaveJS
3. Cache upsteam jsons into Z2M ota repo (and apply manual fixes to them if needed)

What you gain:

- For one offs, the user does not have to go though all the steps of overriding the json files (assuming they can override them at all) and removing it afterwards, which can save time dependent on exactly how many devices they are updating
- It does not require the assumption that the upstream jsons are well-formed as far as versioning is concerned, we could hypothetically wait for ikea to fix their json file entries as it is an issue on their end, but we have no control over when it will be fixed

> According to the zigbee spec, z2m is correct, current = `"fileVersion":65572`, latest is `"fileVersion":581`. What you can do is force the OTA update, see https://www.zigbee2mqtt.io/guide/usage/ota_updates.html#local-ota-index-and-firmware-files on how to do that.

[image]

Bolukan, 2023-07-31:
>     * restarted zigbee2mqtt after changing `ikea.js`
Only restarting Zigbee2MQTT is not sufficient. I had to restart the container.

ertechdesign, 2023-08-28:
I also have this problem with E2002. I think it's time to implement a feature allowing users to upload a firmware file through the UI and deploy it to their respective devices. I know there is a way through editing a JSON file but for many users it's too fiddly. Finding the newest compatible firmware for number of different brands/devices is already enough of a project (since not all producers publish them). Ask users to maintain a file with device IDs/addresses and having to add correct firmware links, many users (including me) just give up.

chrizzlibaer, 2023-09-29:
+1 for uploading or instead choosing the last firmware avaliable for a forced install in the zige2mqtt ui. This would be awesome!

closed 2024-06-04 by github-actions[bot]

woutput, 2024-12-08:
It looks like I have a similar issue with an IKEA [LED2005R5/LED2106R3](https://www.zigbee2mqtt.io/devices/LED2005R5_LED2106R3.html) because most of my devices of that type run firmware version 3.0.21 with build date 20240226 while one device runs firmware version 1.0.012 with build date 20201111. For that single bulb, when I click "Check for new updates" I get:
z2m: Checking if update available for 'ID-Lamp'
z2m: No update available for 'ID-Lamp'

@Koenkk  What do you think of [the proposal by TheManii](https://github.com/Koenkk/zigbee2mqtt/issues/18023#issuecomment-1628288310), especially suggestion 1: install anyway / force update?