Koenkk/zigbee2mqtt #18024: Same (Candeo Zigbee Dimmer Switches) - Model C202 - Having Different Manufactorers/Vendors, Some Supported, And Some Unsupported (Candeo,Shyugj)

maans2001, 2023-06-14:
### What happened?

Recently I bought 3 [Candeo Zigbee No-Neutral Dimmer Switches](https://candeo.io/store/zigbee-dimmer-rotary-switch-c202/),
After I installed the first one, The Device showed up as `Unsupported` on Zigbee2MQTT
However, I updated Zigbee2MQTT and the device was supported as expected

I then installed the second Dimmer Switch,
this time after I updated Zigbee2MQTT and made sure support was added to the device
This Time, the device again showed up as Unsupported

upon further investigation and comparison between the first and second dimmer switches
I noticed that the First dimmer(supported dimmer) had the following:
- Zigbee Model: Candeo Zigbee Dimmer
- Zigbee Manufacturer: Candeo
- Manufacturer: Candeo
- Model: HK-DIM-A

And the second(unsupported) dimmer:
- Zigbee Model: HK_DIM_A
- Zigbee Manufacturer: Shyugj
- Manufacturer: Shyugj
- Model: HK_DIM_A

Despite both switches having been bought from the same [official candeo website](https://candeo.io/) and also at the same time
I find it strange that 2 seemingly identical devices, would report having different manufacturers; `Candeo` and `Shyugj`

After consulting the [Zigbee2MQTT Discord Server](https://discord.gg/NyseBeK)
Looking through the [official Candeo converter](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/candeo.ts)
And finding this pre-release custom converter by @Koenkk in #16613

I have made/modified the following custom converter which seems to have added support to the unsupported device:
````js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['HK_DIM_A'],
    model: 'HK_DIM_A',
    vendor: 'Shyugj',
    description: 'Zigbee LED dimmer smart switch',
    extend: extend.light_onoff_brightness({noConfigure: true, disableEffect: true}),
    configure: async (device, coordinatorEndpoint, logger) => {
        await extend.light_onoff_brightness().configure(device, coordinatorEndpoint, logger);
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
        await reporting.onOff(endpoint);

        // The default reporting from the Candeo dimmer can result in a lot of Zigbee traffic
        // to the extent that reported brighness values can arrive at the endpoint out of order
        // giving the appearance that the values are jumping around.
        // Limit the reporting to once a second to give a smoother reporting of brightness.
        await reporting.brightness(endpoint, {min: 1});
    },
};

module.exports = definition;
````

I hope the info I provided is useful in integrating/handling this strange thing on the official Zigbee2MQTT release

### What did you expect to happen?

Well, Since I read that there was support for this particular Zigbee No-Neutral Dimmer
I was just expecting that I was on an older version, and updating would fix the issue
Which it did, for the first Dimmer Switch
I expected that the second dimmer switch of the same make and model (as apparent from the box and website and appearance)
Would be paired without issue after I did update to the newest Zigbee2MQTT version  

### How to reproduce it (minimal and precise)

I see no way of reproducing this, other than buying multiple [Candeo Zigbee Dimmer Switches
](https://candeo.io/store/zigbee-dimmer-rotary-switch-c202/)and pairing them, and see if the manufacturer is different.  

### Zigbee2MQTT version

1.31.2 commit: 21f5125

### Adapter firmware version

20220219

### Adapter

Sonoff Zigbee Dongle

### Debug log

I do not think the entire log is needed ... but I can give it if you think so
```txt
warn  2023-06-14 03:04:55: Received message from unsupported device with Zigbee model 'HK_DIM_A' and manufacturer name 'Shyugj'
warn  2023-06-14 03:04:58: Received message from unsupported device with Zigbee model 'HK_DIM_A' and manufacturer name 'Shyugj'
warn  2023-06-14 03:05:01: Received message from unsupported device with Zigbee model 'HK_DIM_A' and manufacturer name 'Shyugj'
warn  2023-06-14 03:05:04: Received message from unsupported device with Zigbee model 'HK_DIM_A' and manufacturer name 'Shyugj'
warn  2023-06-14 03:05:07: Received message from unsupported device with Zigbee model 'HK_DIM_A' and manufacturer name 'Shyugj'
warn  2023-06-14 03:15:08: Received message from unsupported device with Zigbee model 'HK_DIM_A' and manufacturer name 'Shyugj'
warn  2023-06-14 03:15:09: Received message from unsupported device with Zigbee model 'HK_DIM_A' and manufacturer name 'Shyugj'
warn  2023-06-14 03:15:12: Received message from unsupported device with Zigbee model 'HK_DIM_A' and manufacturer name 'Shyugj'
warn  2023-06-14 03:15:15: Received message from unsupported device with Zigbee model 'HK_DIM_A' and manufacturer name 'Shyugj'
warn  2023-06-14 03:15:18: Received message from unsupported device with Zigbee model 'HK_DIM_A' and manufacturer name 'Shyugj'
warn  2023-06-14 03:15:21: Received message from unsupported device with Zigbee model 'HK_DIM_A' and manufacturer name 'Shyugj'
```

Koenkk, 2023-06-14:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-06-14 by Koenkk

agrayson47, 2024-10-11:
I have the C202 but it is being recognised as a C201 which is a different device. If I use 1.39.0 is is recognised as a HK-DIM-A which is closer but still not the right device.