Koenkk/zigbee2mqtt #18031: Legrand 199161 power socket not showing power consumption

jplelievre, 2023-06-15:
### What happened?

Hello,

I am trying to configure the Legrand 199161 power socket with Zigbee2mqtt.
This new reference is the replacement reference for 067725 which is the same as 067775 color exepted.

The power socket is recognized as Legrand model 067775/741811
I can switch it on and off but power consumption is not working
I have the following error on the logs :
error 2023-06-15 20:29:42Failed to configure 'Prise lave-vaiselle', attempt 2 (Error: Read 0x0004740000099a33/1 haElectricalMeasurement(["acVoltageMultiplier","acVoltageDivisor","acCurrentMultiplier"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:421:28) at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:563:22) at Object.readEletricalMeasurementMultiplierDivisors (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.js:26:5) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/legrand.js:307:13) at Configure.configure (/app/lib/extension/configure.ts:117:13))

Also there is a warning about the firmware version :
warning 2023-06-15 20:34:25Firmware on '0x0004740000099a33' is newer than latest firmware online.

Thanks in advance
Jean-Pascal

### Zigbee2MQTT version

1.31.2

### Adapter firmware version

20210708

### Adapter

Sonoff Zigbee 3.0

### Debug log

[logs-legrand-199161.txt](https://github.com/Koenkk/zigbee2mqtt/files/11761748/logs-legrand-199161.txt)
[image]
[image]
[image]

jplelievre, 2023-06-19:
Hello,

I don't know if I have to use the support new device documentation as it is recognized as Legrand model 067775/741811.
Any help will be appreciated on how to make the Legrand 199161 entirely workable.

Thanks in advance,
Jean-Pascal

SilentT-FR, 2023-06-27:
Its the same product with the same firmware and the same cluster

SilentT-FR, 2023-06-27:
i think you need to upgrade to an upper firmware but not this online because it dosent work, https://github.com/Koenkk/zigbee-OTA/issues/328

in the meantime you can upgrade by using this file 

https://github.com/Koenkk/zigbee-OTA/blob/e69bc5232152872ef2216f79ea1d32cac134e697/images/Legrand/1021-0011-4a4203-NLP-74.zigbee

 ```
   {
        "fileVersion": 4866563,
        "fileSize": 250039,
        "manufacturerCode": 4129,
        "imageType": 17,
        "sha512": "295ecf710c20c2b6f7c6c12c6dd22c8cf9e063c041c816786feb830113b9aa4b7cf93e426004da8f4d161ba1ebf2d1a4675547f44b65fd385f10a05035547810",
        "url": "https://github.com/Koenkk/zigbee-OTA/raw/e69bc5232152872ef2216f79ea1d32cac134e697/images/Legrand/1021-0011-4a4203-NLP-74.zigbee",
        "path": "images/Legrand/1021-0011-4a4203-NLP-74.zigbee"
    },
```

jplelievre, 2023-06-30:
Thanks a lot!
It resolved my problem.
No more error after this manual upgrade and the power consumption is working fine.

closed 2023-06-30 by jplelievre

SilentT-FR, 2023-06-30:
don't forget to remove the value put in manual for the firmware link to be able to take advantage of the latest version in the future, on the other hand, you have to be careful, the future version of the power sockets is still not functional

SilentT-FR, 2023-06-30:
an opened issue is in progress

jplelievre, 2023-07-03:
> don't forget to remove the value put in manual for the firmware link to be able to take advantage of the latest version in the future, on the other hand, you have to be careful, the future version of the power sockets is still not functional

Sure, I have reverted changes in the configuration file and I am not uprading my power socket as long as the current issue is resolved.