Koenkk/zigbee2mqtt #18032: [New device support]: Avatto ZWSM16-B4

cvroque, 2023-06-16:
### Link

https://aliexpress.com/item/1005003051876791.html

### Database entry

{"id":19,"type":"Router","ieeeAddr":"0xa4c138a830afb472","nwkAddr":8603,"manufId":4417,"manufName":"_TZ3000_5ajpkyq6","powerSource":"Mains (single phase)","modelId":"TS0004","epList":[1,2,3,4,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u000e\u0000\u0000\u0000\u0012","65506":56,"65508":0,"65534":0,"modelId":"TS0004","manufacturerName":"_TZ3000_5ajpkyq6","powerSource":1,"zclVersion":3,"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":""}},"manuSpecificBosch":{"attributes":{"53251":"AAAAAgAABAAABgAA"}},"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0,"moesStartUpOnOff":0}},"manuSpecificTuya_3":{"attributes":{"53248":0,"53249":0,"53250":0,"53251":0,"53252":0,"53253":0,"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b6a112","endpointID":1}],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[4,5,6],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b6a112","endpointID":1}],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":256,"inClusterList":[4,5,6],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b6a112","endpointID":1}],"configuredReportings":[],"meta":{}},"4":{"profId":260,"epId":4,"devId":256,"inClusterList":[4,5,6],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b6a112","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":889996980},"lastSeen":1680635736208,"defaultSendRequestWhen":"immediate"}

### Comments

I saw that the 1ch and 2ch versions got support, here's 4ch.

Koenkk, 2023-06-17:
`_TZ3000_5ajpkyq6` is supported in z2m 1.31.2

closed 2023-06-17 by Koenkk

XDOM-JR, 2025-04-26:
How can I activate the Interlock function of this device?