Koenkk/zigbee2mqtt #18034: Aqara ZNXDD01LM, ZNDDMK11LM problem

esgu1019, 2023-06-16:
### What happened?

Aqara LED drivers and ceiling lights have problems when setting the transition.
When toggling the light, it does not remember the previous brightness, and toggles it by fixing one brightness. If I don't set the transition it works fine again.

In the case of the ZNDDMK11LM, only light 2 has problems, light 1 is fine.

ZNXDD01LM has the same problem and also
Setting power_on_behavior doesn't work.

### How to reproduce it (minimal and precise)

After setting the transition, after changing the brightness to 50%, turning the light off and on again turns it on at 100%, not 50%.

### Zigbee2MQTT version

1.31.2

### Adapter firmware version

20230507

### Adapter

sonoff dongle P

### Debug log

2023-06-16 17:35:36Received MQTT message on 'zigbee2mqtt/0x54ef44100065b498/set' with data '{"brightness_l2":254}'
2023-06-16 17:35:36Publishing 'set' 'brightness' to '0x54ef44100065b498'
2023-06-16 17:35:36Received MQTT message on 'zigbee2mqtt/0x54ef44100065b498/set' with data '{"brightness_l2":97}'
2023-06-16 17:35:36Publishing 'set' 'brightness' to '0x54ef44100065b498'
2023-06-16 17:35:36Received Zigbee message from '0x54ef44100065b498', type 'attributeReport', cluster 'genLevelCtrl', data '{"245":8624384,"currentLevel":254}' from endpoint 2 with groupID 0
2023-06-16 17:35:38Received Zigbee message from '0x54ef44100065b498', type 'attributeReport', cluster 'genLevelCtrl', data '{"245":8624384,"currentLevel":97}' from endpoint 2 with groupID 0
2023-06-16 17:35:40Received MQTT message on 'zigbee2mqtt/0x54ef44100065b498/set' with data '{"state_l2":"OFF"}'
2023-06-16 17:35:40Publishing 'set' 'state' to '0x54ef44100065b498'
2023-06-16 17:35:42Received Zigbee message from '0x54ef44100065b498', type 'attributeReport', cluster 'genOnOff', data '{"245":8624384,"onOff":0}' from endpoint 2 with groupID 0
2023-06-16 17:35:42Received Zigbee message from '0x54ef44100065b498', type 'attributeReport', cluster 'genLevelCtrl', data '{"245":8624384,"currentLevel":1}' from endpoint 2 with groupID 0
2023-06-16 17:35:42Received MQTT message on 'zigbee2mqtt/0x54ef44100065b498/set' with data '{"state_l2":"ON"}'
2023-06-16 17:35:42Publishing 'set' 'state' to '0x54ef44100065b498'
2023-06-16 17:35:42Received Zigbee message from '0x54ef44100065b498', type 'attributeReport', cluster 'genAnalogInput', data '{"presentValue":8.153059005737305}' from endpoint 21 with groupID 0
2023-06-16 17:35:43Received Zigbee message from '0x54ef44100065b498', type 'attributeReport', cluster 'genOnOff', data '{"245":8624384,"onOff":1}' from endpoint 2 with groupID 0
2023-06-16 17:35:44Received Zigbee message from '0x54ef44100065b498', type 'attributeReport', cluster 'genLevelCtrl', data '{"245":8624384,"currentLevel":254}' from endpoint 2 with groupID 0
2023-06-16 17:35:51Received Zigbee message from '0x54ef44100065b498', type 'attributeReport', cluster 'genAnalogInput', data '{"presentValue":12.43221664428711}' from endpoint 21 with groupID 0

asapkenzy, 2023-07-06:
I have the same issue  ><

asapkenzy, 2023-09-13:
Have the same issue with power_on_behavior on ZNXDD01LM.

closed 2023-10-22 by github-actions[bot]

hACKrus, 2023-10-28:
Have same issue:

`Publish 'get' 'power_on_behavior' to 'Серверная Люстра' failed: 'Error: Read 0xfffffffffffffffffffffffff/1 genOnOff(["startUpOnOff"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
`

hACKrus, 2023-11-20:
Seems, in ZHA problem was solved https://github.com/zigpy/zha-device-handlers/issues/2649

kevintong116, 2023-11-21:
Not sure if this fix the issue
 [Koenkk/[zigbee-herdsman-converters]#6450](https://github.com/Koenkk/zigbee-herdsman-converters/pull/6450)