Koenkk/zigbee2mqtt #18040: npm start gives SyntaxError: Unexpected token '(' on fresh installed Rpi with Ubuntu 22.04 LTS

Alrd1, 2023-06-16:
### What happened?

Fresh installed Raspberry PI 4 with Ubuntu 22.04 LTS

Gives as error: 

ubuntu@ubuntu-desktop:/opt/zigbee2mqtt$ npm start

> zigbee2mqtt@1.31.2 start
> node index.js

/opt/zigbee2mqtt/node_modules/rimraf/node_modules/minimatch/dist/cjs/ast.js:86
    #fillNegs() {
             ^

SyntaxError: Unexpected token '('
    at wrapSafe (internal/modules/cjs/loader.js:915:16)
    at Module._compile (internal/modules/cjs/loader.js:963:27)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1027:10)
    at Module.load (internal/modules/cjs/loader.js:863:32)
    at Function.Module._load (internal/modules/cjs/loader.js:708:14)
    at Module.require (internal/modules/cjs/loader.js:887:19)
    at require (internal/modules/cjs/helpers.js:74:18)
    at Object.<anonymous> (/opt/zigbee2mqtt/node_modules/rimraf/node_modules/minimatch/dist/cjs/index.js:9:18)
    at Module._compile (internal/modules/cjs/loader.js:999:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1027:10)
ubuntu@ubuntu-desktop:/opt/zigbee2mqtt$

ubuntu@ubuntu-desktop:/opt/zigbee2mqtt$ ls -l /dev/serial/by-id
total 0
lrwxrwxrwx 1 root root 13 mrt 20 15:33 usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B0008B9C236-if00 -> ../../ttyACM0
ubuntu@ubuntu-desktop:/opt/zigbee2mqtt$ npm -v
8.5.1

### What did you expect to happen?

Just start as the manual describes.

### How to reproduce it (minimal and precise)

Installed according manual https://www.zigbee2mqtt.io/guide/installation/01_linux.html#installing (Linux) installation

### Zigbee2MQTT version

1.31.2

### Adapter firmware version

CC2531_USB_CDC___0X00124B0008B9C236-if00

### Adapter

TI CC2531

Koenkk, 2023-06-17:
Are you using any external converters (`external_converters` in `configuration.yaml) or user extensions (a `data/extension` folder?

Alrd1, 2023-06-17:
This is the configuration.yaml which I am using:

# Home Assistant integration (MQTT discovery)
homeassistant: false

# allow new devices to join
permit_join: true

# MQTT settings
mqtt:
  # MQTT base topic for zigbee2mqtt MQTT messages
  base_topic: zigbee2mqtt
  # MQTT server URL
  server: 'mqtt://localhost'
  # MQTT server authentication, uncomment if required:
  # user: my_user
  # password: my_password

# Serial settings
serial:
  # Location of CC2531 USB sniffer
  port: /dev/ttyACM0
#frontend: true

Alrd1, 2023-06-17:
I am using this configuration.yaml:

# Home Assistant integration (MQTT discovery)
homeassistant: false

# allow new devices to join
permit_join: true

# MQTT settings
mqtt:
  # MQTT base topic for zigbee2mqtt MQTT messages
  base_topic: zigbee2mqtt
  # MQTT server URL
  server: 'mqtt://localhost'
  # MQTT server authentication, uncomment if required:
  # user: my_user
  # password: my_password

# Serial settings
serial:
  # Location of CC2531 USB sniffer
  port: /dev/ttyACM0
#frontend: true

Alrd1, 2023-06-17:
I am using the attached configuration.yaml
[configuration.txt](https://github.com/Koenkk/zigbee2mqtt/files/11778414/configuration.txt)

Koenkk, 2023-06-17:
What node version are you using ? (`node --version`)

Alrd1, 2023-06-17:
I am using v12.22.9

lux73, 2023-06-17:
this is likely unsupported - pls update npm to minimum v14.x - better update to v18.x

btw CC2531 is considered as weak & unstable -> pls upgrade your Coordinator to more powerful & stable CC2652x

Koenkk, 2023-06-18:
12.22 is not supported indeed, updating to 18 will fix your issue.

closed 2023-06-18 by Koenkk

Alrd1, 2023-06-18:
Thanks guys! That was the issue, but for somehow reasons the package would not upgraded. Anyway eventually I managed to do it.