Koenkk/zigbee2mqtt #18041: WHD02 wrong current

timewolf, 2023-06-16:
### What happened?

hi
i've a WHD02  (on zigbee2mqtt it shows as TS000F_power) that have a wrong "current" value, i got 200A, i suppose should be 2A.

### Zigbee2MQTT version

1.31.2 commit: 21f5125

### Adapter firmware version

6.10.3.0 build 297

### Adapter

sonoff zb

Koenkk, 2023-06-19:
It seems the current needs to be divided by 2000. When you connect a higher load, e.g. 100 watt, is the current +- 5x higher?

timewolf, 2023-06-20:
i will connect some higher and test...now there is only a sky decoder :)

timewolf, 2023-06-22:
ok i got the point, i've added to tuya, it's mA no A.

[image]

Koenkk, 2023-06-22:
So the value show in z2m should be divided by 1000?

timewolf, 2023-06-22:
yes

Koenkk, 2023-06-23:
Pushed a fix!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-06-23 by Koenkk