Koenkk/zigbee2mqtt #18043: [New device support]:

pk802, 2023-06-17:
### Link

https://www.alibaba.com/product-detail/10-year-Battery-life-Movable-Zigbee_1600788864565.html?spm=a2700.shop_plgr.41413.3.78f13e5fNn7x8l

### Database entry

{"id":169,"type":"GreenPower","ieeeAddr":"0x00000000fe8029f8","nwkAddr":10744,"manufId":null,"modelId":"GreenPower_2","epList":[242],"endpoints":{"242":{"epId":242,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":true,"meta":{},"lastSeen":1686808711115,"defaultSendRequestWhen":"immediate"}

### Comments

I tried using converters of other GreenPower devices but it seems it has totally different message format.

A single press gives me below Messages:
2023-06-17 15:51:54Received Zigbee message from '6Scene Switch1', type 'commandNotification', cluster 'greenPower', data '{"commandFrame":{"raw":{"data":[6,0,6,12,1,1,93,127,139],"type":"Buffer"}},"commandID":160,"frameCounter":80,"gppGddLink":6,"gppNwkAddr":6,"options":22544,"payloadSize":6,"srcID":4269812216}' from endpoint 242 with groupID 10744
2023-06-17 15:51:54No converter available for 'HVSC18-Z-6S-3.0' with cluster 'greenPower' and type 'commandNotification' and data '{"commandFrame":{"raw":{"data":[6,0,6,12,1,1,93,127,139],"type":"Buffer"}},"commandID":160,"frameCounter":80,"gppGddLink":6,"gppNwkAddr":6,"options":22544,"payloadSize":6,"srcID":4269812216}'
2023-06-17 15:51:55Received Zigbee message from '6Scene Switch1', type 'commandNotification', cluster 'greenPower', data '{"commandFrame":{"raw":{"data":[6,0,6,12,1,1,221,6,134],"type":"Buffer"}},"commandID":160,"frameCounter":81,"gppGddLink":6,"gppNwkAddr":6,"options":22544,"payloadSize":6,"srcID":4269812216}' from endpoint 242 with groupID 10744
2023-06-17 15:51:55No converter available for 'HVSC18-Z-6S-3.0' with cluster 'greenPower' and type 'commandNotification' and data '{"commandFrame":{"raw":{"data":[6,0,6,12,1,1,221,6,134],"type":"Buffer"}},"commandID":160,"frameCounter":81,"gppGddLink":6,"gppNwkAddr":6,"options":22544,"payloadSize":6,"srcID":4269812216}'
2023-06-17 15:51:57Received Zigbee message from '6Scene Switch1', type 'commandNotification', cluster 'greenPower', data '{"commandFrame":{"raw":{"data":[6,0,6,12,1,1,143,30,134],"type":"Buffer"}},"commandID":160,"frameCounter":82,"gppGddLink":6,"gppNwkAddr":6,"options":22544,"payloadSize":6,"srcID":4269812216}' from endpoint 242 with groupID 10744
2023-06-17 15:51:57No converter available for 'HVSC18-Z-6S-3.0' with cluster 'greenPower' and type 'commandNotification' and data '{"commandFrame":{"raw":{"data":[6,0,6,12,1,1,143,30,134],"type":"Buffer"}},"commandID":160,"frameCounter":82,"gppGddLink":6,"gppNwkAddr":6,"options":22544,"payloadSize":6,"srcID":4269812216}'
2023-06-17 15:51:58Received Zigbee message from '6Scene Switch1', type 'commandNotification', cluster 'greenPower', data '{"commandFrame":{"raw":{"data":[6,0,6,12,1,1,241,177,135],"type":"Buffer"}},"commandID":160,"frameCounter":83,"gppGddLink":6,"gppNwkAddr":6,"options":22544,"payloadSize":6,"srcID":4269812216}' from endpoint 242 with groupID 10744
2023-06-17 15:51:58No converter available for 'HVSC18-Z-6S-3.0' with cluster 'greenPower' and type 'commandNotification' and data '{"commandFrame":{"raw":{"data":[6,0,6,12,1,1,241,177,135],"type":"Buffer"}},"commandID":160,"frameCounter":83,"gppGddLink":6,"gppNwkAddr":6,"options":22544,"payloadSize":6,"srcID":4269812216}'

Pressing Button 1:
2023-06-17 15:51:54Received Zigbee message from '6Scene Switch1', type 'commandNotification', cluster 'greenPower', data '{"commandFrame":{"raw":{"data":[6,0,6,12,1,1,93,127,139],"type":"Buffer"}},"commandID":160,"frameCounter":80,"gppGddLink":6,"gppNwkAddr":6,"options":22544,"payloadSize":6,"srcID":4269812216}' from endpoint 242 with groupID 10744
2023-06-17 15:51:55Received Zigbee message from '6Scene Switch1', type 'commandNotification', cluster 'greenPower', data '{"commandFrame":{"raw":{"data":[6,0,6,12,1,1,221,6,134],"type":"Buffer"}},"commandID":160,"frameCounter":81,"gppGddLink":6,"gppNwkAddr":6,"options":22544,"payloadSize":6,"srcID":4269812216}' from endpoint 242 with groupID 10744
2023-06-17 15:51:57Received Zigbee message from '6Scene Switch1', type 'commandNotification', cluster 'greenPower', data '{"commandFrame":{"raw":{"data":[6,0,6,12,1,1,143,30,134],"type":"Buffer"}},"commandID":160,"frameCounter":82,"gppGddLink":6,"gppNwkAddr":6,"options":22544,"payloadSize":6,"srcID":4269812216}' from endpoint 242 with groupID 10744
2023-06-17 15:51:58Received Zigbee message from '6Scene Switch1', type 'commandNotification', cluster 'greenPower', data '{"commandFrame":{"raw":{"data":[6,0,6,12,1,1,241,177,135],"type":"Buffer"}},"commandID":160,"frameCounter":83,"gppGddLink":6,"gppNwkAddr":6,"options":22544,"payloadSize":6,"srcID":4269812216}' from endpoint 242 with groupID 10744

Pressing Button 2:
2023-06-17 15:58:19Received Zigbee message from '6Scene Switch1', type 'commandNotification', cluster 'greenPower', data '{"commandFrame":{"raw":{"data":[6,0,6,14,2,1,112,9,143],"type":"Buffer"}},"commandID":160,"frameCounter":88,"gppGddLink":6,"gppNwkAddr":6,"options":22544,"payloadSize":6,"srcID":4269812216}' from endpoint 242 with groupID 10744
2023-06-17 15:58:20Received Zigbee message from '6Scene Switch1', type 'commandNotification', cluster 'greenPower', data '{"commandFrame":{"raw":{"data":[6,0,6,14,2,1,20,36,135],"type":"Buffer"}},"commandID":160,"frameCounter":89,"gppGddLink":6,"gppNwkAddr":6,"options":22544,"payloadSize":6,"srcID":4269812216}' from endpoint 242 with groupID 10744
2023-06-17 15:58:21Received Zigbee message from '6Scene Switch1', type 'commandNotification', cluster 'greenPower', data '{"commandFrame":{"raw":{"data":[6,0,6,14,2,1,20,36,136],"type":"Buffer"}},"commandID":160,"frameCounter":90,"gppGddLink":6,"gppNwkAddr":6,"options":22544,"payloadSize":6,"srcID":4269812216}' from endpoint 242 with groupID 10744
2023-06-17 15:58:23Received Zigbee message from '6Scene Switch1', type 'commandNotification', cluster 'greenPower', data '{"commandFrame":{"raw":{"data":[6,0,6,14,2,1,248,130,141],"type":"Buffer"}},"commandID":160,"frameCounter":91,"gppGddLink":6,"gppNwkAddr":6,"options":22544,"payloadSize":6,"srcID":4269812216}' from endpoint 242 with groupID 10744

Pressing Button 3:
2023-06-17 15:59:50Received Zigbee message from '6Scene Switch1', type 'commandNotification', cluster 'greenPower', data '{"commandFrame":{"raw":{"data":[6,0,6,15,3,1,112,9,142],"type":"Buffer"}},"commandID":160,"frameCounter":92,"gppGddLink":6,"gppNwkAddr":6,"options":22544,"payloadSize":6,"srcID":4269812216}' from endpoint 242 with groupID 10744
2023-06-17 15:59:51Received Zigbee message from '6Scene Switch1', type 'commandNotification', cluster 'greenPower', data '{"commandFrame":{"raw":{"data":[6,0,6,15,3,1,221,6,137],"type":"Buffer"}},"commandID":160,"frameCounter":93,"gppGddLink":6,"gppNwkAddr":6,"options":22544,"payloadSize":6,"srcID":4269812216}' from endpoint 242 with groupID 10744
2023-06-17 15:59:52Received Zigbee message from '6Scene Switch1', type 'commandNotification', cluster 'greenPower', data '{"commandFrame":{"raw":{"data":[6,0,6,15,3,1,95,238,134],"type":"Buffer"}},"commandID":160,"frameCounter":94,"gppGddLink":6,"gppNwkAddr":6,"options":22544,"payloadSize":6,"srcID":4269812216}' from endpoint 242 with groupID 10744
2023-06-17 15:59:54Received Zigbee message from '6Scene Switch1', type 'commandNotification', cluster 'greenPower', data '{"commandFrame":{"raw":{"data":[6,0,6,15,3,1,143,30,138],"type":"Buffer"}},"commandID":160,"frameCounter":95,"gppGddLink":6,"gppNwkAddr":6,"options":22544,"payloadSize":6,"srcID":4269812216}' from endpoint 242 with groupID 10744

Pressing Button 4:
2023-06-17 16:00:55Received Zigbee message from '6Scene Switch1', type 'commandNotification', cluster 'greenPower', data '{"commandFrame":{"raw":{"data":[6,0,6,16,4,1,20,36,138],"type":"Buffer"}},"commandID":160,"frameCounter":96,"gppGddLink":6,"gppNwkAddr":6,"options":22544,"payloadSize":6,"srcID":4269812216}' from endpoint 242 with groupID 10744
2023-06-17 16:00:57Received Zigbee message from '6Scene Switch1', type 'commandNotification', cluster 'greenPower', data '{"commandFrame":{"raw":{"data":[6,0,6,16,4,1,54,151,134],"type":"Buffer"}},"commandID":160,"frameCounter":97,"gppGddLink":6,"gppNwkAddr":6,"options":22544,"payloadSize":6,"srcID":4269812216}' from endpoint 242 with groupID 10744
[... cut ...]
            const id = meta.device.modelID === 'GreenPower_2' ? msg.data[0] : msg.data[msg.data.length - 1];
            return {action: `scene_${id}`};
        },
    },
};
const definition = {
    zigbeeModel: ['GreenPower_2'],
    model: 'HVSC18-Z-6S-3.0',
    vendor: 'AMELECH',
    description: '6Scene Movable Switch',
    fromZigbee: [fzLocal.scenes_recall_scene_greenPower],
    exposes: [e.battery(),e.action(['scene_1', 'scene_2', 'scene_3', 'scene_4', 'scene_5', 'scene_6'])],
    toZigbee: [],
};

module.exports = definition;
```

pk802, 2023-06-17:
Hello,

What I have observed is in the message : [6,0,6,15,(X),1,112,9,142], X is what changes when corresponding button pressed.

Let me know if any additional info is required.

pk802, 2023-06-18:
Thanks in advance.

Not sure if I am missing something here. Any guidance.

Koenkk, 2023-06-18:
Your ext converter looks good, what is not working?

pk802, 2023-06-18:
This is what I see in the exposes of device:
battery   Null%
action    Null
linkquality  40lqi

I am unsure if converter is converter is unable parse the message properly.

Koenkk, 2023-06-19:
It looks correct to me, what is logged in the debug log when pressing a button?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

pk802, 2023-06-19:
Here is the debug log when I press button 1.
2023-06-20 01:47:29Received Zigbee message from '6Scene Switch1', type 'commandNotification', cluster 'greenPower', data '{"commandFrame":{"raw":{"data":[6,0,6,31,1,1,54,151,138],"type":"Buffer"}},"commandID":160,"frameCounter":143,"gppGddLink":6,"gppNwkAddr":6,"options":22544,"payloadSize":6,"srcID":4269812216}' from endpoint 242 with groupID 10744
2023-06-20 01:47:29No converter available for 'HVSC18-Z-6S-3.0' with cluster 'greenPower' and type 'commandNotification' and data '{"commandFrame":{"raw":{"data":[6,0,6,31,1,1,54,151,138],"type":"Buffer"}},"commandID":160,"frameCounter":143,"gppGddLink":6,"gppNwkAddr":6,"options":22544,"payloadSize":6,"srcID":4269812216}'

Here is the debug log when I press button 2.
2023-06-20 01:50:14Received Zigbee message from '6Scene Switch1', type 'commandNotification', cluster 'greenPower', data '{"commandFrame":{"raw":{"data":[6,0,6,34,2,1,248,130,145],"type":"Buffer"}},"commandID":160,"frameCounter":149,"gppGddLink":6,"gppNwkAddr":6,"options":22544,"payloadSize":6,"srcID":4269812216}' from endpoint 242 with groupID 10744
2023-06-20 01:50:14No converter available for 'HVSC18-Z-6S-3.0' with cluster 'greenPower' and type 'commandNotification' and data '{"commandFrame":{"raw":{"data":[6,0,6,34,2,1,248,130,145],"type":"Buffer"}},"commandID":160,"frameCounter":149,"gppGddLink":6,"gppNwkAddr":6,"options":22544,"payloadSize":6,"srcID":4269812216}'

Koenkk, 2023-06-20:
In your converter, `type: ['raw', 'attributeReport'],` has to change to `type: ['commandNotification'],`

pk802, 2023-06-20:
This is what I get if I change as suggested. 

Debug 2023-06-20 22:33:51Received Zigbee message from '6Scene Switch1', type 'commandNotification', cluster 'greenPower', data '{"commandFrame":{"raw":{"data":[6,0,6,181,1,1,143,30,135],"type":"Buffer"}},"commandID":160,"frameCounter":233,"gppGddLink":6,"gppNwkAddr":6,"options":22544,"payloadSize":6,"srcID":4269812216}' from endpoint 242 with groupID 10744
Info 2023-06-20 22:33:51MQTT publish: topic 'zigbee2mqtt/6Scene Switch1', payload '{"action":"scene_undefined","battery":null,"last_seen":"2023-06-20T22:33:51+05:30","linkquality":43}'
Info 2023-06-20 22:33:51MQTT publish: topic 'zigbee2mqtt/6Scene Switch1', payload '{"action":"","battery":null,"last_seen":"2023-06-20T22:33:51+05:30","linkquality":43}'
Info 2023-06-20 22:33:51MQTT publish: topic 'zigbee2mqtt/6Scene Switch1/action', payload 'scene_undefined'

Koenkk, 2023-06-20:
Now `const id = meta.device.modelID === 'GreenPower_2' ? msg.data[0] : msg.data[msg.data.length - 1];` has to be updated, I think to `const id = msg.data[4];`

pk802, 2023-06-20:
It's still not working. 

Here is code as per suggestion.

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const fzLocal = {
	scenes_recall_scene_greenPower: {
        cluster: 'greenPower',
        type: ['commandNotification'],
        convert: (model, msg, publish, options, meta) => {
            const id = msg.data[4];
            return {action: `scene_${id}`};
        },
    },
};
const definition = {
    zigbeeModel: ['GreenPower_2'],
    model: 'HVSC18-Z-6S-3.0',
    vendor: 'AMELECH',
    description: '6Scene Movable Switch',
    fromZigbee: [fzLocal.scenes_recall_scene_greenPower],
    exposes: [e.battery(),e.action(['scene_1', 'scene_2', 'scene_3', 'scene_4', 'scene_5', 'scene_6'])],
    toZigbee: [],
};

module.exports = definition;`

Here is the debug log

`Debug 2023-06-20 22:46:45Received Zigbee message from '6Scene Switch1', type 'commandNotification', cluster 'greenPower', data '{"commandFrame":{"raw":{"data":[6,0,6,185,1,1,241,177,134],"type":"Buffer"}},"commandID":160,"frameCounter":249,"gppGddLink":6,"gppNwkAddr":6,"options":22544,"payloadSize":6,"srcID":4269812216}' from endpoint 242 with groupID 10744
Info 2023-06-20 22:46:45MQTT publish: topic 'zigbee2mqtt/6Scene Switch1', payload '{"action":"scene_undefined","battery":null,"last_seen":"2023-06-20T22:46:45+05:30","linkquality":40}'
Info 2023-06-20 22:46:45MQTT publish: topic 'zigbee2mqtt/6Scene Switch1', payload '{"action":"","battery":null,"last_seen":"2023-06-20T22:46:45+05:30","linkquality":40}'
Info 2023-06-20 22:46:45MQTT publish: topic 'zigbee2mqtt/6Scene Switch1/action', payload 'scene_undefined'`

pk802, 2023-06-21:
I am made some changes. 
This is working.

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const fzLocal = {
	scenes_recall_scene_greenPower: {
        cluster: 'greenPower',
        type: ['commandNotification'],
        convert: (model, msg, publish, options, meta) => {
            const id = msg.data.commandFrame.raw[4];
            return {action: `scene_${id}`};
        },
    },
};
const definition = {
    zigbeeModel: ['GreenPower_2'],
    model: 'HVSC18-Z-6S-3.0',
    vendor: 'AMELECH',
    description: '6Scene Movable Switch',
    fromZigbee: [fzLocal.scenes_recall_scene_greenPower],
    exposes: [e.battery(),e.action(['scene_1', 'scene_2', 'scene_3', 'scene_4', 'scene_5', 'scene_6'])],
    toZigbee: [],
};

module.exports = definition;
```

Now the problem is every button is sending 4 messages. 1 message each second. 
So I have to wait for 4 second before I press any other scene button. 
Do you have any idea what may be causing it?

closed 2023-07-29 by github-actions[bot]