Koenkk/zigbee2mqtt #18048: [New device support]:  SAMSUNG SDS Doorlock

sdhuaren, 2023-06-17:
### Link

https://samsungsmarthome.cyberbiz.co/products/shp-dp728-gold

### Database entry

{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b000be8b1a1","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,8,10,11,12,13,47,110,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"8":{"profId":260,"epId":8,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"10":{"profId":260,"epId":10,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[1281,10],"outClusterList":[1280,1282],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"12":{"profId":49246,"epId":12,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"13":{"profId":260,"epId":13,"devId":5,"inClusterList":[25],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"47":{"profId":260,"epId":47,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"110":{"profId":260,"epId":110,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":true,"meta":{},"lastSeen":null,"defaultSendRequestWhen":"immediate"} {"id":2,"type":"EndDevice","ieeeAddr":"0x000d6f000f836d9f","nwkAddr":20964,"manufId":3,"manufName":"SAMSUNG SDS","powerSource":"Battery","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":10,"inClusterList":[0,1,3,4,5,9,257],"outClusterList":[25],"clusters":{"closuresDoorLock":{"attributes":{"lockState":2}},"genPowerCfg":{"attributes":{"batteryVoltage":86}},"genBasic":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":83,"hwVersion":1,"zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1687043229783,"defaultSendRequestWhen":"immediate"}

### Comments

Hello,
I have paired the door lock with my CC2531 and interviewed successfully. The door lock is defined with finger print.
I tried to write an converter but failed.

`<html xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/DELL/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/DELL/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

debug | 2023/6/18 | 0:25:15 | Received MQTT message on   'zigbee2mqtt/bridge/config' with data   '{"commit":"unknown","coordinator":{"meta":{"maintrel":2,"majorrel":2,"minorrel":7,"product":2,"revision":20190425,"transportrev":2},"type":"zStack30x"},"log_level":"debug","network":{"channel":11,"extendedPanID":"0x00124b000be8b1a1","panID":6754},"permit_join":false,"version":"1.30.4"}'
-- | -- | -- | --
debug | 2023/6/18 | 0:25:15 | Received MQTT message on   'homeassistant/sensor/0x000d6f000f836d9f/linkquality/config' with data   '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0x000d6f000f836d9f/availability"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0x000d6f000f836d9f"],"manufacturer":"SAMSUNG   SDS","model":"Samsung Door Lock (SHP-DP728)","name":"0x000d6f000f836d9f"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/0x000d6f000f836d9f","name":"0x000d6f000f836d9f   linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/0x000d6f000f836d9f","unique_id":"0x000d6f000f836d9f_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{   value_json.linkquality }}"}'
debug | 2023/6/18 | 0:25:15 | Received MQTT message on   'homeassistant/sensor/0x000d6f000f836d9f/linkquality/config' with data   '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0x000d6f000f836d9f/availability"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0x000d6f000f836d9f"],"manufacturer":"SAMSUNG   SDS","model":"Samsung Door Lock (SHP-DP728)","name":"0x000d6f000f836d9f"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/0x000d6f000f836d9f","name":"0x000d6f000f836d9f   linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/0x000d6f000f836d9f","unique_id":"0x000d6f000f836d9f_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{   value_json.linkquality }}"}'
debug | 2023/6/18 | 0:26:14 | No converter available for 'SHP-DP728' with cluster 'closuresDoorLock'   and type 'raw' and data   '{"data":[9,81,32,2,14,255,255,0,211,242,141,100],"type":"Buffer"}'
debug | 2023/6/18 | 0:26:45 | No converter available for 'SHP-DP728' with cluster 'genAlarms' and type   'commandAlarm' and data '{"alarmcode":7,"clusterid":257}'
debug | 2023/6/18 | 0:52:20 | No converter available for 'SHP-DP728' with cluster 'closuresDoorLock'   and type 'raw' and data   '{"data":[9,84,32,2,10,255,255,0,241,248,141,100],"type":"Buffer"}'
debug | 2023/6/18 | 0:52:32 | No converter available for 'SHP-DP728' with cluster 'closuresDoorLock'   and type 'raw' and data   '{"data":[9,85,32,0,2,0,0,0,253,248,141,100],"type":"Buffer"}'
debug | 2023/6/18 | 0:52:37 | No converter available for 'SHP-DP728' with cluster 'closuresDoorLock'   and type 'raw' and data   '{"data":[9,86,32,2,10,255,255,0,2,249,141,100],"type":"Buffer"}'
debug | 2023/6/18 | 0:52:44 | No converter available for 'SHP-DP728' with cluster 'closuresDoorLock'   and type 'raw' and data   '{"data":[9,87,32,4,2,31,0,0,9,249,141,100],"type":"Buffer"}'
debug | 2023/6/18 | 0:53:13 | No converter available for 'SHP-DP728' with cluster 'genAlarms' and type   'commandAlarm' and data '{"alarmcode":7,"clusterid":257}'

</body>

</html>
`

### External converter

```shell
const fzLocal = {
    Samsung_easycode_action: {
        cluster: 'closuresDoorLock',
        type: 'raw',
        convert: (model, msg, publish, options, meta) => {
            const lookup = {
                10: 'lock',
                14: 'inside_unlock',
                1: 'command_unlock',
				        2: 'manual_unlock',
				        4: 'fingerpri_unlock',
                5: 'bluetooth_unlock',
                3: 'rfid_unlock',
                0: 'keypad_unlock',
            };
            const value = lookup[msg.data[5]];
            if (value == 'lock' || value == 'inside_unlock') {
                return {action: value};
            } else {
                return {action: lookup[msg.data[4]]};
            }
        },
    },
```

sdhuaren, 2023-07-08:
@Koenkk Could you pls help with it?

closed 2023-08-16 by github-actions[bot]

tazeemn, 2024-07-15:
Hi @sdhuaren I'm also interested in adding a ZigBee module to my SHP-728. Did you have any luck sorting this out? Where did you source the ZigBee module? I'm in Australia and I can't find it on any local sites.

sdhuaren, 2024-07-15:
Hi there, I am glad that you have the same doorlock from SAMSUNG. I got the module from Taobao, a local online market. Can you sort it out in Alibaba?
 

Best Regards,
Wei Zhang