Koenkk/zigbee2mqtt #18051: TONGOU TO-Q-SY2-163JZT. The values the device publishes and breaks on are not normal operation values.

garry0garry, 2023-06-17:
### What happened?

https://github.com/home-assistant/core/issues/94781

```
info 2023-06-17 23:06:37MQTT publish: topic 'zigbee2mqtt/0xe0798dfffe0eb6e8', payload '{"current":1.04,"energy":0.51,"indicator_mode":"off/on","last_seen":"2023-06-17T23:06:37+03:00","linkquality":88,"over_current_breaker":"OFF","over_current_threshold":0,"over_voltage_breaker":"OFF","over_voltage_threshold":0,"power":170,"power_breaker":"OFF","power_outage_memory":"restore","power_threshold":13,"state":"ON","temperature":27.08,"temperature_breaker":"OFF","temperature_threshold":0,"under_voltage_breaker":"OFF","under_voltage_threshold":0,"update":{"installed_version":-1,"latest_version":-1,"state":"idle"},"update_available":null,"voltage":224}'
debug 2023-06-17 23:06:46Received Zigbee message from '0xe0798dfffe0eb6e8', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,52]}' from endpoint 1 with groupID 0
info 2023-06-17 23:06:46MQTT publish: topic 'zigbee2mqtt/0xe0798dfffe0eb6e8', payload '{"current":1.04,"energy":0.52,"indicator_mode":"off/on","last_seen":"2023-06-17T23:06:46+03:00","linkquality":84,"over_current_breaker":"OFF","over_current_threshold":0,"over_voltage_breaker":"OFF","over_voltage_threshold":0,"power":170,"power_breaker":"OFF","power_outage_memory":"restore","power_threshold":13,"state":"ON","temperature":27.08,"temperature_breaker":"OFF","temperature_threshold":0,"under_voltage_breaker":"OFF","under_voltage_threshold":0,"update":{"installed_version":-1,"latest_version":-1,"state":"idle"},"update_available":null,"voltage":224}'
debug 2023-06-17 23:06:46Received Zigbee message from '0xe0798dfffe0eb6e8', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsVoltage":219}' from endpoint 1 with groupID 0
info 2023-06-17 23:06:46MQTT publish: topic 'zigbee2mqtt/0xe0798dfffe0eb6e8', payload '{"current":1.04,"energy":0.52,"indicator_mode":"off/on","last_seen":"2023-06-17T23:06:46+03:00","linkquality":80,"over_current_breaker":"OFF","over_current_threshold":0,"over_voltage_breaker":"OFF","over_voltage_threshold":0,"power":170,"power_breaker":"OFF","power_outage_memory":"restore","power_threshold":13,"state":"ON","temperature":27.08,"temperature_breaker":"OFF","temperature_threshold":0,"under_voltage_breaker":"OFF","under_voltage_threshold":0,"update":{"installed_version":-1,"latest_version":-1,"state":"idle"},"update_available":null,"voltage":219}'
debug 2023-06-17 23:06:46Received Zigbee message from '0xe0798dfffe0eb6e8', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,52]}' from endpoint 1 with groupID 0
info 2023-06-17 23:06:46MQTT publish: topic 'zigbee2mqtt/0xe0798dfffe0eb6e8', payload '{"current":1.04,"energy":0.52,"indicator_mode":"off/on","last_seen":"2023-06-17T23:06:46+03:00","linkquality":80,"over_current_breaker":"OFF","over_current_threshold":0,"over_voltage_breaker":"OFF","over_voltage_threshold":0,"power":170,"power_breaker":"OFF","power_outage_memory":"restore","power_threshold":13,"state":"ON","temperature":27.08,"temperature_breaker":"OFF","temperature_threshold":0,"under_voltage_breaker":"OFF","under_voltage_threshold":0,"update":{"installed_version":-1,"latest_version":-1,"state":"idle"},"update_available":null,"voltage":219}'
debug 2023-06-17 23:07:39Received Zigbee message from '0xe0798dfffe0eb6e8', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsVoltage":224}' from endpoint 1 with groupID 0
info 2023-06-17 23:07:39MQTT publish: topic 'zigbee2mqtt/0xe0798dfffe0eb6e8', payload '{"current":1.04,"energy":0.52,"indicator_mode":"off/on","last_seen":"2023-06-17T23:07:39+03:00","linkquality":80,"over_current_breaker":"OFF","over_current_threshold":0,"over_voltage_breaker":"OFF","over_voltage_threshold":0,"power":170,"power_breaker":"OFF","power_outage_memory":"restore","power_threshold":13,"state":"ON","temperature":27.08,"temperature_breaker":"OFF","temperature_threshold":0,"under_voltage_breaker":"OFF","under_voltage_threshold":0,"update":{"installed_version":-1,"latest_version":-1,"state":"idle"},"update_available":null,"voltage":224}'
debug 2023-06-17 23:09:13Received Zigbee message from '0xe0798dfffe0eb6e8', type 'attributeReport', cluster 'genBasic', data '{"65506":31,"65508":0,"appVersion":67}' from endpoint 1 with groupID 0
info 2023-06-17 23:09:13MQTT publish: topic 'zigbee2mqtt/0xe0798dfffe0eb6e8', payload '{"current":1.04,"energy":0.52,"indicator_mode":"off/on","last_seen":"2023-06-17T23:09:13+03:00","linkquality":112,"over_current_breaker":"OFF","over_current_threshold":0,"over_voltage_breaker":"OFF","over_voltage_threshold":0,"power":170,"power_breaker":"OFF","power_outage_memory":"restore","power_threshold":13,"state":"ON","temperature":27.08,"temperature_breaker":"OFF","temperature_threshold":0,"under_voltage_breaker":"OFF","under_voltage_threshold":0,"update":{"installed_version":-1,"latest_version":-1,"state":"idle"},"update_available":null,"voltage":224}'
debug 2023-06-17 23:10:22Received Zigbee message from '0xe0798dfffe0eb6e8', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,53]}' from endpoint 1 with groupID 0
info 2023-06-17 23:10:22MQTT publish: topic 'zigbee2mqtt/0xe0798dfffe0eb6e8', payload '{"current":1.04,"energy":0.53,"indicator_mode":"off/on","last_seen":"2023-06-17T23:10:22+03:00","linkquality":80,"over_current_breaker":"OFF","over_current_threshold":0,"over_voltage_breaker":"OFF","over_voltage_threshold":0,"power":170,"power_breaker":"OFF","power_outage_memory":"restore","power_threshold":13,"state":"ON","temperature":27.08,"temperature_breaker":"OFF","temperature_threshold":0,"under_voltage_breaker":"OFF","under_voltage_threshold":0,"update":{"installed_version":-1,"latest_version":-1,"state":"idle"},"update_available":null,"voltage":224}'
debug 2023-06-17 23:10:22Received Zigbee message from '0xe0798dfffe0eb6e8', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,53]}' from endpoint 1 with groupID 0
info 2023-06-17 23:10:22MQTT publish: topic 'zigbee2mqtt/0xe0798dfffe0eb6e8', payload '{"current":1.04,"energy":0.53,"indicator_mode":"off/on","last_seen":"2023-06-17T23:10:22+03:00","linkquality":80,"over_current_breaker":"OFF","over_current_threshold":0,"over_voltage_breaker":"OFF","over_voltage_threshold":0,"power":170,"power_breaker":"OFF","power_outage_memory":"restore","power_threshold":13,"state":"ON","temperature":27.08,"temperature_breaker":"OFF","temperature_threshold":0,"under_voltage_breaker":"OFF","under_voltage_threshold":0,"update":{"installed_version":-1,"latest_version":-1,"state":"idle"},"update_available":null,"voltage":224}'
debug 2023-06-17 23:11:54Received Zigbee message from '0xe0798dfffe0eb6e8', type 'attributeReport', cluster 'genBasic', data '{"65506":31,"65508":0,"appVersion":67}' from endpoint 1 with groupID 0
info 2023-06-17 23:11:54MQTT publish: topic 'zigbee2mqtt/0xe0798dfffe0eb6e8', payload '{"current":1.04,"energy":0.53,"indicator_mode":"off/on","last_seen":"2023-06-17T23:11:54+03:00","linkquality":72,"over_current_breaker":"OFF","over_current_threshold":0,"over_voltage_breaker":"OFF","over_voltage_threshold":0,"power":170,"power_breaker":"OFF","power_outage_memory":"restore","power_threshold":13,"state":"ON","temperature":27.08,"temperature_breaker":"OFF","temperature_threshold":0,"under_voltage_breaker":"OFF","under_voltage_threshold":0,"update":{"installed_version":-1,"latest_version":-1,"state":"idle"},"update_available":null,"voltage":224}'
```

```
info 2023-06-17 23:06:49MQTT publish: topic 'zigbee2mqtt/0xe0798dfffe0eb6bd', payload '{"current":1.74,"energy":12.76,"indicator_mode":"off/on","last_seen":"2023-06-17T23:06:49+03:00","linkquality":76,"over_current_breaker":"OFF","over_current_threshold":35,"over_voltage_breaker":"OFF","over_voltage_threshold":255,"power":331,"power_breaker":"OFF","power_outage_memory":"restore","power_threshold":13,"state":"ON","temperature":24.52,"temperature_breaker":"OFF","temperature_threshold":100,"under_voltage_breaker":"OFF","under_voltage_threshold":0,"update":{"installed_version":-1,"latest_version":-1,"state":"idle"},"update_available":null,"voltage":223}'
debug 2023-06-17 23:07:55Received Zigbee message from '0xe0798dfffe0eb6bd', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
[... cut ...] r_outage_memory":"restore","power_threshold":13,"state":"ON","temperature":24.52,"temperature_breaker":"OFF","temperature_threshold":100,"under_voltage_breaker":"OFF","under_voltage_threshold":0,"update":{"installed_version":-1,"latest_version":-1,"state":"idle"},"update_available":null,"voltage":223}'

```

### Zigbee2MQTT version

1.31.2 commit: 21f5125

### Adapter firmware version

6.10.3.0 build 297

### Adapter

https://www.zigbee2mqtt.io/devices/TO-Q-SY2-163JZT.html#tongou-to-q-sy2-163jzt

garry0garry, 2023-06-17:
Home Assistant:
```
Logger: homeassistant.components.mqtt.number
Source: components/mqtt/number.py:203
Integration: MQTT (documentation, issues)
First occurred: 19:52:40 (3768 occurrences)
Last logged: 23:19:29

Invalid value for number.0xe0798dfffe0eb6e8_temperature_threshold: 0 (range 40.0 - 100.0)
Invalid value for number.0xe0798dfffe0eb6e8_over_current_threshold: 0 (range 1.0 - 64.0)
Invalid value for number.0xe0798dfffe0eb6e8_over_voltage_threshold: 0 (range 220.0 - 260.0)
Invalid value for number.0xe0798dfffe0eb6e8_under_voltage_threshold: 0 (range 76.0 - 240.0)
Invalid value for number.0xe0798dfffe0eb6bd_under_voltage_threshold: 0 (range 76.0 - 240.0)
```

garry0garry, 2023-06-18:
I carefully looked at the logs and realized that this is not an error!
Restriction (temperature_threshold, power_threshold, over_current_threshold etc.) values are unknown on first connection. Zigbee2MQTT sets 0. You need to set any value from the allowed ones. There is no need to include restrictions.

closed 2023-06-18 by garry0garry

WPANGEA, 2024-08-22:
For 63A version (new one) you need once the device is conected via zigbee2mqtt to keep the Tongou button physcal switch  pressed 2 or 3 seconds in order to let the home assistant tongou device web interface exposes to write the threshold values for A, kW, UV, OV and ºC .and to keep them written