Koenkk/zigbee2mqtt #18060: OTA updates are broken on latest dev

rccoleman, 2023-06-18:
### What happened?

On the latest dev branch, OTA updates seem to be broken.  With a bit of debug that I added, I see this in the logs when Zigbee2MQTT properly identifies that there's an updates and I try to proceed with the update:

```
debug 2023-06-18 15:25:55: ZigbeeOTA: downloaded main index
debug 2023-06-18 15:25:55: getNewImage for '0x54ef4410005ca518', meta {"fileVersion":10,"fileSize":196238,"url":"https://github.com/Koenkk/zigbee-OTA/raw/master/images/Xiaomi/20230130180718_OTA_lumi.motion.ac02_0.0.0_0010_20230104_390E3D.ota","sha512":"22fb7b610d169f433ef0bad5a6c6e5f0abfb5b09a1ed2155bc5da23b6cfd204659441136947270b98514581e97233cd2db33002ec277796e1e2ae1572abf2853"}
info  2023-06-18 15:25:55: UPDATE: https://github.com/Koenkk/zigbee-OTA/raw/master/images/Xiaomi/20230130180718_OTA_lumi.motion.ac02_0.0.0_0010_20230104_390E3D.ota
info  2023-06-18 15:25:55: Not Valid URL
debug 2023-06-18 15:25:55: ZigbeeOTA: getting local firmware file /app/data/https:/github.com/Koenkk/zigbee-OTA/raw/master/images/Xiaomi/20230130180718_OTA_lumi.motion.ac02_0.0.0_0010_20230104_390E3D.ota
debug 2023-06-18 15:25:55: Update of 'DNH Motion Sensor' failed (Error: ENOENT: no such file or directory, open '/app/data/https:/github.com/Koenkk/zigbee-OTA/raw/master/images/Xiaomi/20230130180718_OTA_lumi.motion.ac02_0.0.0_0010_20230104_390E3D.ota')
```
The `UPDATE: xxx` line shows the URL that was passed into `isValidUrl()`, and `Not Valid URL` is printed if it fails the check.  A little more sleuthing shows that `parsed = uri_js_1.default.parse(url);` seems to throw an exception, which ends up returning `false`, despite the fact that it's perfectly normal URL.

### What did you expect to happen?

I expected the device to be updated by downloading the firmware update from the provided URL.

### How to reproduce it (minimal and precise)

1. Pair a device that has a firmware update available, such as a [RTCGQ14LM](https://www.zigbee2mqtt.io/devices/RTCGQ14LM.html#xiaomi-rtcgq14lm).
2. Check for a firmware update, and note that the UI indicates that one is available.
3. Attempt to update to the latest firmware version by clicking the button.
4. Note that a dialog appears with something like
```
Debug 2023-06-18 14:27:46Update of 'DNH Motion Sensor' failed (Error: ENOENT: no such file or directory, open '/app/data/https:/github.com/Koenkk/zigbee-OTA/raw/master/images/Xiaomi/
```

### Zigbee2MQTT version

1.31.2-dev commit: 57aaa5a

### Adapter firmware version

20221226

### Adapter

Tubeszb USB

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/11782759/log.txt)

rccoleman, 2023-06-19:
Figured it out.  Fix is this:

```
diff --git a/src/lib/ota/zigbeeOTA.ts b/src/lib/ota/zigbeeOTA.ts
index d5049a94..25012a1a 100644
--- a/src/lib/ota/zigbeeOTA.ts
+++ b/src/lib/ota/zigbeeOTA.ts
@@ -3,7 +3,7 @@ import * as common from './common';
 import {Logger, Zh, Ota, KeyValueAny} from '../types';
 const axios = common.getAxios();
 import fs from 'fs';
-import URI from 'uri-js';
+import * as URI from 'uri-js';
 import path from 'path';
 
 let overrideIndexFileName: string = null;
```

Koenkk, 2023-06-19:
@rccoleman many thanks for the fix, integrated it!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-06-19 by Koenkk