Koenkk/zigbee2mqtt #18064: [New device support]: Tuya TS0601  _TZE200_5kynszlt 4 gang switch with 4 scene control

Qianli-Ma, 2023-06-19:
### Link

https://www.alibaba.com/product-detail/SEXANKA-KNX-TUYA-Zigbee-Wifi-RS485_1600470231757.html?spm=a2700.wholesale.0.0.40404bc1Lxg2wq

### Database entry

{"id":4,"type":"Router","ieeeAddr":"0x847127fffeaff81d","nwkAddr":11413,"manufId":4098,"manufName":"_TZE200_5kynszlt","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0005","65506":31,"65508":1,"65534":0,"modelId":"TS0601","manufacturerName":"_TZE200_5kynszlt","powerSource":1,"zclVersion":3,"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{}},"manuSpecificTuya_3":{"attributes":{}},"manuSpecificTuya_2":{"attributes":{}},"genOnOffSwitchCfg":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-708457359},"lastSeen":1687187131581,"defaultSendRequestWhen":"immediate"}

### Comments

The following external converter only works for l1-l6, I'm having trouble making l7 and l8 work.
Pretty sure that corrisponding datapoints are 7 and 8
```
debug 2023-06-19 23:26:01: Received Zigbee message from '0x847127fffeaff81d', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":7}],"seq":19456}' from endpoint 1 with groupID 0
debug 2023-06-19 23:26:01: Datapoint 7 not defined for '_TZE200_5kynszlt' with data {"dp":7,"datatype":4,"data":{"type":"Buffer","data":[0]}}
debug 2023-06-19 23:26:02: Received Zigbee message from '0x847127fffeaff81d', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":8}],"seq":19712}' from endpoint 1 with groupID 0
debug 2023-06-19 23:26:02: Datapoint 8 not defined for '_TZE200_5kynszlt' with data {"dp":8,"datatype":4,"data":{"type":"Buffer","data":[0]}}
debug 2023-06-19 23:26:03: Received Zigbee message from '0x847127fffeaff81d', type 'attributeReport', cluster 'genBasic', data '{"65506":31,"65508":0,"appVersion":68}' from endpoint 1 with groupID 0
```
log for buttons 7 and 8 after removing relevent exposes.

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [
        {
            modelID: "TS0601",
            manufacturerName: "_TZE200_5kynszlt",
        }
    ],
    model: 'G3 Smart Switch',
    vendor: 'TuYa',
    description: 'TuYa Smart Switch (4 gang + 2 scene) with neutral wire',
    powerSource: 'Mains (single phase)',
    fromZigbee: [tuya.fz.datapoints],
        toZigbee: [tuya.tz.datapoints],
        configure: tuya.configureMagicPacket,
        exposes: [
            tuya.exposes.switch().withEndpoint('l1'),
            tuya.exposes.switch().withEndpoint('l2'),
            tuya.exposes.switch().withEndpoint('l3'),
            tuya.exposes.switch().withEndpoint('l4'),
            tuya.exposes.switch().withEndpoint('l5'),
            tuya.exposes.switch().withEndpoint('l6'),
            tuya.exposes.switch().withEndpoint('l7'),
            tuya.exposes.switch().withEndpoint('l8'),
        ],
        endpoint: (device) => {
            return {'l1': 1, 'l2': 1, 'l3': 1, 'l4': 1, 'l5': 1, 'l6': 1, 'l7': 1, 'l8': 1};
        },
        meta: {
            multiEndpoint: true,
            tuyaDatapoints: [
                [24, 'state_l1', tuya.valueConverter.onOff],
                [25, 'state_l2', tuya.valueConverter.onOff],
                [26, 'state_l3', tuya.valueConverter.onOff],
                [27, 'state_l4', tuya.valueConverter.onOff],
                [28, 'state_l5', tuya.valueConverter.onOff],
                [29, 'state_l6', tuya.valueConverter.onOff],
                [7, 'state_l7', tuya.valueConverter.onOff],
                [8, 'state_l8', tuya.valueConverter.onOff],
            ],
        },
};

module.exports = definition;
```

Koenkk, 2023-06-19:
Shouldn't these be:

```
                [30, 'state_l7', tuya.valueConverter.onOff],
                [31, 'state_l8', tuya.valueConverter.onOff],
```

Qianli-Ma, 2023-06-20:
That was what I ithought as well, however using 
```
                [30, 'state_l7', tuya.valueConverter.onOff],
                [31, 'state_l8', tuya.valueConverter.onOff],
```
then pressing the corresponding switches it gives out the following log
```
Zigbee2MQTT:debug 2023-06-20 08:23:50: Received Zigbee message from '0x847127fffeaff81d', type 'attributeReport', cluster 'genBasic', data '{"65506":31,"65508":0,"appVersion":68}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2023-06-20 08:23:53: Received Zigbee message from '0x847127fffeaff81d', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":7}],"seq":20224}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2023-06-20 08:23:53: Datapoint 7 not defined for '_TZE200_5kynszlt' with data {"dp":7,"datatype":4,"data":{"type":"Buffer","data":[0]}}
Zigbee2MQTT:debug 2023-06-20 08:23:54: Received Zigbee message from '0x847127fffeaff81d', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":8}],"seq":20480}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2023-06-20 08:23:54: Datapoint 8 not defined for '_TZE200_5kynszlt' with data {"dp":8,"datatype":4,"data":{"type":"Buffer","data":[0]}}

```
Using the following
```
                [7, 'state_l7', tuya.valueConverter.onOff],
                [8, 'state_l8', tuya.valueConverter.onOff],
```
It throws out these errors when pressing  the buttons
```
Zigbee2MQTT:debug 2023-06-20 08:24:33: Received Zigbee message from '0x847127fffeaff81d', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":7}],"seq":20736}' from endpoint 1 with groupID 0
Zigbee2MQTT:error 2023-06-20 08:24:33: Exception while calling fromZigbee converter: Value '0' is not allowed, expected one of true,false}
Zigbee2MQTT:debug 2023-06-20 08:24:33: Error: Value '0' is not allowed, expected one of true,false
    at Object.from (E:\zigbee2mqtt\zigbee2mqtt\node_modules\zigbee-herdsman-converters\src\lib\tuya.js:984:35)
    at Object.convert (E:\zigbee2mqtt\zigbee2mqtt\node_modules\zigbee-herdsman-converters\src\lib\tuya.js:1624:57)
    at Receive.onDeviceMessage (E:\zigbee2mqtt\zigbee2mqtt\lib\extension\receive.ts:150:51)
    at EventEmitter.emit (node:events:525:35)
    at EventBus.emitDeviceMessage (E:\zigbee2mqtt\zigbee2mqtt\lib\eventBus.ts:102:22)
    at Controller.<anonymous> (E:\zigbee2mqtt\zigbee2mqtt\lib\zigbee.ts:106:27)
    at Controller.emit (node:events:513:28)
    at Controller.selfAndDeviceEmit (E:\zigbee2mqtt\zigbee2mqtt\node_modules\zigbee-herdsman\src\controller\controller.ts:515:14)
    at Controller.onZclOrRawData (E:\zigbee2mqtt\zigbee2mqtt\node_modules\zigbee-herdsman\src\controller\controller.ts:726:18)
    at ZStackAdapter.<anonymous> (E:\zigbee2mqtt\zigbee2mqtt\node_modules\zigbee-herdsman\src\controller\controller.ts:144:70)
Zigbee2MQTT:debug 2023-06-20 08:24:36: Received Zigbee message from '0x847127fffeaff81d', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":8}],"seq":20992}' from endpoint 1 with groupID 0
Zigbee2MQTT:error 2023-06-20 08:24:36: Exception while calling fromZigbee converter: Value '0' is not allowed, expected one of true,false}
Zigbee2MQTT:debug 2023-06-20 08:24:36: Error: Value '0' is not allowed, expected one of true,false
    at Object.from (E:\zigbee2mqtt\zigbee2mqtt\node_modules\zigbee-herdsman-converters\src\lib\tuya.js:984:35)
    at Object.convert (E:\zigbee2mqtt\zigbee2mqtt\node_modules\zigbee-herdsman-converters\src\lib\tuya.js:1624:57)
    at Receive.onDeviceMessage (E:\zigbee2mqtt\zigbee2mqtt\lib\extension\receive.ts:150:51)
    at EventEmitter.emit (node:events:525:35)
    at EventBus.emitDeviceMessage (E:\zigbee2mqtt\zigbee2mqtt\lib\eventBus.ts:102:22)
    at Controller.<anonymous> (E:\zigbee2mqtt\zigbee2mqtt\lib\zigbee.ts:106:27)
    at Controller.emit (node:events:513:28)
    at Controller.selfAndDeviceEmit (E:\zigbee2mqtt\zigbee2mqtt\node_modules\zigbee-herdsman\src\controller\controller.ts:515:14)
    at Controller.onZclOrRawData (E:\zigbee2mqtt\zigbee2mqtt\node_modules\zigbee-herdsman\src\controller\controller.ts:726:18)
    at ZStackAdapter.<anonymous> (E:\zigbee2mqtt\zigbee2mqtt\node_modules\zigbee-herdsman\src\controller\controller.ts:144:70)

```

Koenkk, 2023-06-20:
@Qianli-Ma but does the toggling from z2m work with 30 and 31?

Qianli-Ma, 2023-06-20:
@Koenkk unfortunately no

Koenkk, 2023-06-21:
Can you try with `[7, 'state_l7', tuya.valueConverter.raw],` and publish to `zigbee2mqtt/DEVICE_ID/set` payload `{"state_l7": 1}`, does it work?

Qianli-Ma, 2023-06-26:
@Koenkk I have determined with a TuYa bridge the following DP values. 
| DP ID | DP Name          | Identifier   | Data Transfer Type | Data Type | Properties                                                   |
| ----- | ---------------- | ------------ | ------------------ | --------- | ------------------------------------------------------------ |
| 1     | Scene 1          | scene_1      | Send and Report    | enum      | Enum Value: scene                                            |
| 2     | Scene 2          | scene_2      | Send and Report    | enum      | Enum Value: scene                                            |
| 3     | Scene 3          | scene_3      | Send and Report    | enum      | Enum Value: scene                                            |
| 4     | Scene 4          | scene_4      | Send and Report    | enum      | Enum Value: scene                                            |
| 5     | Scene 5          | scene_5      | Send and Report    | enum      | Enum Value: scene                                            |
| 6     | Scene 6          | scene_6      | Send and Report    | enum      | Enum Value: scene                                            |
| 7     | Scene 7          | scene_7      | Send and Report    | enum      | Enum Value: scene                                            |
| 8     | Scene 8          | scene_8      | Send and Report    | enum      | Enum Value: scene                                            |
| 18    | Mode 1           | mode_1       | Send and Report    | enum      | Enum Value: switch_1, scene_1                                |
| 19    | Mode 2           | mode_2       | Send and Report    | enum      | Enum Value: switch_2, scene_2                                |
| 20    | Mode 3           | mode_3       | Send and Report    | enum      | Enum Value: switch_3, scene_3                                |
| 21    | Mode 4           | mode_4       | Send and Report    | enum      | Enum Value: switch_4, scene_4                                |
| 22    | Mode 5           | mode_5       | Send and Report    | enum      | Enum Value: switch_5, scene_5                                |
| 23    | Mode 6           | mode_6       | Send and Report    | enum      | Enum Value: switch_6, scene_6                                |
| 24    | Switch 1         | switch_1     | Send and Report    | bool      |                                                              |
| 25    | Switch 2         | switch_2     | Send and Report    | bool      |                                                              |
| 26    | Switch 3         | switch_3     | Send and Report    | bool      |                                                              |
| 27    | Switch 4         | switch_4     | Send and Report    | bool      |                                                              |
| 28    | Switch 5         | switch_5     | Send and Report    | bool      |                                                              |
| 29    | Switch 6         | switch_6     | Send and Report    | bool      |                                                              |
| 30    | Timer 1          | countdown_1  | Send and Report    | value     | Value Range: 0-43200,            Pitch: 1, Scale:  0,            Unit: s |
| 31    | Timer 2          | countdown_2  | Send and Report    | value     | Value Range: 0-43200,            Pitch: 1, Scale:  0,            Unit: s |
| 32    | Timer 3          | countdown_3  | Send and Report    | value     | Value Range: 0-43200,            Pitch: 1, Scale:  0,            Unit: s |
| 33    | Timer 4          | countdown_4  | Send and Report    | value     | Value Range: 0-43200,            Pitch: 1, Scale:  0,            Unit: s |
| 34    | Timer 5          | countdown_5  | Send and Report    | value     | Value Range: 0-43200,            Pitch: 1, Scale:  0,            Unit: s |
| 35    | Timer 6          | countdown_6  | Send and Report    | value     | Value Range: 0-43200,            Pitch: 1, Scale:  0,            Unit: s |
| 37    | Indicator Status | light_mode   | Send and Report    | enum      | Enum Value: relay, pos, none                                 |
| 38    | Restart Status   | relay_status | Send and Report    | enum      | Enum Value: off, on, memory                                  |

Qianli-Ma, 2023-06-26:
> Can you try with `[7, 'state_l7', tuya.valueConverter.raw],` and publish to `zigbee2mqtt/DEVICE_ID/set` payload `{"state_l7": 1}`, does it work?

I believe it works (No errors are thrown), however, since it is classified as a scene switch, the switch does not reflect a status change.

Koenkk, 2023-06-26:
It seems this device has only 6 switches and not 8?

Qianli-Ma, 2023-06-27:
I belive the other switches act as scence controllers, something similar like U86KCJ-ZP in tuya.ts
However, I'm not familiar how to convert the following  code for external converter use
```
    scenes_recall_scene_65029: {
        cluster: '65029',
        type: ['raw', 'attributeReport'],
        convert: (model, msg, publish, options, meta) => {
            const id = meta.device.modelID === '005f0c3b' ? msg.data[0] : msg.data[msg.data.length - 1];
            return {action: `scene_${id}`};
        },
    } as Fz.Converter,
```

Qianli-Ma, 2023-06-28:
```
debug 2023-06-28 20:26:37: Skipping message, definition is undefined and still interviewing
info  2023-06-28 20:26:37: Successfully interviewed '0x847127fffeaff81d', device has successfully been paired
info  2023-06-28 20:26:38: Device '0x847127fffeaff81d' is supported, identified as: TuYa TuYa Smart Switch (4 gang + 2 scene) with neutral wire (G3 Smart Switch)
info  2023-06-28 20:26:38: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"TuYa Smart Switch (4 gang + 2 scene) with neutral wire","exposes":[{"endpoint":"l1","features":[{"access":3,"description":"On/off state of the switch","endpoint":"l1","name":"state","property":"state_l1","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"endpoint":"l2","features":[{"access":3,"description":"On/off state of the switch","endpoint":"l2","name":"state","property":"state_l2","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"endpoint":"l3","features":[{"access":3,"description":"On/off state of the switch","endpoint":"l3","name":"state","property":"state_l3","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"endpoint":"l4","features":[{"access":3,"description":"On/off state of the switch","endpoint":"l4","name":"state","property":"state_l4","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"endpoint":"l5","features":[{"access":3,"description":"On/off state of the switch","endpoint":"l5","name":"state","property":"state_l5","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"endpoint":"l6","features":[{"access":3,"description":"On/off state of the switch","endpoint":"l6","name":"state","property":"state_l6","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"endpoint":"l7","features":[{"access":3,"description":"On/off state of the switch","endpoint":"l7","name":"state","property":"state_l7","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"endpoint":"l8","features":[{"access":3,"description":"On/off state of the switch","endpoint":"l8","name":"state","property":"state_l8","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"G3 Smart Switch","options":[],"supports_ota":false,"vendor":"TuYa"},"friendly_name":"0x847127fffeaff81d","ieee_address":"0x847127fffeaff81d","status":"successful","supported":true},"type":"device_interview"}'
info  2023-06-28 20:26:38: Configuring '0x847127fffeaff81d'
debug 2023-06-28 20:26:38: Received Zigbee message from '0x847127fffeaff81d', type 'readResponse', cluster 'genBasic', data '{"65534":0,"appVersion":68,"manufacturerName":"_TZE200_5kynszlt","modelId":"TS0601","powerSource":1,"zclVersion":3}' from endpoint 1 with groupID 0
info  2023-06-28 20:26:38: Successfully configured '0x847127fffeaff81d'
debug 2023-06-28 20:26:42: Received Zigbee message from '0x847127fffeaff81d', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
debug 2023-06-28 20:26:42: Received Zigbee message from '0x847127fffeaff81d', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":30}],"seq":12288}' from endpoint 1 with groupID 0
debug 2023-06-28 20:26:42: Datapoint 30 not defined for '_TZE200_5kynszlt' with data {"dp":30,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,0]}}
debug 2023-06-28 20:26:42: Received Zigbee message from '0x847127fffeaff81d', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":31}],"seq":12544}' from endpoint 1 with groupID 0
debug 2023-06-28 20:26:42: Datapoint 31 not defined for '_TZE200_5kynszlt' with data {"dp":31,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,0]}}
debug 2023-06-28 20:26:43: Received Zigbee message from '0x847127fffeaff81d', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":32}],"seq":12800}' from endpoint 1 with groupID 0
debug 2023-06-28 20:26:43: Datapoint 32 not defined for '_TZE200_5kynszlt' with data {"dp":32,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,0]}}
debug 2023-06-28 20:26:43: Received Zigbee message from '0x847127fffeaff81d', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":33}],"seq":13056}' from endpoint 1 with groupID 0
debug 2023-06-28 20:26:43: Datapoint 33 not defined for '_TZE200_5kynszlt' with data {"dp":33,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,0]}}
debug 2023-06-28 20:26:43: Received Zigbee message from '0x847127fffeaff81d', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":34}],"seq":13312}' from endpoint 1 with groupID 0
debug 2023-06-28 20:26:43: Datapoint 34 not defined for '_TZE200_5kynszlt' with data {"dp":34,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,0]}}
[... cut ...]
    at Receiver.dataMessage (E:\zigbee2mqtt\zigbee2mqtt\node_modules\ws\lib\receiver.js:541:14)
debug 2023-06-28 20:27:31: Saving state to file E:\zigbee2mqtt\zigbee2mqtt\data\state.json
info  2023-06-28 20:27:31: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'
info  2023-06-28 20:27:31: Disconnecting from MQTT server
info  2023-06-28 20:27:31: Stopping zigbee-herdsman...
info  2023-06-28 20:27:40: Stopped zigbee-herdsman
info  2023-06-28 20:27:40: Stopped Zigbee2MQTT

```

Koenkk, 2023-06-28:
Updated https://gist.github.com/Koenkk/fb34825c92fd6f1e2d03e25d73c13e24, it shoudl publish the correction actions now (scene_7 and scene_8).

Note that you cannot set the state as these are not switches, this device only supports switch 1 till 7 (https://github.com/Koenkk/zigbee2mqtt/issues/18064#issuecomment-1606862118) (I see you tried this: `Received MQTT message on 'zigbee2mqtt/0x847127fffeaff81d/set' with data '{"state_l7":"ON"}'`)

Qianli-Ma, 2023-06-30:
Seems to be working. I have countdown for each of the switch working as well. However I think I am currently stuck on changing the mode of the switch (changing from switch to scene  control) with DP 18-23, changing the indicator led with DP37, and the power on behavior with DP38.
I tried using `tuya.exposes.indicatorModeNoneRelayPos(),` to change the backlight however It gives the following error:
`Zigbee2MQTT:error 2023-07-01 03:41:44: No converter available for 'power_outage_memory' ("off")`

Koenkk, 2023-07-02:
For the mode, try:
- `{'switch_1': 0, 'scene_1': 1}` -> `{'switch': tuya.enum(0), 'scene': tuya.enum(1)}`
- exposes: `e.enum('mode_1', ea.STATE_SET, ['switch', 'scene'])`

Qianli-Ma, 2023-07-02:
> For the mode, try:
> 
> * `{'switch_1': 0, 'scene_1': 1}` -> `{'switch': tuya.enum(0), 'scene': tuya.enum(1)}`
> * exposes: `e.enum('mode_1', ea.STATE_SET, ['switch', 'scene'])`
By default this throws out an error
`Zigbee2MQTT:error 2023-07-02 15:42:11: No converter available for 'mode_1' ("switch")`
By adding `mode_1` to `tuya.tz.datapoints` I was able to change it between switch and scene modes.

Qianli-Ma, 2023-07-03:
This device also supports individual control of power_on_behavior thru DP39-DP44, i tried using the following and it works
exposes: `new exposes.Enum('power_on_behavior_1', ea.STATE_SET, ['off', 'on', 'memory']).withDescription('Changes the relay_status_1 mode'),`
DP: `[39, 'power_on_behavior_1', tuya.valueConverterBasic.lookup({'off': tuya.enum(0), 'on': tuya.enum(1), 'memory': tuya.enum(2)})]`
and adding `power_on_behavior_1` to `tuya.tz.datapoints`
Should I update `tuya.tz.datapoints` to support these two functions? Or are there workarounds.
Thanks in adavance @Koenkk

Koenkk, 2023-07-04:
> Should I update tuya.tz.datapoints to support these two functions? Or are there workarounds.

Yes, please update

Koenkk, 2023-07-04:
Instead of `power_on_behavior_1`/`mode_1`, use `power_on_behavior_l1`/`mode_l1`

closed 2023-08-12 by github-actions[bot]