Koenkk/zigbee2mqtt #18065: Group binding doesn't work

tomasbedrich, 2023-06-19:
### What happened?

Zigbee2MQTT (or zigbee-herdsman?) always produces `Bind Request` with Group ID 0, despite selecting another Group ID. Please see the Wireshark capture after requesting bind to **Group ID 2**.

[image]

## Notes
- As an obvious consequence, the binding doesn't work as expected. The binding is ACKd by the device, but it is bound to the wrong Group.
- Unicast binding (binding to a single device) works as expected.
- Target group is enumerated and reporting for its members is configured properly.

### What did you expect to happen?

`Destination` field in `Bind Request` should be filled according to the ID of the selected Group in Zigbee2MQTT.

### How to reproduce it (minimal and precise)

HW setup:
- 2x [IKEA LED2003G10](https://www.zigbee2mqtt.io/devices/LED2003G10.html#ikea-led2003g10)
- 1x [Ubisys C4](https://www.zigbee2mqtt.io/devices/C4.html#ubisys-c4)

0. Pair the devices, let the default configure flow happen.
1. Create a group with ID other than `0`. Assign the bulbs to the group.
2. In Ubisys C4 binding, bind arbitrary endpoint (e.g. 2) to the group. I tried the following to overcome possible frontend bug:
  ```
MQTT pub zigbee2mqtt/bridge/request/device/bind:
{"from": "Ubisys C4/2", "to": "IKEA Group", "clusters": ["genOnOff"]}
```
4. Observe the Wireshark capture.

### Zigbee2MQTT version

1.31.2

### Adapter firmware version

7.1.1.0 build 273

### Adapter

SkyConnect (EZSP v9)

tomasbedrich, 2023-06-19:
Relevant info: today I switched from Sonoff Dongle P (CC2652P) to SkyConnect EZSP. Before the switch, I didn't observe any issues with group binding. So very likely it is a bug on the EZSP code-level.

maguiresf, 2023-06-20:
I just came to open an issue about this exact thing. Started testing z2m with SkyConnect, every time I tried to bind, messages were only sent to groupID 0 and the lights didn't respond to button clicks even though binding appeared successful. If I then created a group with ID 0 and added some different lights to it, those would start responding to presses, even though it was supposed to be bound to the original group with an ID of 2. Tried with multiple dimmers, always the same thing. Dimmers Hue are RWL021 and various Hue bulbs tested. Binding directly to a bulb worked perfectly.

I swapped the SkyConnect stick to a ConBee, no other changes at all and group binding works perfectly.

tomasbedrich, 2023-06-20:
@maguiresf I assume you needed to re-pair all the devices after swap to ConBee, right?

maguiresf, 2023-06-20:
@tomasbedrich I actually didn't. I only have a couple of things in z2m at the moment, in the process of moving over from ZHA, but the 8 or so things I had in there carried on working ok after just changing the configuration to point to the new adapter and changing the adapter type. I understood from the documentation it wasn't supposed to work so I was kinda surprised. I repaired the dimmer and light, deleted the group then recreated the group and re-added the light to get the binding working though, that wasn't working properly after the change, I guess the bind to group ID 0 thing was still stuck there.

tomasbedrich, 2023-06-21:
It is zigbee-hersman issue indeed (feel free to transfer this issue). According to Zigbee spec, Bind Request should be: "... 16-bit group address for DstAddr", while it was passed to the coordinator as a plain int. This is the relevant log line before/after applying [5aaea04](https://github.com/bbcontrol/zigbee-herdsman/commit/5aaea04be0ef44a507f7dd6b37cfa170b32b46ae):

```
# before
2023-06-21T08:33:22.673Z zigbee-herdsman:adapter:ezsp:driv ZDO Bind_req params: {"sourceEui":{"_value":{"type":"Buffer","data":[0,31,238,0,0,0,139,228]}},"sourceEp":2,"clusterId":6,"destAddr":{"addrmode":1,"ieee":1,"endpoint":null}}

# after
2023-06-21T09:39:34.496Z zigbee-herdsman:adapter:ezsp:driv ZDO Bind_req params: {"sourceEui":{"_value":{"type":"Buffer","data":[0,31,238,0,0,0,139,228]}},"sourceEp":2,"clusterId":6,"destAddr":{"addrmode":1,"ieee":{"type":"Buffer","data":[1,0]},"endpoint":null}}
```

closed 2023-06-21 by Koenkk