Koenkk/zigbee2mqtt #18070: Lidl HG08164 when pushed gives 3 actions

feu77, 2023-06-19:
### What happened?

I used to use this button for a long time, never got any problem toggling a switch. Since last update, the switch gets triggered randomly and the reason is that the button has changed the way action is published: now I get 3 actions when pushed:  2x "on" and once "". This breaks my rule in NodeRed. Why is the button acting strange since last update?

BTW: I love your work!!

`info 2023-06-19 23:18:20MQTT publish: topic 'zigbee2mqtt/sw_markise', payload '{"action":"on","battery":87,"linkquality":134}'
info 2023-06-19 23:18:20MQTT publish: topic 'zigbee2mqtt/sw_markise/linkquality', payload '134'
info 2023-06-19 23:18:20MQTT publish: topic 'zigbee2mqtt/sw_markise/battery', payload '87'
info 2023-06-19 23:18:20MQTT publish: topic 'zigbee2mqtt/sw_markise/action', payload 'on'
info 2023-06-19 23:18:20MQTT publish: topic 'zigbee2mqtt/sw_markise', payload '{"action":"","battery":87,"linkquality":134}'
info 2023-06-19 23:18:20MQTT publish: topic 'zigbee2mqtt/sw_markise/action', payload 'on'
info 2023-06-19 23:18:20MQTT publish: topic 'zigbee2mqtt/sw_markise/linkquality', payload '134'
info 2023-06-19 23:18:20MQTT publish: topic 'zigbee2mqtt/sw_markise/battery', payload '87'
info 2023-06-19 23:18:20MQTT publish: topic 'zigbee2mqtt/sw_markise/action', payload ''`

### What did you expect to happen?

if pressed ONE published message "on"

### How to reproduce it (minimal and precise)

press the button

### Zigbee2MQTT version

1.31.2 commit: 21f5125

### Adapter firmware version

20220219

### Adapter

Slaesh's

Koenkk, 2023-06-20:
I've pushed some fixes for this device in the latest-dev, can you check with that? (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html). After updating, please re-pair the device.

feu77, 2023-06-21:
Hi, thanks for the quick action.
@Koenkk 
I dl the dev, repair the device, now I get 2 action entries, the "" has gone.
[image]

Koenkk, 2023-06-21:
Can you provide the debug log of this?

feu77, 2023-06-21:
@Koenkk 

- debug 2023-06-21 19:40:07Received Zigbee message from 'sw_markise', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0

- info 2023-06-21 19:40:07MQTT publish: topic 'zigbee2mqtt/sw_markise', payload '{"action":"on","battery":91,"linkquality":145,"voltage":2900}'

- info 2023-06-21 19:40:07MQTT publish: topic 'zigbee2mqtt/sw_markise/linkquality', payload '145'

- info 2023-06-21 19:40:07MQTT publish: topic 'zigbee2mqtt/sw_markise/battery', payload '91'

- info 2023-06-21 19:40:07MQTT publish: topic 'zigbee2mqtt/sw_markise/voltage', payload '2900'

- info 2023-06-21 19:40:07MQTT publish: topic 'zigbee2mqtt/sw_markise/action', payload 'on'

- info 2023-06-21 19:40:07MQTT publish: topic 'homeassistant/device_automation/0x2c1165fffe68bf2d/action_on/config', payload '{"automation_type":"trigger","device":{"identifiers":["zigbee2mqtt_0x2c1165fffe68bf2d"],"manufacturer":"Lidl","model":"Silvercrest smart button (HG08164)","name":"sw_markise"},"payload":"on","subtype":"on","topic":"zigbee2mqtt/sw_markise/action","type":"action"}'

- info 2023-06-21 19:40:07MQTT publish: topic 'zigbee2mqtt/sw_markise/action', payload 'on'

- debug 2023-06-21 19:40:07Received MQTT message on 'homeassistant/device_automation/0x2c1165fffe68bf2d/action_on/config' with data '{"automation_type":"trigger","device":{"identifiers":["zigbee2mqtt_0x2c1165fffe68bf2d"],"manufacturer":"Lidl","model":"Silvercrest smart button (HG08164)","name":"sw_markise"},"payload":"on","subtype":"on","topic":"zigbee2mqtt/sw_markise/action","type":"action"}'

Koenkk, 2023-06-22:
I see the duplicate publish indeed. Can you try the following?

- force remove device through z2m frontend
- restart z2m
- re-pair device

If it doesn't fix the issue, provide me the herdsman debug log of the duplicate action.

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

feu77, 2023-06-24:
Hello,

I did as you told me: 

1. force remove, restart, re-join…problem persists.
2. herdsman logs: 0x2c1165fffe68bf2d is the lidl switch and it joined on line 1485.

￼

Hope it helps!
Thanks for debugging!

Regards,
Christian

> I see the duplicate publish indeed. Can you try the following?
>
> force remove device through z2m frontend
> restart z2m
> re-pair device
> If it doesn't fix the issue, provide me the herdsman debug log of [...]

Using '/app/data' as data directory
Zigbee2MQTT:info  2023-06-24 22:18:16: Logging to console and directory: '/app/data/log/2023-06-24.22-18-16' filename: log.txt
Zigbee2MQTT:info  2023-06-24 22:18:16: Starting Zigbee2MQTT version 1.31.2-dev (commit #5777d76)
Zigbee2MQTT:info  2023-06-24 22:18:16: Starting zigbee-herdsman (0.15.0)
Zigbee2MQTT:info  2023-06-24 22:18:28: zigbee-herdsman started (resumed)
Zigbee2MQTT:info  2023-06-24 22:18:28: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20220219,"transportrev":2},"type":"zStack3x0"}'
Zigbee2MQTT:info  2023-06-24 22:18:28: Currently 71 devices are joined:
Zigbee2MQTT:info  2023-06-24 22:18:28: temp_3d_printer (0x00158d00054220cb): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
Zigbee2MQTT:info  2023-06-24 22:18:28: temp_filament_box (0x000d6ffffe68b2f8): TT001ZAV20 - TuYa Temperature & humidity sensor (EndDevice)
Zigbee2MQTT:info  2023-06-24 22:18:28: vibro_buero_o (0x00158d000577d794): DJT11LM - Xiaomi Aqara vibration sensor (EndDevice)
Zigbee2MQTT:info  2023-06-24 22:18:28: vibro_buero_n (0x00158d000577db60): DJT11LM - Xiaomi Aqara vibration sensor (EndDevice)
Zigbee2MQTT:info  2023-06-24 22:18:28: ikea_onoff_2 (0x842e14fffe78e043): E1743 - IKEA TRADFRI ON/OFF switch (EndDevice)
Zigbee2MQTT:info  2023-06-24 22:18:28: vibro_vorratsraum (0x00158d000543509c): DJT11LM - Xiaomi Aqara vibration sensor (EndDevice)
Zigbee2MQTT:info  2023-06-24 22:18:28: ps_ventil_zisterne (0x588e81fffe61d7bd): E1603/E1702/E1708 - IKEA TRADFRI control outlet (Router)
Zigbee2MQTT:info  2023-06-24 22:18:28: bw_vorratsraum (0x00158d00046026d3): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
Zigbee2MQTT:info  2023-06-24 22:18:28: vibro_waschkueche (0x00158d0003469e19): DJT11LM - Xiaomi Aqara vibration sensor (EndDevice)
Zigbee2MQTT:info  2023-06-24 22:18:28: motion_sensor_treppe (0x588e81fffe1e9c3e): E1525/E1745 - IKEA TRADFRI motion sensor (EndDevice)
Zigbee2MQTT:info  2023-06-24 22:18:28: sw_garagentor (0x842e14fffef82eaf): HG06668 - Lidl Silvercrest smart wireless door bell button (EndDevice)
Zigbee2MQTT:info  2023-06-24 22:18:28: bw_lidl (0xec1bbdfffe69cb4f): HG06335/HG07310 - Lidl Silvercrest smart motion sensor (EndDevice)
Zigbee2MQTT:info  2023-06-24 22:18:28: vibro_luke_1 (0x00158d000577bc60): DJT11LM - Xiaomi Aqara vibration sensor (EndDevice)
Zigbee2MQTT:info  2023-06-24 22:18:28: ikea_onoff_1 (0x588e81fffe6c8ab2): E1743 - IKEA TRADFRI ON/OFF switch (EndDevice)
Zigbee2MQTT:info  2023-06-24 22:18:28: motion_sensor_wc (0x588e81fffeae208f): E1525/E1745 - IKEA TRADFRI motion sensor (EndDevice)
Zigbee2MQTT:info  2023-06-24 22:18:28: osram_e27_rgbw (0x7cb03eaa00a929e8): AC03645 - OSRAM LIGHTIFY LED CLA60 E27 RGBW (Router)
Zigbee2MQTT:info  2023-06-24 22:18:28: vibro_luke_2 (0x00158d000577dbb7): DJT11LM - Xiaomi Aqara vibration sensor (EndDevice)
Zigbee2MQTT:info  2023-06-24 22:18:28: ikea_ps_3 (0x588e81fffe61dc34): E1603/E1702/E1708 - IKEA TRADFRI control outlet (Router)
Zigbee2MQTT:info  2023-06-24 22:18:28: ikea_ps_1 (0x588e81fffe5f359a): E1603/E1702/E1708 - IKEA TRADFRI control outlet (Router)
Zigbee2MQTT:info  2023-06-24 22:18:28: osram_ps_1 (0x84182600000f0f80): AB3257001NJ - OSRAM Smart+ plug (Router)
Zigbee2MQTT:info  2023-06-24 22:18:28: osram_ps_2 (0x84182600000c2fcb): AB3257001NJ - OSRAM Smart+ plug (Router)
Zigbee2MQTT:info  2023-06-24 22:18:28: temp_garage (0x00158d00067cb28a): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
Zigbee2MQTT:info  2023-06-24 22:18:28: temp_holzlager (0x00158d00068b3baf): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
Zigbee2MQTT:info  2023-06-24 22:18:28: waterleak_technik (0x00158d00067971ae): SJCGQ11LM - Xiaomi Aqara water leak sensor (EndDevice)
Zigbee2MQTT:info  2023-06-24 22:18:28: waterleak_waschkueche (0x00158d0006726108): SJCGQ11LM - Xiaomi Aqara water leak sensor (EndDevice)
Zigbee2MQTT:info  2023-06-24 22:18:28: temp_dachboden (0x00158d0006ca2acf): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
Zigbee2MQTT:info  2023-06-24 22:18:28: temp_technik (0x00158d0006a08baa): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
Zigbee2MQTT:info  2023-06-24 22:18:28: temp_schapp (0x00158d0006ca2693): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
Zigbee2MQTT:info  2023-06-24 22:18:28: temp_zisterne (0x00158d0006b88ce1): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
Zigbee2MQTT:info  2023-06-24 22:18:28: psm_wasserpumpe (0x60a423fffef8dce1): TS0121_plug - TuYa 10A UK or 16A EU smart plug (Router)
Zigbee2MQTT:info  2023-06-24 22:18:28: contact_sensor_garagentor (0x00158d0006c0408b): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
Zigbee2MQTT:info  2023-06-24 22:18:28: smoke_detector_ben (0xa4c1389c7941c483): PA-44Z - TuYa Smoke detector (EndDevice)
[... cut ...] evice_class":"temperature","enabled_by_default":true,"entity_category":"diagnostic","name":"bw_vorratsraum_device_temperature","state_class":"measurement","state_topic":"zigbee2mqtt/bw_vorratsraum","unique_id":"0x00158d00046026d3_device_temperature_zigbee2mqtt","unit_of_measurement":"°C","value_template":"{{ value_json.device_temperature }}"}'
Zigbee2MQTT:info  2023-06-24 22:18:29: MQTT publish: topic 'homeassistant/sensor/0x00158d00046026d3/voltage/config', payload '{"availability":[{"topic":"z

Koenkk, 2023-06-25:
> Since last update, the switch gets triggered randomly

Can you go back to the latest working z2m version and confirm that it still works there?

feu77, 2023-06-25:
I went back to 1.29.0, problem persists.

closed 2023-08-02 by github-actions[bot]