Koenkk/zigbee2mqtt #18076: [Feature request]: Add motor speed setting for Aqara Roller Shade Driver E1

GreyEarl, 2023-06-20:
### Is your feature request related to a problem? Please describe

The heavier and bigger the curtain/shade, the better it is to slowly roll down or roll up the shade. It avoids unecassary noise and possibly damage to hardware including the shade. Aqara officially support changing the motor speed.

Before I migrated to Zigbee2mqtt I was using DeCONZ. There it was possible to change the speed:

I noticed lot more (crackling) noises from my more heavy shades since my migration as the device is now using the default motor speed.

### Describe the solution you'd like

I would like to select the desired speed of the motor. That would be basically low, medium or high.

### Describe alternatives you've considered

Unfortuntally there is no alternative. I'm not going back to DeCONZ :)

### Additional context

Information about official Aqara support:
https://homekitnews.com/2021/11/21/aqara-add-more-functionality-to-the-roller-shade-driver-e1/

Information on how DeCONZ documenting the motor speed:
https://dresden-elektronik.github.io/deconz-rest-doc/devices/xiaomi/lumi_curtain_acn002

WorlockM, 2023-09-17:
Also waiting for this function. Can this be re-considered?

aebpower, 2024-01-20:
anyone helping with the spped setting of the motor?available in deconz and also in proprietrary app ( aqara app) . shouldnt be hard to insert then ?

aebpower, 2024-01-20:
> ### Is your feature request related to a problem? Please describe
> The heavier and bigger the curtain/shade, the better it is to slowly roll down or roll up the shade. It avoids unecassary noise and [...]

found out anything to make speed change?

Busta999, 2024-02-01:
This would be helpful as the only way to change the motor speed is to remove from zigbee2MQTT and pair with Aqara make the change then re pair with zigbee2MQTT.

sorryusernameisalreadytaken, 2024-02-11:
@Koenkk can you please reopen this ticket.

Puh, I am a bit unsure how to get this function. I had no aqara base at the moment.
I think this would be the right area:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/36a31e8bab8ce6b9cba29698f1ae1df4a5a734b8/src/devices/lumi.ts#L2337

When I compare to zemismart blind motor, I think we need a new expose:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/36a31e8bab8ce6b9cba29698f1ae1df4a5a734b8/src/devices/zemismart.ts#L177

Did I need to change something on these files to?
https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/lib/lumi.ts
https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/converters/fromZigbee.ts
https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/converters/toZigbee.ts

chaayac, 2024-03-13:
https://github.com/Koenkk/zigbee-herdsman-converters/pull/7109 is available now in v1.36

miwitt123, 2024-04-11:
I can see that it is exposed in the zigbee2mqtt integration, but as far as I can see, changing it does nothing to the motor speed, what are the speed increments that should be expected?

jkurian, 2024-04-23:
Also curious if this is working for anyone else, I am seeing errors like the below:

```
Zigbee2MQTT:error 2024-04-23 08:14:12: Failed to configure '0x1111111111111111', attempt 3 (Error: Device 0x1111111111111111 has no input cluster manuSpecificLumi
    at getEndpointsWithCluster (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:28:15)
    at setupAttributes (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:63:55)
    at configure (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:104:23)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:162:27)
    at Configure.configure (/app/lib/extension/configure.ts:122:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))
```

sandervandenberg, 2024-06-01:
Just checked the motor speed setting and it is working for me. Difference in speed is not very big though.

BUT: in Home Assistant there are now TWO entities with the same name "Motor Speed" created:
* a "select" under "Configuration", which is enabled and 
* a "sensor" under "Diagnostic" that is disabled. 
Enabling the sensor messes up Mosquitto/MQTT and made all my devices go unavailable. Took some trial and error to get it all working again.

closed 2024-11-29 by GreyEarl