Koenkk/zigbee2mqtt #18077: ORVIBO MixSwitch 3 gangs (Model: T40W3Z) Not working physically

mnonelove, 2023-06-20:
### What happened?

I successfully added ORVIBO MixSwitch 3 gangs (Model: T40W3Z) via Zigbee2mqtt (Using Sonoff Dongle Plus). I noticed that I could not control them manually after that. I searched for a long time with no success to solve it. When reset the switch and connect it to Orvibo MixPad (Zigbee Hub), it is working via Orvibo app and manually! Anyone can help?

### What did you expect to happen?

I expected that I can control the switch to On/Off physically like before when connected to Orvibo MixPad Mini via Zigbee.

### How to reproduce it (minimal and precise)

Let the switch working physically along with being smart or any alternative you might suggest.  

**Note: I updated the below log after pressing the 3 buttons to see the received zigbee message from the coordinator.**

### Zigbee2MQTT version

1.31.2 

### Adapter firmware version

20210708

### Adapter

 Sonoff Zigbee 3.0 USB Dongle Plus - Sonoff Zigbee 3.0 USB Dongle Plus

### Debug log

[log (1).txt](https://github.com/Koenkk/zigbee2mqtt/files/11809012/log.1.txt)

closed 2023-07-29 by github-actions[bot]

szl006dl, 2023-11-20:
I still have this problem.

szl006dl, 2025-11-17:
> I still have this problem.
>
> Hi, I’m also considering using the ORVIBO MixSwitch series. May I ask if
> this issue has been resolved on your side?

Apologies,
 I haven't solved it either, but based on my testing, this issue is related
to the firmware version. If your version is 7.04, then this situation will
occur. If your version number is less than 7.04, there's no problem.
However, my capabilities are limited, and I don't know how to flash the
firmware version

RainySat, 2025-11-17:
> Apologies,
>  I haven't solved it either, but based on my testing, this issue is related
> to the firmware version. If your version is 7.04, then this situation will
> occur. If your version number is less than 7.04, there's no problem.
> However, my capabilities are limited, and I don't know how to flash the
> firmware version
> 
> […](#)
> 
>  <[#18077 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/18077#issuecomment-3540655123)>
> 
>  I still have this problem.
> 
>  Hi, I’m also considering using the ORVIBO MixSwitch series. May I ask if
>  this issue has been resolved on your side?
> 
>  <[#18077 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/18077#issuecomment-3540655123)>,

Hi, thanks a lot for your reply and for sharing your testing results — that’s really helpful!

May I ask one more thing?
Do you also have the 1-gang or 2-gang versions of the ORVIBO MixSwitch, and if so, do they work normally for you (including physical button control) with Zigbee2MQTT?

Thanks again!