Koenkk/zigbee2mqtt #18078: Battery Icons - Green vs Gray and Flashing vs Solid

sdschramm, 2023-06-21:
### Link

https://www.samsung.com/us/support/smart-home/smartthings/sensors/moisture-sensor/

### Database entry

{"id":16,"type":"EndDevice","ieeeAddr":"0x24fd5b0001045d04","nwkAddr":8322,"manufId":4362,"manufName":"SmartThings","powerSource":"Battery","modelId":"moisturev4","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,15,32,1026,1280],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"moisturev4","manufacturerName":"SmartThings","powerSource":0,"zclVersion":1,"appVersion":27,"swBuildId":"R860_#11"}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b0026b88665","zoneState":1}},"genPollCtrl":{"attributes":{"checkinInterval":13200}},"genPowerCfg":{"attributes":{"batteryVoltage":26}},"msTemperatureMeasurement":{"attributes":{"measuredValue":2978}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b88665","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b88665","endpointID":1},{"cluster":1026,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b88665","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":32,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0},{"cluster":1026,"attrId":0,"minRepIntval":10,"maxRepIntval":3600,"repChange":100}],"meta":{}}},"appVersion":27,"swBuildId":"R860_#11","zclVersion":1,"interviewCompleted":true,"meta":{"configured":-700028871},"lastSeen":1687308888294,"defaultSendRequestWhen":"active","checkinInterval":3300}

### Comments

Can someone please explain the difference between a green battery vs a gray battery and a solid battery vs a flashing one? One could guess a flashing battery would mean "critically low" but the icon shows the battery is full compared to others that are half, etc. Is this behavior expected?

Example:
https://github.com/Koenkk/zigbee2mqtt/assets/38740977/f626d525-f6f2-4ceb-9099-953986ccdf3f

sdschramm, 2023-07-22:
Still an issue

sdschramm, 2023-08-23:
Still zero progress on this.

sdschramm, 2023-08-23:
Hi @sjorge, thanks for looking into this. 
It would be nice if there was a way to simplify this. A blinking full battery icon makes no sense and neither do different colors aside from green = good and red = bad. 
You mentioned this is something implemented in the front end, which is not this repo. Can you please point me to the correct place to report this? 
Thanks!

sjorge, 2023-08-23:
https://github.com/nurikk/zigbee2mqtt-frontend

sdschramm, 2023-08-24:
Thanks! I will close this out and open a report over on the frontend page.

closed 2023-08-24 by sdschramm