Koenkk/zigbee2mqtt #18081: Candeo dimmer model 202 has stopped working

JasonBSteele, 2023-06-21:
### What happened?

My Candeo dimmers have stopped working and are now showing in the logs as unsupported.
Link to device details https://candeo.io/support/support-c202/

They used to show up as EcoDim dimmers and you can see in the log below that the Zigbee model "Dimmer-Switch-ZB3.0" is the same as that for EcoDim devices.

### What did you expect to happen?

It would continue to work as an EcoDim device or be recognised as a Candeo device

### Zigbee2MQTT version

1.31.2-dev commit: 38de19b

### Adapter firmware version

20210708

### Adapter

Sonoff Zigbee 3.0 Dongle Plus (SO-ZBD3)

### Debug log

When I reset it and then attempted to rejoin I got the following log

```
Info 2023-06-21 18:49:35Device 'Chandelier' joined
Info 2023-06-21 18:49:35MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Chandelier","ieee_address":"0x000d6ffffec25236"},"type":"device_joined"}'
Info 2023-06-21 18:49:35Starting interview of 'Chandelier'
Info 2023-06-21 18:49:35MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Chandelier","ieee_address":"0x000d6ffffec25236","status":"started"},"type":"device_interview"}'
Info 2023-06-21 18:49:38MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Chandelier","ieee_address":"0x000d6ffffec25236"},"type":"device_announce"}'
Info 2023-06-21 18:49:46Successfully interviewed 'Chandelier', device has successfully been paired
Warning 2023-06-21 18:49:46Device 'Chandelier' with Zigbee model 'Dimmer-Switch-ZB3.0' and manufacturer name 'Candeo' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Info 2023-06-21 18:49:46MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"Chandelier","ieee_address":"0x000d6ffffec25236","status":"successful","supported":false},"type":"device_interview"}'
```

Koenkk, 2023-06-22:
Added the device!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-06-22 by Koenkk

JasonBSteele, 2023-06-24:
Just to let you know that it works great, thanks for getting this resolved so quickly.

A small point, but the icons in the web UI have remained as the generic Zigbee Z. Do I need to do anything to get them to refresh.

If it helps, it looks identical to the HK-DIM-A one: https://www.zigbee2mqtt.io/devices/HK-DIM-A.html