Koenkk/zigbee2mqtt #18082: [New device support]: ZNQBKG39LM Xiaomi Aqara D1 2 gang smart wall switch

jomud9, 2023-06-21:
### Link

https://www.aqara.com/cn/Canon_Smart_Wall_Switch_Z1_overview

### Database entry

{"id":3,"type":"EndDevice","ieeeAddr":"0x54ef4410008ea781","nwkAddr":47837,"manufId":4447,"manufName":"Aqara","powerSource":"DC Source","modelId":"lumi.switch.acn049","epList":[1,2,3,4,21],"endpoints":{"1":{"profId":260,"epId":1,"devId":0,"inClusterList":[2820,1794,18,5,4,6,3,0,64704],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"modelId":"lumi.switch.acn049","manufacturerName":"Aqara","powerSource":4,"zclVersion":3,"appVersion":21,"stackVersion":27,"hwVersion":1,"dateCode":"20230526"}},"aqaraOpple":{"attributes":{"4":4,"247":{"type":"Buffer","data":[3,40,38,5,33,11,0,9,33,0,0,10,33,132,210,12,32,10,13,35,21,0,0,0,17,35,1,0,0,0,100,16,1,101,16,1,149,57,172,71,225,60,150,57,51,171,14,69,152,57,82,184,53,66,151,57,205,108,180,67,154,32,0]},"512":1,"617":1,"645":0,"646":1,"648":0,"1303":1}},"genOnOff":{"attributes":{"245":62577920,"onOff":1}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":0,"inClusterList":[18,5,4,6,64704],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"245":117440538,"onOff":1}},"aqaraOpple":{"attributes":{"512":0,"617":1,"645":0}},"genMultistateInput":{"attributes":{"presentValue":1}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":0,"inClusterList":[64704],"outClusterList":[],"clusters":{"aqaraOpple":{"attributes":{"512":1,"617":1,"645":0}}},"binds":[],"configuredReportings":[],"meta":{}},"4":{"profId":260,"epId":4,"devId":0,"inClusterList":[64704],"outClusterList":[],"clusters":{"aqaraOpple":{"attributes":{"617":0}}},"binds":[],"configuredReportings":[],"meta":{}},"21":{"profId":260,"epId":21,"devId":0,"inClusterList":[12],"outClusterList":[],"clusters":{"genAnalogInput":{"attributes":{"presentValue":45.91999816894531}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":21,"stackVersion":27,"hwVersion":1,"dateCode":"20230526","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2114939527},"lastSeen":1687378993322,"defaultSendRequestWhen":"immediate"}

### Comments

It is my first attempt to create an external converter, and finally I made at first usable version.

Something still missing
- I found multi action did not work. Tried to search old posts and one suggested I need to put some code from fromzigbee.js and/or tozigbee.js inside the converter. I might need some inspiring on that
- similar issue for flip_indicator_light 
> Publishing get 'get' 'flip_indicator_light' to 'switchz1_readingroom'
Received Zigbee message from 'switchz1_readingroom', type 'readResponse', cluster 'aqaraOpple', data '{}' from endpoint 1 with groupID 0
lumi.switch.acn049: Processed data into payload {"flip_indicator_light":"OFF"}
Publish 'get' 'flip_indicator_light' to 'switchz1_readingroom' failed: 'Error: Read 0x54ef4410008f17de/1 aqaraOpple([240], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
- ...and power_outage_memory
> Received MQTT message on 'zigbee2mqtt/switchz1_readingroom/get' with data '{"power_outage_memory":""}'
> No converter available for 'power_outage_memory' ("")

Also one more question - Is vendor support required for router feature? I just realise it shown as "End Device" instead of "Router" in the current setup. I understand it is new release product with (maybe) premature firmware, but just want to know if there are option/parameter in z2m to make it full functional. 

Thank you!

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['lumi.switch.acn049'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'ZNQBKG39LM', // Vendor model number, look on the device for a model number
    vendor: 'Xiaomi', // Vendor of the device (only used for documentation and startup logging)
    description: 'Aqara Z1 2 gang smart wall switch', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.on_off, fz.aqara_opple, fz.xiaomi_multistate_action, fz.xiaomi_power], 
    toZigbee: [tz.on_off, tz.xiaomi_switch_operation_mode_opple, tz.xiaomi_flip_indicator_light], meta: {multiEndpoint: true},
        endpoint: (device) => {
            return {'left': 1, 'right': 2};
        },
    exposes: [
        e.switch().withEndpoint('left'), e.switch().withEndpoint('right'), 
        exposes.enum('operation_mode', ea.ALL, ['control_relay', 'decoupled'])
                .withDescription('Decoupled mode for left button')
                .withEndpoint('left'),
        exposes.enum('operation_mode', ea.ALL, ['control_relay', 'decoupled'])
                .withDescription('Decoupled mode for right button')
                .withEndpoint('right'),
        e.action(['single_left', 'double_left', 'single_right', 'double_right', 'single_both', 'double_both']),
        e.power(), e.voltage(), e.device_temperature(), e.flip_indicator_light(), e.power_outage_memory(),
        ],
    configure: async (device, coordinatorEndpoint, logger) => {
        await device.getEndpoint(1).write('aqaraOpple', {'mode': 1}, {manufacturerCode: 0x115f, disableResponse: true});
        },
              
};

module.exports = definition;
```

jomud9, 2023-06-23:
Inspired by https://github.com/Koenkk/zigbee2mqtt/discussions/7048, I tried to handcraft the things which need to be updated on fromzigbee.js, tozigbee.js and xiaomi.ts .

Good news:
- now it could display as single_left and single_right instead of single when decoupled
     - no double or both, still. (early firmware issue??)

Still pending / need help
- flip_indicator_light
> Publish 'get' 'flip_indicator_light' to 'switchz1_kitchen' failed: 'Error: Read 0x54ef4410008ea781/1 aqaraOpple([240], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'

- power_outage_memory
> Publish 'get' 'power_outage_memory' to 'switchz1_kitchen' failed: 'Error: Read 0x54ef4410008ea781/1 aqaraOpple([513], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'

- click_mode
Publish 'get' 'click_mode' to 'switchz1_kitchen' failed: 'Error: Read 0x54ef4410008ea781/1 aqaraOpple([293], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'

Updated converter code
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

// Updated local Fz/Tz - inspired from https://github.com/Koenkk/zigbee2mqtt/discussions/7048

const { hasAlreadyProcessedMessage } = require('zigbee-herdsman-converters/lib/utils');

const fzLocal = {
    xiaomi_multistate_action: {
        cluster: 'genMultistateInput',
        type: ['attributeReport'],
        convert: (model, msg, publish, options, meta) => {
            if (hasAlreadyProcessedMessage(msg, model)) return;
            let actionLookup = {0: 'hold', 1: 'single', 2: 'double', 3: 'triple', 255: 'release'};
            if (model.model === 'WXKG12LM') {
                actionLookup = {...actionLookup, 16: 'hold', 17: 'release', 18: 'shake'};
            }
            if (['WXKG02LM_rev2', 'WXKG07LM', 'WXKG15LM', 'WXKG17LM', 'ZNQBKG39LM'].includes(model.model)) {
                buttonLookup = {1: 'left', 2: 'right', 3: 'both'};
            }
            if (['QBKG12LM', 'QBKG24LM'].includes(model.model)) buttonLookup = {5: 'left', 6: 'right', 7: 'both'};
            if (['QBKG39LM', 'QBKG41LM', 'WS-EUK02', 'WS-EUK04', 'QBKG20LM', 'QBKG28LM', 'QBKG31LM'].includes(model.model)) {
                buttonLookup = {41: 'left', 42: 'right', 51: 'both'};
            }
            if (['QBKG25LM', 'QBKG26LM', 'QBKG29LM', 'QBKG34LM', 'ZNQBKG31LM', 'ZNQBKG26LM'].includes(model.model)) {
                buttonLookup = {
                    41: 'left', 42: 'center', 43: 'right',
                    51: 'left_center', 52: 'left_right', 53: 'center_right',
                    61: 'all',
                };
            }
            if (['WS-USC02', 'WS-USC04'].includes(model.model)) {
                buttonLookup = {41: 'top', 42: 'bottom', 51: 'both'};
            }

            const action = actionLookup[msg.data['presentValue']];
            if (buttonLookup) {
                const button = buttonLookup[msg.endpoint.ID];
                if (button) {
                    return {action: `${action}_${button}`};
                }
            } else {
                return {action};
            }
        },
    },
};
[... 106 lines cut ...]
    },
    meta: {multiEndpoint: true},
    exposes: [
        e.power(), e.voltage(), e.device_temperature(), e.flip_indicator_light(), e.power_outage_memory(),
        e.switch().withEndpoint('left'), e.switch().withEndpoint('right'), 
        exposes.enum('operation_mode', ea.ALL, ['control_relay', 'decoupled'])
                .withDescription('Decoupled mode for left button')
                .withEndpoint('left'),
        exposes.enum('operation_mode', ea.ALL, ['control_relay', 'decoupled'])
                .withDescription('Decoupled mode for right button')
                .withEndpoint('right'),
        exposes.enum('click_mode', ea.ALL, ['fast', 'multi'])
                .withDescription('Click mode, fast: only supports single click which will be send immediately after clicking.' +
                    'multi: supports more events like double and hold'),
        e.action(['single_left', 'double_left', 'single_right', 'double_right', 'single_both', 'double_both']),
    ],
    onEvent: preventReset,
    configure: async (device, coordinatorEndpoint, logger) => {
        await device.getEndpoint(1).write('aqaraOpple', {'mode': 1}, {manufacturerCode: 0x115f, disableResponse: true});
    },

};

module.exports = definition;
```

ShadowFistOkami, 2023-06-26:
@jomud9 first thing I would do is make sure your zigbee coordinator firmware is updated.
Had issues with missing actions & some entities on my H1M switch and found that out thanks to [this issue](https://github.com/Koenkk/zigbee2mqtt/issues/17477).
Once that's done, then you can properly test whether the converter you're using works properly

jomud9, 2023-06-27:
> @jomud9 first thing I would do is make sure your zigbee coordinator firmware is updated.
> Had issues with missing actions & some entities on my H1M switch and found that out thanks to [this [...]

Just did a quick check and I updated my Sonoff Dongle firmware to 20230507 - still the same.

However I did some local code to check how xiaomi_multistate_action() do. From debug logs I did not see action when I double / triple / hold buttons. So I am thinking if it is related to switch firmware, too.

Anyway I did some change to the code as I am also installed a ZNQBKG40LM today

ShadowFistOkami, 2023-06-27:
@jomud9 I know you've probably done this already, but try force removing the device from z2mqtt, holding any button on the switch till you get a "device left the network" message, then re-add it.
You really shouldn't be getting that switch detected as an end device, so something's off.
Side note - how did you manage to get your hands on one? I found out about these switches ONE day after I bought my H1M, and the extra scene switch on the 4-gang version would be perfect for my needs!

jomud9, 2023-06-28:
> @jomud9 I know you've probably done this already, but try force removing the device from z2mqtt, holding any button on the switch till you get a "device left the network" message, then re-add it. [...]

I am from Hong Kong so it is "still" easy to get access via Taobao (Local Ali*). However it is new enough I did not get my 3 gang on the same batch.

And yes, due to my "try and error" in fact I did lots of un-pair and re-add in order to remove some entry...

BTW Do you also have idea on this?
>{
    "consumption": 0,
    "current": 0.04,
    "device_temperature": 38,
    "energy": 0,
    "flip_indicator_light": null,
    "linkquality": 98,
    "mode_switch": "anti_flicker_mode",
    "operation_mode": "control_relay",
    "operation_mode_left": "control_relay",
    "operation_mode_right": "decoupled",
    "power": 2.72,
    "power_outage_count": 14,
    "power_outage_memory": null,
    "state": "ON",
    "state_left": "ON",
    "state_right": "ON",
    "voltage": 231.56,
    "action": null,
    "click_mode": null,
    "operation_mode_button_1": "control_relay",
    "operation_mode_button_2": "decoupled",
    "state_button_1": "ON",
    "state_button_2": "OFF",
}

Seems   "state_left",  "state_right",   "operation_mode_left", "operation_mode_right" are not refer to anything in UI (instead state_button_n and operation_mode_button_n should be in place.  Would like to see how to polish that. :-)

jomud9, 2023-06-28:
### Updated code for 20230628

```shell
// changlog 20230628 
//. 0. Suggested by https://github.com/ShadowFistOkami to check Zigbee firmware on USB Dongle, updated.
//. 1. still thinking this version of firmware only support
//.     - action: single (no double, triple, hold)
//.     - no multiple button pressed together (no both, all)
//. 2. as button arranged in "top to buttom" instead of "standard" left to right, I renamed as button_1 instead of left
//. 3. Need help on function to make these working
//.   - flip_indicator_light, power_outage_memory and click mode
//.   - Seems address got changed but I have no idea
//.   - Seems related address are difference, (early) firmware fault? 

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');

//. PLEASE READ: These are present from device/xiaomi.ts, only needed for external converter
const ota = require('zigbee-herdsman-converters/lib/ota');
const herdsman = require('zigbee-herdsman');
const e = exposes.presets;
const ea = exposes.access;

const manufacturerOptions = {
    xiaomi: {manufacturerCode: herdsman.Zcl.ManufacturerCode.LUMI_UNITED_TECH, disableDefaultResponse: true},
};

const { hasAlreadyProcessedMessage } = require('zigbee-herdsman-converters/lib/utils');
// PLEASE READ: THE END OF THE BLOCK THAT CAN BE DELELED

//. PLEASE READ: These should be update to FromZigbee.js 
const fzLocal = {
    xiaomi_multistate_action: {
        cluster: 'genMultistateInput',
        type: ['attributeReport'],
        convert: (model, msg, publish, options, meta) => {
            if (hasAlreadyProcessedMessage(msg, model)) return;
            let actionLookup = {0: 'hold', 1: 'single', 2: 'double', 3: 'triple', 255: 'release'};
            if (model.model === 'WXKG12LM') {
                actionLookup = {...actionLookup, 16: 'hold', 17: 'release', 18: 'shake'};
            }
            if (['WXKG02LM_rev2', 'WXKG07LM', 'WXKG15LM', 'WXKG17LM'].includes(model.model)) {
                buttonLookup = {1: 'left', 2: 'right', 3: 'both'};
            }
            if (['ZNQBKG39LM'].includes(model.model)) {
                buttonLookup = {1: 'button_1', 2: 'button_2'};
            }
            if (['ZNQBKG40LM'].includes(model.model)) {
                buttonLookup = {1: 'button_1', 2: 'button_2', 3: 'button_3'};
            }
            if (['QBKG12LM', 'QBKG24LM'].includes(model.model)) buttonLookup = {5: 'left', 6: 'right', 7: 'both'};
            if (['QBKG39LM', 'QBKG41LM', 'WS-EUK02', 'WS-EUK04', 'QBKG20LM', 'QBKG28LM', 'QBKG31LM'].includes(model.model)) {
                buttonLookup = {41: 'left', 42: 'right', 51: 'both'};
            }
            if (['QBKG25LM', 'QBKG26LM', 'QBKG29LM', 'QBKG34LM', 'ZNQBKG31LM', 'ZNQBKG26LM'].includes(model.model)) {
                buttonLookup = {
                    41: 'left', 42: 'center', 43: 'right',
                    51: 'left_center', 52: 'left_right', 53: 'center_right',
                    61: 'all',
                };
            }
            if (['WS-USC02', 'WS-USC04'].includes(model.model)) {
                buttonLookup = {41: 'top', 42: 'bottom', 51: 'both'};
            }

            const action = actionLookup[msg.data['presentValue']];
            if (buttonLookup) {
[... 116 lines cut ...]
    meta: {multiEndpoint: true},
    exposes: [
        e.power(), e.voltage(), e.device_temperature(), e.flip_indicator_light(), e.power_outage_memory(),
        e.switch().withEndpoint('button_1'), e.switch().withEndpoint('button_2'),
        exposes.enum('operation_mode', ea.ALL, ['control_relay', 'decoupled'])
                .withDescription('Decoupled mode for button1 (Top) button')
                .withEndpoint('button_1'),
        exposes.enum('operation_mode', ea.ALL, ['control_relay', 'decoupled'])
                .withDescription('Decoupled mode for button 2 (Buttom) button')
                .withEndpoint('button_2'),
        exposes.enum('click_mode', ea.ALL, ['fast', 'multi'])
                .withDescription('Click mode, fast: only supports single click which will be send immediately after clicking.' +
                    'multi: supports more events like double and hold'),
        e.action(['button_1_single', 'button_2_single', 'button_1_double', 'button_2_double', 'all_single', 'all_double']),
    ],
    onEvent: preventReset,
    configure: async (device, coordinatorEndpoint, logger) => {
        await device.getEndpoint(1).write('aqaraOpple', {'mode': 1}, {manufacturerCode: 0x115f, disableResponse: true});
//        await device.getEndpoint(1).write('aqaraOpple', {0x0125: {value: 0x02, type: 0x20}}, {manufacturerCode: 0x115f});
    },

};

module.exports = definition;
```

jomud9, 2023-07-02:
> @jomud9 lucky you! Here, try this converter instead. I removed some imports (you can always add them back later), fixed some issues, and moved your buttonLookup to point to a different button. If it works, this will cause your switch to report left, right & both again. Don't worry about that for now - we can either change it later or just rename the entities in HA. Word of warning - I am far from an expert on this, obviously cannot test it, and it's been a looong day at work, so might have missed something, but give this a try:

Thanks for your suggestion. After more re-try I understand if I did not remove the entity completely, some old value will be still there, but it for my 2 gang and 3 gang, my code is running run. 

But you help me to spot some copy and paste error. 

```shell
//. changlog 20230702 
//. 1. Confirm button1, button2 is running as expected
//. 2. fix typo on fromZigbee and toZigbee section, but still not working for flip_indicator_light / power_outage_memory() and clickmode()
//.

//. changlog 20230628 
//. 0. Suggested by https://github.com/ShadowFistOkami to check Zigbee firmware on USB Dongle, updated.
//. 1. still thinking this version of firmware only support
//.     - action: single (no double, triple, hold)
//.     - no multiple button pressed together (no both, all)
//. 2. as button arranged in "top to buttom" instead of "standard" left to right, I renamed as button_1 instead of left
//. 3. Need help on function to make these working
//.   - flip_indicator_light, power_outage_memory and click mode
//.   - Seems address got changed but I have no idea
//.   - Seems related address are difference, (early) firmware fault? 

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');

//. PLEASE READ: These are present from device/xiaomi.ts, only needed for external converter
const ota = require('zigbee-herdsman-converters/lib/ota');
const herdsman = require('zigbee-herdsman');
const e = exposes.presets;
const ea = exposes.access;

const manufacturerOptions = {
    xiaomi: {manufacturerCode: herdsman.Zcl.ManufacturerCode.LUMI_UNITED_TECH, disableDefaultResponse: true},
};

const { hasAlreadyProcessedMessage } = require('zigbee-herdsman-converters/lib/utils');

//. PLEASE READ: These should be update to FromZigbee.js 
const fzLocal = {
    xiaomi_multistate_action: {
        cluster: 'genMultistateInput',
        type: ['attributeReport'],
        convert: (model, msg, publish, options, meta) => {
            if (hasAlreadyProcessedMessage(msg, model)) return;
            let actionLookup = {0: 'hold', 1: 'single', 2: 'double', 3: 'triple', 255: 'release'};
            if (model.model === 'WXKG12LM') {
                actionLookup = {...actionLookup, 16: 'hold', 17: 'release', 18: 'shake'};
            }
            if (['WXKG02LM_rev2', 'WXKG07LM', 'WXKG15LM', 'WXKG17LM'].includes(model.model)) {
                buttonLookup = {1: 'left', 2: 'right', 3: 'both'};
            }
            if (['ZNQBKG39LM'].includes(model.model)) {
                buttonLookup = {1: 'button_1', 2: 'button_2'};
            }
            if (['ZNQBKG40LM'].includes(model.model)) {
                buttonLookup = {1: 'button_1', 2: 'button_2', 3: 'button_3'};
            }
            if (['QBKG12LM', 'QBKG24LM'].includes(model.model)) buttonLookup = {5: 'left', 6: 'right', 7: 'both'};
            if (['QBKG39LM', 'QBKG41LM', 'WS-EUK02', 'WS-EUK04', 'QBKG20LM', 'QBKG28LM', 'QBKG31LM'].includes(model.model)) {
                buttonLookup = {41: 'left', 42: 'right', 51: 'both'};
            }
            if (['QBKG25LM', 'QBKG26LM', 'QBKG29LM', 'QBKG34LM', 'ZNQBKG31LM', 'ZNQBKG26LM'].includes(model.model)) {
                buttonLookup = {
                    41: 'left', 42: 'center', 43: 'right',
                    51: 'left_center', 52: 'left_right', 53: 'center_right',
                    61: 'all',
[... 122 lines cut ...]
        return {'button_1': 1, 'button_2': 2};
    },
    meta: {multiEndpoint: true},
    exposes: [
        e.power(), e.voltage(), e.device_temperature(), e.power_outage_memory(), e.led_disabled_night(), e.flip_indicator_light(), 
        e.switch().withEndpoint('button_1'), e.switch().withEndpoint('button_2'),
        exposes.enum('operation_mode', ea.ALL, ['control_relay', 'decoupled'])
                .withDescription('Decoupled mode for button1 (Top) button').withEndpoint('button_1'),
        exposes.enum('operation_mode', ea.ALL, ['control_relay', 'decoupled'])
                .withDescription('Decoupled mode for button 2 (Buttom) button').withEndpoint('button_2'),
        exposes.enum('click_mode', ea.ALL, ['fast', 'multi'])
                .withDescription('Click mode, fast: only supports single click which will be send immediately after clicking.' +
                    'multi: supports more events like double and hold'),
        e.action(['button_1_single', 'button_2_single', 'button_1_double', 'button_2_double', 'all_single', 'all_double']),
    ],
    onEvent: preventReset,
    configure: async (device, coordinatorEndpoint, logger) => {
        await device.getEndpoint(1).write('aqaraOpple', {'mode': 1}, {manufacturerCode: 0x115f, disableResponse: true});
//        await device.getEndpoint(1).write('aqaraOpple', {0x0125: {value: 0x02, type: 0x20}}, {manufacturerCode: 0x115f});
    },

};

module.exports = definition;
```

ShadowFistOkami, 2023-07-03:
@jomud9 based on your latest changelog, are you saying that now you're getting **all** actions ('button_1_single', 'button_2_single', 'button_1_double', 'button_2_double', 'all_single', 'all_double') in decoupled mode?
I'm asking because all recent models use buttonLookup 41,42 & 51, while you're only using 1 & 2 (no all).

I can't help with the missing power_outage_memory and clickmode, since those aren't exposed on my H1M (though I believe click_mode is only used for wireless switches, plus you're not declaring it in toZigbee).

Regarding flip_indicator_light, I have no idea why it wouldn't work. It's not tied to a particular model, so toggling that setting should turn on/off the indicator light on your switch. Are you getting any errors in debug log when you do that?

jomud9, 2023-07-04:
> @jomud9 based on your latest changelog, are you saying that now you're getting **all** actions ('button_1_single', 'button_2_single', 'button_1_double', 'button_2_double', 'all_single', [...]

I did not have concrete answer for that.
However I tried to make fun with debug mode and I defined a 1 to 100 mapping to see if I could find something interesting, result is nothing. That's why I draw to conclusion it might be a firmware limitation. I second that idea as I also pair another spare Z1 2 gang with Aqara Home - Nothing besides single click button1 and button2 are allow in Virtual Button Setting.

For power_outage_memory/clickmode I might forgot to manage it got an error - Attribute not found or something. As it is something new (as least from AD) I am unuse something got changed on this part

jomud9, 2023-07-04:
Let's share the fun for figuring out what's these. UNSUPPORTED_ATTRIBUTE means. :-)

```shell

Received MQTT message on 'zigbee2mqtt/switchz1_livingroom/set' with data '{"flip_indicator_light":"ON"}'
Publishing 'set' 'flip_indicator_light' to 'switchz1_livingroom'
Publish 'set' 'flip_indicator_light' to 'switchz1_livingroom' failed: 'Error: Write 0x54ef4410008f61d3/1 aqaraOpple({"240":{"value":1,"type":32}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'

Publishing 'set' 'power_outage_memory' to 'switchz1_livingroom'
Publish 'set' 'power_outage_memory' to 'switchz1_livingroom' failed: 'Error: Write 0x54ef4410008f61d3/1 aqaraOpple({"513":{"value":1,"type":16}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
Error: Write 0x54ef4410008f61d3/1 aqaraOpple({"513":{"value":1,"type":16}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:419:28) at Endpoint.write (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:488:22) at Object.convertSet (/app/data/extension/externally-loaded.js:115:17) at Publish.onMQTTMessage (/app/lib/extension/publish.ts:248:36)
```

jomud9, 2023-07-12:
@Koenkk and brothers - I got a new Dongle-E and I think I could try to do some sniffing. Could you inspire me how to get require information to fix the mentioned "UNSUPPORTED_ATTRIBUTE"? These looked OK on use with Aqara Home + Hub setup.

ShadowFistOkami, 2023-07-12:
@jomud9 Other more knowledgeable people will probably show up and correct me, but if you have the original Aqara hub, you could try sniffing data through wireshark. 
[This](https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html) guide should work, though you will need a cc2531 adapter and it looks complicated.

Can I ask a favour of you since you have the original hub/app? Can you check if you have a "Preventing accidental deletion" setting (either on the device page or the hub setting page)? That setting is used by Aqara to ignore reset on long press of the switch, and if we could get it to work in Z2M, we might be able to use hold actions on switches without them resetting.

closed 2023-08-20 by github-actions[bot]