Koenkk/zigbee2mqtt #18083: RTCGQ11LM Aqara Motion Sensor no_occupancy_since message malformed

mateuszkozakiewicz, 2023-06-21:
### What happened?

When `no_occupancy_since` is set a malformed message arrives after a specified time period, example:
`Received MQTT message on 'zigbee2mqtt/bridge/request/device/options' with data '{"id":"0x00158d0007758324","options":{"no_occupancy_since":[{"undefined":30}]},"transaction":"fjz2v-14"}'`
Such message is ignored by HASS and the attribute `no_occupancy_since` is not added to the device.

### What did you expect to happen?

Correct format should be per the field description:
```
no_occupancy_since
Sends a message the last time occupancy (occupancy: true) was detected. 
When setting this for example to [10, 60] a `{"no_occupancy_since": 10}` will be send after 10 seconds 
and a `{"no_occupancy_since": 60}` after 60 seconds.
```
So my understanding is I should've received:
``Received MQTT message on 'zigbee2mqtt/bridge/request/device/options' with data '{"id":"0x00158d0007758324","options":{"no_occupancy_since":30},"transaction":"fjz2v-14"}'``
instead.

### How to reproduce it (minimal and precise)

Configure `no_occupancy_since` for Aqara Motion Sensor device.

### Zigbee2MQTT version

1.31.2

### Adapter firmware version

6.7.10.0 build 423

### Adapter

Xiaomi Gateway 3

mateuszkozakiewicz, 2023-06-22:
Correction: it still works kinda odd, observe the following log:
(`occupancy_timeout: 60` `no_occupancy_since: 30`)
```
Zigbee2MQTT:info  2023-06-22 09:42:05: MQTT publish: topic 'zigbee2mqtt/Motion Sensor', payload '{"battery":100,"device_temperature":32,"illuminance":26,"illuminance_lux":26,"linkquality":76,"occupancy":true,"power_outage_count":10,"voltage":3045}'
Zigbee2MQTT:info  2023-06-22 09:42:35: MQTT publish: topic 'zigbee2mqtt/Motion Sensor', payload '{"battery":100,"device_temperature":32,"illuminance":26,"illuminance_lux":26,"linkquality":76,"no_occupancy_since":30,"occupancy":true,"power_outage_count":10,"voltage":3045}'
Zigbee2MQTT:info  2023-06-22 09:43:35: MQTT publish: topic 'zigbee2mqtt/Motion Sensor', payload '{"battery":100,"device_temperature":32,"illuminance":26,"illuminance_lux":26,"linkquality":76,"occupancy":false,"power_outage_count":10,"voltage":3045}'
```

So currently the process is to start counting the `no_occupancy_since` on the rising edge of `occupancy`, so I receive `no_occupancy_since` WHILE `occupancy` is still true which makes very little sense and contradicts the original issue:
https://github.com/Koenkk/zigbee2mqtt/issues/1195#issuecomment-470627505
>When motion true, reset to 0 as long as movement is detected:
>
>{"no_occupancy_since": 0}

Which makes much more sense.
Is current behavior expected or is it a bug?

mateuszkozakiewicz, 2023-06-22:
The attribute is also not showing in HASS but I am unsure where is the problem here.

paifre, 2023-06-24:
Similar behavior reported in #17957

ybizeul, 2023-07-10:
There seem to be an issue with how z2m UI writes the configuration, which it does as follow in the current version exposing this bug :
```
    no_occupancy_since:
      - undefined: 30
      - undefined: 60
      - undefined: 90
```

Instead of :
```
    no_occupancy_since:
      - 30
      - 60
      - 90
```
or :
```
    no_occupancy_since: [30, 60, 90]
```

Fixing the config file seems to fix the issue at least for me.

merlinpimpim, 2023-08-06:
The bug is still there for this module : https://www.zigbee2mqtt.io/devices/RTCGQ01LM.html#xiaomi-rtcgq01lm
(But, as mentioned above, a manual fix in the configuration file provides a workaround).
A real bug fix soon?

closed 2023-09-13 by github-actions[bot]

mpuff, 2023-10-21:
i have the same problem now, also on the motion sensor, the gui is writing it wrong into the configuration file

stuffofmine, 2023-10-22:
Same for me, the config is saved as per below when setting the values through z2m gui. 

```
...
no_occupancy_since:
  - undefined: 300
  - undefined: 600
...
```
Editing/cleaning the config manually is required.

werty1st, 2023-12-01:
Will this not be fixed? Has somebody identified the wrong code?

fcassata, 2024-07-11:
Would love to have this resolved :)