Koenkk/zigbee2mqtt #18090: Failed to start zigbee: get network parameters Error:TIMEOUT

Simonh98, 2023-06-22:
### What happened?

With a freshly installed Homeassistant addon, the startup process sometimes fails with a `Error: get network parameters Error:TIMEOUT` error message. This error does not always occur, but in most cases it does. On the Raspberry Pi 4B it runs without problems so far.

Adapter: ConBee II
Hardware: GK3 Plus Mini PC, Intel Alder Lake-N95
Configuration:
[image]

### What did you expect to happen?

Reliable startup

### How to reproduce it (minimal and precise)

Difficult to say, sometimes the startup process is successful after a reboot or after removing and reinserting the adapter.

### Zigbee2MQTT version

1.31.2-1

### Adapter firmware version

26580700

### Adapter

ConBee II

### Debug log

debug 2023-06-22 20:16:09: Can't load state from file /config/zigbee2mqtt/state.json (doesn't exist)
info  2023-06-22 20:16:09: Logging to console and directory: '/config/zigbee2mqtt/log/2023-06-22.20-16-09' filename: log.txt
info  2023-06-22 20:16:09: Starting Zigbee2MQTT version 1.31.2 (commit #unknown)
info  2023-06-22 20:16:09: Starting zigbee-herdsman (0.14.117)
debug 2023-06-22 20:16:09: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"adapter":"deconz","path":"/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2421591-if00"}}'
info  2023-06-22 20:16:09: zigbee-herdsman started (resumed)
info  2023-06-22 20:16:12: Coordinator firmware version: '{"meta":{"maintrel":0,"majorrel":38,"minorrel":88,"product":0,"revision":"0x26580700","transportrev":0},"type":"ConBee2/RaspBee2"}'
error 2023-06-22 20:16:22: Failed to start zigbee
error 2023-06-22 20:16:22: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
error 2023-06-22 20:16:22: Exiting...
error 2023-06-22 20:16:22: Error: get network parameters Error:TIMEOUT
    at DeconzAdapter.getNetworkParameters (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:1002:35)
    at runMicrotasks (<anonymous>)
    at runNextTicks (node:internal/process/task_queues:61:5)
    at listOnTimeout (node:internal/timers:528:9)
    at processTimers (node:internal/timers:502:7)
    at Controller.getNetworkParameters (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:348:44)
    at Zigbee.start (/app/lib/zigbee.ts:111:62)
    at Controller.start (/app/lib/controller.ts:101:27)
    at start (/app/index.js:107:5)

closed 2023-07-30 by github-actions[bot]

psroczyk, 2023-08-09:
I Have exactly the same issue on same gk3 plus n100

Risc534, 2023-09-03:
Same exact issue here "sometimes the startup process is successful after a reboot or after removing and reinserting the adapter"
Adapter: ConBee II
Hardware: T8Plus Mini PC, Intel N100

Zigbee2MQTT:error 2023-09-03 18:14:02: Failed to start zigbee
Zigbee2MQTT:error 2023-09-03 18:14:02: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2023-09-03 18:14:02: Exiting...
Zigbee2MQTT:error 2023-09-03 18:14:02: Error: get network parameters Error:TIMEOUT
    at DeconzAdapter.getNetworkParameters (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:1002:35)
    at runMicrotasks (<anonymous>)
    at runNextTicks (node:internal/process/task_queues:61:5)
    at listOnTimeout (node:internal/timers:528:9)
    at processTimers (node:internal/timers:502:7)
    at Controller.getNetworkParameters (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:360:44)
    at Zigbee.start (/app/lib/zigbee.ts:113:62)
    at Controller.start (/app/lib/controller.ts:101:27)
    at start (/app/index.js:107:5)

WRLPDZ, 2023-09-12:
Hi @Risc534 did you resolve the problem?
I have the same equipent, and the same error wint Conbee 2.

Risc534, 2023-09-12:
Yes...  :-/ 
I switched to the SONOFF-ZB USB Dongle, and it works perfectly. Same configuration, except adapter parameter from deconz to ezsp. I think the Conbee Dongle definitely has a problem with HA/Zigbee2MQTT and some hardware...

przemowg, 2023-09-25:
Hi, I had the same issue but just update HA to 2023.9.3 (today's release) and it seems to be working.

Kirjy, 2023-11-04:
Hi,
Same issue here. It used to worked before on my raspberry with the same dongle.

Simonh98, 2023-11-04:
> I Have exactly the same issue on same gk3 plus n100 @Simonh98 have you resolved the issue?

No, I switched to the Sonoff dongle. So far it's working fine.

Daniel-WW, 2023-12-04:
Same issue here. Since a few weeks, z2m doesn't work reliable with deconz stick. Sometimes it works for a few hours and then it crashes again. Today z2m docker container was not able to start, becaus it couldn't connect with conbee stick. I did a complete system restart, but only after several hours it was able to connect to the stick again. When the connbee stick is connected, it seems like the transfer takes a long time. Some devices can be addressed, others do not respond at all. For example dimmable lights. For example, I can't use dimmable light bulbs at all: I always get error reports. - So far I haven't been able to narrow down the error. Now I'm doing a test with a longer cable and I've hung the stick away from all the technical stuff.

anthosz, 2023-12-04:
Same error here with conbee2:
After upgrade to another mini pc 1 week ago -> I get this error -> after few docker restart, it worked well ... until this evening, the error continue since more 1h...

It seems the issue started this night when I tried to enabling new appearing...

x3ntr0, 2023-12-15:
Just like in another thread: Service is already running, stop it and start it again.
VM: zigbee2mqtt 
$: sudo systemctl stop zigbee2mqtt
$: npm start

anthosz, 2023-12-15:
> Just like in another thread: Service is already running, stop it and start it again. VM: zigbee2mqtt $: sudo systemctl stop zigbee2mqtt $: npm start

It's not the same error. In my case, npm is not running at all so docker restart...

anthosz, 2023-12-15:
> Same issue here. Since a few weeks, z2m doesn't work reliable with deconz stick. Sometimes it works for a few hours and then it crashes again. Today z2m docker container was not able to start, [...]

By curiosity, did you change your device on which is plugged your stick? Or it's the same?

romain-nio, 2023-12-24:
> Yes... :-/ I switched to the SONOFF-ZB USB Dongle, and it works perfectly. Same configuration, except adapter parameter from deconz to ezsp. I think the Conbee Dongle definitely has a problem with HA/Zigbee2MQTT and some hardware...

> > I Have exactly the same issue on same gk3 plus n100 @Simonh98 have you resolved the issue?
> 
> No, I switched to the Sonoff dongle. So far it's working fine.

@Simonh98  thanks ! do you have to resync all your devices and reconfigure all z2M ? or just changing the adapter ?

anthosz, 2023-12-24:
> @Simonh98 thanks ! do you have to resync all your devices and reconfigure all z2M ? or just changing the adapter ?

It's not an issue with adapter. If you check on a rpi by example, it works.

You can check the another ticket for details

Daniel-WW, 2023-12-24:
@anthosz 
Hi, I didn't change my device. After a lot of tests and restarts I decided to use a longer usb-cable and I put the deconz-stick more than one meter away from all other technical stuff, high on the wall.
By that I obviously could reach a much better connectivity. Since 10 Days my system runs again without bigger problems.

anthosz, 2023-12-24:
> @anthosz Hi, I didn't change my device. After a lot of tests and restarts I decided to use a longer usb-cable and I put the deconz-stick more than one meter away from all other technical stuff, high [...]

I don't think we speak of the same issue, the USB cable is standard (I want to say mandatory) to avoid interference.

The issue mentioned here is the fact that z2m crash before start (even with no db/no device paired)

Kirjy, 2023-12-26:
> @anthosz Hi, I didn't change my device. After a lot of tests and restarts I decided to use a longer usb-cable and I put the deconz-stick more than one meter away from all other technical stuff, high [...]

I tried the same with a 2m cable but that didn't work for me. Best options were sonoff dongle or using deconz. Z2m and ha didn't work no matter what.

anthosz, 2023-12-27:
> I tried the same with a 2m cable but that didn't work for me. Best options were sonoff dongle or using deconz. Z2m and ha didn't work no matter what.

Unless with raspberry or probably another computer type

romain-nio, 2023-12-30:
> I tried the same with a 2m cable but that didn't work for me. Best options were sonoff dongle or using deconz. Z2m and ha didn't work no matter what.

ok I will probably try to buy a sonoff dongle. Do I need to reconfigure/re-pair all the devices if I'm restoring a backup with all the configured devices in z2m and configure the new sonoff dongle?

Thanks

vdiogo, 2024-02-07:
Same issue here after installing z2m in LXC following the instructions here https://smarthomescene.com/guides/how-to-separate-zigbee2mqtt-from-home-assistant-in-proxmox/
```
Zigbee2MQTT:info  2024-02-07 10:52:33: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2024-02-07.10-52-33' filename: log.txt
Zigbee2MQTT:info  2024-02-07 10:52:34: Starting Zigbee2MQTT version 1.35.3 (commit #fe0742a)
Zigbee2MQTT:info  2024-02-07 10:52:34: Starting zigbee-herdsman (0.33.8)
Zigbee2MQTT:info  2024-02-07 10:52:37: zigbee-herdsman started (resumed)
Zigbee2MQTT:info  2024-02-07 10:52:37: Coordinator firmware version: 'undefined'
Zigbee2MQTT:error 2024-02-07 10:52:47: Failed to start zigbee
Zigbee2MQTT:error 2024-02-07 10:52:47: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2024-02-07 10:52:47: Exiting...
Zigbee2MQTT:error 2024-02-07 10:52:47: Error: get network parameters Error:TIMEOUT
    at DeconzAdapter.getNetworkParameters (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:1017:35)
    at runNextTicks (node:internal/process/task_queues:60:5)
    at listOnTimeout (node:internal/timers:540:9)
    at processTimers (node:internal/timers:514:7)
    at Controller.getNetworkParameters (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:362:44)
    at Zigbee.start (/opt/zigbee2mqtt/lib/zigbee.ts:120:62)
    at Controller.start (/opt/zigbee2mqtt/lib/controller.ts:109:27)
    at start (/opt/zigbee2mqtt/index.js:107:5)
root@zigbee2mqtt:/opt/zigbee2mqtt# 
```

a-cesari, 2024-07-09:
Same issue here.
Home Assistant OS + zigbee2mqtt 1.39.0 + conbee2 (26780700 firm) on a Mini PC intel N95.
`[2024-07-09 08:27:56] error: 	z2m: Failed to start zigbee
[2024-07-09 08:27:56] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-07-09 08:27:56] error: 	z2m: Exiting...
[2024-07-09 08:27:56] error: 	z2m: Error: get network parameters Error:TIMEOUT
    at DeconzAdapter.getNetworkParameters (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:1015:35)
    at Controller.getNetworkParameters (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:379:44)
    at Zigbee.start (/app/lib/zigbee.ts:125:62)
    at Controller.start (/app/lib/controller.ts:139:27)
    at start (/app/index.js:154:5)`

It's working correctly if using deconz Addon.

Same hardware and zigbee2mqtt was working fine on raspberry pi4.

a-cesari, 2024-07-10:
> Same issue here. Home Assistant OS + zigbee2mqtt 1.39.0 + conbee2 (26780700 firm) on a Mini PC intel N95.
>
> ```
> [2024-07-09 08:27:56] error: 	z2m: Failed to start zigbee
> [2024-07-09 08:27:56] error: [...]

I solved the problem by attaching the ConBee 2 to a USB 2.0 cable

anthosz, 2024-07-10:
> I solved the problem by attaching the ConBee 2 to a USB 2.0 cable

N100 + USB cable 2 0 = I reproduce the issue

havardthom, 2024-08-09:
Same error here with conbee 2 stick on Beelink S12 Pro minipc (N100). HA OS LXC from https://tteck.github.io/Proxmox and USB device passthrough. Also had a couple of other startup errors (`undefined` and `USB device not found`), this hardware combination feels very unstable for Z2M in general. I managed to start Z2M successfully a couple of times when replugging the usb and restarting HAOS container, but not always.. Luckily I've ordered a SMLIGHT SLZB-06 which is in transit, hopefully that device works.

````
[00:11:55] INFO: Preparing to start...
[00:11:55] INFO: Socat not enabled
[00:11:55] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2024-08-10 00:11:56] info: 	z2m: Logging to console, file (filename: log.log)
[2024-08-10 00:11:56] debug: 	z2m: Loaded state from file /config/zigbee2mqtt/state.json
[2024-08-10 00:11:56] info: 	z2m: Starting Zigbee2MQTT version 1.39.1 (commit #unknown)
[2024-08-10 00:11:56] info: 	z2m: Starting zigbee-herdsman (0.55.3)
[2024-08-10 00:11:56] debug: 	z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":15924,\"extendedPanID\":[134,75,67,70,30,52,170,195],\"channelList\":[11],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/config/zigbee2mqtt/database.db\",\"databaseBackupPath\":\"/config/zigbee2mqtt/database.db.backup\",\"backupPath\":\"/config/zigbee2mqtt/coordinator_backup.json\",\"serialPort\":{\"path\":\"/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2194195-if00\",\"adapter\":\"deconz\"},\"adapter\":{\"concurrent\":null,\"delay\":null,\"disableLED\":false}}"'
[2024-08-10 00:11:57] debug: 	zh:adapter: Failed to validate path: 'Error: spawn udevadm ENOENT'
[2024-08-10 00:11:57] debug: 	zh:deconz:driver: Set delay to 0
[2024-08-10 00:11:57] debug: 	zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":"HIDDEN","panID":15924,"extendedPanID":[134,75,67,70,30,52,170,195],"channelList":[11]},"serialPort":{"path":"/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2194195-if00","adapter":"deconz"},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}'
[2024-08-10 00:11:57] debug: 	zh:deconz:driver: Opening with /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2194195-if00
[2024-08-10 00:11:57] debug: 	zh:deconz:driver: Serialport opened
[2024-08-10 00:11:57] debug: 	zh:controller: Started with result 'resumed'
[2024-08-10 00:11:57] debug: 	zh:deconz:driver: send read parameter request from queue. seqNr: 1 paramId: 5
[2024-08-10 00:11:57] debug: 	zh:deconz:frameparser: PANID: 3e34
[2024-08-10 00:11:57] debug: 	zh:deconz:driver: send read parameter request from queue. seqNr: 2 paramId: 11
[2024-08-10 00:11:57] debug: 	zh:deconz:frameparser: APS_EXT_PANID: 0x864b43461e34aac3
[2024-08-10 00:11:57] debug: 	zh:deconz:driver: send read parameter request from queue. seqNr: 3 paramId: 28
[2024-08-10 00:12:00] debug: 	zh:deconz: checking coordinator simple descriptor
[2024-08-10 00:12:00] debug: 	zh:deconz:driver: DATA_REQUEST - destAddr: 0x0 EP:0 SeqNr. 4 request id: 1
[2024-08-10 00:12:00] debug: 	zh:deconz:frameparser: CHANNEL: 11
[2024-08-10 00:12:00] debug: 	zh:deconz:driver: send read parameter request from queue. seqNr: 5 paramId: 24
[2024-08-10 00:12:00] debug: 	zh:deconz:frameparser: DATA_REQUEST RESPONSE - request id: 1 status: 6
[2024-08-10 00:12:00] debug: 	zh:deconz:driver: networkstate: 0 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 0
[2024-08-10 00:12:10] debug: 	zh:deconz:driver: Timeout for request - CMD: 0xa seqNr: 5
[2024-08-10 00:12:10] debug: 	zh:deconz: get network parameters Error:TIMEOUT
[2024-08-10 00:12:10] error: 	z2m: Error while starting zigbee-herdsman
[2024-08-10 00:12:10] error: 	z2m: Failed to start zigbee
[2024-08-10 00:12:10] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-08-10 00:12:10] error: 	z2m: Exiting...
[2024-08-10 00:12:10] error: 	z2m: Error: get network parameters Error:TIMEOUT
    at DeconzAdapter.getNetworkParameters (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:1153:35)
    at processTicksAndRejections (node:internal/process/task_queues:95:5)
    at runNextTicks (node:internal/process/task_queues:64:3)
    at listOnTimeout (node:internal/timers:538:9)
    at processTimers (node:internal/timers:512:7)
    at Controller.getNetworkParameters (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:394:44)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:135:31)
    at Zigbee.start (/app/lib/zigbee.ts:63:27)
    at Controller.start (/app/lib/controller.ts:139:27)
    at start (/app/index.js:154:5)

/app/node_modules/winston/node_modules/readable-stream/lib/_stream_writable.js:264
  var er = new ERR_STREAM_WRITE_AFTER_END();
           ^
Error: write after end
    at writeAfterEnd (/app/node_modules/winston/node_modules/readable-stream/lib/_stream_writable.js:264:12)
    at DerivedLogger.Writable.write (/app/node_modules/winston/node_modules/readable-stream/lib/_stream_writable.js:300:21)
    at DerivedLogger.log (/app/node_modules/winston/lib/winston/logger.js:231:12)
    at Logger.log (/app/lib/util/logger.ts:189:25)
    at Logger.debug (/app/lib/util/logger.ts:210:14)
    at Driver.processQueue (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/driver.ts:489:24)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/driver.ts:169:22)
    at listOnTimeout (node:internal/timers:569:17)
    at processTimers (node:internal/timers:512:7)
```

Wikibear, 2024-08-30:
Update to the latest firmware version will fix this.

anthosz, 2024-08-31:
> Update to the latest firmware version will fix this.

What do you mean? The latest firmware of the conbee? It's from 2022.. Prior to N100 availability...

Wikibear, 2024-08-31:
Sorry for my shorten message. You must update the Conbee II Firmware from the Stick. In my case it's working. But it's a little bit tricky without phoscon. You must download firmware and flasher from git. You must compile the flasher and the you can flash the stick. There are amount of site they explain this.

anthosz, 2024-08-31:
> Sorry for my shorten message. You must update the Conbee II Firmware from the Stick. In my case it's working. But it's a little bit tricky without phoscon. You must download firmware and flasher [...]

The big question is: where is this firmware? Do you have the link? It's official? Because on phoscon, nothing available since 2022

Wikibear, 2024-08-31:
https://deconz.dresden-elektronik.de/deconz-firmware/
Yes be sure that you have the latest. It's from 2022 sure, but I had installed from 2021 before. The most guys never upgrade the firmware. So, after upgrade the firmware to latest version, it's working for me.