Koenkk/zigbee2mqtt #18096: Develco Siren SIRZB-110,  ssIasZone  alarm is not cleared

pomacanthus, 2023-06-23:
### What happened?

Hi,
Information from the Ias Zone updates only once. After this update, the Ias Zone message is repeated and the fields do not toggle back again.

_example:_ ac_status gets true when device is unplugged from power, and then multiple messages like this are sent *even if the device is plugged back in*

`MQTT publish: topic 'zigbee2mqtt/0x0015bc004100XXXX', payload '{"ac_status":true,"battery":100,"battery_low":false,"elapsed":5012,"enrolled":true,"ias_alarm":false,"ias_cie_address":"0x00124b00257XXXX","linkquality":182,"max_duration":900,"restore_reports":true,"supervision_reports":true,"tamper":false,"test":false,"update":{"installed_version":131074,"latest_version":131073,"state":"available"},"voltage":3900,"zone_id":23}'`

The tamper bit has the same behaviour. Tamper will turn to true and then never toggle back to false.

This seems to be an issue with the default response :extract from the siren's technical  manual [https://www.develcoproducts.com/media/1999/sirzb-110-technical-manual-smart-siren.pdf](url)
`Note: How to clear a alarm in the “Zone status”
The device requests ZCL Default Response on the Zone Status Change notification, if any new Alarm bit has been set. Until the IAS CIE has acknowledged the received alarm by sending the mandated Default Response, the Alarm bits are not cleared – even if there is no longer an alarm situation. When the Default Response is received, a new Zone Status Change notification is sent with the Alarm bits cleared, if the alarm situation has disappeared since sending the Zone Status message with alarm set.
`

### What did you expect to happen?

ac_status should toggle when plugged or unplugged
tamper should  toggle to false when device is open or closed 

### How to reproduce it (minimal and precise)

(1) pair a SIREN SIRZB-110 
(2) unplug it, check ac_status from log (true)
-> multiple identical messages while a single message is expected 
(3) plug it back in to power, check ac_status from log (got true, expected false)
(4) open it, check tamper from log (true)
-> multiple identical messages while a single message is expected 
(5) close it , check tamper from log (got true, expected false)

### Zigbee2MQTT version

1.31.8

### Adapter firmware version

20221226

### Adapter

CC2652

### Debug log

`Info 2023-06-23 09:26:42MQTT publish: topic 'zigbee2mqtt/0x0015bc004100XXXX', payload '{"ac_status":true,"battery":100,"battery_low":false,"elapsed":180914,"enrolled":true,"ias_alarm":false,"ias_cie_address":"0x00124b002573XXXX","linkquality":167,"max_duration":900,"restore_reports":true,"supervision_reports":true,"tamper":false,"test":false,"update":{"installed_version":131074,"latest_version":131073,"state":"available"},"voltage":3900,"zone_id":23}'
Info 2023-06-23 09:26:47MQTT publish: topic 'zigbee2mqtt/0x0015bc004100XXXX', payload '{"ac_status":true,"battery":100,"battery_low":false,"elapsed":4946,"enrolled":true,"ias_alarm":false,"ias_cie_address":"0x00124b002573XXXX","linkquality":171,"max_duration":900,"restore_reports":true,"supervision_reports":true,"tamper":false,"test":false,"update":{"installed_version":131074,"latest_version":131073,"state":"available"},"voltage":3900,"zone_id":23}'
Info 2023-06-23 09:26:52MQTT publish: topic 'zigbee2mqtt/0x0015bc004100XXXX', payload '{"ac_status":true,"battery":100,"battery_low":false,"elapsed":5007,"enrolled":true,"ias_alarm":false,"ias_cie_address":"0x00124b002573XXXX","linkquality":171,"max_duration":900,"restore_reports":true,"supervision_reports":true,"tamper":false,"test":false,"update":{"installed_version":131074,"latest_version":131073,"state":"available"},"voltage":3900,"zone_id":23}'
Info 2023-06-23 09:26:57MQTT publish: topic 'zigbee2mqtt/0x0015bc004100XXXX', payload '{"ac_status":true,"battery":100,"battery_low":false,"elapsed":4998,"enrolled":true,"ias_alarm":false,"ias_cie_address":"0x00124b002573XXXX","linkquality":171,"max_duration":900,"restore_reports":true,"supervision_reports":true,"tamper":false,"test":false,"update":{"installed_version":131074,"latest_version":131073,"state":"available"},"voltage":3900,"zone_id":23}'`

Koenkk, 2023-06-23:
Can you provide the herdsman debug log when the device send a zone status change notification? I expect that zigbee2mqtt already sends the default response.

pomacanthus, 2023-06-23:
Thanks for your help @Koenkk, it is much appreciated.

I believe this is the log you requested :
``` - Logs begin at Fri 2023-06-23 11:19:15 UTC. --
Jun 23 16:10:07 device-19300 node[5015]: [device] Received zigbee2mqtt/0x0015bc0031014fa9/availability message: "{\"state\":\"online\"}"
Jun 23 16:10:11 device-19300 node[5015]: 2023-06-23T16:10:11.205Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,29,68,129,0,0,0,5,106,93,35,11,0,163,0,188,176,223,0,0,9,25,170,0]
Jun 23 16:10:11 device-19300 node[5015]: 2023-06-23T16:10:11.206Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,29,68,129,0,0,0,5,106,93,35,11,0,163,0,188,176,223,0,0,9,25,170,0]
Jun 23 16:10:11 device-19300 node[5015]: 2023-06-23T16:10:11.208Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [48,0,0,23,28,80,106,93,29,73]
Jun 23 16:10:11 device-19300 node[5015]: 2023-06-23T16:10:11.209Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,29,68,129,0,0,0,5,106,93,35,11,0,163,0,188,176,223,0,0,9,25,170,0,48,0,0,23,28,80,106,93,29,73]
Jun 23 16:10:11 device-19300 node[5015]: 2023-06-23T16:10:11.209Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 29 - 2 - 4 - 129 - [0,0,0,5,106,93,35,11,0,163,0,188,176,223,0,0,9,25,170,0,48,0,0,23,28,80,106,93,29] - 73
Jun 23 16:10:11 device-19300 node[5015]: 2023-06-23T16:10:11.211Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":1280,"srcaddr":23914,"srcendpoint":35,"dstendpoint":11,"wasbroadcast":0,"linkquality":163,"securityuse":0,"timestamp":14659772,"transseqnumber":0,"len":9,"data":{"type":"Buffer","data":[25,170,0,48,0,0,23,28,80]}}
Jun 23 16:10:11 device-19300 node[5015]: 2023-06-23T16:10:11.226Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":170,"manufacturerCode":null,"commandIdentifier":0},"Payload":{"zonestatus":48,"extendedstatus":0},"Command":{"ID":0,"parameters":[{"name":"zonestatus","type":33},{"name":"extendedstatus","type":32}],"name":"statusChangeNotification"}},"address":23914,"endpoint":35,"linkquality":163,"groupID":0,"wasBroadcast":false,"destinationEndpoint":11}'
Jun 23 16:10:11 device-19300 node[5015]: 2023-06-23T16:10:11.255Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
Jun 23 16:10:11 device-19300 node[5015]: Zigbee2MQTT:info  2023-06-23 16:10:11: MQTT publish: topic 'zigbee2mqtt/0x0015bc001a01e982', payload '{"battery_low":false,"ias_alarm":false,"linkquality":163,"occupancy":false,"tamper":false}'
Jun 23 16:10:33 device-19300 node[5015]: 2023-06-23T16:10:33.730Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,29,68,129,0,0,0,5,26,241,43,11,0,145,0,46]
Jun 23 16:10:33 device-19300 node[5015]: 2023-06-23T16:10:33.731Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,29,68,129,0,0,0,5,26,241,43,11,0,145,0,46]
Jun 23 16:10:33 device-19300 node[5015]: 2023-06-23T16:10:33.734Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [44,5,0,0,9,9,3,0]
Jun 23 16:10:33 device-19300 node[5015]: 2023-06-23T16:10:33.735Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,29,68,129,0,0,0,5,26,241,43,11,0,145,0,46,44,5,0,0,9,9,3,0]
Jun 23 16:10:33 device-19300 node[5015]: 2023-06-23T16:10:33.737Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [240,0,0,23,60,163,26,241,29,13]
Jun 23 16:10:33 device-19300 node[5015]: 2023-06-23T16:10:33.737Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,29,68,129,0,0,0,5,26,241,43,11,0,145,0,46,44,5,0,0,9,9,3,0,240,0,0,23,60,163,26,241,29,13]
Jun 23 16:10:33 device-19300 node[5015]: 2023-06-23T16:10:33.738Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 29 - 2 - 4 - 129 - [0,0,0,5,26,241,43,11,0,145,0,46,44,5,0,0,9,9,3,0,240,0,0,23,60,163,26,241,29] - 13
Jun 23 16:10:33 device-19300 node[5015]: 2023-06-23T16:10:33.739Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":1280,"srcaddr":61722,"srcendpoint":43,"dstendpoint":11,"wasbroadcast":0,"linkquality":145,"securityuse":0,"timestamp":338990,"transseqnumber":0,"len":9,"data":{"type":"Buffer","data":[9,3,0,240,0,0,23,60,163]}}
Jun 23 16:10:33 device-19300 node[5015]: 2023-06-23T16:10:33.744Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":3,"manufacturerCode":null,"commandIdentifier":0},"Payload":{"zonestatus":240,"extendedstatus":0},"Command":{"ID":0,"parameters":[{"name":"zonestatus","type":33},{"name":"extendedstatus","type":32}],"name":"statusChangeNotification"}},"address":61722,"endpoint":43,"linkquality":145,"groupID":0,"wasBroadcast":false,"destinationEndpoint":11}'
Jun 23 16:10:33 device-19300 node[5015]: 2023-06-23T16:10:33.758Z zigbee-herdsman:controller:endpoint DefaultResponse 0x0015bc00410siren/43 1280(0, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
Jun 23 16:10:33 device-19300 node[5015]: 2023-06-23T16:10:33.772Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x0015bc00410siren:61722/43 (0,0,1)
[... cut ...]
Jun 23 16:11:43 device-19300 node[5015]: 2023-06-23T16:11:43.445Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,14,200]
Jun 23 16:11:43 device-19300 node[5015]: 2023-06-23T16:11:43.445Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,14] - 200
Jun 23 16:11:43 device-19300 node[5015]: 2023-06-23T16:11:43.446Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":14}
Jun 23 16:11:43 device-19300 node[5015]: 2023-06-23T16:11:43.447Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
```

Koenkk, 2023-06-24:
Based on the logs, it seems a default response is send:

```
Jun 23 16:10:33 device-19300 node[5015]: 2023-06-23T16:10:33.739Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":1280,"srcaddr":61722,"srcendpoint":43,"dstendpoint":11,"wasbroadcast":0,"linkquality":145,"securityuse":0,"timestamp":338990,"transseqnumber":0,"len":9,"data":{"type":"Buffer","data":[9,3,0,240,0,0,23,60,163]}}
Jun 23 16:10:33 device-19300 node[5015]: 2023-06-23T16:10:33.744Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":3,"manufacturerCode":null,"commandIdentifier":0},"Payload":{"zonestatus":240,"extendedstatus":0},"Command":{"ID":0,"parameters":[{"name":"zonestatus","type":33},{"name":"extendedstatus","type":32}],"name":"statusChangeNotification"}},"address":61722,"endpoint":43,"linkquality":145,"groupID":0,"wasBroadcast":false,"destinationEndpoint":11}'
Jun 23 16:10:33 device-19300 node[5015]: 2023-06-23T16:10:33.758Z zigbee-herdsman:controller:endpoint DefaultResponse 0x0015bc00410siren/43 1280(0, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
Jun 23 16:10:33 device-19300 node[5015]: 2023-06-23T16:10:33.772Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x0015bc00410siren:61722/43 (0,0,1)
Jun 23 16:10:33 device-19300 node[5015]: 2023-06-23T16:10:33.779Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":61722,"destendpoint":43,"srcendpoint":1,"clusterid":1280,"transid":2,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[24,3,11,0,0]}}
```

pomacanthus, 2023-06-24:
> Based on the logs, it seems a default response is send:
> [...]
{"dstaddr":61722,"destendpoint":43,"srcendpoint":1,"clusterid":1280,"transid":2,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[24,3,11,0,0]}}
> ```

This above looks like the default response, yet the siren kept sending same status, as if the default response is not the one the siren expects. 

Nota: _I could confirm that the same siren toggles the ac status on a proprietary gateway._

pomacanthus, 2023-06-24:
@Koenkk I played with the various fields and managed to have the siren toggle by forcing the "options.direction" to 0 Zcl.Direction.CLIENT_TO_SERVER

- the default response 

> Jun 24 20:28:35 device-19300 node[14431]: 2023-06-24T20:28:35.517Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x0015bc0041siren:61722/43 (0,0,1)
> Jun 24 20:28:35 device-19300 node[14431]: 2023-06-24T20:28:35.519Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":61722,"destendpoint":43,"srcendpoint":1,"clusterid":1280,"transid":15,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[ **16** ,15,11,0,0]}}
> Jun 24 20:28:35 device-19300 node[14431]: 2023-06-24T20:28:35.520Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,15,36,1,26,241,43,1,0,5,15,0,30,5,**16,**15,11,0,0,238]
> 

- code for the proof of concept: 

endpoint.ts :
>   public async defaultResponse(
>         commandID: number, status: number, clusterID: number, transactionSequenceNumber: number, options?: Options
>     ): Promise<void> {
>         assert(!options || !options.hasOwnProperty('transactionSequenceNumber'), 'Use parameter');
>         options = this.getOptionsWithDefaults(options, true, Zcl.Direction.SERVER_TO_CLIENT, null);
>         const payload = {cmdId: commandID, statusCode: status};
>         const frame = Zcl.ZclFrame.create(
>             Zcl.FrameType.GLOBAL, **0**, options.disableDefaultResponse,
>             options.manufacturerCode, transactionSequenceNumber  , 'defaultRsp', clusterID, payload, options.reservedBits
>         );
>         [...]

Now, I am not sure if we should enforce that opposite direction for that SIRZB-110 model only or for that ssIasZone cluster. I am not sure either how to do it.

Koenkk, 2023-06-25:
Great investigation! I pushed a fix in https://github.com/Koenkk/zigbee-herdsman/pull/726, let me know if it works.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-06-25 by Koenkk

pomacanthus, 2023-06-25:
Fix looks and works fine, at least for the siren SIRZB-110

I will test other sensors I have tomorrow.
Thanks