Koenkk/zigbee2mqtt #18097: [New device support]:

dankarization, 2023-06-23:
### Link

https://www.becasmart.com/product/51.html

### Database entry

same as all other BHT Family thermostats

### Comments

I have a BHT-006GBZB theermostat, functioning the same as old BHT-002, but under a new manufacturer name
{modelID: 'TS0601', manufacturerName: '_TZE204_aoclfnxz'},

Please add it to the list of others under this thermostat converter :3

Koenkk, 2023-06-23:
added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-06-23 by Koenkk

denveronly, 2023-08-27:
I recently added same thermostat and in logs i see continuos errors like these

```
2023-08-27 20:58:23.883 ERROR (MainThread) [homeassistant.components.mqtt.models] Exception raised when updating state of sensor.poolpump_program, topic: 'zigbee2mqtt/PoolPump' with payload: b'{"child_lock":"UNLOCK","current_heating_setpoint":33,"deadzone_temperature":null,"heat":"ON","last_seen":"2023-08-27T17:58:23.878Z","linkquality":72,"local_temperature":27.5,"local_temperature_calibration":-2,"max_temperature":45,"max_temperature_limit":null,"min_temperature_limit":4,"preset":"hold","preset_mode":"hold","program":{"saturday_p1_hour":6,"saturday_p1_minute":0,"saturday_p1_temperature":35,"saturday_p2_hour":12,"saturday_p2_minute":0,"saturday_p2_temperature":35,"saturday_p3_hour":14,"saturday_p3_minute":30,"saturday_p3_temperature":35,"saturday_p4_hour":17,"saturday_p4_minute":30,"saturday_p4_temperature":35,"sunday_p1_hour":6,"sunday_p1_minute":0,"sunday_p1_temperature":35,"sunday_p2_hour":12,"sunday_p2_minute":30,"sunday_p2_temperature":35,"sunday_p3_hour":14,"sunday_p3_minute":30,"sunday_p3_temperature":21,"sunday_p4_hour":18,"sunday_p4_minute":30,"sunday_p4_temperature":35,"weekdays_p1_hour":6,"weekdays_p1_minute":0,"weekdays_p1_temperature":35,"weekdays_p2_hour":11,"weekdays_p2_minute":26,"weekdays_p2_temperature":35,"weekdays_p3_hour":13,"weekdays_p3_minute":30,"weekdays_p3_temperature":35,"weekdays_p4_hour":17,"weekdays_p4_minute":40,"weekdays_p4_temperature":35},"running_state":"heat","sensor":"OU","system_mode":"heat"}'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/mqtt/models.py", line 270, in process_write_state_requests
    entity.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 746, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 865, in _async_write_ha_state
    hass.states.async_set(entity_id, state, attr, self.force_update, self._context)
  File "/usr/src/homeassistant/homeassistant/core.py", line 1650, in async_set
    state = State(
            ^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 1266, in __init__
    raise InvalidStateError(
homeassistant.exceptions.InvalidStateError: Invalid state encountered for entity ID: sensor.poolpump_program. State max length is 255 characters.
2023-08-27 20:58:23.960 ERROR (MainThread) [homeassistant.components.mqtt.models] Exception raised when updating state of sensor.poolpump_program, topic: 'zigbee2mqtt/PoolPump' with payload: b'{"child_lock":"UNLOCK","current_heating_setpoint":33,"deadzone_temperature":null,"heat":"ON","last_seen":"2023-08-27T17:58:23.955Z","linkquality":72,"local_temperature":27.5,"local_temperature_calibration":-2,"max_temperature":45,"max_temperature_limit":null,"min_temperature_limit":4,"preset":"hold","preset_mode":"hold","program":{"saturday_p1_hour":6,"saturday_p1_minute":0,"saturday_p1_temperature":35,"saturday_p2_hour":12,"saturday_p2_minute":0,"saturday_p2_temperature":35,"saturday_p3_hour":14,"saturday_p3_minute":30,"saturday_p3_temperature":35,"saturday_p4_hour":17,"saturday_p4_minute":30,"saturday_p4_temperature":35,"sunday_p1_hour":6,"sunday_p1_minute":0,"sunday_p1_temperature":35,"sunday_p2_hour":12,"sunday_p2_minute":30,"sunday_p2_temperature":35,"sunday_p3_hour":14,"sunday_p3_minute":30,"sunday_p3_temperature":21,"sunday_p4_hour":18,"sunday_p4_minute":30,"sunday_p4_temperature":35,"weekdays_p1_hour":6,"weekdays_p1_minute":0,"weekdays_p1_temperature":35,"weekdays_p2_hour":11,"weekdays_p2_minute":26,"weekdays_p2_temperature":35,"weekdays_p3_hour":13,"weekdays_p3_minute":30,"weekdays_p3_temperature":35,"weekdays_p4_hour":17,"weekdays_p4_minute":40,"weekdays_p4_temperature":35},"running_state":"heat","sensor":"OU","system_mode":"heat"}'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/mqtt/models.py", line 270, in process_write_state_requests
    entity.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 746, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 865, in _async_write_ha_state
    hass.states.async_set(entity_id, state, attr, self.force_update, self._context)
  File "/usr/src/homeassistant/homeassistant/core.py", line 1650, in async_set
    state = State(
            ^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 1266, in __init__
    raise InvalidStateError(
homeassistant.exceptions.InvalidStateError: Invalid state encountered for entity ID: sensor.poolpump_program. State max length is 255 characters.
2023-08-27 20:58:24.063 ERROR (MainThread) [homeassistant.components.mqtt.models] Exception raised when updating state of sensor.poolpump_program, topic: 'zigbee2mqtt/PoolPump' with payload: b'{"child_lock":"UNLOCK","current_heating_setpoint":33,"deadzone_temperature":null,"heat":"ON","last_seen":"2023-08-27T17:58:24.058Z","linkquality":72,"local_temperature":27.5,"local_temperature_calibration":-2,"max_temperature":45,"max_temperature_limit":null,"min_temperature_limit":4,"preset":"hold","preset_mode":"hold","program":{"saturday_p1_hour":6,"saturday_p1_minute":0,"saturday_p1_temperature":35,"saturday_p2_hour":12,"saturday_p2_minute":0,"saturday_p2_temperature":35,"saturday_p3_hour":14,"saturday_p3_minute":30,"saturday_p3_temperature":35,"saturday_p4_hou [... cut ...]
    hass.states.async_set(entity_id, state, attr, self.force_update, self._context)
  File "/usr/src/homeassistant/homeassistant/core.py", line 1650, in async_set
    state = State(
            ^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 1266, in __init__
    raise InvalidStateError(
homeassistant.exceptions.InvalidStateError: Invalid state encountered for entity ID: sensor.poolpump_program. State max length is 255 characters.
```

I guess it is happening becasuse sensor program is being fed by a long json instead of 255 character

Nikolay-Ch, 2024-01-09:
Hi,
This device contains two addition display info: weather and humidity...
Is it possible to transfer this
[image]
 information to the device?