Koenkk/zigbee2mqtt #18101: [Feature request]: Reducing the number of writes to the SD card.

krzbor, 2023-06-23:
### Is your feature request related to a problem? Please describe

I have installed zigbee2mqtt on Raspberry Pi. The software is on the SD card. I try to minimize the number of records on the card. I set cache_state_persistent: false but I noticed that the database.db file is also changed frequently.

### Describe the solution you'd like

Is it possible to disable saving states to this file? The idea is that it will only be modified when adding/removing devices. Perhaps setting cache_state_persistent: false should also apply to database.db modifications?

### Describe alternatives you've considered

You can also add a new option in configuration.yaml that blocks saving states and is independent of cache_state_persistent

### Additional context

-

lux73, 2023-06-26:
in my opinion this is not a funcionality that should z2m provide

Google for zRam, this is what you searching for

With all pro & cons 🙃

krzbor, 2023-06-26:
I noticed that the database.db contains the exact device visibility times. I thought every such change writes to disk. After more thorough testing, it turned out that database.db is written once an hour. It suits me perfectly.

closed 2023-06-26 by krzbor