Koenkk/zigbee2mqtt #18103: [New device support]: Zigbee Curtain Blinds Controller YH004

helgek, 2023-06-24:
### Link

https://expo.tuya.com/product/952002

### Database entry

{"id":18,"type":"EndDevice","ieeeAddr":"0x540f57fffebc02d3","nwkAddr":46329,"manufId":4098,"manufName":"_TZE200_zvo63cmo","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":" o',\u0012 o',\u0012","65506":31,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE200_zvo63cmo","powerSource":3,"zclVersion":3,"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genPowerCfg":{"attributes":{}},"genOta":{"attributes":{}},"genGroups":{"attributes":{"nameSupport":0}},"genTime":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1687459642130,"defaultSendRequestWhen":"immediate"}

### Comments

Hi,

I would very much appreciate your support to set up the external converter file. I've managed to obtain Tuya data points already. I hope this is a good starting point.

Thanks,
Helge

--

Device:
https://expo.tuya.com/product/952002

[image]

[image]

What makes this device very interesting and different from the existing covers for blinds in the [database](https://zigbee.blakadder.com/covers.html) is:

- Faster rotation speed
- larger diameter of the driving gear
- 1.35 Nm torque instead of 0.45 Nm
- it communicates battery level which isn't the case for the other Zigbee motors I'm using.

The reason why I bought this device in addition to the 0.45 Nm devices that I already have is that one big blind in my apartment stutters when it's being moved up and down slowly. Now with this device it moves the blind much faster and the stuttering of the blind is almost gone.

Manufacturer:
https://expo.tuya.com/store/en/contacts/P9kmp7y2lylv1

More devices from this manufacturer:
https://expo.tuya.com/store/en/shop/P9kmp7y2lylv1

**This is the data points investigation outcome through Tuya Gateway:** 
[Based on the new Tuya device instructions](https://www.zigbee2mqtt.io/advanced/support-new-devices/02_support_new_tuya_devices.html):

{"1":"Control","2":"Curtain position setting","3":"Current curtain position","5":"Motor Direction","7":"Work State","8":"Left time","9":"Left Time","10":"Fault","101":"电池电量"}

The Chines text translates to "Battery Power" (according to Google Translate) and battery power information is available through Tuya gateway.

Screenshot:

[image]

<html>
<body>
<div class="ant-tabs-nav" style="border: 0px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-variant-numeric: inherit; font-variant-east-asian: inherit; font-variant-alternates: inherit; font-weight: 400; font-stretch: inherit; font-size: 14px; line-height: inherit; font-family: Calibri, Arial, Helvetica, sans-serif; font-optical-sizing: inherit; font-kerning: inherit; font-feature-settings: inherit; font-variation-settings: inherit; margin: 0px; padding: 0px; vertical-align: baseline; color: rgb(39, 46, 59); letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal; text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial; box-sizing: border-box; align-items: center; display: flex; flex: 0 0 auto; background-color: rgb(255, 255, 255);"><div class="ant-tabs-nav-wrap" style="border: 0px; font: inherit; margin: 0px; padding: 0px; vertical-align: baseline; color: inherit; box-sizing: border-box; align-self: stretch; display: flex; flex: 1 1 auto; overflow: hidden;"><div class="ant-tabs-nav-list" style="border: 0px; font: inherit; margin: 0px; padding: 0px; vertical-align: baseline; color: inherit; box-sizing: border-box; display: flex;"><div class="ant-tabs-tab ant-tabs-tab-active" style="border: 0px; font: inherit; margin: 0px 0px 0px 20px; padding: 12px 10px; vertical-align: baseline; color: inherit; box-sizing: border-box; align-items: center; background: transparent; display: inline-flex; outline: none;"><span class="ant-tabs-tab-btn ContentPasted2" tabindex="0" style="border: 0px; font: inherit; margin: 0px; padding: 0px; vertical-align: baseline; color: rgb(0, 85, 255); box-sizing: border-box; outline: none;">Standard Instruction Set</span></div><div class="ant-tabs-tab" style="border: 0px; font: inherit; margin: 0px 0px 0px 32px; padding: 12px 10px; vertical-align: baseline; color: inherit; box-sizing: border-box; align-items: center; background: transparent; display: inline-flex; outline: none;"><div class="ant-tabs-tab-btn ContentPasted2" tabindex="0" style="border: 0px; font: inherit; margin: 0px; padding: 0px; vertical-align: baseline; color: inherit; box-sizing: border-box; outline: none;"></div></div><div class="ant-tabs-ink-bar" style="border: 0px; font: inherit; margin: 0px; padding: 0px; vertical-align: baseline; color: inherit; box-sizing: border-box; background: rgb(0, 85, 255); height: 2px; bottom: 0px; left: 20px;"></div></div></div><div class="ant-tabs-nav-operations ant-tabs-nav-operations-hidden" style="border: 0px; font: inherit; margin: 0px; padding: 0px; vertical-align: baseline; color: inherit; box-sizing: border-box; align-self: stretch; display: flex; visibility: hidden;"><button type="button" class="ant-tabs-nav-more" tabindex="-1" style="border: 0px; font: inherit; margin: 0px; padding: 8px 16px; vertical-align: baseline; color: inherit; box-sizing: border-box; background: transparent; order: 1;"><span class="anticon anticon-ellipsis" style="border: 0px; font-style: inherit; font-variant: inherit; font-weight: inherit; font-stretch: inherit; font-size: inherit; line-height: 0; font-family: inherit; font-optical-sizing: inherit; font-kerning: inherit; font-feature-settings: inherit; font-variation-settings: inherit; margin: 0px; padding: 0px; vertical-align: -0.125em; color: inherit; box-sizing: border-box; display: inline-block; text-align: center;"><span data-icon="ellipsis" width="1em" height="1em" style="border: 0px; font: inherit; margin: 0px; padding: 0px; vertical-align: baseline; color: inherit;"><span style="border: 0px; font: inherit; margin: 0px; padding: 0px; vertical-align: baseline; color: inherit;"></span></span></span></button></div></div><div class="ant-tabs-content-holder" style="border: 0px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-variant-numeric: inherit; font-variant-east-asian: inherit; font-variant-alternates: inherit; font-weight: 400; font-stretch: inherit; font-size: 14px; line-height: inherit; font-family: Calibri, Arial, Helvetica, sans-serif; font-optical-sizing: inherit; font-kerning: inherit; font-feature-settings: inherit; font-variation-settings: inherit; margin: 0px; padding: 0px; vertical-align: baseline; color: rgb(39, 46, 59); letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal; text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial; box-sizing: border-box; flex: 1 1 auto; min-height: 0px; min-width: 0px; background-color: rgb(255, 255, 255);"><div class="ant-tabs-content ant-tabs-content-top" style="border: 0px; font: inherit; margin: 0px; padding: 0px; vertical-align: baseline; color: inherit; box-sizing: border-box; display: flex; flex: 1 1 0%; overflow: auto;"><div tabindex="0" class="ant-tabs-tabpane ant-tabs-tabpane-active" style="border: 0px; font: inherit; margin: 0px; padding: 16px 0px 16px 16px; vertical-align: baseline; color: inherit; box-sizing: border-box; flex: 0 0 auto; outline: none;"><div class="ant-table-wrapper ProTable_m-pro-table__Y-PGl " style="border: 0px; font: inherit; margin: 0px; padding: 0px; vertical-align: baseline; color: inherit; box-sizing: border-box; clear: both; max-width: 100%;"><div class="ant-spin-nested-loading" style="border: 0px; font: inherit; margin: 0px; padding: 0px; vertical-align: baseline; color: inherit; box-sizing: border-box;"><div class="ant-spin-container" style="border: 0px; font: inherit; margin: 0px; padding: 0px; vertical-align: baseline; color: inherit; box-sizing: border-box;"><div class="ant-table" style="border: 0px; font-style: inherit; font-variant: tabular-nums; font-weight: inherit; fon [... cut ...]
countdown | Enum | {   "range": [     "cancel",     "1",     "2",     "3",     "4"   ] }
countdown_left | Integer | {   "unit": "s",   "min": 0,   "max": 86400,   "scale": 0,   "step": 1 }
fault | Bitmap | {   "label": [     "motor_fault"   ] }
time_total | Integer | {   "unit": "ms",   "min": 0,   "max": 120000,   "scale": 0,   "step": 1 }
battery_percentage | Integer | {   "unit": "%",   "min": 0,   "max": 100,   "scale": 0,   "step": 1 }

</div></div></div></div></div></div></div></div></div>
</body>
</html>

helgek, 2023-08-04:
Please reopen this issue.

helgek, 2023-08-18:
@Koenkk please kindly advise how I should proceed. Thank you.

Koenkk, 2023-09-07:
Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/79780d4380c01fff06c93769e99e2e2c

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

helgek, 2023-09-08:
@Koenkk Thank you for your follow-up and reopening the issue.

With your ext_converter.js configuration the device is now being discovered as 'supported'. When I press buttons on the device there's no additional log entry in debug mode. Also when I adjust controls in Z2M the motor doesn't react.

The full log looks like this when the device has completed pairing with Z2M:

```
info 2023-09-08 02:43:46MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x540f57fffebc02d3"},"type":"device_connected"}'
info 2023-09-08 02:43:46MQTT publish: topic 'homeassistant/cover/0x540f57fffebc02d3/cover/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"command_topic":"zigbee2mqtt/0x540f57fffebc02d3/set","device":{"identifiers":["zigbee2mqtt_0x540f57fffebc02d3"],"manufacturer":"TuYa","model":"Cover motor (TS0601_cover_7)","name":"0x540f57fffebc02d3"},"json_attributes_topic":"zigbee2mqtt/0x540f57fffebc02d3","name":"0x540f57fffebc02d3","position_template":"{{ value_json.position }}","position_topic":"zigbee2mqtt/0x540f57fffebc02d3","set_position_template":"{ \"position\": {{ position }} }","set_position_topic":"zigbee2mqtt/0x540f57fffebc02d3/set","state_closed":"CLOSE","state_open":"OPEN","state_stopped":"STOP","state_topic":"zigbee2mqtt/0x540f57fffebc02d3","unique_id":"0x540f57fffebc02d3_cover_zigbee2mqtt","value_template":"{{ value_json.state }}"}'
info 2023-09-08 02:43:46MQTT publish: topic 'homeassistant/sensor/0x540f57fffebc02d3/battery/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x540f57fffebc02d3"],"manufacturer":"TuYa","model":"Cover motor (TS0601_cover_7)","name":"0x540f57fffebc02d3"},"device_class":"battery","enabled_by_default":true,"entity_category":"diagnostic","json_attributes_topic":"zigbee2mqtt/0x540f57fffebc02d3","name":"0x540f57fffebc02d3 battery","state_class":"measurement","state_topic":"zigbee2mqtt/0x540f57fffebc02d3","unique_id":"0x540f57fffebc02d3_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
info 2023-09-08 02:43:46MQTT publish: topic 'homeassistant/sensor/0x540f57fffebc02d3/motor_direction/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x540f57fffebc02d3"],"manufacturer":"TuYa","model":"Cover motor (TS0601_cover_7)","name":"0x540f57fffebc02d3"},"enabled_by_default":false,"json_attributes_topic":"zigbee2mqtt/0x540f57fffebc02d3","name":"0x540f57fffebc02d3 motor direction","state_topic":"zigbee2mqtt/0x540f57fffebc02d3","unique_id":"0x540f57fffebc02d3_motor_direction_zigbee2mqtt","value_template":"{{ value_json.motor_direction }}"}'
info 2023-09-08 02:43:46MQTT publish: topic 'homeassistant/select/0x540f57fffebc02d3/motor_direction/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"command_topic":"zigbee2mqtt/0x540f57fffebc02d3/set/motor_direction","device":{"identifiers":["zigbee2mqtt_0x540f57fffebc02d3"],"manufacturer":"TuYa","model":"Cover motor (TS0601_cover_7)","name":"0x540f57fffebc02d3"},"json_attributes_topic":"zigbee2mqtt/0x540f57fffebc02d3","name":"0x540f57fffebc02d3 motor direction","options":["normal","reversed"],"state_topic":"zigbee2mqtt/0x540f57fffebc02d3","unique_id":"0x540f57fffebc02d3_motor_direction_zigbee2mqtt","value_template":"{{ value_json.motor_direction }}"}'
info 2023-09-08 02:43:46MQTT publish: topic 'homeassistant/sensor/0x540f57fffebc02d3/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x540f57fffebc02d3"],"manufacturer":"TuYa","model":"Cover motor (TS0601_cover_7)","name":"0x540f57fffebc02d3"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/0x540f57fffebc02d3","name":"0x540f57fffebc02d3 linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/0x540f57fffebc02d3","unique_id":"0x540f57fffebc02d3_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
info 2023-09-08 02:43:46Starting interview of '0x540f57fffebc02d3'
info 2023-09-08 02:43:46MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x540f57fffebc02d3","ieee_address":"0x540f57fffebc02d3","status":"started"},"type":"device_interview"}'
info 2023-09-08 02:43:46MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x540f57fffebc02d3"},"type":"pairing"}'
debug 2023-09-08 02:43:46Received MQTT message on 'homeassistant/cover/0x540f57fffebc02d3/cover/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"command_topic":"zigbee2mqtt/0x540f57fffebc02d3/set","device":{"identifiers":["zigbee2mqtt_0x540f57fffebc02d3"],"manufacturer":"TuYa","model":"Cover motor (TS0601_cover_7)","name":"0x540f57fffebc02d3"},"json_attributes_topic":"zigbee2mqtt/0x540f57fffebc02d3","name":"0x540f57fffebc02d3","position_template":"{{ value_json.position }}","position_topic":"zigbee2mqtt/0x540f57fffebc02d3","set_position_template":"{ \"position\": {{ position }} }","set_position_topic":"zigbee2mqtt/0x540f57fffebc02d3/set","state_closed":"CLOSE","state_open":"OPEN","state_stopped":"STOP","state_topic":"zigbee2mqtt/0x540f57fffebc02d3","unique_id":"0x540f57fffebc02d3_cover_zigbee2mqtt","value_template":"{{ value_json.state }}"}'
[... cut ...]
debug 2023-09-08 02:43:51Received Zigbee message from '0x540f57fffebc02d3', type 'attributeReport', cluster 'genBasic', data '{"65503":"�F�,\u0012�F�,\u0012"}' from endpoint 1 with groupID null
debug 2023-09-08 02:43:51Received Zigbee message from '0x540f57fffebc02d3', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID null
debug 2023-09-08 02:43:56Received Zigbee message from '0x540f57fffebc02d3', type 'commandMcuSyncTime', cluster 'manuSpecificTuya', data '{"payloadSize":3328}' from endpoint 1 with groupID null
```

Koenkk, 2023-09-08:
I've updated https://gist.github.com/Koenkk/79780d4380c01fff06c93769e99e2e2c, can you try again after re-pairing the device?

helgek, 2023-09-08:
Thank you Koen. I think I made a mistake and your first version could have maybe worked too. Actually the device only reacts and sends commands after setting upper and lower limit on the device itself. This is also stated in the manual. Now the device works well it seems. Thank you again!

The only thing that doesn't work is 'motor_direction'. Switching it doesn't have any effect. Example debug log when I change 'motor_direction' and run 'open' command:

```
Debug 2023-09-08 20:13:57Received MQTT message on 'zigbee2mqtt/0x540f57fffebc02d3/set' with data '{"motor_direction":"reversed"}'
Debug 2023-09-08 20:13:57Publishing 'set' 'motor_direction' to '0x540f57fffebc02d3'
Info 2023-09-08 20:13:57MQTT publish: topic 'zigbee2mqtt/0x540f57fffebc02d3', payload '{"battery":7,"linkquality":255,"motor_direction":"reversed","position":0,"state":"STOP"}'
Debug 2023-09-08 20:14:09Received MQTT message on 'zigbee2mqtt/0x540f57fffebc02d3/set' with data '{"state":"OPEN"}'
Debug 2023-09-08 20:14:09Publishing 'set' 'state' to '0x540f57fffebc02d3'
Info 2023-09-08 20:14:09MQTT publish: topic 'zigbee2mqtt/0x540f57fffebc02d3', payload '{"battery":7,"linkquality":255,"motor_direction":"reversed","position":0,"state":"OPEN"}'
Debug 2023-09-08 20:14:09Received Zigbee message from '0x540f57fffebc02d3', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":1}],"seq":39424}' from endpoint 1 with groupID null
Info 2023-09-08 20:14:09MQTT publish: topic 'zigbee2mqtt/0x540f57fffebc02d3', payload '{"battery":7,"linkquality":255,"motor_direction":"reversed","position":0,"state":"OPEN"}'
Debug 2023-09-08 20:14:23Received MQTT message on 'zigbee2mqtt/0x540f57fffebc02d3/set' with data '{"motor_direction":"normal"}'
Debug 2023-09-08 20:14:23Publishing 'set' 'motor_direction' to '0x540f57fffebc02d3'
Info 2023-09-08 20:14:23MQTT publish: topic 'zigbee2mqtt/0x540f57fffebc02d3', payload '{"battery":7,"linkquality":255,"motor_direction":"normal","position":0,"state":"OPEN"}'
Debug 2023-09-08 20:14:26Received MQTT message on 'zigbee2mqtt/0x540f57fffebc02d3/set' with data '{"state":"OPEN"}'
Debug 2023-09-08 20:14:26Publishing 'set' 'state' to '0x540f57fffebc02d3'
Info 2023-09-08 20:14:26MQTT publish: topic 'zigbee2mqtt/0x540f57fffebc02d3', payload '{"battery":7,"linkquality":255,"motor_direction":"normal","position":0,"state":"OPEN"}'
Debug 2023-09-08 20:14:27Received Zigbee message from '0x540f57fffebc02d3', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":1}],"seq":39680}' from endpoint 1 with groupID null
Info 2023-09-08 20:14:27MQTT publish: topic 'zigbee2mqtt/0x540f57fffebc02d3', payload '{"battery":7,"linkquality":255,"motor_direction":"normal","position":0,"state":"OPEN"}'
```

Your adjusted ext_converter.js brought an improvement with regards to the "reconfigure" command. This command led to an error message with the previous version. Now the log for this command is:

```
Debug 2023-09-08 19:41:17Received MQTT message on 'zigbee2mqtt/bridge/request/device/configure' with data '{"id":"0x540f57fffebc02d3","transaction":"i8p66-4"}'
Info 2023-09-08 19:41:17Configuring '0x540f57fffebc02d3'
Debug 2023-09-08 19:41:18Received Zigbee message from '0x540f57fffebc02d3', type 'readResponse', cluster 'genBasic', data '{"65534":0,"appVersion":72,"manufacturerName":"_TZE200_zvo63cmo","modelId":"TS0601","powerSource":3,"zclVersion":3}' from endpoint 1 with groupID null
Info 2023-09-08 19:41:18Successfully configured '0x540f57fffebc02d3'
Info 2023-09-08 19:41:18MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"0x540f57fffebc02d3"},"status":"ok","transaction":"i8p66-4"}'
```

For your updated ext_converter.js the debug log for pairing the device looks like this:

```
Info 2023-09-08 19:37:43Device '0x540f57fffebc02d3' joined
Info 2023-09-08 19:37:44MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x540f57fffebc02d3","ieee_address":"0x540f57fffebc02d3"},"type":"device_joined"}'
Info 2023-09-08 19:37:44MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x540f57fffebc02d3"},"type":"device_connected"}'
Info 2023-09-08 19:37:44MQTT publish: topic 'homeassistant/cover/0x540f57fffebc02d3/cover/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"command_topic":"zigbee2mqtt/0x540f57fffebc02d3/set","device":{"identifiers":["zigbee2mqtt_0x540f57fffebc02d3"],"manufacturer":"TuYa","model":"Cover motor (TS0601_cover_7)","name":"0x540f57fffebc02d3"},"json_attributes_topic":"zigbee2mqtt/0x540f57fffebc02d3","name":"0x540f57fffebc02d3","position_template":"{{ value_json.position }}","position_topic":"zigbee2mqtt/0x540f57fffebc02d3","set_position_template":"{ \"position\": {{ position }} }","set_position_topic":"zigbee2mqtt/0x540f57fffebc02d3/set","state_closed":"CLOSE","state_open":"OPEN","state_stopped":"STOP","state_topic":"zigbee2mqtt/0x540f57fffebc02d3","unique_id":"0x540f57fffebc02d3_cover_zigbee2mqtt","value_template":"{{ value_json.state }}"}'
Info 2023-09-08 19:37:44MQTT publish: topic 'homeassistant/sensor/0x540f57fffebc02d3/battery/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x540f57fffebc02d3"],"manufacturer":"TuYa","model":"Cover motor (TS0601_cover_7)","name":"0x540f57fffebc02d3"},"device_class":"battery","e [... cut ...]
Debug 2023-09-08 19:37:49Received Zigbee message from '0x540f57fffebc02d3', type 'attributeReport', cluster 'genBasic', data '{"65503":"�4�,\u0012�4�,\u0012"}' from endpoint 1 with groupID null
Debug 2023-09-08 19:37:49Received Zigbee message from '0x540f57fffebc02d3', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID null
Debug 2023-09-08 19:37:53Received Zigbee message from '0x540f57fffebc02d3', type 'commandMcuSyncTime', cluster 'manuSpecificTuya', data '{"payloadSize":5120}' from endpoint 1 with groupID null
```
Updated screenshot:
[image]

Koenkk, 2023-09-09:
Great! I'm not sure why the motor direction doesn't work, my ext converter matches the datapoints and is also the same as how it works for other TuYa devices. To debug this testing/sniffing is needed with the original Tuya gateway is needed (https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html). But I'm also fine leaving this out and adding out-of-the-box support.

helgek, 2023-09-11:
@Koenkk For me it would be good enough to skip the motor direction for now. I've checked the tutorial but it looks like a bit more effort. Or is there maybe another trial & error way to test through a couple of parameters which could work too? Also I could contact the manufacturer. I'd just need to know how I should formulate the technical question in the correct way.

Koenkk, 2023-09-11:
Added the device! You can remove the ext converter after updating to the next release.

I have no good experience with manufacturers, the only way to release know this is by sniffing.

closed 2023-09-13 by Koenkk