Koenkk/zigbee2mqtt #18104: [Feature request]: Integration of Solum E-INK displays / dmitry firmware

slimline33, 2023-06-24:
### Is your feature request related to a problem? Please describe

Integration of Solum E-INK displays (electronic price tags)

It would be great if the EPS displays could be integrated into the zigbee system.

https://github.com/jjwbruijn/OpenEPaperLink

[image]
[image]
[image]
[image]

### Describe the solution you'd like

Addressing via zigbee mqtt with templates for different values

### Describe alternatives you've considered

Nothing

### Additional context

Nothing

closed 2023-08-01 by github-actions[bot]

timoknapp, 2024-04-08:
@slimline33 did you make zigbee2mqtt work with those price tags? I have also a couple of them and would be interested too.

@Koenkk ever played around with those tags?

slimline33, 2024-04-08:
@timoknapp No, the displays transmit via Zigbee, but use a different protocol. An extra Zigbee coordinator is absolutely necessary!

timoknapp, 2024-04-08:
@slimline33 thanks for the details. I do already have a dedicated AP for the price tags but I am currently unhappy with the network coverage of the price tags within my home. I was therefore wondering if I could leverage the existing smart home ZigBee network to extend the ZigBee network of my price tags. Do you have any idea on improving signal quality?

slimline33, 2024-04-08:
@timoknapp

What kind of access point are you using? You can use several displays with one esp32 s3, then they will mesh the display network! Why don't you join the oepl discord group, they will be happy to help!