Koenkk/zigbee2mqtt #18105: Expose Temperature for SmartThings STSS-MULT-001

ridizy, 2023-06-24:
### Link

https://venustitan-guidance.readthedocs.io/en/latest/_static/src/door_sensor/smartsense_multi_doorWindow_sensor.html

### Database entry

{"id":7,"type":"EndDevice","ieeeAddr":"0xd052a80038e40003","nwkAddr":54046,"manufId":4362,"manufName":"SmartThings","powerSource":"Unknown","modelId":"PGC313","epList":[1,2],"endpoints":{"1":{"profId":260,"epId":1,"devId":313,"inClusterList":[0],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"powerSource":0,"zclVersion":1,"appVersion":4,"stackVersion":71,"hwVersion":6,"dateCode":"130401"}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":64513,"epId":2,"devId":313,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":4,"stackVersion":71,"hwVersion":6,"dateCode":"130401","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1687589396495,"defaultSendRequestWhen":"immediate"}

### Comments

I was using this sensor previously with SmartThings, where it supported temperature.  Would it be possible to expose temperature through Zigbee2MQTT?  Thank you.

Koenkk, 2023-06-24:
Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/2dfdb942054054d06a884c317270a21c

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed, if it doesn't work, try to re-pair the device.

ridizy, 2023-06-25:
Thank you for the quick response.  Unfortunately, this did not work, even after re-pairing.

I now see `temperature` as exposed, but the value remains as `N/A`

Koenkk, 2023-06-25:
I updated https://gist.github.com/Koenkk/2dfdb942054054d06a884c317270a21c , can you try again?

Make sure the configure succeeds, you can trigger it manually by pressing the yellow refresh button in the z2m frontend, if it fails wakeup the device (by triggering) right before pressing the configure button.

ridizy, 2023-06-25:
Thank you but still no luck.  I tried reconfiguring while being sure the keep the device awake.  I also tried re-pairing.  

I should have also noted earlier that the battery percentage is also not being exposed.  Not as critical for me for this application, but it would be helpful if that was also available.  Thanks again.

Koenkk, 2023-06-25:
Did you get a message that the configure succeed or an error?

ridizy, 2023-06-25:
I tried several times.  I only received an error message once:

```
Failed to configure
(ConfigureReporting
0x052a8003840003/2
msTemperatureMeasurement (["
{"sendWhen" "immediate" "timed
failed (Timeout - 43479 - 2 - 3
- 1026 - 7 after 10000ms))
```

Koenkk, 2023-06-26:
Did you already try to re-pair the device close to the coordinator? (and see if configure works after that when waking up the device by triggering it)

Koenkk, 2023-06-26:
```
No converter available for 'STSS-MULT-001' with cluster 'manuSpecificPhilips2' and type 'raw' and data '{"data":[29,10,17,87,5,71,0,6,0,255,3],"type":"Buffer"}'
```

It seems they use some proprietary format for this, but I have no clue what the format is.

ridizy, 2023-06-26:
I believe this was the SmartThings Device Type Handler used with it: https://github.com/SmartThingsCommunity/SmartThingsPublic/blob/78747fa5a3b42a0c801f1095f8f7521da968ba42/devicetypes/smartthings/smartsense-multi-sensor.src/smartsense-multi-sensor.groovy

Not sure if that's of any help.

closed 2023-08-04 by github-actions[bot]