Koenkk/zigbee2mqtt #18111: [New device support]: Adeo ZBEK-22 Floodlight Lamp 21W

samuele2723, 2023-06-24:
### Link

https://www.leroymerlin.it/prodotti/illuminazione/illuminazione-per-esterno/luci-per-giardino/faro-da-esterno-enki-in-alluminio-nero-modulo-led-21w-ip65-inspire-89844722.html

### Database entry

{"id":36,"type":"Router","ieeeAddr":"0x943469fffe9e4d3c","nwkAddr":10108,"manufId":4727,"manufName":"Adeo","powerSource":"Mains (single phase)","modelId":"ZBEK-22","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,4096,64599,64642],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"zclVersion":3}},"genOnOff":{"attributes":{"onOff":1}},"lightingColorCtrl":{"attributes":{"colorTemperature":370,"colorTempPhysicalMin":153,"colorTempPhysicalMax":370,"options":0}},"genLevelCtrl":{"attributes":{"currentLevel":87,"options":0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":2,"hwVersion":1,"dateCode":"20220720","swBuildId":"2.12.01","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1687644347630,"defaultSendRequestWhen":"immediate"}

### Comments

RGBCCT Lamp from "enki"

on the box is coded "Yonker"
[image]

### Supported color modes

color temperature, rgb

### Color temperature range

{"colorTempPhysicalMax":370,"colorTempPhysicalMin":153}

Koenkk, 2023-06-25:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-06-25 by Koenkk

samuele2723, 2023-06-25:
Awesome! You are fantastic.

when will the updtate be released on hassio components release?

Koenkk, 2023-06-25:
Next release is on 1 July

samuele2723, 2023-07-04:
Hello @Koenkk thanks for the integration, i updated the Z2m and finally works!

only note is that on the box they were supposed to be also RGB but seems only dimming and CCT feature is available. 

Any chance there is a mistake on that?

samuele2723, 2023-07-04:
[image]

samuele2723, 2023-07-04:
[image]

Koenkk, 2023-07-04:
Added the colors.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)