Koenkk/zigbee2mqtt #18113: Unable to start Zigbee2Mqtt after update

bipsendk, 2023-06-25:
### What happened?

Upgraded Zigbee2Mqtt to 1.31.2 on HomeAssistant.
Now Z2M cannot start, and log reveals:

`[08:54:16] INFO: Preparing to start...
[08:54:16] INFO: Socat not enabled
[08:54:17] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:info  2023-06-25 08:54:21: Logging to console and directory: '/config/zigbee2mqtt/log/2023-06-25.08-54-21' filename: log.txt
Zigbee2MQTT:info  2023-06-25 08:54:21: Starting Zigbee2MQTT version 1.31.2 (commit #unknown)
Zigbee2MQTT:info  2023-06-25 08:54:21: Starting zigbee-herdsman (0.14.117)
Zigbee2MQTT:error 2023-06-25 08:54:21: Configuration is not consistent with adapter state/backup!
Zigbee2MQTT:error 2023-06-25 08:54:21: - PAN ID: configured=6754, adapter=6754
Zigbee2MQTT:error 2023-06-25 08:54:21: - Extended PAN ID: configured=00124b0026e01c10, adapter=00124b0026e01c10
Zigbee2MQTT:error 2023-06-25 08:54:21: - Network Key: configured=01030507090b0d0f00020406080a0c0d, adapter=01030507090b0d0f00020406080a0c0d
Zigbee2MQTT:error 2023-06-25 08:54:21: - Channel List: configured=20, adapter=11
Zigbee2MQTT:error 2023-06-25 08:54:21: Please update configuration to prevent further issues.
Zigbee2MQTT:error 2023-06-25 08:54:21: If you wish to re-commission your network, please remove coordinator backup at /config/zigbee2mqtt/coordinator_backup.json.
Zigbee2MQTT:error 2023-06-25 08:54:21: Re-commissioning your network will require re-pairing of all devices!
Zigbee2MQTT:error 2023-06-25 08:54:21: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2023-06-25 08:54:21: Failed to start zigbee
Zigbee2MQTT:error 2023-06-25 08:54:21: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2023-06-25 08:54:21: Exiting...
Zigbee2MQTT:error 2023-06-25 08:54:21: Error: startup failed - configuration-adapter mismatch - see logs above for more information
    at ZnpAdapterManager.determineStrategy (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:229:35)
    at ZnpAdapterManager.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:63:26)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:58:27)
    at Controller.start (/app/lib/controller.ts:101:27)
    at start (/app/index.js:107:5)
`

### What did you expect to happen?

Z2M would start as usual after upgrade

### How to reproduce it (minimal and precise)

Config:

data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
mqtt:
  server: mqtt://core-mosquitto:1883
  user: mqtt-user
  password: secret
serial:
  port: >-
    /dev/serial/by-id/usb-Silicon_Labs_Sonoff_Zigbee_3.0_USB_Dongle_Plus_0001-if00-port0

### Zigbee2MQTT version

1.31.2

### Adapter firmware version

20210708

### Adapter

Sonoff P 

### Debug log

info  2023-06-25 09:00:30: Logging to console and directory: '/config/zigbee2mqtt/log/2023-06-25.09-00-30' filename: log.txt
info  2023-06-25 09:00:30: Starting Zigbee2MQTT version 1.31.2 (commit #unknown)
info  2023-06-25 09:00:30: Starting zigbee-herdsman (0.14.117)
error 2023-06-25 09:00:30: Configuration is not consistent with adapter state/backup!
error 2023-06-25 09:00:30: - PAN ID: configured=6754, adapter=6754
error 2023-06-25 09:00:30: - Extended PAN ID: configured=00124b0026e01c10, adapter=00124b0026e01c10
error 2023-06-25 09:00:30: - Network Key: configured=01030507090b0d0f00020406080a0c0d, adapter=01030507090b0d0f00020406080a0c0d
error 2023-06-25 09:00:30: - Channel List: configured=20, adapter=11
error 2023-06-25 09:00:30: Please update configuration to prevent further issues.
error 2023-06-25 09:00:30: If you wish to re-commission your network, please remove coordinator backup at /config/zigbee2mqtt/coordinator_backup.json.
error 2023-06-25 09:00:30: Re-commissioning your network will require re-pairing of all devices!
error 2023-06-25 09:00:30: Error while starting zigbee-herdsman
error 2023-06-25 09:00:30: Failed to start zigbee
error 2023-06-25 09:00:30: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
error 2023-06-25 09:00:30: Exiting...
error 2023-06-25 09:00:30: Error: startup failed - configuration-adapter mismatch - see logs above for more information
    at ZnpAdapterManager.determineStrategy (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:229:35)
    at ZnpAdapterManager.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:63:26)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:58:27)
    at Controller.start (/app/lib/controller.ts:101:27)
    at start (/app/index.js:107:5)

DaN66O, 2023-06-25:
> Error: startup failed - configuration-adapter mismatch

https://community.home-assistant.io/t/no-dev-serial-by-id-entries-after-latest-update/562195

In short, you need to use /dev/serial/by-path instead of /dev/serial/by-id

bipsendk, 2023-06-25:
Changing configuration use use /dev/ttyUSB0 instead does not change anything.

bipsendk, 2023-06-25:
Removed the backup json file, and will start from scratch... At this point only a few devices were attached...

closed 2023-06-25 by bipsendk