Koenkk/zigbee2mqtt #18119: Aurora AOne model AU-A1ZBPIAB (uk smart plug, partially working, incorrectly identified)

jimmer74, 2023-06-26:
### What happened?

AU-A1ZBPIAB pairs correctly, but is identified SingleSocket50AU (which blakadder associates with AU-A1ZBPIA) as rather than SmartPlug51AU (as per the blakadder info page for AU-A1ZBPIAB). The picture of the pair product in z2m is also that of the AU-A1ZBPIA/SingleSocket50AU rather than that for AU-A1ZBPIAB/SmartPlug51AU.

Further, current and voltage information is reported as null. If I go into dev console and prepare a packet for rmsVoltage and rmsCurrent, I can get these values back from device, so I am wondering if it's requesting/misinterpreting the wrong voltage/current information when it reads device.

### What did you expect to happen?

The AU-A1ZBPIAB is correct identified as such, with correct product image. See below for correct image on blakadder:

[image]

Also that voltage/current is correctly read from device (without having to manually send a rmsVoltage/rmsCurrent packet. How product appears in z2m, with incorrect image and null readings:

[image]

### How to reproduce it (minimal and precise)

Pair device - it will be incorrectly detected (have reproduced across 45 of these devices at this point).

Check device status - current and voltage will be NULL.

### Zigbee2MQTT version

1.31.2 commit: 21f5125

### Adapter firmware version

20210708

### Adapter

SONOFF Universal Zigbee 3.0 dongle-P

### Debug log

Am pairing some more of these from tomorrow (if they arrive on time), will attach a debug log of the discovery/pairing/current/voltage information then.

jimmer74, 2023-06-30:
Sorry this took so long!

`debug 2023-06-29 12:29:42: Received MQTT message on 'zigbee2mqtt/0x00158d0004709566/1/set' with data '{"read":{"attributes":["rmsVoltage","rmsCurrent","activePower"],"cluster":"haElectricalMeasurement","options":{}}}'
debug 2023-06-29 12:29:42: Received Zigbee message from '0x00158d0004709566', type 'readResponse', cluster 'haElectricalMeasurement', data '{"activePower":0,"rmsCurrent":4,"rmsVoltage":239}' from endpoint 1 with groupID 0
info  2023-06-29 12:29:42: Read result of 'haElectricalMeasurement': {"rmsVoltage":239,"rmsCurrent":4,"activePower":0}`

This is me manually sending the current/voltage/active power packet to the plug and getting the correct current/voltage response. Not sure if helpful though.

Will attach the full debug log with the pairing info above

jimmer74, 2023-07-04:
Here's the section of the log pertaining to pairing one of these plugs:

> info  2023-06-29 12:28:48: Zigbee: allowing new devices to join.
info  2023-06-29 12:28:48: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"5moe1-2"}'
info  2023-06-29 12:28:49: Device '0x00158d0004709566' joined
info  2023-06-29 12:28:49: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00158d0004709566","ieee_address":"0x00158d0004709566"},"type":"device_joined"}'
info  2023-06-29 12:28:49: MQTT publish: topic 'zigbee2mqtt/0x00158d0004709566/availability', payload '{"state":"online"}'
info  2023-06-29 12:28:49: Starting interview of '0x00158d0004709566'
info  2023-06-29 12:28:49: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00158d0004709566","ieee_address":"0x00158d0004709566","status":"started"},"type":"device_interview"}'
debug 2023-06-29 12:28:49: Device '0x00158d0004709566' announced itself
info  2023-06-29 12:28:49: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00158d0004709566","ieee_address":"0x00158d0004709566"},"type":"device_announce"}'
debug 2023-06-29 12:28:50: Received Zigbee message from '0x00158d0004709566', type 'readResponse', cluster 'genBasic', data '{"modelId":"SingleSocket50AU"}' from endpoint 1 with groupID 0
debug 2023-06-29 12:28:50: Skipping message, definition is undefined and still interviewing
debug 2023-06-29 12:28:50: Received Zigbee message from '0x00158d0004709566', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"Aurora"}' from endpoint 1 with groupID 0
debug 2023-06-29 12:28:50: Skipping message, definition is undefined and still interviewing
debug 2023-06-29 12:28:50: Received Zigbee message from '0x00158d0004709566', type 'readResponse', cluster 'genBasic', data '{"powerSource":1}' from endpoint 1 with groupID 0
debug 2023-06-29 12:28:50: Skipping message, definition is undefined and still interviewing
debug 2023-06-29 12:28:50: Received Zigbee message from '0x00158d0004709566', type 'readResponse', cluster 'genBasic', data '{"zclVersion":1}' from endpoint 1 with groupID 0
debug 2023-06-29 12:28:50: Skipping message, definition is undefined and still interviewing
debug 2023-06-29 12:28:50: Received Zigbee message from '0x00158d0004709566', type 'readResponse', cluster 'genBasic', data '{"appVersion":3}' from endpoint 1 with groupID 0
debug 2023-06-29 12:28:50: Skipping message, definition is undefined and still interviewing
debug 2023-06-29 12:28:50: Received Zigbee message from '0x00158d0004709566', type 'readResponse', cluster 'genBasic', data '{"stackVersion":2}' from endpoint 1 with groupID 0
debug 2023-06-29 12:28:50: Skipping message, definition is undefined and still interviewing
debug 2023-06-29 12:28:50: Received Zigbee message from '0x00158d0004709566', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID 0
debug 2023-06-29 12:28:50: Skipping message, definition is undefined and still interviewing
debug 2023-06-29 12:28:50: Received Zigbee message from '0x00158d0004709566', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
debug 2023-06-29 12:28:50: Skipping message, definition is undefined and still interviewing
debug 2023-06-29 12:28:50: Received Zigbee message from '0x00158d0004709566', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
debug 2023-06-29 12:28:50: Skipping message, definition is undefined and still interviewing
info  2023-06-29 12:28:50: Successfully interviewed '0x00158d0004709566', device has successfully been paired
info  2023-06-29 12:28:50: Device '0x00158d0004709566' is supported, identified as: Aurora Lighting Power plug Zigbee EU (AU-A1ZBPIAB)
info  2023-06-29 12:28:50: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Power plug Zigbee EU","exposes":[{"features":[{"access":7,"description":"On/off state of the switch","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"access":1,"description":"Instantaneous measured power","name":"power","property":"power","type":"numeric","unit":"W"},{"access":1,"description":"Measured electrical potential value","name":"voltage","property":"voltage","type":"numeric","unit":"V"},{"access":1,"description":"Instantaneous measured electrical current","name":"current","property":"current","type":"numeric","unit":"A"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"AU-A1ZBPIAB","options":[{"access":2,"description":"State actions will also be published as 'action' when true (default false).","name":"state_action","property":"state_action","type":"binary","value_off":false,"value_on":true},{"access":2,"description":"Calibrates the power value (percentual offset), takes into effect on next report of device.","name":"power_calibration","property":"power_calibration","type":"numeric"},{"access":2,"description":"Number of digits after decimal point for power, takes into effect on next report of device.","name":"power_precision","property":"power_precision","type":"numeric","value_max":3,"value_min":0},{"access":2,"description":"Calibrates the current value (percen [... cut ...] :"all","device":{"identifiers":["zigbee2mqtt_0x00158d0004709566"],"manufacturer":"Aurora Lighting","model":"Power plug Zigbee EU (AU-A1ZBPIAB)","name":"0x00158d0004709566"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","name":"0x00158d0004709566 linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/0x00158d0004709566","unique_id":"0x00158d0004709566_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'

jimmer74, 2023-07-04:
Here is a regular poll of an already paired device:

> debug 2023-06-29 12:28:49: Received Zigbee message from 'Timeless Bar-7's 1', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":58}' from endpoint 1 with groupID 0
info  2023-06-29 12:28:49: MQTT publish: topic 'zigbee2mqtt/Timeless Bar-7's 1', payload '{"current":null,"linkquality":111,"power":58,"state":"ON","voltage":null}'

Note that current and voltage are returning null values. If I manually request rmsVoltage and rmsCurrent using the dev console from such a device, then subsequent automatic "reads" by z2m will show the instantaneous manually measured value (i.e. if I turn smart switch off, current will still be reported as being drawn at the value measured manually before).

The value of any current measured manually is also off by a factor of a 1000 (i.e. if switch is "off" it will record 4A, and when "on" will record in the 100's-1000's of amps).

Please let me know if there any tests I can carry out of any modifications I can try to help trouble-shoot/narrow-down these issues.

closed 2023-08-13 by github-actions[bot]

viharm, 2023-12-25:
@jimmer74 , were you able to resolve this? I have the exact same model; and it is even detected correctly as [AU-A1ZBPIAB](https://www.zigbee2mqtt.io/devices/AU-A1ZBPIAB.html#aurora%2520lighting-au-a1zbpiab), however I am still not getting the current and the voltage, exactly as per your screenshot.

DJBenson, 2025-04-04:
Just got a couple of these and have been trying with an external converter to get them to report the current and voltage values but it just doesn't work. As others have said, the values are returned when the plug is polled from the dev console (and seemingly on restart of z2m) but it will not proactively report the values.