Koenkk/zigbee2mqtt #18120: Cant change MQTT Server

Fillip38, 2023-06-26:
### What happened?

When trying to change the MQTT server via the WebUI, I get an error message

Request 'zigbee2mqtt/bridge/request/options' failed with error: 'Cannot read properties of null (reading 'length')'
Cannot read properties of null (reading 'length')

### How to reproduce it (minimal and precise)

Try to change the MQTT Server / Settings

### Zigbee2MQTT version

1.31.2

### Adapter firmware version

20220219

### Adapter

Zigbee 3.0 USB Dongle Plus, TI CC2652P

### Debug log

error 2023-06-26 13:00:11Request 'zigbee2mqtt/bridge/request/options' failed with error: 'Cannot read properties of null (reading 'length')'

info 2023-06-26 13:00:11MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{},"error":"Cannot read properties of null (reading 'length')","status":"error","transaction":"tezjz-1"}'

lux73, 2023-06-26:
try to change directly in configuration.yaml

Fillip38, 2023-06-26:
Change via configuration.yaml works!

mist0706, 2023-07-30:
This is still an issue, haven't tried changing via configuration.yaml yet

closed 2023-09-07 by github-actions[bot]

Jibbonator, 2023-10-07:
It´s Still a problem

MrSpoocy, 2023-11-12:
Same issue here.

tdn131, 2023-11-17:
I have the same problem. I can't change the MQTT server via Ingress. Zigbee2mqtt keeps using mqtt://core-mosquitto:1883. When I change in the interface (settings > MQTT) I get the same error. I even tried a clean start but it insists using mqtt://core-mosquitto:1883.

It doesn't change configuration.yaml. Even when I change the mqtt server in configuration.yaml, zigbee2mqtt uses mqtt://core-mosquitto:1883

@Koenkk I am not sure why this is closed, but it's still available. Would you mind taking a look at this. Thanks!

Erdnussflipp, 2023-11-18:
> I have the same problem. I can't change the MQTT server via Ingress. Zigbee2mqtt keeps using mqtt://core-mosquitto:1883. When I change in the interface (settings > MQTT) I get the same error as the others. I even tried a clean start but it insists using mqtt://core-mosquitto:1883.
> 
> It doesn't change configuration.yaml. Even when I change the mqtt server in configuration.yaml, zigbee2mqtt uses mqtt://core-mosquitto:1883.
> 
> @Koenkk I am not sure why this is closed, but it's still available. Would you mind taking a look at this. Thanks!
> 
> //Edit: When I change it in the configuration tab in Ingress it does accept 192.168.131.131. It looks like it checks if the server is online. I can change it to 192.168.131.131 which is the Mosquitto plugin. I can't change it to 192.168.131.131:1883. Rarely it works. I think it's easy to reproduce. If you want I can make a few screens of it.

I have exactly the same issues. Nothing works. when i change it via configuration.yaml it does take to effect. 

What is to do here?

tdn131, 2023-11-20:
Can we re-open this issue or should we open a new issue? I am not sure that it will get follow up if it's "closed". Maybe @Fillip38 can re-open?

mist0706, 2023-11-20:
Agreed, it's still an issue as i reported several months ago

Shipsik132, 2023-11-23:
Had the same issue, did not have password and user set up on a separate mosquitto running on a Raspberry pi. The config file also did not allow no user and no password but after setting the user and password this config worked. The web ui never worked. 

[image]

Stefkeys, 2023-11-27:
Is there any progress on this issue? The bot has closed this automatically by mistake
@Koenkk

boardlord1, 2024-01-02:
> Had the same issue, did not have password and user set up on a separate mosquitto running on a Raspberry pi. The config file also did not allow no user and no password but after setting the user and password this config worked. The web ui never worked.
> 
> ![image22222](https://private-user-images.githubusercontent.com/115375414/285227741-cc95dda3-7150-4604-a407-366b7fe757b9.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQxOTExMTEsIm5iZiI6MTcwNDE5MDgxMSwicGF0aCI6Ii8xMTUzNzU0MTQvMjg1MjI3NzQxLWNjOTVkZGEzLTcxNTAtNDYwNC1hNDA3LTM2NmI3ZmU3NTdiOS5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTAyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDEwMlQxMDIwMTFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT05NDM1ZWIzOTU4NGQ2ZTEyOWE3MmQ5MDQxM2VlYWI0NTNhNmQ5ODhjM2JhNjQ1M2ZmMWNkOTZjMTdlN2VlYjU3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.Knp4jIMIZTdsslA7T55ULZ0cbJAuozsjiLE2vvc3i_Y)

I also had this issue. I use the Z2M addon, and wanted to switch to EMQX. I tried to set the new MQTT server address in the Z2M UI and got the same error as the others. Setting the server address in the config YAML didn't work either, Z2M insisted on trying to connect to mosquitto.

The only way I could make Z2M use the new MQTT server was to set it via the addon config tab as per the post of @Shipsik132

arnonel, 2024-01-15:
Same issue for me. Messed everything up, so reverting to old config

Jezza34000, 2024-01-15:
i reopen new issue i have same problem
https://github.com/Koenkk/zigbee2mqtt/issues/20819

robao23, 2024-01-27:
I encountered this issue on the latest version running on a Raspberry Pi, and I was only able to resolve it as follows:

Stop zigbee2mqtt

`sudo systemctl stop zigbee2mqtt
`
Then, open the configuration file for editing:

`nano /opt/zigbee2mqtt/data/configuration.yaml

`If you face issues editing, try using root.

Edit the IP directly in the YAML file.

Save your changes and exit (usually done with keys X, Y,).

Finally, restart Zigbee2MQTT:

`sudo systemctl start zigbee2mqtt

`I understand the limitation that prevents changing it through the web interface since the service is in use, but there should be an option to make changes and trigger a restart to apply the modification.

dragonis41, 2024-02-28:
I have the same issue here on a Raspberry Pi with HAOS.
I have changed the address from configuration.conf, rebooted 5 times, but Z2M don't care and still connect to local courtier (mosquitto in my case)

wimdo, 2024-02-29:
I have the same issue. Despite changing the server in the YAML file and several reboots, the MQTT setting remains unchanged.

00107, 2024-03-02:
I have the same issue. After changing the mqtt server in the configuration.yaml I cannot access the UI. If I enter the old server and restart z2m the UI is available again.

mihsu81, 2024-05-09:
An alternative is to modify the Add-On configuration. But I do agree that a real fix is needed.

```yaml
base_topic: zigbee2mqtt
server: mqtt://192.168.45.201:1883
user: xxxxxxx
password: xxxxxxxxxxxx
include_device_information: true
```

druss67, 2024-06-23:
Still an issue, even with modification in the Add-on configuration, in Yaml or anything else...

druss67, 2024-06-23:
I succeed, here the steps:
Reset configuration from GUI
Edit configuration yaml in zigbee2mqtt folder
Shutdown RPI4
Wait 10 minutes 
Power on

Works for me