Koenkk/zigbee2mqtt #18121: [New device support]: ubisys H10/230 10way relays (heating)

stefanschaedeli, 2023-06-26:
### Link

https://www.ubisys.de/produkte/produkte-klima/produkte-klima-h10/

### Database entry

{"id":100,"type":"Router","ieeeAddr":"0x001fee0000003610","nwkAddr":8354,"manufId":4338,"manufName":"ubisys","powerSource":"Mains (single phase)","modelId":"H10","epList":[1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,232,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":769,"inClusterList":[0,3,4,5,513],"outClusterList":[513,1026,1029,1030],"clusters":{"genBasic":{"attributes":{"modelId":"H10","manufacturerName":"ubisys","powerSource":1,"zclVersion":2,"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"20190107-DE-FB0"}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b000be89cf0","endpointID":1}],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":769,"inClusterList":[0,3,4,5,513],"outClusterList":[513,1026,1029,1030],"clusters":{"genBasic":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":769,"inClusterList":[0,3,4,5,513],"outClusterList":[513,1026,1029,1030],"clusters":{"genBasic":{"attributes":{}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b000be89cf0","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b000be89cf0","endpointID":1}],"configuredReportings":[],"meta":{}},"4":{"profId":260,"epId":4,"devId":769,"inClusterList":[0,3,4,5,513],"outClusterList":[513,1026,1029,1030],"clusters":{"genBasic":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}},"5":{"profId":260,"epId":5,"devId":769,"inClusterList":[0,3,4,5,513],"outClusterList":[513,1026,1029,1030],"clusters":{"genBasic":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}},"6":{"profId":260,"epId":6,"devId":769,"inClusterList":[0,3,4,5,513],"outClusterList":[513,1026,1029,1030],"clusters":{"genBasic":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}},"7":{"profId":260,"epId":7,"devId":769,"inClusterList":[0,3,4,5,513],"outClusterList":[513,1026,1029,1030],"clusters":{"genBasic":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}},"8":{"profId":260,"epId":8,"devId":769,"inClusterList":[0,3,4,5,513],"outClusterList":[513,1026,1029,1030],"clusters":{"genBasic":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}},"9":{"profId":260,"epId":9,"devId":769,"inClusterList":[0,3,4,5,513],"outClusterList":[513,1026,1029,1030],"clusters":{"genBasic":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}},"10":{"profId":260,"epId":10,"devId":769,"inClusterList":[0,3,4,5,513],"outClusterList":[513,1026,1029,1030],"clusters":{"genBasic":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}},"11":{"profId":260,"epId":11,"devId":768,"inClusterList":[0,3,4,5,6,8],"outClusterList":[513],"clusters":{"genBasic":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}},"12":{"profId":260,"epId":12,"devId":768,"inClusterList":[0,3,4,5,6,8],"outClusterList":[513],"clusters":{"genBasic":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}},"13":{"profId":260,"epId":13,"devId":768,"inClusterList":[0,3,4,5,6,8],"outClusterList":[513],"clusters":{"genBasic":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}},"14":{"profId":260,"epId":14,"devId":768,"inClusterList":[0,3,4,5,6,8],"outClusterList":[513],"clusters":{"genBasic":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}},"15":{"profId":260,"epId":15,"devId":768,"inClusterList":[0,3,4,5,6,8],"outClusterList":[513],"clusters":{"genBasic":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}},"16":{"profId":260,"epId":16,"devId":768,"inClusterList":[0,3,4,5,6,8],"outClusterList":[513],"clusters":{"genBasic":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}},"17":{"profId":260,"epId":17,"devId":768,"inClusterList":[0,3,4,5,6,8],"outClusterList":[513],"clusters":{"genBasic":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}},"18":{"profId":260,"epId":18,"devId":768,"inClusterList":[0,3,4,5,6,8],"outClusterList":[513],"clusters":{"genBasic":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}},"19":{"profId":260,"epId":19,"devId":768,"inClusterList":[0,3,4,5,6,8],"outClusterList":[513],"clusters":{"genBasic":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}},"20":{"profId":260,"epId":20,"devId":768,"inClusterList":[0,3,4,5,6,8],"outClusterList":[513],"clusters":{"genBasic":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}},"21":{"profId":260,"epId":21,"devId":7,"inClusterList":[0,3],"outClusterList":[1026],"clusters":{"genBasic":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}},"232":{"profId":260,"epId":232,"devId":1287,"inClusterList":[0,21],"outClusterList":[3,25],"clusters":{"genBasic":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":102,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"20190107-DE-FB0","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1687779553766,"defaultSendRequestWhen":"immediate"}

### Comments

Tried to create my external converter but seem to struggle...

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['H10'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'H10', // Vendor model number, look on the device for a model number
    vendor: 'ubisys', // Vendor of the device (only used for documentation and startup logging)
    description: 'Heatingcontrol 10Way', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.on_off], // <-- added here all clusters reported from zigbee
    toZigbee: [tz.on_off], // <-- add here all clusters to send commands to zigbee
    exposes: [e.switch()], // <-- this will define which fields will be exposed in the definition message to configure a front end (e.g. the z2m frontend, Home Assistant, Domoticz)
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);
        await reporting.onOff(endpoint);
    }
};

module.exports = definition;
```

Koenkk, 2023-07-06:
Pushed a fix, let me know if it works.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

sjorge, 2023-07-06:
It's not an easy device to add support for, but ubisys has great technical documentation https://www.ubisys.de/wp-content/uploads/ubisys-h10-technical-reference.pdf

That is always a good place to start. I had a quick look over the tech reference:
- ep 1-10 are thermostat endpoints
- ep 11-20 are heater/cooler endpoints
- ep 21 is the master control endpoint

So there is a lot to configure on here, It also seems to work with an optional additional unit for pump control which is confusing me.

It looks ep 1 + 11, 2 + 12, ... work as pairs where the thermostat endpoint will use temperature data while the heater/cooler will bypass the former and allow direct control.

It's very unclear from just looking at the docs where it gets the temperature info form that is used by the thermostat endpoints. I was hoping to be able to give some pointers but this is a rather complex device that seems to require a basic HVAC installation knowledge which I unfortunately lack.

I did add support for the H1 which was a simple valve control but the H10 is a whole different beast.

stefanschaedeli, 2023-07-07:
@sjorge: at the end, its just 10 seperate relays. the device can be operated in cooling mode, then there are always two relays bundled so only one is on while the other is of. Or you run it in heating mode, then its just 10 relays that are on or off to turn on the valve.

The temperature module is optional and I think would just report but not automatically act. the native module does not not have any thermostat inputs.

The master control endpoint should be the one switching between cooling and heating option.

Happy to do the footwork on the 10 relays if you could help me to understand how to create the converter...

sjorge, 2023-07-07:
```node
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['H10'],
    model: 'H10', 
    vendor: 'Ubisys',
    description: 'Heatingcontrol 10Way', 
    meta: {thermostat: {dontMapPIHeatingDemand: true}, multiEndpoint: true},
    fromZigbee: [fz.on_off, fz.thermostat, fz.thermostat_weekly_schedule],
    toZigbee: [
      tz.on_off, tz.thermostat_occupied_heating_setpoint,
      tz.thermostat_unoccupied_heating_setpoint, tz.thermostat_local_temperature,
      tz.thermostat_system_mode, tz.thermostat_weekly_schedule,
      tz.thermostat_clear_weekly_schedule, tz.thermostat_running_mode,
      tz.thermostat_pi_heating_demand,
    ],
    // FIXME: @koenkk do I add th1-10 to z2m/utils/utils.ts 
    endpoint: (device) => {
        return {
            //'th1': 1, 'th2', 2, 'th3': 3, 'th4': 4, 'th5': 5,
            //'th6': 6, 'th7', 7, 'th8': 8, 'th9': 9, 'th10': 10,
            'l1': 11, 'l2': 12, 'l3': 13, 'l4': 14, 'l5': 15,
            'l6': 16, 'l7': 17, 'l8': 18, 'l9': 19, 'l10': 20,
            default: 21
        };
    },
    exposes: [
        e.switch().withEndpoint('l1'), e.switch().withEndpoint('l2'),
        e.switch().withEndpoint('l3'), e.switch().withEndpoint('l4'),
        e.switch().withEndpoint('l5'), e.switch().withEndpoint('l6'),
        e.switch().withEndpoint('l7'), e.switch().withEndpoint('l8'),
        e.switch().withEndpoint('l9'), e.switch().withEndpoint('l10'),
    ],
    configure: async (device, coordinatorEndpoint, logger) => {     
        // setup ep 1-10 as thermostats
        // FIXME: doesn't look like fz/tz.thermostat and friends support
        //        multiEndpoint
        /*
        const thermostatBinds = ['hvacThermostat'];
        for (let ep = 1; ep <= 10; ep++) { 
            const endpoint = device.getEndpoint(ep);
            await reporting.bind(endpoint, coordinatorEndpoint, thermostatBinds);
            await reporting.thermostatSystemMode(endpoint);
            await reporting.thermostatTemperature(endpoint,
                {min: 0, max: constants.repInterval.HOUR, change: 50});
            await reporting.thermostatOccupiedHeatingSetpoint(endpoint,
                {min: 0, max: constants.repInterval.HOUR, change: 50});
            await reporting.thermostatPIHeatingDemand(endpoint,
                {min: 15, max: constants.repInterval.HOUR, change: 1});
        }
        */

        // setup ep 11-20 as on/off switches
        const heaterCoolerBinds = ['genOnOff'];
        for (let ep = 11; ep <= 20; ep++) { 
            const endpoint = device.getEndpoint(ep);
            await reporting.bind(endpoint, coordinatorEndpoint, heaterCoolerBinds);
            await reporting.onOff(endpoint);
        }
    }
};

module.exports = definition;
```

@stefanschaedeli didn't test this as I don't have a device so I just write this erm freehand. For now it only does ep11-20 as onOff switches as that is what they are according to the tech ref. fz/tz.on_off also support multiEndpoint so  those should *knocks on wood* work.

I had to comment ep1-10 as  the fz/tz thermostat converters do not support multiEndpoint and there are also no real names available for the endpoints.

Nothing to do be done for ep 21 (master) as it only is a binding target for a thermal sensor and has genTime cluster which z2m should handle on it's own.

So let's see if that stuff at least works.

@Koenkk how do you feel about adding more endpoints to `utils/utils.ts`  for the thermostat endpoints? I used l1-l10 for the heater/cooler (genOnOff zigbee cluster wise) so that is at least inline with other 'switches' we have although it's not a perfect fit. I was thinking adding 'th1' -> 'th10' but other suggestions welcome.

stefanschaedeli, 2023-07-07:
@sjorge Thank you, I will test once I'm back from holidays and report :)

stefanschaedeli, 2023-07-20:
@sjorge I did test your code and it seems to create the 10 switches, but they are not switching the relay yet ... here the log output when i request the  first relay to switch:
`debug 2023-07-20 21:35:10Received MQTT message on 'zigbee2mqtt/0x001fee0000003610/set' with data '{"state_l1":"ON"}'
debug 2023-07-20 21:35:10Publishing 'set' 'state' to '0x001fee0000003610'
info 2023-07-20 21:35:10MQTT publish: topic 'zigbee2mqtt/0x001fee0000003610', payload '{"last_seen":"2023-07-20T21:35:10+02:00","linkquality":42,"state":"OFF","state_l1":"OFF","state_l10":"OFF","state_l2":"OFF","state_l3":"OFF","state_l4":"OFF","state_l5":"OFF","state_l6":"OFF","state_l7":"OFF","state_l8":"OFF","state_l9":"OFF"}'
info 2023-07-20 21:35:10MQTT publish: topic 'zigbee2mqtt/0x001fee0000003610', payload '{"last_seen":"2023-07-20T21:35:10+02:00","linkquality":42,"state":"OFF","state_l1":"ON","state_l10":"OFF","state_l2":"OFF","state_l3":"OFF","state_l4":"OFF","state_l5":"OFF","state_l6":"OFF","state_l7":"OFF","state_l8":"OFF","state_l9":"OFF"}'
debug 2023-07-20 21:35:10Received Zigbee message from '0x001fee0000003610', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 11 with groupID 0
info 2023-07-20 21:35:10MQTT publish: topic 'zigbee2mqtt/0x001fee0000003610', payload '{"action":"on_l1","elapsed":46900,"last_seen":"2023-07-20T21:35:10+02:00","linkquality":39,"state":"OFF","state_l1":"ON","state_l10":"OFF","state_l2":"OFF","state_l3":"OFF","state_l4":"OFF","state_l5":"OFF","state_l6":"OFF","state_l7":"OFF","state_l8":"OFF","state_l9":"OFF"}'
info 2023-07-20 21:35:10MQTT publish: topic 'zigbee2mqtt/0x001fee0000003610', payload '{"action":"","last_seen":"2023-07-20T21:35:10+02:00","linkquality":39,"state":"OFF","state_l1":"ON","state_l10":"OFF","state_l2":"OFF","state_l3":"OFF","state_l4":"OFF","state_l5":"OFF","state_l6":"OFF","state_l7":"OFF","state_l8":"OFF","state_l9":"OFF"}'
info 2023-07-20 21:35:10MQTT publish: topic 'zigbee2mqtt/0x001fee0000003610/action', payload 'on_l1'`

stefanschaedeli, 2023-07-20:
big surprise, it suddenly worked, and then resetet itself.... the device seems to toggle between the setting mode and operation mode:

debug 2023-07-20 21:40:11Publishing 'set' 'state' to '0x001fee0000003610'
info 2023-07-20 21:40:11MQTT publish: topic 'zigbee2mqtt/0x001fee0000003610', payload '{"last_seen":"2023-07-20T21:40:11+02:00","linkquality":42,"state":"OFF","state_l1":"ON","state_l10":"OFF","state_l2":"ON","state_l3":"OFF","state_l4":"OFF","state_l5":"OFF","state_l6":"OFF","state_l7":"OFF","state_l8":"OFF","state_l9":"OFF"}'
info 2023-07-20 21:40:11MQTT publish: topic 'zigbee2mqtt/0x001fee0000003610', payload '{"last_seen":"2023-07-20T21:40:11+02:00","linkquality":42,"state":"OFF","state_l1":"ON","state_l10":"OFF","state_l2":"ON","state_l3":"ON","state_l4":"OFF","state_l5":"OFF","state_l6":"OFF","state_l7":"OFF","state_l8":"OFF","state_l9":"OFF"}'
debug 2023-07-20 21:40:11Received Zigbee message from '0x001fee0000003610', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 13 with groupID 0
info 2023-07-20 21:40:11MQTT publish: topic 'zigbee2mqtt/0x001fee0000003610', payload '{"elapsed":824,"last_seen":"2023-07-20T21:40:11+02:00","linkquality":42,"state":"OFF","state_l1":"ON","state_l10":"OFF","state_l2":"ON","state_l3":"ON","state_l4":"OFF","state_l5":"OFF","state_l6":"OFF","state_l7":"OFF","state_l8":"OFF","state_l9":"OFF"}'
debug 2023-07-20 21:40:11Received MQTT message on 'zigbee2mqtt/0x001fee0000003610/set' with data '{"state_l1":"ON","state_l2":"ON","state_l3":"ON","state_l4":"ON"}'
debug 2023-07-20 21:40:11Publishing 'set' 'state' to '0x001fee0000003610'
info 2023-07-20 21:40:11MQTT publish: topic 'zigbee2mqtt/0x001fee0000003610', payload '{"last_seen":"2023-07-20T21:40:11+02:00","linkquality":42,"state":"OFF","state_l1":"ON","state_l10":"OFF","state_l2":"ON","state_l3":"ON","state_l4":"OFF","state_l5":"OFF","state_l6":"OFF","state_l7":"OFF","state_l8":"OFF","state_l9":"OFF"}'
debug 2023-07-20 21:40:11Publishing 'set' 'state' to '0x001fee0000003610'
info 2023-07-20 21:40:11MQTT publish: topic 'zigbee2mqtt/0x001fee0000003610', payload '{"last_seen":"2023-07-20T21:40:11+02:00","linkquality":42,"state":"OFF","state_l1":"ON","state_l10":"OFF","state_l2":"ON","state_l3":"ON","state_l4":"OFF","state_l5":"OFF","state_l6":"OFF","state_l7":"OFF","state_l8":"OFF","state_l9":"OFF"}'
debug 2023-07-20 21:40:11Publishing 'set' 'state' to '0x001fee0000003610'
info 2023-07-20 21:40:11MQTT publish: topic 'zigbee2mqtt/0x001fee0000003610', payload '{"last_seen":"2023-07-20T21:40:11+02:00","linkquality":42,"state":"OFF","state_l1":"ON","state_l10":"OFF","state_l2":"ON","state_l3":"ON","state_l4":"OFF","state_l5":"OFF","state_l6":"OFF","state_l7":"OFF","state_l8":"OFF","state_l9":"OFF"}'
debug 2023-07-20 21:40:11Publishing 'set' 'state' to '0x001fee0000003610'
info 2023-07-20 21:40:11MQTT publish: topic 'zigbee2mqtt/0x001fee0000003610', payload '{"last_seen":"2023-07-20T21:40:11+02:00","linkquality":42,"state":"OFF","state_l1":"ON","state_l10":"OFF","state_l2":"ON","state_l3":"ON","state_l4":"OFF","state_l5":"OFF","state_l6":"OFF","state_l7":"OFF","state_l8":"OFF","state_l9":"OFF"}'
info 2023-07-20 21:40:11MQTT publish: topic 'zigbee2mqtt/0x001fee0000003610', payload '{"last_seen":"2023-07-20T21:40:11+02:00","linkquality":42,"state":"OFF","state_l1":"ON","state_l10":"OFF","state_l2":"ON","state_l3":"ON","state_l4":"ON","state_l5":"OFF","state_l6":"OFF","state_l7":"OFF","state_l8":"OFF","state_l9":"OFF"}'
debug 2023-07-20 21:40:11Received Zigbee message from '0x001fee0000003610', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 14 with groupID 0
info 2023-07-20 21:40:11MQTT publish: topic 'zigbee2mqtt/0x001fee0000003610', payload '{"elapsed":561,"last_seen":"2023-07-20T21:40:11+02:00","linkquality":42,"state":"OFF","state_l1":"ON","state_l10":"OFF","state_l2":"ON","state_l3":"ON","state_l4":"ON","state_l5":"OFF","state_l6":"OFF","state_l7":"OFF","state_l8":"OFF","state_l9":"OFF"}'
debug 2023-07-20 21:40:12Received MQTT message on 'zigbee2mqtt/0x001fee0000003610/set' with data '{"state_l1":"ON","state_l2":"ON","state_l3":"ON","state_l4":"ON","state_l5":"ON"}'
debug 2023-07-20 21:40:12Publishing 'set' 'state' to '0x001fee0000003610'
info 2023-07-20 21:40:12MQTT publish: topic 'zigbee2mqtt/0x001fee0000003610', payload '{"last_seen":"2023-07-20T21:40:12+02:00","linkquality":42,"state":"OFF","state_l1":"ON","state_l10":"OFF","state_l2":"ON","state_l3":"ON","state_l4":"ON","state_l5":"OFF","state_l6":"OFF","state_l7":"OFF","state_l8":"OFF","state_l9":"OFF"}'
debug 2023-07-20 21:40:12Publishing 'set' 'state' to '0x001fee0000003610'
info 2023-07-20 21:40:12MQTT publish: topic 'zigbee2mqtt/0x001fee0000003610', payload '{"last_seen":"2023-07-20T21:40:12+02:00","linkquality":42,"state":"OFF","state_l1":"ON","state_l10":"OFF","state_l2":"ON","state_l3":"ON","state_l4":"ON","state_l5":"OFF","state_l6":"OFF","state_l7":"OFF","state_l8":"OFF","state_l9":"OFF"}'
debug 2023-07-20 21:40:12Publishing 'set' 'state' to '0x001fee0000003610'
debug 2023-07-20 21:40:12Received Zigbee message from 'ug_waschkuche_entfeuchter2', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":184}' from endpoint 1 with groupID 0
info 2023-07-20 21:40:12MQTT publish: topic 'zigbee2mqtt/ug_waschkuche_entfeuchter2', payload '{"child_lock":"UNLOCK","current":0,"elapsed":158425,"energy":380.8,"indicator_mode":"on/off","last_seen":"2023-07-20T21:40:12+02:00","linkquality" [... cut ...]
debug 2023-07-20 21:41:31Received Zigbee message from 'ug_waschkuche_tumbler', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,14423]}' from endpoint 1 with groupID 0
info 2023-07-20 21:41:31MQTT publish: topic 'zigbee2mqtt/ug_waschkuche_tumbler', payload '{"child_lock":"UNLOCK","current":1.87,"elapsed":59212,"energy":144.23,"indicator_mode":"off/on","last_seen":"2023-07-20T21:41:31+02:00","linkquality":18,"power":404,"power_outage_memory":"on","state":"ON","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":false,"voltage":235}'

stefanschaedeli, 2023-07-20:
https://share.icloud.com/photos/0c4_WgkAAkglJgnuDE4o9_BtA

sjorge, 2023-07-20:
The device, at least based on the attributeReports seems to be reacting correctly but only your first attempt. It looks like the frontend is somehow setting weird extra entries the more state you try and toggle?

Can you try with just sending it via mqtt (not using the frontend?)

```
# turn l1 on
mosquitto_pub -t zigbee2mqtt/0x001fee0000003610/set/state_l1 -m 'on'
# turn l2 on
mosquitto_pub -t zigbee2mqtt/0x001fee0000003610/set/state_l2 -m 'on'
# ...
# turn l1 off
mosquitto_pub -t zigbee2mqtt/0x001fee0000003610/set/state_l1 -m 'off'
...
```

And check if it gets a correct report from the device between each one, those shoud look something like:
```
debug 2023-07-20 21:35:10Received Zigbee message from '0x001fee0000003610', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 11 with groupID 0 info 2023-07-20 21:35:10MQTT publish: topic 'zigbee2mqtt/0x001fee0000003610', payload '{"action":"on_l1","elapsed":46900,"last_seen":"2023-07-20T21:35:10+02:00","linkquality":39,"state":"OFF","state_l1":"ON","state_l10":"OFF","state_l2":"OFF","state_l3":"OFF","state_l4":"OFF","state_l5":"OFF","state_l6":"OFF","state_l7":"OFF","state_l8":"OFF","state_l9":"OFF"}'
```

where endpoint XX should what whatever l<x> you turned on or off (mapping is in the converter but copied here for easy of access)
```
            'l1': 11, 'l2': 12, 'l3': 13, 'l4': 14, 'l5': 15,
            'l6': 16, 'l7': 17, 'l8': 18, 'l9': 19, 'l10': 20,
```

stefanschaedeli, 2023-07-21:
@sjorge 
Its correctly setting all swtiches I choose, but its restarting the device here and there... very strange... 

I just reintervied the device, here the full log:
`info 2023-07-21 19:46:15MQTT publish: topic 'zigbee2mqtt/0x001fee0000003610', payload '{"last_seen":"2023-07-21T19:46:15+02:00","linkquality":135,"state":"OFF","state_l1":"ON","state_l10":"OFF","state_l2":"ON","state_l3":"ON","state_l4":"ON","state_l5":"ON","state_l6":"ON","state_l7":"OFF","state_l8":"OFF","state_l9":"OFF"}'
info 2023-07-21 19:46:15MQTT publish: topic 'zigbee2mqtt/0x001fee0000003610', payload '{"last_seen":"2023-07-21T19:46:15+02:00","linkquality":135,"state":"OFF","state_l1":"ON","state_l10":"OFF","state_l2":"ON","state_l3":"ON","state_l4":"ON","state_l5":"ON","state_l6":"ON","state_l7":"OFF","state_l8":"OFF","state_l9":"OFF"}'
debug 2023-07-21 19:46:16Received Zigbee message from '0x001fee0000003610', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 11 with groupID 0
info 2023-07-21 19:46:16MQTT publish: topic 'zigbee2mqtt/0x001fee0000003610', payload '{"elapsed":34221,"last_seen":"2023-07-21T19:46:15+02:00","linkquality":132,"state":"OFF","state_l1":"OFF","state_l10":"OFF","state_l2":"ON","state_l3":"ON","state_l4":"ON","state_l5":"ON","state_l6":"ON","state_l7":"OFF","state_l8":"OFF","state_l9":"OFF"}'
debug 2023-07-21 19:46:16Received Zigbee message from '0x001fee0000003610', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 12 with groupID 0
info 2023-07-21 19:46:16MQTT publish: topic 'zigbee2mqtt/0x001fee0000003610', payload '{"elapsed":11,"last_seen":"2023-07-21T19:46:15+02:00","linkquality":135,"state":"OFF","state_l1":"OFF","state_l10":"OFF","state_l2":"OFF","state_l3":"ON","state_l4":"ON","state_l5":"ON","state_l6":"ON","state_l7":"OFF","state_l8":"OFF","state_l9":"OFF"}'
debug 2023-07-21 19:46:16Received Zigbee message from '0x001fee0000003610', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 13 with groupID 0
info 2023-07-21 19:46:16MQTT publish: topic 'zigbee2mqtt/0x001fee0000003610', payload '{"elapsed":48,"last_seen":"2023-07-21T19:46:16+02:00","linkquality":132,"state":"OFF","state_l1":"OFF","state_l10":"OFF","state_l2":"OFF","state_l3":"OFF","state_l4":"ON","state_l5":"ON","state_l6":"ON","state_l7":"OFF","state_l8":"OFF","state_l9":"OFF"}'
info 2023-07-21 19:46:16MQTT publish: topic 'zigbee2mqtt/0x001fee0000003610', payload '{"last_seen":"2023-07-21T19:46:16+02:00","linkquality":132,"state":"OFF","state_l1":"OFF","state_l10":"OFF","state_l2":"OFF","state_l3":"OFF","state_l4":"ON","state_l5":"ON","state_l6":"ON","state_l7":"OFF","state_l8":"OFF","state_l9":"OFF"}'
debug 2023-07-21 19:46:16Received Zigbee message from '0x001fee0000003610', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 15 with groupID 0
info 2023-07-21 19:46:16MQTT publish: topic 'zigbee2mqtt/0x001fee0000003610', payload '{"elapsed":293,"last_seen":"2023-07-21T19:46:16+02:00","linkquality":132,"state":"OFF","state_l1":"OFF","state_l10":"OFF","state_l2":"OFF","state_l3":"OFF","state_l4":"ON","state_l5":"OFF","state_l6":"ON","state_l7":"OFF","state_l8":"OFF","state_l9":"OFF"}'
info 2023-07-21 19:46:16MQTT publish: topic 'zigbee2mqtt/0x001fee0000003610', payload '{"last_seen":"2023-07-21T19:46:16+02:00","linkquality":129,"state":"OFF","state_l1":"OFF","state_l10":"OFF","state_l2":"OFF","state_l3":"OFF","state_l4":"ON","state_l5":"OFF","state_l6":"ON","state_l7":"OFF","state_l8":"OFF","state_l9":"OFF"}'
debug 2023-07-21 19:46:16Received Zigbee message from '0x001fee0000003610', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 14 with groupID 0
info 2023-07-21 19:46:16MQTT publish: topic 'zigbee2mqtt/0x001fee0000003610', payload '{"elapsed":106,"last_seen":"2023-07-21T19:46:16+02:00","linkquality":132,"state":"OFF","state_l1":"OFF","state_l10":"OFF","state_l2":"OFF","state_l3":"OFF","state_l4":"OFF","state_l5":"OFF","state_l6":"ON","state_l7":"OFF","state_l8":"OFF","state_l9":"OFF"}'
info 2023-07-21 19:46:16MQTT publish: topic 'zigbee2mqtt/0x001fee0000003610', payload '{"last_seen":"2023-07-21T19:46:16+02:00","linkquality":129,"state":"OFF","state_l1":"OFF","state_l10":"OFF","state_l2":"OFF","state_l3":"OFF","state_l4":"OFF","state_l5":"OFF","state_l6":"ON","state_l7":"OFF","state_l8":"OFF","state_l9":"OFF"}'
info 2023-07-21 19:46:16Successfully configured '0x001fee0000003610'
debug 2023-07-21 19:46:17Received Zigbee message from '0x001fee0000003610', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 17 with groupID 0
debug 2023-07-21 19:46:17Received Zigbee message from '0x001fee0000003610', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 16 with groupID 0
debug 2023-07-21 19:46:17Received Zigbee message from '0x001fee0000003610', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 18 with groupID 0
debug 2023-07-21 19:46:17Received Zigbee message from '0x001fee0000003610', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 19 with groupID 0
info 2023-07-21 19:46:17MQTT publish: topic 'zigbee2mqtt/0x001fee0000003610', payload '{"elapsed":1377,"last_seen":"2023-07-21T19:46:17+02:00","linkquality":132,"state":"OFF","state_l1":"OFF","state_l10":"OFF","state_l2":"OFF","state_l3":"OFF","state_l4":"OFF","state_l5":"OFF","state_l6":"ON","state_l7":"OFF","state_l8":"OFF","state_l9":"OFF"}'
[... cut ...]
info 2023-07-21 19:47:31MQTT publish: topic 'zigbee2mqtt/0x001fee0000003610', payload '{"elapsed":73030,"last_seen":"2023-07-21T19:47:31+02:00","linkquality":135,"state":"OFF","state_l1":"ON","state_l10":"OFF","state_l2":"OFF","state_l3":"OFF","state_l4":"OFF","state_l5":"OFF","state_l6":"OFF","state_l7":"OFF","state_l8":"OFF","state_l9":"OFF"}'`

what is really strange is that it actually works... the relays are toggling... but its not showing popperly on the leds and also the status is odd in home assistant...

stefanschaedeli, 2023-07-21:
slight correction, the device works in home asssistant too... it just seems to behave odd.. I have a case open with the manufacturer, maybe they can help. 
In any case... I can consider the device working for me in this case already ! :) thank you

sjorge, 2023-07-21:
> ```js
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const reporting = require('zigbee-herdsman-converters/lib/reporting');
> const extend = require('zigbee-herdsman-converters/lib/extend');
> const e = exposes.presets;
> const ea = exposes.access;
> 
> const definition = {
>     zigbeeModel: ['H10'],
>     model: 'H10', 
>     vendor: 'Ubisys',
>     description: 'Heatingcontrol 10Way', 
>     meta: {thermostat: {dontMapPIHeatingDemand: true}, multiEndpoint: true},
>     fromZigbee: [fz.on_off, fz.thermostat, fz.thermostat_weekly_schedule],
>     toZigbee: [
>       tz.on_off, tz.thermostat_occupied_heating_setpoint,
>       tz.thermostat_unoccupied_heating_setpoint, tz.thermostat_local_temperature,
>       tz.thermostat_system_mode, tz.thermostat_weekly_schedule,
>       tz.thermostat_clear_weekly_schedule, tz.thermostat_running_mode,
>       tz.thermostat_pi_heating_demand,
>     ],
>     // FIXME: @koenkk do I add th1-10 to z2m/utils/utils.ts 
>     endpoint: (device) => {
>         return {
>             //'th1': 1, 'th2', 2, 'th3': 3, 'th4': 4, 'th5': 5,
>             //'th6': 6, 'th7', 7, 'th8': 8, 'th9': 9, 'th10': 10,
>             'l1': 11, 'l2': 12, 'l3': 13, 'l4': 14, 'l5': 15,
>             'l6': 16, 'l7': 17, 'l8': 18, 'l9': 19, 'l10': 20,
>             default: 21
>         };
>     },
>     exposes: [
>         e.switch().withEndpoint('l1'), e.switch().withEndpoint('l2'),
>         e.switch().withEndpoint('l3'), e.switch().withEndpoint('l4'),
>         e.switch().withEndpoint('l5'), e.switch().withEndpoint('l6'),
>         e.switch().withEndpoint('l7'), e.switch().withEndpoint('l8'),
>         e.switch().withEndpoint('l9'), e.switch().withEndpoint('l10'),
>     ],
>     configure: async (device, coordinatorEndpoint, logger) => {     
>         // setup ep 1-10 as thermostats
>         // FIXME: doesn't look like fz/tz.thermostat and friends support
>         //        multiEndpoint
>         /*
>         const thermostatBinds = ['hvacThermostat'];
>         for (let ep = 1; ep <= 10; ep++) { 
>             const endpoint = device.getEndpoint(ep);
>             await reporting.bind(endpoint, coordinatorEndpoint, thermostatBinds);
>             await reporting.thermostatSystemMode(endpoint);
>             await reporting.thermostatTemperature(endpoint,
>                 {min: 0, max: constants.repInterval.HOUR, change: 50});
>             await reporting.thermostatOccupiedHeatingSetpoint(endpoint,
>                 {min: 0, max: constants.repInterval.HOUR, change: 50});
>             await reporting.thermostatPIHeatingDemand(endpoint,
>                 {min: 15, max: constants.repInterval.HOUR, change: 1});
>         }
>         */
> 
>         // setup ep 11-20 as on/off switches
>         const heaterCoolerBinds = ['genOnOff'];
>         for (let ep = 11; ep <= 20; ep++) { 
>             const endpoint = device.getEndpoint(ep);
>             await reporting.bind(endpoint, coordinatorEndpoint, heaterCoolerBinds);
>             await reporting.onOff(endpoint);
>         }
>     }
> };
> 
> module.exports = definition;
> ```
> 
> @stefanschaedeli didn't test this as I don't have a device so I just write this erm freehand. For now it only does ep11-20 as onOff switches as that is what they are according to the tech ref. fz/tz.on_off also support multiEndpoint so those should _knocks on wood_ work.
> 
> I had to comment ep1-10 as the fz/tz thermostat converters do not support multiEndpoint and there are also no real names available for the endpoints.
> 
> Nothing to do be done for ep 21 (master) as it only is a binding target for a thermal sensor and has genTime cluster which z2m should handle on it's own.
> 
> So let's see if that stuff at least works.
> 
> @Koenkk how do you feel about adding more endpoints to `utils/utils.ts` for the thermostat endpoints? I used l1-l10 for the heater/cooler (genOnOff zigbee cluster wise) so that is at least inline with other 'switches' we have although it's not a perfect fit. I was thinking adding 'th1' -> 'th10' but other suggestions welcome.

@Koenkk are you OK with me adding the th1 tot th10 endpoint names?

If not, I'll do a PR to get at least the basic relay part commited to zhc, since that seems to be working, minus the on device leds not updating which doesn't look like it's a z2m problem.

Koenkk, 2023-07-23:
@sjorge that's fine!

sjorge, 2023-07-23:
@stefanschaedeli once #18405 is merged, can you switch to the dev branch and try this converter? ([https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html))

```node
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['H10'],
    model: 'H10', 
    vendor: 'Ubisys',
    description: 'Heatingcontrol 10Way', 
    meta: {thermostat: {dontMapPIHeatingDemand: true}, multiEndpoint: true},
    fromZigbee: [fz.on_off, fz.thermostat, fz.thermostat_weekly_schedule],
    toZigbee: [
      tz.on_off, tz.thermostat_occupied_heating_setpoint,
      tz.thermostat_unoccupied_heating_setpoint, tz.thermostat_local_temperature,
      tz.thermostat_system_mode, tz.thermostat_weekly_schedule,
      tz.thermostat_clear_weekly_schedule, tz.thermostat_running_mode,
      tz.thermostat_pi_heating_demand,
    ],
    endpoint: (device) => {
        return {
            'th1': 1, 'th2', 2, 'th3': 3, 'th4': 4, 'th5': 5,
            'th6': 6, 'th7', 7, 'th8': 8, 'th9': 9, 'th10': 10,
            'l1': 11, 'l2': 12, 'l3': 13, 'l4': 14, 'l5': 15,
            'l6': 16, 'l7': 17, 'l8': 18, 'l9': 19, 'l10': 20,
            default: 21
        };
    },
    exposes: [
        e.switch().withEndpoint('l1'), e.switch().withEndpoint('l2'),
        e.switch().withEndpoint('l3'), e.switch().withEndpoint('l4'),
        e.switch().withEndpoint('l5'), e.switch().withEndpoint('l6'),
        e.switch().withEndpoint('l7'), e.switch().withEndpoint('l8'),
        e.switch().withEndpoint('l9'), e.switch().withEndpoint('l10'),
        e.climate().withEndpoint('th1')
            .withSystemMode(['off', 'heat'], ea.ALL).withRunningMode(['off', 'heat'])
            .withSetpoint('occupied_heating_setpoint', 7, 30, 0.5)
            .withLocalTemperature().withPiHeatingDemand(ea.STATE_GET),
        e.climate().withEndpoint('th2')
            .withSystemMode(['off', 'heat'], ea.ALL).withRunningMode(['off', 'heat'])
            .withSetpoint('occupied_heating_setpoint', 7, 30, 0.5)
            .withLocalTemperature().withPiHeatingDemand(ea.STATE_GET),
        e.climate().withEndpoint('th3')
            .withSystemMode(['off', 'heat'], ea.ALL).withRunningMode(['off', 'heat'])
            .withSetpoint('occupied_heating_setpoint', 7, 30, 0.5)
            .withLocalTemperature().withPiHeatingDemand(ea.STATE_GET),
        e.climate().withEndpoint('th4')
            .withSystemMode(['off', 'heat'], ea.ALL).withRunningMode(['off', 'heat'])
            .withSetpoint('occupied_heating_setpoint', 7, 30, 0.5)
            .withLocalTemperature().withPiHeatingDemand(ea.STATE_GET),
        e.climate().withEndpoint('th5')
            .withSystemMode(['off', 'heat'], ea.ALL).withRunningMode(['off', 'heat'])
            .withSetpoint('occupied_heating_setpoint', 7, 30, 0.5)
            .withLocalTemperature().withPiHeatingDemand(ea.STATE_GET),
        e.climate().withEndpoint('th6')
            .withSystemMode(['off', 'heat'], ea.ALL).withRunningMode(['off', 'heat'])
            .withSetpoint('occupied_heating_setpoint', 7, 30, 0.5)
            .withLocalTemperature().withPiHeatingDemand(ea.STATE_GET),
        e.climate().withEndpoint('th7')
            .withSystemMode(['off', 'heat'], ea.ALL).withRunningMode(['off', 'heat'])
            .withSetpoint('occupied_heating_setpoint', 7, 30, 0.5)
            .withLocalTemperature().withPiHeatingDemand(ea.STATE_GET),
        e.climate().withEndpoint('th8')
            .withSystemMode(['off', 'heat'], ea.ALL).withRunningMode(['off', 'heat'])
[... 17 lines cut ...]
            const endpoint = device.getEndpoint(ep);
            await reporting.bind(endpoint, coordinatorEndpoint, thermostatBinds);
            await reporting.thermostatSystemMode(endpoint);
            await reporting.thermostatTemperature(endpoint,
                {min: 0, max: constants.repInterval.HOUR, change: 50});
            await reporting.thermostatOccupiedHeatingSetpoint(endpoint,
                {min: 0, max: constants.repInterval.HOUR, change: 50});
            await reporting.thermostatPIHeatingDemand(endpoint,
                {min: 15, max: constants.repInterval.HOUR, change: 1});
        }

        // setup ep 11-20 as on/off switches
        const heaterCoolerBinds = ['genOnOff'];
        for (let ep = 11; ep <= 20; ep++) { 
            const endpoint = device.getEndpoint(ep);
            await reporting.bind(endpoint, coordinatorEndpoint, heaterCoolerBinds);
            await reporting.onOff(endpoint);
        }
    }
};

module.exports = definition;
```

The thermostat stuff might not work without more changes but this would be the first step I'd try.

stefanschaedeli, 2023-07-23:
sure :)

stefanschaedeli, 2023-07-24:
getting these erros in the logs:
`Using '/app/data' as data directory
/app/data/extension/externally-loaded.js:25
            'th1': 1, 'th2', 2, 'th3': 3, 'th4': 4, 'th5': 5,
                      ^^^^^
SyntaxError: Unexpected string
    at new Script (node:vm:94:7)
    at createScript (node:vm:251:10)
    at Object.runInNewContext (node:vm:292:10)
    at loadModuleFromText (/app/lib/util/utils.ts:151:8)
    at loadModuleFromFile (/app/lib/util/utils.ts:158:12)
    at Object.getExternalConvertersDefinitions (/app/lib/util/utils.ts:168:25)
    at getExternalConvertersDefinitions.next (<anonymous>)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:12:20)
    at new Controller (/app/lib/controller.ts:84:58)
    at start (/app/index.js:106:18)
Using '/app/data' as data directory`

stefanschaedeli, 2023-07-24:
`
Using '/app/data' as data directory
/app/data/extension/externally-loaded.js:25
            'th1': 1, 'th2', 2, 'th3': 3, 'th4': 4, 'th5': 5,
                      ^^^^^
SyntaxError: Unexpected string
    at new Script (node:vm:94:7)
    at createScript (node:vm:251:10)
    at Object.runInNewContext (node:vm:292:10)
    at loadModuleFromText (/app/lib/util/utils.ts:151:8)
    at loadModuleFromFile (/app/lib/util/utils.ts:158:12)
    at Object.getExternalConvertersDefinitions (/app/lib/util/utils.ts:168:25)
    at getExternalConvertersDefinitions.next (<anonymous>)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:12:20)
    at new Controller (/app/lib/controller.ts:84:58)
    at start (/app/index.js:106:18)
Using '/app/data' as data directory
`

stefanschaedeli, 2023-07-24:
```
function test() {
  console.log("notice the blank line before this function?");
}
```

sjorge, 2023-07-24:
I'll have a look tonight unless I get home late, probably made an error somewhere.

sjorge, 2023-07-24:
That is the one on the correct commit, so that should be fine.

sjorge, 2023-07-24:
```node
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['H10'],
    model: 'H10',
    vendor: 'Ubisys',
    description: 'Heatingcontrol 10Way',
    meta: {thermostat: {dontMapPIHeatingDemand: true}, multiEndpoint: true},
    fromZigbee: [fz.on_off, fz.thermostat, fz.thermostat_weekly_schedule],
    toZigbee: [
      tz.on_off, tz.thermostat_occupied_heating_setpoint,
      tz.thermostat_unoccupied_heating_setpoint, tz.thermostat_local_temperature,
      tz.thermostat_system_mode, tz.thermostat_weekly_schedule,
      tz.thermostat_clear_weekly_schedule, tz.thermostat_running_mode,
      tz.thermostat_pi_heating_demand,
    ],
    endpoint: (device) => {
        return {
            'th1': 1, 'th2': 2, 'th3': 3, 'th4': 4, 'th5': 5,
            'th6': 6, 'th7': 7, 'th8': 8, 'th9': 9, 'th10': 10,
            'l1': 11, 'l2': 12, 'l3': 13, 'l4': 14, 'l5': 15,
            'l6': 16, 'l7': 17, 'l8': 18, 'l9': 19, 'l10': 20,
            default: 21
        };
    },
    exposes: [
        e.switch().withEndpoint('l1'), e.switch().withEndpoint('l2'),
        e.switch().withEndpoint('l3'), e.switch().withEndpoint('l4'),
        e.switch().withEndpoint('l5'), e.switch().withEndpoint('l6'),
        e.switch().withEndpoint('l7'), e.switch().withEndpoint('l8'),
        e.switch().withEndpoint('l9'), e.switch().withEndpoint('l10'),
        e.climate()
            .withSystemMode(['off', 'heat'], ea.ALL).withRunningMode(['off', 'heat'])
            .withSetpoint('occupied_heating_setpoint', 7, 30, 0.5)
            .withLocalTemperature().withPiHeatingDemand(ea.STATE_GET)
            .withEndpoint('th1'),
        e.climate()
            .withSystemMode(['off', 'heat'], ea.ALL).withRunningMode(['off', 'heat'])
            .withSetpoint('occupied_heating_setpoint', 7, 30, 0.5)
            .withLocalTemperature().withPiHeatingDemand(ea.STATE_GET)
            .withEndpoint('th2'),
        e.climate()
            .withSystemMode(['off', 'heat'], ea.ALL).withRunningMode(['off', 'heat'])
            .withSetpoint('occupied_heating_setpoint', 7, 30, 0.5)
            .withLocalTemperature().withPiHeatingDemand(ea.STATE_GET)
            .withEndpoint('th3'),
        e.climate()
            .withSystemMode(['off', 'heat'], ea.ALL).withRunningMode(['off', 'heat'])
            .withSetpoint('occupied_heating_setpoint', 7, 30, 0.5)
            .withLocalTemperature().withPiHeatingDemand(ea.STATE_GET)
            .withEndpoint('th4'),
        e.climate()
            .withSystemMode(['off', 'heat'], ea.ALL).withRunningMode(['off', 'heat'])
            .withSetpoint('occupied_heating_setpoint', 7, 30, 0.5)
            .withLocalTemperature().withPiHeatingDemand(ea.STATE_GET)
            .withEndpoint('th5'),
        e.climate()
            .withSystemMode(['off', 'heat'], ea.ALL).withRunningMode(['off', 'heat'])
            .withSetpoint('occupied_heating_setpoint', 7, 30, 0.5)
            .withLocalTemperature().withPiHeatingDemand(ea.STATE_GET)
            .withEndpoint('th6'),
        e.climate()
            .withSystemMode(['off', 'heat'], ea.ALL).withRunningMode(['off', 'heat'])
            .withSetpoint('occupied_heating_setpoint', 7, 30, 0.5)
            .withLocalTemperature().withPiHeatingDemand(ea.STATE_GET)
            .withEndpoint('th7'),
        e.climate()
            .withSystemMode(['off', 'heat'], ea.ALL).withRunningMode(['off', 'heat'])
            .withSetpoint('occupied_heating_setpoint', 7, 30, 0.5)
            .withLocalTemperature().withPiHeatingDemand(ea.STATE_GET)
            .withEndpoint('th8'),
        e.climate()
            .withSystemMode(['off', 'heat'], ea.ALL).withRunningMode(['off', 'heat'])
            .withSetpoint('occupied_heating_setpoint', 7, 30, 0.5)
            .withLocalTemperature().withPiHeatingDemand(ea.STATE_GET)
            .withEndpoint('th9'),
        e.climate()
            .withSystemMode(['off', 'heat'], ea.ALL).withRunningMode(['off', 'heat'])
            .withSetpoint('occupied_heating_setpoint', 7, 30, 0.5)
            .withLocalTemperature().withPiHeatingDemand(ea.STATE_GET)
            .withEndpoint('th10'),
    ],
    configure: async (device, coordinatorEndpoint, logger) => {
        // setup ep 1-10 as thermostats
        // FIXME: doesn't look like fz/tz.thermostat and friends support
        //        multiEndpoint
        const thermostatBinds = ['hvacThermostat'];
        for (let ep = 1; ep <= 10; ep++) {
            const endpoint = device.getEndpoint(ep);
            await reporting.bind(endpoint, coordinatorEndpoint, thermostatBinds);
            await reporting.thermostatSystemMode(endpoint);
            await reporting.thermostatTemperature(endpoint,
                {min: 0, max: constants.repInterval.HOUR, change: 50});
            await reporting.thermostatOccupiedHeatingSetpoint(endpoint,
                {min: 0, max: constants.repInterval.HOUR, change: 50});
            await reporting.thermostatPIHeatingDemand(endpoint,
                {min: 15, max: constants.repInterval.HOUR, change: 1});
        }

        // setup ep 11-20 as on/off switches
        const heaterCoolerBinds = ['genOnOff'];
        for (let ep = 11; ep <= 20; ep++) {
            const endpoint = device.getEndpoint(ep);
            await reporting.bind(endpoint, coordinatorEndpoint, heaterCoolerBinds);
            await reporting.onOff(endpoint);
        }
    }
};

module.exports = definition;
```

stefanschaedeli, 2023-07-24:
works now 
[image]

but would there be a way to have this on or off? because the temperature sensors is additional module that can be connected through kind of a "special connector". but if you dont have this, all of these do not really exist

sjorge, 2023-07-25:
There is no easy way to turn on or off the thermostat exposes. But are they actually working properly?

I'm not sure the thermostat converters (and expose) understand the endpoint stuff.

closed 2023-09-01 by github-actions[bot]

ivanfrolovmd, 2025-01-04:
@sjorge, are there plans to include the code of H10 converter from the comments in the `master`? I have this as a custom converter, and it works fine, but I don't see this converter in the `master` [here](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/ubisys.ts). Did it just fall through the cracks?

sjorge, 2025-01-04:
> @sjorge, are there plans to include the code of H10 converter from the comments in the `master`? I have this as a custom converter, and it works fine, but I don't see this converter in the `master` [...]

I never got an answer if the version https://github.com/Koenkk/zigbee2mqtt/issues/18121#issuecomment-1648388512 actually works properly when I asked here https://github.com/Koenkk/zigbee2mqtt/issues/18121#issuecomment-1649690382

stefanschaedeli, 2025-01-04:
@sjorge Sorry for that, it does work all fine in my setup

sjorge, 2025-01-04:
I'm busy this weekend, but I'll try to get a PR out before the next release