Koenkk/zigbee2mqtt #18122: Centralite Keypas 3400-D not working

GLehnhoff, 2023-06-26:
### What happened?

Centralite Keypas 3400-D does not work and should not be in the list of supported devices.

CAUTION! 
This issue only concerns the 3400-D. It is not a 3400 or 3400-something. As far as I can remember, the D version is a special version for Germany, for Deutsche Telekom and their smarthome system "Telekom Magenta SmartHome". My 3400-D worked perfectly with Telekom Magenta SmartHome. It may have a different firmware, as e.g. 3400 (without "-D") does not work with Telekom Magenta SmartHome.

### What did you expect to happen?

### Pairing
Pairing works with option 2 described here: https://www.zigbee2mqtt.io/devices/3400-D.html.

### Show temperature
Works.

### Show battery
Works.

### Occupancy
Working. By the way, this means that all lights are flashing. This is the correct behaviour when the device does not know the status of the alarm system.

### Send code
Works. Action, action code, etc. is sent to the Home Assistant and can be easily picked up by Node-RED. However, milliseconds after the first massage, two more messages are sent, all with None or Null. So I would be able to process the code and action by picking up only the first message. 

### Device disconnects
However, after the three messages have been sent, the device seems to disconnect. At least the connected light at the bottom left is off and no more messages are being sent. It reconnects after about 5 minutes.

### Sending a message to the device
I could imagine that the device needs a message from the Home Assistant to let it know what state it is in. Unfortunately this does not work, the log says that the device is not found after sending a message via zigbee2mqtt/FRIENDLY_NAME/set.

### Arm Away button does not work
Only Disarm (left button) and Arm Night (middle button) work. Pressing Arm Away (right button) has no effect.

### Summary
I can live with three messages sent, as I would only work with messages with correct values. However, the unit is then dead for about 5 minutes, which is unacceptable. In addition, the Away button does not work and I have been unable to send a message to the unit. I would be very interested to know if anyone has got this unit working properly. If not, it should not be on the list of supported devices here: https://www.zigbee2mqtt.io/devices/3400-D.html.

### How to reproduce it (minimal and precise)

Always reproducible.

### Zigbee2MQTT version

1.31.2-1

### Adapter firmware version

20210708

### Adapter

Sonoff_Zigbee_3.0_USB stick

GLehnhoff, 2023-07-12:
I got it to work with ZHA. See here: https://community.home-assistant.io/t/centralite-3400-keypad/297353/11?u=glehnhoff

tardich, 2023-08-24:
I'm able to arm or disarm Alarmo through zigbee2mqtt without any problem. But I'm unable to set the state so the keypad shows the appropriate button on  :-(

GLehnhoff, 2023-09-25:
> I'm able to arm or disarm Alarmo through zigbee2mqtt without any problem. But I'm unable to set the state so the keypad shows the appropriate button on :-(

@tardich, how does your integration with Alamo works reg. what is written above in ...

1. Send Code
2. Device disconnects
3. Send a message to the device => This has been answered by you already, that it does not work
4. Arm Night button does not work

closed 2023-11-04 by github-actions[bot]