Koenkk/zigbee2mqtt #18129: TuYa TS0215A (_remote) no battery report

Isy99, 2023-06-27:
### What happened?

Hello folks,
after joining to zigbee2mqtt the 4 nobs of the remote are **working as expected.**
After weeks of operation, several deletions, resets and re-joinings there is still no battery report.

I tried as well to enable a report (reports menu, endpoint "1" cluster "PowerCfg" attribut "batteryPercentageRemaining"
2023-06-27 09:25:19Error: Bind 0xbc33acfffe5d2942/1 genPowerCfg from '0x00124b0018df3ffb/1' failed (AREQ - ZDO - bindRsp after 10000ms) at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (node:internal/timers:559:17) at processTimers (node:internal/timers:502:7)

Not a real problem, but perhaps it can be fixed in future releases.

battery
Remaining battery in %, can take up to 24 hours before reported.
N/A%

action
Ausgelöstes Ereignis (z.B. ein Aktivieren eines Tasters)
N/A

linkquality
Signalstärke
134lqi

### What did you expect to happen?

As described it takes longer until remaining battery will be reported.
My device doesn't report battery for a long period of time.

### How to reproduce it (minimal and precise)

No battery report since joining.
Deletion , factory reset and  re-joining of the device does not help

### Zigbee2MQTT version

 1.31.2 commit: 21f51258v

### Adapter firmware version

20211217

### Adapter

zStack3x0

Isy99, 2023-06-27:
**Update:**
zigbee2mqtt log shows the following after pressing a key:
`2023-06-27 10:41:54MQTT publish: topic 'zigbee2mqtt/wi_Schluesselschalter', payload '{"action":"arm_day_zones","action_code":"","action_zone":23,"linkquality":61}'`
`2023-06-27 10:42:25MQTT publish: topic 'zigbee2mqtt/wi_BU_Garage_Neu', payload '{"battery":100,"linkquality":109}'`

It looks like the battery data is sent to mqtt.
The web frontend of zigbee2mqtt on the details page shows the same as above.

Isy99, 2023-07-28:
OK, not really urgent, only a small bug.
73 Helmut

closed 2023-10-05 by github-actions[bot]

nireins, 2023-11-19:
Hi, i have the same issue. After 48 h no Batteryreport. 

info 2023-11-19 10:17:34MQTT publish: topic 'zigbee2mqtt/4T_FB_01', payload '{"action":"emergency","battery":null,"last_seen":"2023-11-19T10:17:34+01:00","linkquality":188}'
info 2023-11-19 10:17:34MQTT publish: topic 'zigbee2mqtt/4T_FB_01', payload '{"action":"","battery":null,"last_seen":"2023-11-19T10:17:34+01:00","linkquality":188}'
info 2023-11-19 10:17:34MQTT publish: topic 'zigbee2mqtt/4T_FB_01/action', payload 'emergency'

What is wrong? or what ca i do?

nireins, 2023-11-19:
EDIT: I test a small soultion: put a new battery into your remote and press a button. Then got to "development-concole" use endpoint 1 cluster "PowerCfg" attribut "batteryPercentageRemaining" and type 100 in the field. Then click "write" and press a button on the remote. In my case the batteryreport read 100%, i hope that helps

[image]

[image]

thepsr, 2024-02-19:
Attempting to write just gave me an error message.  

Then I tried to click "read" and press a button on the remote and from  this moment on it reports a battery value and not just null like before. I am not sure if I will have to periodically perform this read to update the value or if it will now "just work".

Strange behavior in any case though.

fred-c1, 2025-09-13:
Strange:   On my side this problem ("battery":null) appeared after updating Z2M to the latest version yesterday

fred-c1, 2025-12-05:
The solution mentionned above seems to work:
- go to "Dev. concole"  (menu in Z2M GUI/frontend)
- cluster "PowerCfg"
- attribute "batteryPercentageRemaining"
- click "Read"