Koenkk/zigbee2mqtt #18130: [New device support]:

fred-c1, 2023-06-27:
### Link

https://www.aliexpress.com/item/1005005466228073.html

### Database entry

{"id":23,"type":"Router","ieeeAddr":"0xa4c1388dfcc7548a","nwkAddr":26601,"manufId":4417,"manufName":"_TZE204_ac0fhfiq","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65506":56,"65508":0,"65534":0,"modelId":"TS0601","manufacturerName":"_TZE204_ac0fhfiq","powerSource":1,"zclVersion":3,"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-708457359},"lastSeen":1687880173333,"defaultSendRequestWhen":"immediate"}

### Comments

The external converter is working but I don't know how to modify the source code in github.com/Koenkk/zigbee-herdsman-converters

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

// added
const {Buffer} = require('buffer');

const phaseVariantNew = {
        from: (v) => {
            const buf = Buffer.from(v, 'base64');
            return {
            voltage: (   (buf[0] << 8)  | buf[1]    ) / 10,
            current: (   (buf[2] << 16) |  (buf[3] << 8)  |  buf[4]   ) / 1000,
            power : (    (buf[5] << 16) |  (buf[6] << 8)  |  buf[7]   )
            }; 
        },
      };

const definition = {
    fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE204_ac0fhfiq',
        },
    ],
    model: 'TS0601_Bidirectional_Energy_Meter_with_150A_Current_Clamp',
    vendor: 'TuYa',
    description: 'Bidirectional Energy Meter with 150A Current Clamp',
    
    fromZigbee: [tuya.fz.datapoints],
    
    toZigbee: [tuya.tz.datapoints],
    
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    
    configure: tuya.configureMagicPacket,
    
    exposes: [
        // Here you should put all functionality that your device exposes
        
        exposes.numeric('energy_forward', ea.STATE).withUnit('kW.h').withDescription('Total forward energy'),

        exposes.numeric('energy_reverse', ea.STATE).withUnit('kW.h').withDescription('Total forward reverse'),       
        
        e.power(), e.voltage(), e.current(),

        //exposes.numeric('power101', ea.STATE).withUnit('W').withDescription('Power from datapoint 101'),
        // datapoint 101 is not used see datapoint 101 below
        
        exposes.numeric('power_direction', ea.STATE).withDescription('Power direction 0/1 for forward/reverse'),

    ],
    
    meta: {
        tuyaDatapoints: [
            
            [1, 'energy_forward', tuya.valueConverter.divideBy100],
            
            [2, 'energy_reverse', tuya.valueConverter.divideBy100],
            
            // datapoint 3 = monthly energy: "we don't know how to support these" (resquest + response)
            
            // datapoint 4 = daily energy: "we don't know how to support these" (resquest + response)

            [6, null, phaseVariantNew], // voltage, current and power
         
            // datapoint 101 is datatype=number
            //[101, 'power101', tuya.valueConverter.raw],
            // Not used: "power" is taken in datapoint 6. Both provide the same value.
            
            [102, 'power_direction', tuya.valueConverter.raw],

        ],
    },
};

module.exports = definition;
```

Koenkk, 2023-06-27:
`energy_forward` = consumed energy and `energy_reverse` = produced energy?

fred-c1, 2023-06-27:
The spec says forward/reverse but yes, consumed/produced is ok too.

Koenkk, 2023-06-27:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)