Koenkk/zigbee2mqtt #18131: Z2M Map is not showing up anymore

canedje, 2023-06-27:
### What happened?

Since a while the map is not showing up anymore.
After push the button "load map" The wheel is turning for ever and never shows the map
It did work before.
I restarted the docker, the same result

### What did you expect to happen?

A shown map with all devices and connections

### How to reproduce it (minimal and precise)

just put the button "load map" again

### Zigbee2MQTT version

1.31.2

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonof zigbee 3.0 dongle plus-E

closed 2023-08-06 by github-actions[bot]

kcgthb, 2024-07-08:
I have the same issue with 1.39.0, nothing shows in the logs, but the map doesn't load.

tackin, 2024-11-28:
Found the problem in my case:
If I set docker-compose-file
```
environment:
 - Z2M_WATCHDOG=default

```

The watchdog will kill requests to devices which are not available yet. So the timeout will never reach the card-building processing.
Disable that environment-var solves the card-problem for me immediately.