Koenkk/zigbee2mqtt #18135: [New device support]: ZNQBKG40LM Aqara Z1 3 gang smart wall switch

jomud9, 2023-06-28:
### Link

https://www.aqara.com/cn/Canon_Smart_Wall_Switch_Z1_overview

### Database entry

{"id":15,"type":"EndDevice","ieeeAddr":"0x54ef4410008f6086","nwkAddr":22320,"manufId":4447,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.switch.acn054","epList":[1,2,3,4,21],"endpoints":{"1":{"profId":260,"epId":1,"devId":0,"inClusterList":[2820,1794,18,5,4,6,3,0,64704],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"modelId":"lumi.switch.acn054","appVersion":21}},"aqaraOpple":{"attributes":{"4":4,"247":{"type":"Buffer","data":[154,32,0]},"512":0,"617":1,"645":0,"646":1,"648":0,"1303":1}},"genOnOff":{"attributes":{"245":117440526,"onOff":1}},"genMultistateInput":{"attributes":{"presentValue":1}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":0,"inClusterList":[18,5,4,6,64704],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"245":117440563,"onOff":0}},"aqaraOpple":{"attributes":{"512":0,"617":1,"645":0}},"genMultistateInput":{"attributes":{"presentValue":1}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":0,"inClusterList":[18,5,4,6,64704],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"245":117440564,"onOff":0}},"aqaraOpple":{"attributes":{"512":0,"617":1,"645":0}},"genMultistateInput":{"attributes":{"presentValue":1}}},"binds":[],"configuredReportings":[],"meta":{}},"4":{"profId":260,"epId":4,"devId":0,"inClusterList":[64704],"outClusterList":[],"clusters":{"aqaraOpple":{"attributes":{"617":0}}},"binds":[],"configuredReportings":[],"meta":{}},"21":{"profId":260,"epId":21,"devId":0,"inClusterList":[12],"outClusterList":[],"clusters":{"genAnalogInput":{"attributes":{"presentValue":2.059999942779541}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":21,"stackVersion":27,"hwVersion":1,"dateCode":"20230526","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1854471209},"lastSeen":1687935847391,"defaultSendRequestWhen":"immediate"}

### Comments

New Z1 3 gang after setup https://github.com/Koenkk/zigbee2mqtt/issues/18082

Currently it works for basic function 
- Docoupled control
- voltage and power display

Following item still need to be addressed
- action: ONLY single are supported after decoupled (no double, triple, hold) 
- no multiple button pressed together (no both, all)
- flip_indicator_light(), power_outage_memory() and click mode() not working. ?? New Hardware New address??

### External converter

```shell
//. Changlog 20230628 - first ver for ZNQBKG40LM
//. Starting from ZNQBKG39LM - https://github.com/Koenkk/zigbee2mqtt/issues/18082 
//. 
//. 1. still thinking this version of firmware only support
//.     - action: single (no double, triple, hold)
//.     - no multiple button pressed together (no both, all)
//. 2. as button arranged in "top to buttom" instead of "standard" left to right, I renamed as button_1 instead of left
//. 3. Need help on function to make these working
//.   - flip_indicator_light, power_outage_memory and click mode
//.   - Seems address got changed but I have no idea

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');

//. PLEASE READ: These are present from device/xiaomi.ts, only needed for external converter
const herdsman = require('zigbee-herdsman');
const e = exposes.presets;
const ea = exposes.access;

const manufacturerOptions = {
    xiaomi: {manufacturerCode: herdsman.Zcl.ManufacturerCode.LUMI_UNITED_TECH, disableDefaultResponse: true},
};

const { hasAlreadyProcessedMessage } = require('zigbee-herdsman-converters/lib/utils');

const preventReset = async (type, data, device) => { 
    if (
        // options.allow_reset ||
        type !== 'message' ||
        data.type !== 'attributeReport' ||
        data.cluster !== 'genBasic' ||
        !data.data[0xfff0] ||
        // eg: [0xaa, 0x10, 0x05, 0x41, 0x87, 0x01, 0x01, 0x10, 0x00]
        !data.data[0xFFF0].slice(0, 5).equals(Buffer.from([0xaa, 0x10, 0x05, 0x41, 0x87]))
    ) {
        return;
    }
    const options = {manufacturerCode: 0x115f};
    const payload = {[0xfff0]: {
        value: [0xaa, 0x10, 0x05, 0x41, 0x47, 0x01, 0x01, 0x10, 0x01],
        type: 0x41,
    }};
[... 140 lines cut ...]
                .withDescription('Decoupled mode for button1 (Top) button')
                .withEndpoint('button_1'),
        exposes.enum('operation_mode', ea.ALL, ['control_relay', 'decoupled'])
                .withDescription('Decoupled mode for button 2 (Middle) button')
                .withEndpoint('button_2'),
        exposes.enum('operation_mode', ea.ALL, ['control_relay', 'decoupled'])
                .withDescription('Decoupled mode for button 3 (Buttom) button')
                .withEndpoint('button_3'),
        exposes.enum('click_mode', ea.ALL, ['fast', 'multi'])
                .withDescription('Click mode, fast: only supports single click which will be send immediately after clicking.' +
                    'multi: supports more events like double and hold'),
        e.action(['button_1_single', 'button_1_double', 'button_2_single', 'button_2_double', 'button_3_single', 'button_3_double',
                    'button_1_button_2_single', 'button_1_button_2_double', 'button_1_button_3_single', 'button_1_button_3_double',
                    'button_2_button_3_single', 'button_2_button_3_double', 'all_single', 'all_double']),
    ],
    onEvent: preventReset,
    configure: async (device, coordinatorEndpoint, logger) => {
        await device.getEndpoint(1).write('aqaraOpple', {'mode': 1}, {manufacturerCode: 0x115f, disableResponse: true});
//       await device.getEndpoint(1).write('aqaraOpple', {0x0125: {value: 0x02, type: 0x20}}, {manufacturerCode: 0x115f});
    },

};

module.exports = definition;
```

jomud9, 2023-07-07:
Got stuck with same issue reported in https://github.com/Koenkk/zigbee2mqtt/issues/18082#issuecomment-1620483816

@Koenkk not sure if Zigbee sniffing could help or not.

jomud9, 2023-07-12:
No feedback still. Might try to update related files to make it workable, then fix the remaining later

Helloookimi, 2023-07-13:
please also support ZNQBKG41LM which is 4 gang switch thank u very much~

jomud9, 2023-07-17:
> please also support ZNQBKG41LM which is 4 gang switch thank u very much~

If you could help, please try this?

```shell
//. Changlog 20230717 - first ver for ZNQBKG41LM
//. Starting from ZNQBKG39LM - https://github.com/Koenkk/zigbee2mqtt/issues/18082 
//. 
//. 1. still thinking this version of firmware only support
//.     - action: single (no double, triple, hold)
//.     - no multiple button pressed together (no both, all)
//. 2. as button arranged in "top to buttom" instead of "standard" left to right, I renamed as button_1 instead of left
//. 3. Need help on function to make these working
//.   - flip_indicator_light, power_outage_memory and click mode
//.   - Seems address got changed but I have no idea

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');

//. PLEASE READ: These are present from device/xiaomi.ts, only needed for external converter
const herdsman = require('zigbee-herdsman');
const e = exposes.presets;
const ea = exposes.access;

const manufacturerOptions = {
    xiaomi: {manufacturerCode: herdsman.Zcl.ManufacturerCode.LUMI_UNITED_TECH, disableDefaultResponse: true},
};

const { hasAlreadyProcessedMessage } = require('zigbee-herdsman-converters/lib/utils');

const preventReset = async (type, data, device) => { 
    if (
        // options.allow_reset ||
        type !== 'message' ||
        data.type !== 'attributeReport' ||
        data.cluster !== 'genBasic' ||
        !data.data[0xfff0] ||
        // eg: [0xaa, 0x10, 0x05, 0x41, 0x87, 0x01, 0x01, 0x10, 0x00]
        !data.data[0xFFF0].slice(0, 5).equals(Buffer.from([0xaa, 0x10, 0x05, 0x41, 0x87]))
    ) {
        return;
    }
    const options = {manufacturerCode: 0x115f};
    const payload = {[0xfff0]: {
        value: [0xaa, 0x10, 0x05, 0x41, 0x47, 0x01, 0x01, 0x10, 0x01],
        type: 0x41,
    }};
    await device.getEndpoint(1).write('genBasic', payload, options);
}; // PLEASE READ: THE END OF THE BLOCK THAT CAN BE DELELED after released

//. PLEASE READ: These should be update to FromZigbee.js 
const fzLocal = {
    xiaomi_multistate_action: {
        cluster: 'genMultistateInput',
        type: ['attributeReport'],
        convert: (model, msg, publish, options, meta) => {
            if (hasAlreadyProcessedMessage(msg, model)) return;
            let actionLookup = {0: 'hold', 1: 'single', 2: 'double', 3: 'triple', 255: 'release'};
            if (model.model === 'WXKG12LM') {
                actionLookup = {...actionLookup, 16: 'hold', 17: 'release', 18: 'shake'};
            }
            if (['WXKG02LM_rev2', 'WXKG07LM', 'WXKG15LM', 'WXKG17LM'].includes(model.model)) {
                buttonLookup = {1: 'left', 2: 'right', 3: 'both'};
            }
            if (['ZNQBKG39LM'].includes(model.model)) {
                buttonLookup = {1: 'button_1', 2: 'button_2'};
[... 128 lines cut ...]
                .withEndpoint('button_2'),
        exposes.enum('operation_mode', ea.ALL, ['control_relay', 'decoupled'])
                .withDescription('Decoupled mode for button 3 (Buttom) button')
                .withEndpoint('button_3'),
                exposes.enum('operation_mode', ea.ALL, ['control_relay', 'decoupled'])
                .withDescription('Decoupled mode for button 4 (Buttom) button')
                .withEndpoint('button_4'),        
        exposes.enum('click_mode', ea.ALL, ['fast', 'multi'])
                .withDescription('Click mode, fast: only supports single click which will be send immediately after clicking.' +
                    'multi: supports more events like double and hold'),
        e.action(['button_1_single', 'button_1_double', 'button_2_single', 'button_2_double', 'button_3_single', 'button_3_double',
        			'button_4_single', 'button_4_double',
                    'button_1_button_2_single', 'button_1_button_2_double', 'button_1_button_3_single', 'button_1_button_3_double',
                    'button_2_button_3_single', 'button_2_button_3_double', 'all_single', 'all_double']),
    ],
    onEvent: preventReset,
    configure: async (device, coordinatorEndpoint, logger) => {
        await device.getEndpoint(1).write('aqaraOpple', {'mode': 1}, {manufacturerCode: 0x115f, disableResponse: true});
//       await device.getEndpoint(1).write('aqaraOpple', {0x0125: {value: 0x02, type: 0x20}}, {manufacturerCode: 0x115f});
    },

};

module.exports = definition;
```

djante17, 2023-08-06:
Hi can I check if support for this device is done?

wildchai, 2023-09-10:
Hi,
I just got this 3gang switch and it is unsupported in my Z2m. This topic is to add it to z2m?

Koenkk, 2023-09-10:
@jomud9 what issues are you still facing with the current ext converter? Regarding the missing actions, check if you see any "Recieved zigbee message" in the z2m debug log when triggering the action. If you don't see anything, this device doesn't support this action.

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

wildchai, 2023-09-11:
Managed to get this loaded into Z2M and HA but there's a noticeable delay if using HA to activate the buttons. Physically pressing the buttons are quite responsive. This normal?

wildchai, 2023-09-22:
> please also support ZNQBKG41LM which is 4 gang switch thank u very much~

Is it working your 4 gang?

nickgoh, 2023-10-19:
> ZNQBKG41LM

zigbeeModel  **lumi.switch.acn055** instead of **lumi.switch.acn054** For ZNQBKG41LM

**Not Working**
Flip indicator light
Power outage memory
Click mode

**Button 4** 
Operation mode - Not Working
Single Click : OK
Double Click - show single_button_4
Hold - show single_button_4

nickgoh, 2023-10-22:
Hi, I am not sure wether this is correct way to sniff the data btw the gateway & the switch.  if this is the correct way i will sniff the other data as well .

**power_outage_memory**
In aqara app there is 4 options
1) Power-off memory ( After powering on, the device keep the previous state)
2) Electric Appliance is power on after xxx ?? (not sure what it actually do, cant find any manual explaining)
3) Electric Appliance is power off after xxx ??(not sure what it actually do,cant find any manual explaining)
4) The on-off status is reverse. ( reverse power of the previous state )

**1) Power-off memory ( After powering on, the device keep the previous state)**
Write
`
Frame 2: 54 bytes on wire (432 bits), 54 bytes captured (432 bits)
IEEE 802.15.4 Data, Dst: 0xeacb, Src: 0x0000
ZigBee Network Layer Data, Dst: 0xeacb, Src: 0x0000
ZigBee Application Support Layer Data, Dst Endpt: 1, Src Endpt: 1
    Frame Control Field: Data (0x00)
    Destination Endpoint: 1
    Cluster: Manufacturer Specific (0xfcc0)
    Profile: Home Automation (0x0104)
    Source Endpoint: 1
    Counter: 166
ZigBee Cluster Library Frame, Mfr: Lumi United Techology, Ltd Shenzhen (0x115f), Command: Write Attributes, Seq: 111
    Frame Control Field: Profile-wide (0x04)
    Manufacturer Code: Lumi United Techology, Ltd Shenzhen (0x115f)
    Sequence Number: 111
    Command: Write Attributes (0x02)
    Attribute Field, Uint8: 1
        Attribute: 0x0517
        Data Type: 8-Bit Unsigned Integer (0x20)
        Uint8: 1 (0x01)
`
Respond
`
Frame 4: 51 bytes on wire (408 bits), 51 bytes captured (408 bits)
IEEE 802.15.4 Data, Dst: 0x0000, Src: 0xeacb
ZigBee Network Layer Data, Dst: 0x0000, Src: 0xeacb
ZigBee Application Support Layer Data, Dst Endpt: 1, Src Endpt: 1
    Frame Control Field: Data (0x00)
    Destination Endpoint: 1
    Cluster: Manufacturer Specific (0xfcc0)
    Profile: Home Automation (0x0104)
    Source Endpoint: 1
    Counter: 129
ZigBee Cluster Library Frame, Mfr: Lumi United Techology, Ltd Shenzhen (0x115f), Command: Write Attributes Response, Seq: 111
    Frame Control Field: Profile-wide (0x1c)
    Manufacturer Code: Lumi United Techology, Ltd Shenzhen (0x115f)
    Sequence Number: 111
    Command: Write Attributes Response (0x04)
    Status Record
        Status: Success (0x00)
`

**2) Electric Appliance is power on after xxx ?? (not sure what it actually do)**
write
`
Frame 14: 54 bytes on wire (432 bits), 54 bytes captured (432 bits)
IEEE 802.15.4 Data, Dst: 0xeacb, Src: 0x0000
ZigBee Network Layer Data, Dst: 0xeacb, Src: 0x0000
ZigBee Application Support Layer Data, Dst Endpt: 1, Src Endpt: 1
    Frame Control Field: Data (0x00)
    Destination Endpoint: 1
    Cluster: Manufacturer Specific (0xfcc0)
    Profile: Home Automation (0x0104)
    Source Endpoint: 1
    Counter: 167
ZigBee Cluster Library Frame, Mfr: Lumi United Techology, Ltd Shenzhen (0x115f), Command: Write Attributes, Seq: 112
    Frame Control Field: Profile-wide (0x04)
    Manufacturer Code: Lumi United Techology, Ltd Shenzhen (0x115f)
    Sequence Number: 112
[... 88 lines cut ...]
    Attribute Field, Uint8: 3
        Attribute: 0x0517
        Data Type: 8-Bit Unsigned Integer (0x20)
        Uint8: 3 (0x03)
`
respond
`
Frame 35: 51 bytes on wire (408 bits), 51 bytes captured (408 bits)
IEEE 802.15.4 Data, Dst: 0x0000, Src: 0xeacb
ZigBee Network Layer Data, Dst: 0x0000, Src: 0xeacb
ZigBee Application Support Layer Data, Dst Endpt: 1, Src Endpt: 1
    Frame Control Field: Data (0x00)
    Destination Endpoint: 1
    Cluster: Manufacturer Specific (0xfcc0)
    Profile: Home Automation (0x0104)
    Source Endpoint: 1
    Counter: 135
ZigBee Cluster Library Frame, Mfr: Lumi United Techology, Ltd Shenzhen (0x115f), Command: Write Attributes Response, Seq: 114
    Frame Control Field: Profile-wide (0x1c)
    Manufacturer Code: Lumi United Techology, Ltd Shenzhen (0x115f)
    Sequence Number: 114
    Command: Write Attributes Response (0x04)
    Status Record
        Status: Success (0x00)
`

nickgoh, 2023-10-25:
Model **ZNQBKG41LM**

Tested on **ZNQBKG41LM** i only have a 4 button switch

**power_outage_memory** 
- on - **Tested working** (last power outage memory)
- Inverted - **Tested working** (last power outage memory (inverted))
- electric_appliances_on - not sure what it does
- electric_appliances_off - not sure what it does

**Click mode(Tested working)**  - only for button 4 ( Multi : 'button_4_single', 'button_4_double','button_4_hold', 'button_4_release')
**flip_indicator_light (removed)** -  as i cant find any settings in aqara app
**Decoupled mode for button 4 (removed)** -   button 4 is only a wireless switch without relay

**issue :**
**Lock Relay**  - **(not working)**  in aqara app it will lock up physical button & app control button

**External converter**

```javascript
//. Changlog 20230717 - first ver for ZNQBKG41LM
//. Starting from ZNQBKG39LM - https://github.com/Koenkk/zigbee2mqtt/issues/18082 
//. 
//. 1. still thinking this version of firmware only support
//.     - action: single (no double, triple, hold)
//.     - no multiple button pressed together (no both, all)
//. 2. as button arranged in "top to buttom" instead of "standard" left to right, I renamed as button_1 instead of left
//. 3. Need help on function to make these working
//.   - flip_indicator_light, power_outage_memory and click mode
//.   - Seems address got changed but I have no idea

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');

//. PLEASE READ: These are present from device/xiaomi.ts, only needed for external converter
const herdsman = require('zigbee-herdsman');
const e = exposes.presets;
const ea = exposes.access;

const manufacturerOptions = {
    xiaomi: {manufacturerCode: herdsman.Zcl.ManufacturerCode.LUMI_UNITED_TECH, disableDefaultResponse: true},
};

const { hasAlreadyProcessedMessage } = require('zigbee-herdsman-converters/lib/utils');

const preventReset = async (type, data, device) => { 
    if (
        // options.allow_reset ||
        type !== 'message' ||
        data.type !== 'attributeReport' ||
        data.cluster !== 'genBasic' ||
        !data.data[0xfff0] ||
        // eg: [0xaa, 0x10, 0x05, 0x41, 0x87, 0x01, 0x01, 0x10, 0x00]
        !data.data[0xFFF0].slice(0, 5).equals(Buffer.from([0xaa, 0x10, 0x05, 0x41, 0x87]))
    ) {
        return;
    }
    const options = {manufacturerCode: 0x115f};
    const payload = {[0xfff0]: {
        value: [0xaa, 0x10, 0x05, 0x41, 0x47, 0x01, 0x01, 0x10, 0x01],
        type: 0x41,
    }};
    await device.getEndpoint(1).write('genBasic', payload, options);
}; // PLEASE READ: THE END OF THE BLOCK THAT CAN BE DELELED after released

//. PLEASE READ: These should be update to FromZigbee.js 
[... 169 lines cut ...]
        exposes.enum('lock_relay', ea.ALL, ['On', 'Off'])
                .withDescription('Lock Relay for button 3 (Buttom) button')
                .withEndpoint('button_3'),

                
                

        exposes.enum('click_mode', ea.ALL, ['fast', 'multi'])
                .withDescription('Click mode (Button 4 Only), fast: only supports single click which will be send immediately after clicking.' +
                    'multi: supports more events like double and hold'),

        e.action([
        			'button_4_single', 'button_4_double','button_4_hold', 'button_4_release',
                    ]),
    ],
    onEvent: preventReset,
    configure: async (device, coordinatorEndpoint, logger) => {
        await device.getEndpoint(1).write('aqaraOpple', {'mode': 1}, {manufacturerCode: 0x115f, disableResponse: true});
//       await device.getEndpoint(1).write('aqaraOpple', {0x0125: {value: 0x02, type: 0x20}}, {manufacturerCode: 0x115f});
    },

};

module.exports = definition;
```

jomud9, 2023-11-24:
> Hi, I am not sure wether this is correct way to sniff the data btw the gateway & the switch. if this is the correct way i will sniff the other data as well .
>
> **power_outage_memory** In aqara app [...]

Would you share your setup to inspire me to do a sniff on my side too?

djante17, 2023-12-02:
Seems like an error with the converter recently, anybody else getting this error?

TypeError: Cannot read properties of undefined (reading 'options')

jomud9, 2023-12-06:
> Seems like an error with the converter recently, anybody else getting this error?
> 
> TypeError: Cannot read properties of undefined (reading 'options')

Not sure very well, but my z2m unable to start after upgraded to 1.34-01, I rollback to 1.33 and it works

djante17, 2023-12-06:
> > Seems like an error with the converter recently, anybody else getting this error?
> > TypeError: Cannot read properties of undefined (reading 'options')
> 
> Not sure very well, but my z2m unable to start after upgraded to 1.34-01, I rollback to 1.33 and it works

I've investigated this and realize that the exception is thrown because 'fz.aqara_opple' is no longer supported in the zgbee herdsman converter package. I changed it to 'fz.aqara_opple_multistate' and they are all able to start now in 1.34. Im not exactly sure what aqara_opple_multistate or aqara_opple initially does though but it seems to work same as before now.

Koenkk, 2023-12-06:
`fz.aqara_opple` -> `xiaomi.fromZigbee.aqara_opple` and make sure your converter has `const xioami = require('zigbee-herdsman-converters/lib/xioami');` at the top

leeaash, 2023-12-07:
> xiaomi.fromZigbee.aqara_opple

I think i did change as you said, but i still get an error as following when restarting z2m
`node:internal/modules/cjs/loader:1031
  throw err;
  ^

Error: Cannot find module 'zigbee-herdsman-converters/lib/xioami'
Require stack:
- /app/dist/util/utils.js
- /app/dist/util/settings.js
- /app/index.js
    at Function.Module._resolveFilename (node:internal/modules/cjs/loader:1028:15)
    at Function.Module._load (node:internal/modules/cjs/loader:873:27)
    at Module.require (node:internal/modules/cjs/loader:1100:19)
    at require (node:internal/modules/cjs/helpers:119:18)
    at /app/data/extension/externally-loaded.js:12:16
    at Script.runInContext (node:vm:141:12)
    at Script.runInNewContext (node:vm:146:17)
    at Object.runInNewContext (node:vm:306:38)
    at loadModuleFromText (/app/lib/util/utils.ts:154:8)
    at loadModuleFromFile (/app/lib/util/utils.ts:161:12)`

leeaash, 2023-12-07:
> I've investigated this and realize that the exception is thrown because 'fz.aqara_opple' is no longer supported in the zgbee herdsman converter package. I changed it to 'fz.aqara_opple_multistate' [...]

That worked, tks.

djante17, 2023-12-07:
> > `fz.aqara_opple` -> `xiaomi.fromZigbee.aqara_opple` and make sure your converter has `const xioami = require('zigbee-herdsman-converters/lib/xioami');` at the top
> 
> I think i did change as you said, but i still get an error as following when restarting z2m `node:internal/modules/cjs/loader:1031 throw err; ^
> 
> Error: Cannot find module 'zigbee-herdsman-converters/lib/xioami' Require stack:
> 
> * /app/dist/util/utils.js
> * /app/dist/util/settings.js
> * /app/index.js
>   at Function.Module._resolveFilename (node:internal/modules/cjs/loader:1028:15)
>   at Function.Module._load (node:internal/modules/cjs/loader:873:27)
>   at Module.require (node:internal/modules/cjs/loader:1100:19)
>   at require (node:internal/modules/cjs/helpers:119:18)
>   at /app/data/extension/externally-loaded.js:12:16
>   at Script.runInContext (node:vm:141:12)
>   at Script.runInNewContext (node:vm:146:17)
>   at Object.runInNewContext (node:vm:306:38)
>   at loadModuleFromText (/app/lib/util/utils.ts:154:8)
>   at loadModuleFromFile (/app/lib/util/utils.ts:161:12)`

probably because theres a typo it shoudl be 'xiaomi' not 'xioami'

@Koenkk what is the diffenrece between using xiaomi.fromZigbee.aqara_opple vs fz.aqara_opple_multistate?

leeaash, 2024-01-03:
when will Aqara z1 be officially supported by z2m?

jomud9, 2024-01-03:
> when will Aqara z1 be officially supported by z2m?

I afraid it is so difficult to say that "officially". However if the converter work for you and my cat is not harassing me for next couple of hours/days and let me to submit this thingie, Next (few) version for z2m will include that :-) (fingercrossed)

jomud9, 2024-01-03:
> I've investigated this and realize that the exception is thrown because 'fz.aqara_opple' is no longer supported in the zgbee herdsman converter package. I changed it to 'fz.aqara_opple_multistate' [...]

I tried the "multistate" as I applied what Koenkk said and did not work. I kept monitoring and yes it looked working but from long there are something alike

```
Zigbee2MQTT:debug 2024-01-03 17:27:41: Received Zigbee message from 'switchz1_livingroom', type 'attributeReport', cluster 'aqaraOpple', data '{"247":{"data":[3,40,29,5,33,32,0,9,33,0,0,10,33,201,20,12,32,10,13,35,28,0,0,0,17,35,1,0,0,0,100,16,0,101,16,1,102,16,0,149,57,74,66,187,65,150,57,205,76,18,69,152,57,195,245,40,63,151,57,195,245,20,66],"type":"Buffer"}}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2024-01-03 17:27:41: No converter available for 'ZNQBKG40LM' with cluster 'aqaraOpple' and type 'attributeReport' and data '{"247":{"data":[3,40,29,5,33,32,0,9,33,0,0,10,33,201,20,12,32,10,13,35,28,0,0,0,17,35,1,0,0,0,100,16,0,101,16,1,102,16,0,149,57,74,66,187,65,150,57,205,76,18,69,152,57,195,245,40,63,151,57,195,245,20,66],"type":"Buffer"}}'
```

I will have a quick check after cooking and washing. :-)

jomud9, 2024-01-03:
> @Koenkk what is the diffenrece between using xiaomi.fromZigbee.aqara_opple vs fz.aqara_opple_multistate?

@djante17 
That's what I understand.. it is the difference between aliasing, and how to make the code cleaner and easier to maintain. 👍 

https://github.com/Koenkk/zigbee-herdsman-converters/blob/6fd241fb4bbac24620de7b887368d8fd161a6a28/src/converters/fromZigbee.ts#L4125

and yes. After following Koenkk fix (without typo) it works

[Add] const xiaomi = require('zigbee-herdsman-converters/lib/xiaomi');
[Replace] fz.aqara_opple -> xiaomi.fromZigbee.aqara_opple

dinhchinh82, 2024-01-13:
@jomud9 have you got sucessfully on making Aqara Z1 4 gang work yet?

Thank you.

wildchai, 2024-01-19:
> Model **ZNQBKG41LM**
> 
> Tested on **ZNQBKG41LM** i only have a 4 button switch
> 
> **power_outage_memory**
> 
> * on - **Tested working** (last power outage memory)
> * Inverted - **Tested working** (last power outage memory (inverted))
> * electric_appliances_on - not sure what it does
> * electric_appliances_off - not sure what it does
> 
> **Click mode(Tested working)** - only for button 4 ( Multi : 'button_4_single', 'button_4_double','button_4_hold', 'button_4_release') **flip_indicator_light (removed)** - as i cant find any settings in aqara app **Decoupled mode for button 4 (removed)** - button 4 is only a wireless switch without relay
> 
> **issue :** **Lock Relay** - **(not working)** in aqara app it will lock up physical button & app control button
> 
> **External converter**
> 
> ```js
> //. Changlog 20230717 - first ver for ZNQBKG41LM
> //. Starting from ZNQBKG39LM - https://github.com/Koenkk/zigbee2mqtt/issues/18082 
> //. 
> //. 1. still thinking this version of firmware only support
> //.     - action: single (no double, triple, hold)
> //.     - no multiple button pressed together (no both, all)
> //. 2. as button arranged in "top to buttom" instead of "standard" left to right, I renamed as button_1 instead of left
> //. 3. Need help on function to make these working
> //.   - flip_indicator_light, power_outage_memory and click mode
> //.   - Seems address got changed but I have no idea
> 
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const reporting = require('zigbee-herdsman-converters/lib/reporting');
> const extend = require('zigbee-herdsman-converters/lib/extend');
> const ota = require('zigbee-herdsman-converters/lib/ota');
> 
> 
> //. PLEASE READ: These are present from device/xiaomi.ts, only needed for external converter
> const herdsman = require('zigbee-herdsman');
> const e = exposes.presets;
> const ea = exposes.access;
> 
> const manufacturerOptions = {
>     xiaomi: {manufacturerCode: herdsman.Zcl.ManufacturerCode.LUMI_UNITED_TECH, disableDefaultResponse: true},
> };
> 
> const { hasAlreadyProcessedMessage } = require('zigbee-herdsman-converters/lib/utils');
> 
> const preventReset = async (type, data, device) => { 
>     if (
>         // options.allow_reset ||
>         type !== 'message' ||
>         data.type !== 'attributeReport' ||
>         data.cluster !== 'genBasic' ||
>         !data.data[0xfff0] ||
>         // eg: [0xaa, 0x10, 0x05, 0x41, 0x87, 0x01, 0x01, 0x10, 0x00]
>         !data.data[0xFFF0].slice(0, 5).equals(Buffer.from([0xaa, 0x10, 0x05, 0x41, 0x87]))
>     ) {
>         return;
>     }
>     const options = {manufacturerCode: 0x115f};
>     const payload = {[0xfff0]: {
>         value: [0xaa, 0x10, 0x05, 0x41, 0x47, 0x01, 0x01, 0x10, 0x01],
>         type: 0x41,
>     }};
>     await device.getEndpoint(1).write('genBasic', payload, options);
> }; // PLEASE READ: THE END OF THE BLOCK THAT CAN BE DELELED after released
> 
> //. PLEASE READ: These should be update to FromZigbee.js 
> const fzLocal = {
[... 176 lines cut ...]
>                 
> 
>         exposes.enum('click_mode', ea.ALL, ['fast', 'multi'])
>                 .withDescription('Click mode (Button 4 Only), fast: only supports single click which will be send immediately after clicking.' +
>                     'multi: supports more events like double and hold'),
> 
>         e.action([
>         			'button_4_single', 'button_4_double','button_4_hold', 'button_4_release',
>                     ]),
>     ],
>     onEvent: preventReset,
>     configure: async (device, coordinatorEndpoint, logger) => {
>         await device.getEndpoint(1).write('aqaraOpple', {'mode': 1}, {manufacturerCode: 0x115f, disableResponse: true});
> //       await device.getEndpoint(1).write('aqaraOpple', {0x0125: {value: 0x02, type: 0x20}}, {manufacturerCode: 0x115f});
>     },
> 
> };
> 
> module.exports = definition;
> ```

Hi,
Does this replace the 3-gang version external converter? or both can still exists simultaneously? 

Thanks

jomud9, 2024-01-20:
@wildchai While they share almost the same code, one single file can not host two devices.

I would suggest you put them into two files and add these as separate convertors, at this moment.

I am looking forward to coming holidays then I could compare code diff among these time and submit, so no more convertor is needed.

wildchai, 2024-01-20:
> @wildchai While they share almost the same code, one single file can not host two devices.
>
> I would suggest you put them into two files and add these as separate convertors, at this moment.
>
> I am [...]

Thanks for your effort. Looking forward to it. Too bad I'm still new to all this. Would love to contribute more.

djante17, 2024-02-03:
@Koenkk the latest release seems to support this switch:https://www.zigbee2mqtt.io/devices/ZNQBKG41LM.html#aqara-znqbkg41lm. However, it only shows 2 buttons to control the relay instead of 3. Also down not seem to work better than the converters. Is there a way to use the external converters instead of the new one in the update? I cant seem to override it.

Koenkk, 2024-02-03:
@waterlou since you added these devices, could you take a look at this?

waterlou, 2024-02-03:
Yes. Working on this. Will send and update asap.

nickgoh, 2024-02-07:
Hi @waterlou ,

Thanks for the update ,im on

edge 1.35.3-dev 
zigbee_herdsman_converters_version
18.28.3
zigbee_herdsman_version
0.33.8

image and button show up and working correctly for top and bottom, however when triggering button middle shows the error below :

Error 2024-02-07 10:23:14No converter available for 'state_middle' ("OFF")
Error 2024-02-07 10:23:15No converter available for 'state_middle' ("ON")
Error 2024-02-07 10:23:23No converter available for 'operation_mode_middle' ("control_relay")

waterlou, 2024-02-07:
Can you kindly test this external converter?  Sorry my 4 gang switch may available only after lunar new year so cannot test it.

https://gist.github.com/waterlou/8ba3767a339b48a9d75741f5e664894e

nickgoh, 2024-02-07:
> Can you kindly test this external converter? Sorry my 4 gang switch may available only after lunar new year so cannot test it.
> 
> https://gist.github.com/waterlou/8ba3767a339b48a9d75741f5e664894e

Hi @waterlou ,

still showing the same error using the converter above, top and bottom working well

[12:47:54] INFO: Preparing to start...
[12:47:55] INFO: Socat not enabled
[12:47:57] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:error 2024-02-07 12:48:21: No converter available for 'state_middle' ("OFF")
Zigbee2MQTT:error 2024-02-07 12:48:22: No converter available for 'state_middle' ("ON")
Zigbee2MQTT:error 2024-02-07 12:49:30: No converter available for 'state_middle' ("OFF")
Zigbee2MQTT:error 2024-02-07 12:49:31: No converter available for 'state_middle' ("ON")
Zigbee2MQTT:error 2024-02-07 12:49:36: No converter available for 'operation_mode_middle' ("control_relay")
Zigbee2MQTT:error 2024-02-07 12:49:38: No converter available for 'lock_relay_middle' (true)

waterlou, 2024-02-08:
Please try this again
https://gist.github.com/waterlou/aad1f8a2f4521568c4055e4e2a2b998a

nickgoh, 2024-02-08:
> Please try this again https://gist.github.com/waterlou/aad1f8a2f4521568c4055e4e2a2b998a

Hi @waterlou , Thanks for the update. Middle button is working now.

waterlou, 2024-02-14:
Please try this one: 
https://gist.github.com/waterlou/338614a6efb18bad570b8797526855b7

for the debug message, did you have log_level: debug in the configuration.yaml?

wildchai, 2024-02-15:
> Please try this one: https://gist.github.com/waterlou/338614a6efb18bad570b8797526855b7
> 
> for the debug message, did you have log_level: debug in the configuration.yaml?

Tested the new code. works!

yes, i removed the debug line in configuration.yaml. But are those debug logs not affecting the performance?

btw....i noticed that all of my Z1 switches are not routers...is this a firmware issue or it is not meant to be one?

waterlou, 2024-02-17:
I think with debug message, the performance hit is minimal. For endpoint problem, yes it's the design of the switch, likely because it supports "no natural". Dunno if they will change it later with update firmware.

waterlou, 2024-02-18:
@wildchai can you kindly test this new one and see if it works?  It removed multiEndPoint for 1 gang model.

https://gist.github.com/waterlou/bb09c26c843cb3565be51a0a0bf2b02a

wildchai, 2024-02-22:
> @wildchai can you kindly test this new one and see if it works?  It removed multiEndPoint for 1 gang model.
> 
> https://gist.github.com/waterlou/bb09c26c843cb3565be51a0a0bf2b02a

Hi, 

What does the multiendpoint do? What do I need to look out for?

Will check it later tonight.

waterlou, 2024-02-23:
Multi endpoint is for more than one button. So z2m moderator suggest not using it for 1 gang switch. It should work as previous code but better test it and I don't have 1 gang switch in hand.

jomud9, 2024-03-14:
Although it is not a nice completion due to Aqara Firmware, I tried the code submit by @waterlou are all good.

So I think it is time to close this.

Thanks 水哥, 遲D一齊打芒亨Wilds. :-)

closed 2024-03-14 by jomud9