Koenkk/zigbee2mqtt #18137: Mesh does not work (is not shown in map)

mode2k, 2023-06-28:
### What happened?

I have a zigbee switch "Bewässerung" far away from my coordinator. In between there is a router to boost the signal. 
[image]
But according to the map, the zigbee switch trys to connect to the coordinator direct and  not using the router. why is that happening?
I am using Zigbee2MQTT Version
    [1.31.2](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.31.2) as HomeAssitant AddOn. 
Coordinator-Typ    zStack3x0
Coordinator-Version    20221226

Any idea what is going on? Is the map wrong an meshing works as designed?

### What did you expect to happen?

Meshing

### Zigbee2MQTT version

1.31.2

### Adapter firmware version

20221226

### Adapter

Zigbee CC2652P2 TCP Ethernet Coordinator

danieledwardgeorgehitchcock, 2023-06-28:
Some zigbee devices do not comply to the Zigbee standard as much as others and some devices are sensitive to this.

What device did you use when pairing the switch? sometimes you can force the route by pairing with the nearest router as opposed to opening up your whole network and co-ordinator...

mode2k, 2023-06-29:
I re-paired the Device "Bewässerung" with the Router.  Now the map looks like this:
[image]

No direct connection to the coordinator but also no solid line to any devices. But it works fine. Is the map wrong?

closed 2023-08-06 by github-actions[bot]