Koenkk/zigbee2mqtt #18138: Fingerbot Plus Switch Arm Issue (Z2M Edge)

notownblues, 2023-06-28:
### What happened?

Hi there,

I've been using the new Fingerbot Plus Switch Function (Thanks to [Gr8z](https://github.com/Gr8z) and [kirovilya](https://github.com/kirovilya) for pushing the changes btw, awesome new features!) recently merged [here](https://github.com/Koenkk/zigbee-herdsman-converters/pull/5883) from this [feature request](https://github.com/Koenkk/zigbee2mqtt/issues/18011). 

I've noticed that after a couple of ON/OFF switches the arm gets stuck and doesn't fully retract. I'm not sure whether it's a product malfunction or an issue with the newly merged integration but I thought I'd report it as I couldn't see the issue anywhere else. I'm not sure if it's an isolated case or happening to more people either.

It's fairly new so completely understand it will take some time to work properly.

I can't attach the video here due to it being too large but you can watch it [here](https://imgur.com/a/hWpERWC).

The issue happens at the very end of the video.

Many thanks!

### What did you expect to happen?

I was expecting that the Fingerbot Arm would fully go up or down consistently at the level set in the integration.

### How to reproduce it (minimal and precise)

Toggle the ON/OFF switch a couple of times as shown in the video.

### Zigbee2MQTT version

Edge

### Adapter firmware version

20221226

### Adapter

SONOFF Dongle

### Debug log

`Debug 2023-06-28 16:34:48Publishing 'set' 'state' to 'Shower Fan Fingerbot'
Info 2023-06-28 16:34:49MQTT publish: topic 'zigbee2mqtt/Shower Fan Fingerbot', payload '{"battery":100,"delay":0,"linkquality":43,"lower":60,"mode":"switch","program":{"0":255,"1":255,"2":255,"3":0},"reverse":"OFF","state":"ON","touch":"ON","upper":0}'
Debug 2023-06-28 16:34:49Received Zigbee message from 'Shower Fan Fingerbot', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Info 2023-06-28 16:34:49MQTT publish: topic 'zigbee2mqtt/Shower Fan Fingerbot', payload '{"battery":100,"delay":0,"linkquality":43,"lower":60,"mode":"switch","program":{"0":255,"1":255,"2":255,"3":0},"reverse":"OFF","state":"ON","touch":"ON","upper":0}'
Debug 2023-06-28 16:34:50Received MQTT message on 'zigbee2mqtt/Shower Fan Fingerbot/set' with data '{"state":"OFF"}'
Debug 2023-06-28 16:34:50Publishing 'set' 'state' to 'Shower Fan Fingerbot'
Debug 2023-06-28 16:34:52Received Zigbee message from 'Shower Fan Fingerbot', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Info 2023-06-28 16:34:52MQTT publish: topic 'zigbee2mqtt/Shower Fan Fingerbot', payload '{"battery":100,"delay":0,"linkquality":47,"lower":60,"mode":"switch","program":{"0":255,"1":255,"2":255,"3":0},"reverse":"OFF","state":"OFF","touch":"ON","upper":0}'
Info 2023-06-28 16:34:52MQTT publish: topic 'zigbee2mqtt/Shower Fan Fingerbot', payload '{"battery":100,"delay":0,"linkquality":47,"lower":60,"mode":"switch","program":{"0":255,"1":255,"2":255,"3":0},"reverse":"OFF","state":"OFF","touch":"ON","upper":0}'
Debug 2023-06-28 16:34:52Received MQTT message on 'zigbee2mqtt/Shower Fan Fingerbot/set' with data '{"state":"ON"}'
Debug 2023-06-28 16:34:52Publishing 'set' 'state' to 'Shower Fan Fingerbot'
Debug 2023-06-28 16:34:54Received Zigbee message from 'Shower Fan Fingerbot', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Info 2023-06-28 16:34:54MQTT publish: topic 'zigbee2mqtt/Shower Fan Fingerbot', payload '{"battery":100,"delay":0,"linkquality":43,"lower":60,"mode":"switch","program":{"0":255,"1":255,"2":255,"3":0},"reverse":"OFF","state":"ON","touch":"ON","upper":0}'
Info 2023-06-28 16:34:54MQTT publish: topic 'zigbee2mqtt/Shower Fan Fingerbot', payload '{"battery":100,"delay":0,"linkquality":43,"lower":60,"mode":"switch","program":{"0":255,"1":255,"2":255,"3":0},"reverse":"OFF","state":"ON","touch":"ON","upper":0}'
Debug 2023-06-28 16:34:55Received MQTT message on 'zigbee2mqtt/Shower Fan Fingerbot/set' with data '{"state":"OFF"}'
Debug 2023-06-28 16:34:55Publishing 'set' 'state' to 'Shower Fan Fingerbot'
Debug 2023-06-28 16:34:56Received Zigbee message from 'Shower Fan Fingerbot', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Info 2023-06-28 16:34:57MQTT publish: topic 'zigbee2mqtt/Shower Fan Fingerbot', payload '{"battery":100,"delay":0,"linkquality":43,"lower":60,"mode":"switch","program":{"0":255,"1":255,"2":255,"3":0},"reverse":"OFF","state":"OFF","touch":"ON","upper":0}'
Info 2023-06-28 16:34:57MQTT publish: topic 'zigbee2mqtt/Shower Fan Fingerbot', payload '{"battery":100,"delay":0,"linkquality":43,"lower":60,"mode":"switch","program":{"0":255,"1":255,"2":255,"3":0},"reverse":"OFF","state":"OFF","touch":"ON","upper":0}'
Debug 2023-06-28 16:34:57Received MQTT message on 'zigbee2mqtt/Shower Fan Fingerbot/set' with data '{"state":"ON"}'
Debug 2023-06-28 16:34:57Publishing 'set' 'state' to 'Shower Fan Fingerbot'
Info 2023-06-28 16:34:59MQTT publish: topic 'zigbee2mqtt/Shower Fan Fingerbot', payload '{"battery":100,"delay":0,"linkquality":40,"lower":60,"mode":"switch","program":{"0":255,"1":255,"2":255,"3":0},"reverse":"OFF","state":"ON","touch":"ON","upper":0}'
Debug 2023-06-28 16:34:59Received Zigbee message from 'Shower Fan Fingerbot', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Info 2023-06-28 16:34:59MQTT publish: topic 'zigbee2mqtt/Shower Fan Fingerbot', payload '{"battery":100,"delay":0,"linkquality":43,"lower":60,"mode":"switch","program":{"0":255,"1":255,"2":255,"3":0},"reverse":"OFF","state":"ON","touch":"ON","upper":0}'
Debug 2023-06-28 16:35:00Received MQTT message on 'zigbee2mqtt/Shower Fan Fingerbot/set' with data '{"state":"OFF"}'
Debug 2023-06-28 16:35:00Publishing 'set' 'state' to 'Shower Fan Fingerbot'
Info 2023-06-28 16:35:01MQTT publish: topic 'zigbee2mqtt/Shower Fan Fingerbot', payload '{"battery":100,"delay":0,"linkquality":43,"lower":60,"mode":"switch","program":{"0":255,"1":255,"2":255,"3":0},"reverse":"OFF","state":"OFF","touch":"ON","upper":0}'
Debug 2023-06-28 16:35:01Received Zigbee message from 'Shower Fan Fingerbot', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Info 2023-06-28 16:35:01MQTT publish: topic 'zigbee2mqtt/Shower Fan Fingerbot', payload '{"battery":100,"delay":0,"linkquality":43,"lower":60,"mode":"switch","program":{"0":255,"1":255,"2":255,"3":0},"reverse":"OFF","state":"OFF","touch":"ON","upper":0}'
Debug 2023-06-28 16:35:02Received MQTT message on 'zigbee2mqtt/Shower Fan Fingerbot/set' with data '{"state":"ON"}'
Debug 2023-06-28 16:35:02Publishing 'set' 'state' to 'Shower Fan Fingerbot'
Info 2023-06-28 16:35:04MQTT publish: topic 'zigbee2mqtt/Shower Fan Fingerbot', payload '{"battery":100,"delay":0,"linkquality":43,"lower":60,"mode":"switch","program":{"0":255,"1":255,"2":255,"3":0},"reverse":"OFF","state":"ON","touch":"ON","upper":0}'
Debug 2023-06-28 16:35:04Received Zigbee message from 'Shower Fan Fingerbot', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Info 2023-06-28 16:35:04MQTT publish: topic 'zigbee2mqtt/Shower Fan Fingerbot', payload '{"battery":100,"delay":0,"linkquality":47,"lower":60,"mode":"switch","program":{"0":255,"1":255,"2":255,"3":0},"reverse":"OFF","state":"ON","touch":"ON","upper":0}'
Debug 2023-06-28 16:35:04Received MQTT message on 'zigbee2mqtt/Shower Fan Fingerbot/set' with data '{"state":"OFF"}'
Debug 2023-06-28 16:35:04Publishing 'set' 'state' to 'Shower Fan Fingerbot'
Debug 2023-06-28 16:35:06Received Zigbee message from 'Shower Fan Fingerbot', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Info 2023-06-28 16:35:06MQTT publish: topic 'zigbee2mqtt/Shower Fan Fingerbot', payload '{"battery":100,"delay":0,"linkquality":43,"lower":60,"mode":"switch","program":{"0":255,"1":255,"2":255,"3":0},"reverse":"OFF","state":"OFF","touch":"ON","upper":0}'
Info 2023-06-28 16:35:06MQTT publish: topic 'zigbee2mqtt/Shower Fan Fingerbot', payload '{"battery":100,"delay":0,"linkquality":43,"lower":60,"mode":"switch","program":{"0":255,"1":255,"2":255,"3":0},"reverse":"OFF","state":"OFF","touch":"ON","upper":0}'`

closed 2023-08-06 by github-actions[bot]

SimonMayerhofer, 2024-02-11:
Did you find a solution to this? I have the same issue

notownblues, 2024-02-11:
Unfortunately not, I've also posted the issue in the HA forum and on Reddit but didn't get any reply. I've got 2 units and noticed the same problem on both in switch mode.

I've got 1 unit on my thermostat working fine in click mode but the other unit has been sitting in a drawer for months. At least it was inexpensive.

SimonMayerhofer, 2024-02-14:
> I've got 1 unit on my thermostat working fine in click mode 

I tested that but didn't work for me. After a few uses it stopped fully retracting again.

So I did some more testing and found a workaround which works now for me.
Instead of triggering the click, I just switch from click -> switch mode which results in a click. After 4 seconds I switch back to click mode, so it goes back to the unretracted position.
When switching the mode this way it always retracts as much as I need it.

notownblues, 2024-02-14:
Ok cool! I'm definitely gonna give it a try. Thanks for sharing this @SimonMayerhofer 🙏

conrad784, 2024-02-19:
> So I did some more testing and found a workaround which works now for me. Instead of triggering the click, I just switch from click -> switch mode which results in a click. After 4 seconds I switch [...]

Interestingly this workaround also works for mine (which also started to not work in switch mode ~1 week ago). So this should be in some way be fixable in software.

JohnM2000, 2024-02-27:
I too am having this issue.   The fingerbot has been working well for about three months but then stopped retracting.  I did all the usual, new battery, restarted, reinstalled the apps, re-paired, etc.  I was beginning to think it was simply wearing out.  Only after reading this post did I look more closely at the various modes.  After fiddling with "Up movement" and "Down movement" settings the bot seems to suddenly work perfectly.  I didn’t pay much attention to these settings when I first set it up, I just accepted the defaults.  I don't know if anything occurred that could have changed the settings, including me (user error).  So, questions:  
The settings don't seem to Save as I change between modes.  They only save for the mode that is selected when I close the app.  When I reopen the app only the  saved settings are for the mode that was last selected.  All other modes have default values.  When I change between modes (such as the suggested change from click to switch and back to click) it is not simply the change that causes a click action, it is that the 'new' mode has different Up and Down movement settings.  The bot changes to match these settings upon selection.  With each change of mode, the bot changes to the newly selected mode settings.   Is anyone else seeing this same behavior? 
Thanks.

SimonMayerhofer, 2024-03-23:
@notownblues are you able to open the issue again, so that people might look into it?
I notice that the workaround occasionally doesn’t work. But usually everything’s fine.

notownblues, 2024-03-23:
Commenting to re-open. 

The issue has also been happening in Click Mode on 2 different units in the last 2-3 weeks when it was only happening in Switch Mode before.

It got so bad (getting stuck nearly every time) it was unusable and I replaced them with 2 SwitchBot for the time being.

I still have the units so happy to provide logs if needed.

Strauman, 2024-05-09:
Got the same issue!

4rianton, 2024-10-22:
Having the same issue. Bought 2x fingerbots from Ali and both of them don't work or work very unreliably with Z2m